morrownr/USB-WiFi #1: Netgear A6210 adapter recognized but won’t turn on

Semiramide, 2021-03-13:
Upgraded to Ubuntu 20.04 kernel 5.  Connected Netgear A6210 WiFi adapter.  Hardware is recognized in WiFi settings but I am unable to turn it on.  I purchased this model based on your assertion that the driver would work in newest kernel, 5.  The adapter works well in Windows.  I don’t know what I’m missing.  Can you help?

morrownr, 2021-03-15:
It is good to meet you. I have something for you to check out...

Please run the following in a terminal:

$ uname -r

Let's see what version of the kernel you are running. Please post back.

The issue here is that the Ubuntu devs broke the mt7612u driver from about 5.4.59 until 5.4.65 or so. The problem was specific to Ubuntu 20.04 and lasted longer than it should have. If you are not on the latest version of the kernel, run:

$ sudo apt update
$ sudo apy full-upgrade

Let's see if that fixes the problem, if not, we press on with other solutions.

Regards.

morrownr, 2021-03-15:
I happened to think today that maybe you had not updated the Ubuntu 20.04 because you need the A6210 working to do updates. If that is the case, let me suggest that you download Ubuntu 20.10 and burn it to the flash driver or DVD you are using for installation. Boot the system with 20.10 with the A6210 in a USB 3 port and I'll bet you have internet access. If it works well, I will suggest that you then install a fresh copy of 20.10 over the 20.04 installation (format the 20.04 partition during installation to ensure a perfectly clean installation.)

Good luck

Semiramide, 2021-03-15:
Uname -r : 5.4.0-66-generic

I’m using a wired connection for internet.  The old internal wireless card is dead but still shows up in WiFi settings as Intel. By the way I have only usb 2.0 on this semi-old Dell laptop.  I assume the driver will be able to adapt to usb 2.0 without fiddling with obscure settings.

Thanks for a prompt reply.

John Johnson

CAUTION: External Sender

morrownr, 2021-03-15:
I just took one of my adapters with a mt7612u chipset to test on an older system that only has USB 2 ports. It was recognized immediately. I performed a system update and upgrade. No problems noted.

The mt7612u driver is really different than the Realtek drivers. It is like having automatic transmission vs a standard. It automatically detects which type of USB port it is in and provides appropriate support. It always seems to get it right here.

That kernel (66) is what my Linux Mint 20.1 is running and it works well with the mt7612u based adapters. That is causing me to ponder if something else is amiss. Have you "forgotten" all existing connections? I've seen that gum things up.

Semiramide, 2021-03-15:
Was wondering about existing connections but all I could figure is the dead WiFi card inside might be causing a problem.

I first purchased  a Realtek adapter using rtl8812 driver which I installed with a downloaded script but it never worked in 18.04.  The Realtek driver was apparently installed as I plugged it in after upgrading to 20.04.  It showed up in WiFi settings but again I could not turn it on either as with the Netgear adapter.  The Realtek adapter also worked in Windows.

Do you suggest a clean reinstall of 20.04 or first pursue other measures?

John Johnson

morrownr, 2021-03-15:
John,

It sounds like there may be a little upgrade mess going on. As a general rule, if you are doing a major distro upgrade from say 18.04 to 20.04, I recommend you check dkms ( $ sudo dkms status ) to see what is installed via dkms so that it can be uninstalled via dkms before the upgrade takes place. Reinstall once the new version is settled.

My recommendation is install 20.10 while formatting the installation partition during the installation process. That will ensure you have a clean install to work with. Backup any important files you have before installing and if you need help with the installation, come on back and I'll see what I can do.

If you want to do a test run with a flash driver or DVD that has 20.10 on it, go ahead and boot it with the A6210 in a port and see if you have internet access. That would tell you if it is going to work.

Semiramide, 2021-03-15:
Thanks I’ll try that next and get back. Probably take me a day or two.

John Johnson

Semiramide, 2021-03-23:
Finally got back to this project.

Installed 20.10 from usb(in try only mode).  Got same result when Netgear A6210 was plugged in.  Unable to toggle it on in WiFi settings.  See attached images.

[cid:F078887B-D2D0-48A5-AEFC-49D4FAA97C70-L0-001][cid:237C38E7-0B09-4B53-8E26-CE95CD7C42CB-L0-001][cid:6C37C657-E47A-4B07-A8F6-8BE8F069986C-L0-001]
Thanks for your attention.

John Johnson

morrownr, 2021-03-23:
@Semiramide 

John,

The attached pics did not come through. I think if you log into github you can drag and drop the pics into a reply and it will work.

I would like to see the pics as I am flying blind right now. Also could I get you to run...

$ lsusb

and

$ lsusb -t

and

$ lspci

and 

$ rfkill list all

and paste the results in a reply. 

That last command will tell us if your system has a soft or hard block in place. Hey, we might even bring your internal card back too life if the problem is a block.

Nick

Semiramide, 2021-03-23:
pics were just shots to show you I was in 20.10 demo mode, had the netgear attached, and the wifi settings screen showing the recognition of the two wifi interfaces and the toggle button to turn them on.

attached is the terminal output file you requested.

Semiramide, 2021-03-24:
Here is the terminal output you requested:
john@john-Latitude-E7440:~$ lsusb
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0846:9053 NetGear, Inc. A6210
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0c45:64d2 Microdia Integrated Webcam
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
john@john-Latitude-E7440:~$ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=mt76x2u, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 4: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
john@john-Latitude-E7440:~$ lspci
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 09)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection I218-LM (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 1 (rev e4)
00:1c.3 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 4 (rev e4)
00:1c.4 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 5 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 RAID bus controller: Intel Corporation 82801 Mobile SATA Controller [RAID mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
02:00.0 Network controller: Intel Corporation Wireless 7260 (rev 73)
03:00.0 SD Host controller: O2 Micro, Inc. SD/MMC Card Reader Controller (rev 01)
john@john-Latitude-E7440:~$ rfkill list all
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: yes
2: dell-wifi: Wireless LAN
	Soft blocked: no
	Hard blocked: yes
3: dell-bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: yes
4: nfc0: NFC
	Soft blocked: no
	Hard blocked: no
5: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no
john@john-Latitude-E7440:~$

Semiramide, 2021-03-24:
Don't know why the text above has strikethru's.  I'll do it again.

john@john-Latitude-E7440:~$ lsusb
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 0846:9053 NetGear, Inc. A6210
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0c45:64d2 Microdia Integrated Webcam
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
john@john-Latitude-E7440:~$ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=mt76x2u, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 4: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 4: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
john@john-Latitude-E7440:~$ lspci
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 09)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection I218-LM (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 1 (rev e4)
00:1c.3 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 4 (rev e4)
00:1c.4 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 5 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 RAID bus controller: Intel Corporation 82801 Mobile SATA Controller [RAID mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
02:00.0 Network controller: Intel Corporation Wireless 7260 (rev 73)
03:00.0 SD Host controller: O2 Micro, Inc. SD/MMC Card Reader Controller (rev 01)
john@john-Latitude-E7440:~$ rfkill list all
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: yes
2: dell-wifi: Wireless LAN
	Soft blocked: no
	Hard blocked: yes
3: dell-bluetooth: Bluetooth
	Soft blocked: no
	Hard blocked: yes
4: nfc0: NFC
	Soft blocked: no
	Hard blocked: no
5: phy1: Wireless LAN
	Soft blocked: no
	Hard blocked: no
john@john-Latitude-E7440:~$

Semiramide, 2021-03-24:
I placed all this into github thread

morrownr, 2021-03-24:
Hard blocked: yes

The above is a BIG problem. Do you see how it is listed for both wifi devices and bluetooth. Your bluetooth is not working either.

Fn-Alt-F2

Try the above key combination and see what happens. Ref:

https://askubuntu.com/questions/127977/wireless-is-disabled-by-hardware-switch-on-dell-inspiron-1750

This basically means you have Airplane-mode turned on. I'm about 90% confident this is the problem. You will likely have two working wifi devices and bluetooth working soon. 

Nick

morrownr, 2021-03-24:
Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=mt76x2u, 5000M

Smokin' Just thought I'd point out that your A6210 is not only ready to go according the driver but is ready at USB3 speed.

Just have to get that hard block turned off. Let me know how it goes.

Semiramide, 2021-03-24:
sorry but the fix didn't help.  my wifi settings shows airplane mode is off and no adapter is found.

Semiramide, 2021-03-24:
keystroke combination did nothing

morrownr, 2021-03-24:
"keystroke combination did nothing"

I understand. That just means we need to find either the right key combination or the switch. On my old Dell Laptop, the key combination is Fn+F6. I have had laptops that have a little switch on the side. If the switch is set to airplane mode, you can see different results depending on the os. Linux fully respects the setting in the cases I am familiar with but Windows does not so you might see Windows unable to work with the internal card but will work with the USB adapter. I'll let bet your bluetooth is not working in Windows.

Hard block is telling us that a hardware setting, either a key combination or switch, is set to tell the os not to turn wifi and bluetooth on. I did a google search on the following:

"Dell Latitude E7440 airplane mode toggle"

One interesting result is:

https://forums.tomshardware.com/threads/dell-latitude-e7440-stuck-in-flight-mode.3060750/

morrownr, 2021-03-24:
Take heart in the fact that you are far from the only one to ever experience this problem. It is not a Linux problem or a Windows problem per se but a problem caused by a lack of consistent design by system designers. There are dozens of key combinations in use on laptops for airplane mode. Some don't even use key combinations, they use a hardware switch somewhere on the laptop. 

To complicate things, even if you find the correct solution, you may have to reboot to discover if it is the right solution. Can you imagine if things were this messed up on iPhones or Android phones? A lot of phones would go in the trash. You are going to have to find the key combination or switch for your laptop and it may take a lot of trial and error.

Semiramide, 2021-03-24:
Much thanks for your efforts.  The thread you linked about Dell laptops certainly makes it sound confusing and frustrating.  Mainly it’s good to know the WiFi adapter nor driver are the problem.  I’ll try some of those potential solutions.  I’ll check back with results in the near future.

Thanks again.

John Johnson

Semiramide, 2021-03-24:
Retested all bios settings related to wireless and wireless switch settings.  Nothing helped to allow me to activate the Netgear A6210 in Ubuntu 20.04 settings.

Next I removed the dead Intel wireless card from the Dell laptop.  I insulated the card connectors to prevent an accidental short.  Rebooted into Ubuntu 20.04.  In settings page Bluetooth is not available and no network available.  Plugged in the A6210 dongle and immediately it turned itself on in WiFi settings and showed list of available wireless networks.  After turning WiFi services on and logging in to my local network router WiFi is now working well with fast internet connections. 

Apparently the built in WiFi card was also supplying Bluetooth connectivity as well so that’s gone. 

 Earlier when I was unsuccessful in testing the Realtek WiFi dongle I removed the dead internal card and it made no difference.  I assume it might have worked but the driver was the problem there. 
 
I suppose this issue is closed but I’ll wait to close it for a few days.

morrownr, 2021-03-24:
John, you might consider posting the issue in the Ubuntu Networking and Wireless forum:

https://ubuntuforums.org/forumdisplay.php?f=336

Post the results that showed Hard block on and specify your computer model and you might have someone that has the same exact system that you do and knows the answer. I would like to hear back from you once the issues is resolved so you can post the resolution here for others to see.

I'm sorry I could not point you to the exact answer but the adapter is working and the driver is working, I can tell that from what you posted. The problem is that your computer is in Airplane mode. As they used to say on X-Files, the answer is out there.

Good luck.

morrownr, 2021-03-24:
"Apparently the built in WiFi card was also supplying Bluetooth connectivity as well so that’s gone."

Was bluetooth working in Windows before you removed the card? The reason I ask is to try to confirm what the issue was. If bluetooth was working on Windows, that could indicate Airplane Mode was not the issue. Airplane Mode still could be a cause as you have probably pressed several key combinations at this point. When i gave you my 90% estimate that Airplane Mode was the issue, the other 10% was a internal failure of the internal wifi/bt card that is causing system problems.

Since you have now removed the card and things seem to be coming to life. Maybe a hard fail of the internal card was the problem. Stuff happens. For what it is worth and for your future reference: Internal replacement cards are readily available for most situations and are very cheap for the most part. Mark sure the chipset is from Intel which means near 100% probability that Linux has an internal driver ready to go. It would be nice if Intel would sell chipsets into the USB wifi market. The one thing you really have to research is making sure you get a card that will indeed work on your laptop as some laptops limit in the bios which cards will work.

Glad you got the A6210 going. Hopefully it will serve your well for years on many different computers.

Semiramide, 2021-03-24:
Well I got it working after all. See my last  comment in GitHub.   I will post this anyway to Ubuntu forum as well once I figure out how.

John Johnson

Semiramide, 2021-03-24:
I haven’t tried using Bluetooth ever in either O.S.  as I haven’t needed it.  I just know it disappeared as an option once the dead card was removed.   Airplane mode switch operates normally now with the working WiFi usb. Actually I tried no more key combinations in attempting a remedy.  Just removing the internal card solved the hard blocks.

It seems to make sense to me it that it wasn’t a problem with airplane mode but the faulty card.

You still think it’s airplane mode?

John Johnson

Semiramide, 2021-03-24:
I found several used/new? replacement cards on eBay but, as you say, finding the correct one is the problem.  I read several reports of buyers having to return dead cards or not getting the right model.  It seems like a lot more trouble than it’s worth if I can get a working dongle.

John Johnson

morrownr, 2021-03-24:
"It seems to make sense to me it that it wasn’t a problem with airplane mode but the faulty card."

"You still think it’s airplane mode?"

I'm going to agree with you that it sure looks like that internal card bit the dust and did so in a way that was causing other problems. Once in a while the 10% probability is right. Heh heh. What we don't know for sure is if the problem was specifically in the card or in the cables to the card or in part of the mb that supports the card which means we don't know for sure that a new card would fix the problem.... even if you find one that will work. I think your idea to run with the A6210 since it is working is a good idea. 

After you have used the A6210 for a while, could I get you to come back here with a product review so that others can get an idea about that adapter. It is still on the market and the idea here is to let folks know what works. I'd be interested to know if it is a single-state or multi-state adapter. The best way to test is to plug it into a Raspberry Pi and see if it shows up as a CDROM or Flash driver instead of a wifi adapter. Maybe you have a friend with a Raspberry Pi?

Cheers.

Semiramide, 2021-03-25:
Will return with a review.  Alas no Raspberry Pi to play with.

Again thanks for your guidance and comments on the hardware problems we face with laptops.  And thanks to those who maintain and update the Linux kernel.  I’d say another monetary contribution is due to the community.

John Johnson

morrownr, 2021-04-03:
Closing issue. Product review can go in a newly opened issue.

morrownr, 2021-05-21:
@Semiramide 

How is it going? Now that you have had the Netgear A6210 for a while, can I talk you into doing a short review?

If you look in the main README, you will see several short reviews on some of the adapters. Something like that would be great. If you want to do something longer, that is fine also.

Nick