morrownr/USB-WiFi #2: Alfa AWUS036ACM Master Mode

DeityOfChaos, 2021-03-31:
My Alfa AWUS036ACM doesn't support master mode, I thought it worked for AP..?

[image]
[image]

[image]

I noticed this because it's the only adapter that would crash hostapd, then I figured it must be something realted to AP/master mode.
How did you get yours in master mode?

morrownr, 2021-04-01:
https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point-2.md

That link shows my setup on a RasPi4b. I use 2 hostapd.conf files so as to support 5g (the ACM) and 2g (the internal RasPi wifi capability). You can grab the ACM ht and vht settings as needed.

Let me know.

morrownr, 2021-04-01:
For the ACM (mt7612u)

ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]

vht_capab=[RXLDPC][TX-STBC-2BY1][SHORT-GI-80][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

The mt7612u and rtl8812au require very different settings for the above lines. The above lines are for the mt7612u chipset and driver.

DeityOfChaos, 2021-04-02:
> For the ACM (mt7612u)
> 
> ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]
> 
> vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
> 
> The mt7612u and rtl8812au require very different settings for the above lines. The above lines are for the mt7612u chipset and driver.

yes my primary issue was the capability set, and then for unknown reasons problems persisted until I rebooted the device. I commented out 5Ghz functions and simplified the config file to just use 802.11n and seems to work fine, I found your repository regarding configs mt7612u and it was a life saver lol

the AC model and ACH seem to be a little bit more versatile and easier to set up for AP mode  (they worked with whatever config I threw at them) but again the fact RTL8812AU isn't an in-kernel driver can be a deal breaker.

Thanks dude, happy easter mate

morrownr, 2021-04-02:
Happy Easter to you too.

I think using hostapd with the ACM vs ACH is really a matter of familiarity. My RasPi4b's main job is serving as an access point. I do a lot of testing with the setup and can say without hesitation that the Alfa ACM (mt7612u) and ACH (rtl8812au) are the best adapters that I have as far as serving as access points. I am going to have an Alfa ACHM (mt7610u) soon so I'll do some testing with it to see how it stacks up. I will add, in case others read this post, that user looking to use master (AP) and/or monitor modes should stick with adapters based on the mt7612u/mt7610u and rtl8812au/rtl8811au chipsets if they are looking to dual band adapters. The Realtek drivers for the rtl8812bu, rtl8814au and rtl8811cu chipsets do not do master (AP) and monitor modes well at all.

The differences that I see between the ACM and ACH have mostly to do with driver capabilities:

- WPA3 - the Mediatek driver (in-kernel) for the ACM supports WPA3 and WPA3 Transition. The Realtek driver for the ACH does not support WPA3. It says is supports WPA3 but I can't make it work and have found nobody else that can make it work. I am working behind the scenes to try to figure out what the problem is. The hostapd.conf you looked at and had problems with had WPA3 Transition turned on. The ACM will certainly support that but if you then try to access it with an adapter that is using a Realtek driver, the Realtek driver won't work so you think the ACM is in the wrong...which is wrong. If you will post your hostapd.conf that you are using with the ACM, I will help you tweak it.

-  Interface combinations - the ACM driver supports various interface combinations. The Realtek driver does not. You can check for yourself with `$ iw list`. I regularly get questions about how to set up various interface combinations with the Realtek drivers and I just have to say "not supported" by Realtek and it is not something that I can add.

- Support for Linux Wireless Standards. You can see this when using iw and ip. With the Realtek drivers you will notice gaps in information and some bogus information.

On to other stuff:

Quote: "I also noticed the ACM model (400mA max input power) could not have the TX Power adjusted"

I have several things to say here. That low input power requirement of the ACM makes it a dream come true for use with the RasPi4b since the RasPi4b only has a total of 1200 mA for the entire USB subsystem. The ACH can and does take around 800 mA when pushed so if you have other things running from USB, you may be looking at brown out time when pushing things with the ACH. I understand what you are saying about "power adjusted" but let me point out that just because the Realtek driver says the power has been adjusted does not mean that it really has been adjusted.

I'll be glad to help you get your ACM going full speed with hostapd on your RasPi4b.

Regards,

Nick

DeityOfChaos, 2021-04-02:
> just because the Realtek driver says the power has been adjusted does not mean that it really has been adjusted.

I've had this feeling for a while now lol :')

Regarding WPA3 let's hope someone who done it speaks up regarding WPA3 with RTL8812AU.
I'm starting to enjoy the MediaTek chipset, I haven't found any bug with the driver yet and I honestly wished Alfa Networks relied more on MediaTek than Realtek.

My AP setup relies in a Raspberry is RPi 3B+ with Raspbian Lite and even though it isn't even USB 3.0 and is significantly less performing than Rpi4, it is still strong enough to provide a solid network with minor lag and performance loss, pretty awesome