morrownr/USB-WiFi #3: mt7612u DFS support

soyersoyer, 2021-04-29:
Great summary!

You can add some more info:
The mt7612u(n) doesn't support DFS, but it can use that channels with different reg domain.
The multi ap mode (guest ap mode) works too.

morrownr, 2021-04-29:
Quote: "The multi ap mode (guest ap mode) works too."

Do you have an example you would share of setting up the guest mode?

Thanks.

soyersoyer, 2021-04-30:
Of course!
I am using a COMFAST CF-WU782AC adapter.
My hostapd.conf:
```
interface=wlan0
ssid=ssid0
bssid=e2:aa:bb:cc:dd:ee

hw_mode=a
channel=36

ieee80211n=1
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][HT40+][GF][TX-STBC][RX-STBC1]

ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][TX-ANTENNA-PATTERN][RX-ANTENNA-PATTERN]

vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

ieee80211d=1
#ieee80211h=1
country_code=US

wmm_enabled=1

macaddr_acl=0
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
wpa_passphrase=pass0
ap_isolate=1

bss=wlan1
ssid=ssid1
bssid=e6:aa:bb:cc:dd:ee
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
wpa_passphrase=pass1
```
The trick is, you have to manually set the bssids, and the first one first byte have to be masked with 0x02 and the second one first byte with 0x06.

This is because the driver uses the NEW_MULTI_BSSID_MODE:
```
New BSSID numbering rule:
a. Bit0 of MAC address Byte0 is broadcast/multicast bit.
b. Bit1 of MAC address Byte0 is local administration bit
and should be set to 1 in extended multiple BSSIDs’.
c. Bit4:Bit3 of MAC address Byte0 is extended multiple
BSSID index.
For example: in 4-BSSID mode and MAC address is
00:0c:43:28:60:01,
by new rule, the extended 3-BSSID is
02:0c:43:28:60:01 and
06:0c:43:28:60:01 and
0a:0c:43:28:60:01.
```

morrownr, 2021-05-01:
Thanks. I'm going to have to try that on my RasPi4b access point. And I will get it documented.

Any possibility of getting you write a review and providing a link or two for the COMFAST CF-WU782AC adapter? I have a couple of links for it but without having one, there is only so much I can write about.

Cheers,

Nick

soyersoyer, 2021-05-02:
It works well with my rasPi4b. I had to disable the usb_sg. I like this setup because it can route near gigabit (with a tp-link ue300), my mobile devices have 300-500mbit/s download speed too, it has guest wifi, and I can easily switch to a newer wifi adapter later. The rpi runs kodi, a shairport server and a bluetooth sound receiver server too.
I bought the wifi adapter from https://www.aliexpress.com/item/32902591576.html
Bests,
s

morrownr, 2021-07-24:
@soyersoyer 

Quote: "The mt7612u(n) doesn't support DFS"

I finally had time this morning to check this issue. My client system is using an ALFA AWUS036ACM (mt7612u) adapter and the wifi router has been changed to channel 52, which is a DFS channel here in the US. My intent is to make this a long term test and try all of the DFS channels to get results. Can I get you to go over your results again as it looks to me like the mt76 drivers and the adapters do support DFS channels? At least the support appears to be there in client mode. I have not tested AP mode yet.

Medullitus, 2022-02-04:
> @soyersoyer
>
> Quote: "The mt7612u(n) doesn't support DFS"
>
> I finally had time this morning to check this issue. My client system is using an ALFA AWUS036ACM (mt7612u) adapter and the wifi router has [...]

When will you test the AP mode?

morrownr, 2022-02-06:
> When will you test the AP mode?

I have. DFS is not supported currently. I'm trying to work the issue.

morrownr, 2022-03-29:
Hi @soyersoyer 

I need your take on something I am reading:

```
https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.18-Networking
```

See this part:

```
background radar detection for the Mediatek MT76 driver
```

Is that DFS support? Is it just for the new 7921 or is it for the 7612 also? 

Regards

soyersoyer, 2022-03-29:
I think they wrote about these commits:
https://github.com/openwrt/mt76/commit/fae295af31eb463ee3654c5c2a29ec4fe3ba03e2
https://github.com/openwrt/mt76/commit/e0f9479bf8937bc0ff7322043a7a0554b44af29f
https://github.com/openwrt/mt76/commit/822e1135e7e1daba426972b8de6627f17158d8f7

Sami32, 2022-09-18:
MT7610 and MT7612 support DFS in AP mode but it seem that their Linux driver is lacking of such support:
[mtk_wi-fi_softap_software_programming_guide_v4_6.pdf](https://github.com/morrownr/USB-WiFi/files/9594329/mtk_wi-fi_softap_software_programming_guide_v4_6.pdf)

Anthony96922, 2023-02-16:
The PDF you posted makes no claim that the mt7610 or mt7612 support DFS in AP mode.

morrownr, 2023-02-16:
If you are willing to and know how to patch the in-kernel driver/kernel:

```
DFS for mt7612u

 mt76x02_util.c 

@@ -89,6 +89,10 @@ static const struct ieee80211_iface_combination mt76x02u_if_comb[] = {

		.max_interfaces = 2,
		.num_different_channels = 1,
		.beacon_int_infra_match = true,
+                .radar_detect_widths = BIT(NL80211_CHAN_WIDTH_20_NOHT) |
+                                       BIT(NL80211_CHAN_WIDTH_20) |
+                                       BIT(NL80211_CHAN_WIDTH_40) |
+                                       BIT(NL80211_CHAN_WIDTH_80),
	}
};
```

aaronkogan, 2023-02-22:
> If you are willing to and know how to patch the in-kernel driver/kernel:
>
> ```
> DFS for mt7612u
>
>  mt76x02_util.c
>
> @@ -89,6 +89,10 @@ static const struct ieee80211_iface_combination mt76x02u_if_comb[] [...]

When will this patch appear in the mainline kernel?

Anthony96922, 2023-02-22:
I don't know about the mt7612u, but I've tested the mt7610u with "dfs-pulse-tester" and my HackRF SDR and have not seen any radar events on the hostapd log. Seems like it can't sense radar signals. I know my SDR setup is working since I can get my Ubiquiti AP's to react.

morrownr, 2023-02-22:
> I don't know about the mt7612u, but I've tested the mt7610u with "dfs-pulse-tester" and my HackRF SDR and have not seen any radar events on the hostapd log. Seems like it can't sense radar signals. I know my SDR setup is working since I can get my Ubiquiti AP's to react.

If this is the case then support is not going to appear in the mainline kernel.

Anthony96922, 2023-02-22:
Looking at mt76x02_util.c, I see this:

```
void mt76x02_init_device(struct mt76x02_dev *dev)
...
        if (mt76_is_usb(&dev->mt76)) {
                hw->extra_tx_headroom += sizeof(struct mt76x02_txwi) +
                                         MT_DMA_HDR_LEN;
                wiphy->iface_combinations = mt76x02u_if_comb;
                wiphy->n_iface_combinations = ARRAY_SIZE(mt76x02u_if_comb);
        } else {
                INIT_DELAYED_WORK(&dev->wdt_work, mt76x02_wdt_work);

                mt76x02_dfs_init_detector(dev); /* <-- not enabled for usb */

                wiphy->reg_notifier = mt76x02_regd_notifier;
                wiphy->iface_combinations = mt76x02_if_comb;
                wiphy->n_iface_combinations = ARRAY_SIZE(mt76x02_if_comb);

                /* init led callbacks */
                if (IS_ENABLED(CONFIG_MT76_LEDS)) {
                        dev->mt76.led_cdev.brightness_set =
                                        mt76x02_led_set_brightness;
                        dev->mt76.led_cdev.blink_set = mt76x02_led_set_blink;
                }
        }
...
```

mt76x02_dfs_init_detector is only run for PCIe. Might be worth adding it for USB but I doubt the detection logic exists in the hardware.

morrownr, 2023-02-23:
My bet is that the logic is there and the reason the support is not active in the usb driver is because Mediatek did not go through the certification process for the usb adapters. USB is the step child of the wifi world.

VA1DER, 2023-07-10:
> My bet is that the logic is there

Adding mt76x02_dfs_init_detector() to the USB init doesn't work to add dfs support to my mt7610u.  I have also looked for other places where usb handling of dfs is different and so far not found any.  I would love to get dfs working on my mt7610 but  so far I'm failing.

Has anyone else managed to get this working yet?

morrownr, 2025-02-03:
Hi @UfukAKKAYA 

> Looking at the MT7612 eeprom content, there seems to be no DFS support?

It will work just fine with DFS channels in managed mode. DFS is not supported in AP mode but that is actually pretty standard for USB WiFi adapters.

What are you trying to do?

UfukAKKAYA, 2025-02-05:
> Hi [@UfukAKKAYA](https://github.com/UfukAKKAYA)
> 
> > Looking at the MT7612 eeprom content, there seems to be no DFS support?
> 
> It will work just fine with DFS channels in managed mode. DFS is not supported in AP mode but that is actually pretty standard for USB WiFi adapters.
> 
> What are you trying to do?

Hi @morrownr 

I want to buy a stable WiFi adapter that supports DFS. While researching before purchasing, I thought that adapters using the MT7612 chipset didn't support DFS. Do you have any recommendations for a USB adapter with DFS support? It would be sufficient if it works in managed mode.

Thanks.

morrownr, 2025-02-05:
@UfukAKKAYA 

Before I comment, here is a link to another user asking about DFS support:

https://github.com/morrownr/USB-WiFi/issues/560

Let me point something out. The first message in this thread has a comment that can be misleading:

> The mt7612u(n) doesn't support DFS...

That comment is about AP mode, not managed mode. Adapters based on the mt7612u most certainly do support DFS channels in managed mode. In fact, I am not aware of any USB WiFi adapters that cannot do DFS channels in managed mode. AP mode is a different subject and I am not aware of any USB WiFi adapters that can (legally) do DFS channels in AP mode. 

> It would be sufficient if it works in managed mode.

Then you should have no problem. I say that but the channels that are allowed does vary wildly from country to country when it comes to the 5 GHz band. You can take a look at The Plug and Play List as all of the dual band and tri band adapters there should support DFS channels in managed mode with no problem. 

> Also, there are no dfs-related tests in the fccid reports.

DFS tests are only needed for AP mode... where the device would be initiating radiation. In managed mode connecting to an AP, the adapter can passively scan channels to determine capability, it does need to initiate radiation.