morrownr/USB-WiFi #4: (looking for P2P-device mode)

RodBelaFarin, 2021-05-27:
Hi, sorry for bothering here at github, but it looks like this repo is made by the experts that I am looking for. I do need an USB-WiFi adapter that supports P2P-device interface mode.

Supported interface modes
* IBSS
* managed
* AP
* AP/VLAN
* monitor
* mesh point
* P2P-client
* P2P-GO
* P2P-device

Is there any device at all that supports this?

morrownr, 2021-05-27:
I just did a grep for "P2P-device" on the entire Linux 5.13 rc1 kernel source and got nothing. Zilch.

Can you point us toward something that would enlighten us as to what this is?

RodBelaFarin, 2021-05-27:
You should have done grep on one of the drivers you are hosting here ;)

https://github.com/morrownr/88x2bu/blob/5.8.7.4/os_dep/linux/ioctl_cfg80211.c#L621
https://github.com/morrownr/8814au/blob/main/os_dep/linux/ioctl_cfg80211.c#L591
https://github.com/morrownr/8812au/blob/5.9.3.2/os_dep/linux/ioctl_cfg80211.c#L631

This mode is for example by default enabled and integrated for the internal Raspberry Pi 3B+ and Pi4 WiFi. But I could not yet find ANY USB WiFi that has the this functionality.

morrownr, 2021-05-27:
Well, of course I didn't check right under my nose. 

I just checked the Linux kernel again for "P2P_DEVICE" and found a lot. The networking stack appears to have support and it appears your best bet is with the Intel "iwlwifi" driver. I don't see anything that would lead me to think there is support in any USB wifi adapter. I also checked line 591 in the 8814au driver here and I'm not sure what to say. I would have to do some deep research to figure out what needs to be done. I've managed to add and fix a few things here before but it could take a while. FWIW, I started github.com/morrownr about a year ago to stay busy as I was high risk for a bad outcome with C19. I had to dust off my old C books, I originally learned programming with FORTRAN so that should give you an idea of my age and I had not touched the code of anything resembling a device driver for nearly 20 years. What I found in these usb wifi drivers is the most complex stuff you can imagine. Some things can be done but it goes really slowly.

The Raspberry Pi Compute Module 4 and IO Board has a slot that might handle an Intel chipset based wifi card.

morrownr, 2021-05-27:
Hold on...

While looking around the kernel, I did see what appeared to be support in the Atheros drivers and Broadcom drivers so I decided to go look at what my ar9271 chipset based usb wifi adapter... here is a copy and paste of a few things:

Supported commands: * start_p2p_device

Supported RX frame types: * P2P-device: 0x40 0xd0

Supported TX frame types: * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0

It does not mention interface. I can paste the entire $ iw list for the adapter if you want.

Here is the adapter: https://store.rokland.com/collections/wi-fi-usb-adapters/products/alfa-awus036nha-802-11n-wireless-n-usb-wi-fi-adapter-2-watt

morrownr, 2021-05-28:
This just keeps getting more interesting...

Alfa AWUS036ACM Technical information
Supported interface modes
* IBSS
* managed
* AP
* AP/VLAN
* monitor
* mesh point
* P2P-client
* P2P-GO
VHT Capabilities (0x318001b0)
Supported interface combinations
* #{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1, total <= 3, #channels <= 2
* #{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1, total <= 4, #channels <= 1

Look at those last two lines... P2P-device. There is something going on here and we make need to figure it out. The main README has more information and links to the Alfa ACM adapter. The chipset is mt7612u and the driver is the mt76 in-kernel driver. I have and use one of these adapters. Outstanding adapter that works wonderfully with Raspi's.

RodBelaFarin, 2021-05-28:
Well, I am really impressed by your work! I can barely read the code for the drivers.

Just to make sure that we don't misunderstand each other: I do not need this functionality on my Raspberry Pi (instead I do need it for the Revolution Pi, which does not come with internal WiFi). I do need a USB WiFi Dongle for this case, which has the some modes available like the Broadcom chip on the Pi.

Running

`iw list | grep "Supported interface modes" -A 8`

on my Pi3B+ returns:

```
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
		 * P2P-device
```

Interesting is that the ALFA device says:

```
Supported interface combinations
#{ managed } <= 1, #{ P2P-device } <= 1, #{ P2P-client, P2P-GO } <= 1, total <= 3, #channels <= 2
#{ managed } <= 1, #{ AP } <= 1, #{ P2P-client } <= 1, #{ P2P-device } <= 1, total <= 4, #channels <= 1
```
But does not list

`		 * P2P-device`

as an option for the interface modes.

RodBelaFarin, 2021-05-28:
What I also found out is for the Realtek drivers, if you change the following flag from 0 to 1

https://github.com/morrownr/88x2bu/blob/5.8.7.4/os_dep/linux/ioctl_cfg80211.h#L34

than the device will also say that it's capable of 

`		 * P2P-device`

as interface mode.

BUT even if it says so, it still does not work. So it's not enough to just switch the flag from 0 to 1, there must be something else... I'm not sure, if it's software or hardware.

Also I don't know where to find this flag in the Broadcom driver.

Nevertheless your ALFA AWUS036 could be promising!

morrownr, 2021-05-28:
Quote: "BUT even if it says so, it still does not work."

Reply: That doesn't surprise me. I've ran into similar situations a few times in these Realtek drivers. What you found is interesting and I love a good mystery so if you can tell me a simply way to test if what you want is working, I will take a look.

Quote: "Also I don't know where to find this flag in the Broadcom driver."

Reply: I don't even know where the Broadcom source is.

Quote: "Nevertheless your ALFA AWUS036 could be promising!"

Reply: That is interesting. Too bad you don't have one to test... I do have one. Can you guide me through testing it?

FWIW: Below is the driver for the Alfa AWUS036ACM in case you are interested.

https://github.com/torvalds/linux/tree/master/drivers/net/wireless/mediatek/mt76

RodBelaFarin, 2021-05-29:
That's a bit more complex. I will write you a how-to later. Currently it's a nice summer day and I will enjoy the sun ;)

But to name the required steps:

First you need to switch from Network Manager to systemd-networkd and configure for your adapters.

Once that's done and working you can use the commandline tool wpa_cli.

Using that on the specific adapter: wpa_cli -i wlan0 (in case your adapter is named wlan0).

Now the single commands in wpa_cli are:

p2p_find (this will make your adapter discoverable for p2p connections and search for other devices)
now you need an Android phone, go to settings, wifi and wifi direct. Depending on your Android you might need to make your phone discoverable as well. You should probably now see your p2p device listed.

Next step is to connect your phone with your WiFi device. For that use in wpa_cli:

p2p_peers

This should list the mac address of your phone now. To connect:

p2p_connect <phones_mac_adress> pbc (pbc is for push button control)

NOW on your phone click on the device and connect. And now both devices should be connected via WiFi direct.

AND NOW the final check. You can quit wpa_cli and running

ip a

should now list a new adapter named p2p-wlan0-0

If this adapter appears, it works!

RodBelaFarin, 2021-05-31:
### systemd-networkd
**How-To enable systemd-networkd and disable NetworkManager**

```
sudo systemctl disable NetworkManager
sudo systemctl enable systemd-networkd
sudo systemctl enable systemd-resolved
sudo systemctl start systemd-resolved
```

**Configure network connections with system-network**

Create the following 3 files in `/etc/systemd/network`

1. `sudo nano 04-wired.network`
To configure all your wired adapters starting with e* (e.g. eth0, eth1...) copy and paste the following config to this file:

> [Match]
> Name=e*
> [Link]
> RequiredForOnline=no
> [Network]
> LLMNR=no
> LLDP=true
> EmitLLDP=nearest-bridge
> #LinkLocalAddressing=no
> MulticastDNS=yes
> DHCP=ipv4

2. `sudo nano 08-wifi.network`
To configure all your wireless adapters starting with wl* (e.g. wlan0, wlan1...) copy and paste the following config to this file:

> [Match]
> Name=wl*
> [Link]
> RequiredForOnline=no
> [Network]
> LLMNR=no
> #LinkLocalAddressing=no
> MulticastDNS=yes
> DHCP=ipv4

3. `sudo nano 12-p2p-wlan0.network`

To configure a virtual p2p adapter for your physical wlan0 adapter copy and paste the following config to this file:

> [Match]
> Name=p2p-wlan0-*
> [Network]
> Address=192.168.127.1/24
> Address=fd09:11:22:33::1/64
> IPv6PrefixDelegation=true
> DHCPServer=true
> [IPv6Prefix]
> Prefix=fd09:11:22:33::/64

**Configure wpa_supplicant**

`sudo nano /etc/wpa_supplicant/wpa_supplicant-wlan0.conf`
_hint: wlan0 has to be replaced by the name of your wifi adapter!_

Copy and paste the following config. Adjust your wifi ssid and password:

> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP>
> update_config=1
> country=DE
> device_name=$P2PNAME
> 
> p2p_go_intent=10
> 
> p2p_go_ht40=1
> 
> device_type=6-0050F204-1
> driver_param=p2p_device=6
> config_methods=virtual_push_button
> 
> network={
>     ssid="your_ssid"
>     psk="your_password"
>     priority=10
> }

**Create a service that will start wpa_supplicant**

`sudo nano /etc/systemd/system/wpa_supplicant@.service`

Copy and paste the following as it is:

> [Unit]
> Description=WPA supplicant daemon (interface-specific version)
> Requires=sys-subsystem-net-devices-%i.device
> After=sys-subsystem-net-devices-%i.device
> Before=network.target
> Wants=network.target
> 
> 
> [Service]
> Type=simple
> ExecStart=/sbin/wpa_supplicant -u -c/etc/wpa_supplicant/wpa_supplicant-%I.conf -Dnl80211 >
> 
> [Install]
> Alias=multi-user.target.wants/wpa_supplicant@%i.service

**Enable the service to run at system start**

```
sudo systemctl disable wpa_supplicant.service
sudo enable wpa_supplicant@wlan0.service
```
_hint: wlan0 has to be replaced by the name of your wifi adapter!_

RodBelaFarin, 2021-05-31:
After a reboot your system should now start with systemd-networkd and now you can test the p2p-device functionality.

### Test p2p-device

`sudo wpa_cli -i wlan0`
_hint: wlan0 has to be replaced by the name of your wifi adapter!_

Now in the interactive mode run the following step by step:

`p2p_find`

Now you need an Android phone, go to settings, wifi and wifi direct. Depending on your Android you might need to make your phone discoverable as well. You should probably now see your p2p device listed. Do **not** yet click on it.

Next step is to connect your phone with your WiFi device using wpa_cli:

`p2p_peers`

This should list the mac address of your phone now. To connect:

`p2p_connect <phones_mac_adress> pbc`
_hint: pbc is the push button control method._

Now on your phone you should see an invitation to connect. Accept AND NOW the final check. You can quit wpa_cli and run:

`ip a`

This should now list a new adapter named p2p-wlan0-*

If this adapter appears, it works!

morrownr, 2021-05-31:
Now to find time to do this. Hopefully tomorrow.

morrownr, 2021-06-03:
I haven't forgot about this. It is a matter of logistics and time. Life is getting in the way.

RodBelaFarin, 2021-06-04:
Don't worry. Take your time. I'm not in a hurry with that. Nevertheless I am also curious about the result 😁

morrownr, 2021-06-07:
@RodBelaFarin 

I have been overcome by events. I am in the process of moving so most of my little lab is now packed away. I am going to have to put this and most other things that require more than a quick comment on hold for now. If you would be so kind as to write a message to remind me of this in 2-4 weeks, I will hopefully by then be where I can do the testing. Regards.

RodBelaFarin, 2021-06-08:
OK. I will write a reminder in a couple of weeks ;)

RodBelaFarin, 2021-07-16:
Kind reminder to this issue. But it looks like the problem does not exist anymore with systemd version 248. They found a way to avoid the necessaty of P2P-device mode anymore. The P2P communication works fine without P2P-device mode now. So for me this issue is solved, but it's still interesting :D

morrownr, 2021-07-19:
That is interesting. I'm still not fully setup but I did notice in the latest version of the mt76 driver that P2P-device seems to have gone away. I get the feeling that somewhere out of sight there was a discussion about a better way to do this and now we are seeing the results.

closed 2021-07-27 by RodBelaFarin