morrownr/USB-WiFi #6: [help wanted]wpa_supplicant fail with driver wext but success with nl80211

aijunlu-ella, 2021-08-20:
wpa_supplicant v2.4
random: Trying to read entropy from /dev/random
Successfully initialized wpa_supplicant
Initializing interface 'wlan1' conf 'wpa_supplicant.conf' driver 'wext' ctrl_interface 'N/A' bridge 'N/A'
Configuration file 'wpa_supplicant.conf' -> '/home/test/wpa_supplicant.conf'
Reading configuration file '/home/test/wpa_supplicant.conf'
Line: 1 - start of a new network block
ssid - hexdump_ascii(len=7):
     74 65 73 74 61 62 63                              testabc         
PSK - hexdump(len=32): [REMOVED]
Priority group 0
   id=0 ssid='testabc'
WEXT: cfg80211-based driver detected
wext: interface wlan1 phy: phy1
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
rfkill: initial event: idx=2 type=1 op=0 soft=0 hard=0
SIOCGIWRANGE: WE(compiled)=22 WE(source)=16 enc_capa=0xf
  capabilities: key_mgmt 0xf enc 0x1f flags 0x0
ioctl[SIOCSIWAP]: Operation not permitted
WEXT: Failed to clear BSSID selection on disconnect
netlink: Operstate: ifindex=5 linkmode=1 (userspace-control), operstate=5 (IF_OPER_DORMANT)
Add interface wlan1 to a new radio phy1

morrownr, 2021-08-20:
What hardware and driver is involved?

aijunlu-ella, 2021-08-25:
> What hardware and driver is involved?

model: https://www.amazon.com/1200Mbps-Network-Adapter-866Mbps-300Mbps/dp/B07RQG97GX
driver:  https://github.com/morrownr/88x2bu.git