morrownr/USB-WiFi #7: mt7612u bug in 5g band

hac425xxx, 2021-08-23:
Hello, my most recent test found that when mt7612u is in monitor mode, it cannot send data packets in the 5g band, such as channel 36.

Do you know how to solve this problem?

morrownr, 2021-08-23:
Can you write me a little checklist of the steps I can use to duplicate the problem? I should have time to test this today.

Nick

hac425xxx, 2021-08-24:
first set channel of monitor wlan0

```
sudo iwconfig wlan0 channel 36
```

then use libpcap to send packet 

```
./pcap-test wlan0
```

code

```
/* GNU 2016
 *
 * 80211 MGMT Frame Simple Sniffer written in C by
 * Douglas Berdeaux, 2016 - weaknetlabs@gmail.com
 *
 * Listens for a beacon and prints data if one found
 * exits if not.
 *
 * Version 1.1
 *
 */
#include <stdio.h>      // for simple IO
#include <stdlib.h>     // for malloc();
#include <pcap.h>       // for all PCAP functions
#include <netinet/in.h> // for the uint8_t

struct ieee_hdr
{
    uint8_t type;
    uint8_t flags;
    uint16_t duration;
    unsigned char addr1[6];
    unsigned char addr2[6];
    unsigned char addr3[6];
    uint16_t frag_seq;
} __attribute__((packed));

// This struct is the RadioTap header: https://radiotap.org
struct radiotap_header
{                    // RadioTap is the standard for 802.11 reception/transmission/injection
    uint8_t it_rev;  // Revision: Version of RadioTap
    uint8_t it_pad;  // Padding: 0 - Aligns the fields onto natural word boundaries
    uint16_t it_len; // Length: 26 - entire length of RadioTap header
};

unsigned int get_radiotap_size(unsigned char *data, unsigned int len)
{
    if (len > 4)
    {
        return (((uint16_t) * (data + 3)) << 4) | *(data + 2); // little endian to big endian
    }

    return 0;
}

unsigned char beacon_data[] = "\x00\x00\x0f\x00\x2e\x00\x00\x00\x10\x02\x85\x09\xa0\x00\xfa\x80"
                              "\x00\x00\x00\xff\xff\xff\xff\xff\xff\x08\x10\x7b\xbf\xcc\x93\x08"
                              "\x10\x7b\xbf\xcc\x93\xf0\xa9\x12\xa4\x64\x5a\x00\x00\x00\x00\x64"
                              "\x00\x11\x00\x00\x05\x33\x37\x6c\x61\x62\x01\x04\x82\x84\x0b\x16"
                              "\x03\x01\x06\x05\x04\x00\x01\x00\x00\x30\x14\x01\x00\x00\x0f\xac"
                              "\x04\x01\x00\x00\x0f\xac\x04\x01\x00\x00\x0f\xac\x02\x00\x00\xdd"
                              "\x07\x00\xe0\x4c\x00\x02\x00\x04\xca\xaa\xa7\x87";

void usage(void); // function prototypes

// entry point for loader/app:
int main(int argc, char **argv)
{ // main function
    if (argc >= 2)
    { // argument to contain a wireless device name
        int offset = 0;
        char *erbuf;   // for errors (required)
        char *dev;     // place to store device name
        dev = argv[1]; // get wlan device from command line
        pcap_t *handle;
        // printf("%s\n",pcap_lib_version()); // DEBUG
        handle = pcap_open_live(dev, BUFSIZ, 0, 3000, erbuf);
        if (handle == NULL)
        {
            printf("ERROR: %s\n", erbuf);
            exit(1);
        } // was the device ready/readable?

        while (1)
        {

            // struct pcap_pkthdr header;
            // u_char *packet = pcap_next(handle, &header);
            // unsigned int rt_size = get_radiotap_size(packet, header.caplen);
            // struct ieee_hdr *hdr = packet + rt_size;
            // unsigned int pkg_len = header.caplen - rt_size;

            // unsigned char *bssid = hdr->addr2.octet;

            pcap_sendpacket(handle, beacon_data, sizeof(beacon_data));
            usleep(10);

            // fprintf(stdout, "BSSID string: %02X:%02X:%02X:%02X:%02X:%02X\n", bssid[0], bssid[1], bssid[2], bssid[3], bssid[4], bssid[5]);
        }

        return 0; // good bye main()!
    }
    else
    { // no argument, display usage:
        usage();
        return 1;
    }
}

// print how to use the application:
void usage(void)
{                                                        // display how to use application
    fprintf(stderr, "Usage: ./80211sniff deviceName\n"); // function in stdio.h
    return;                                              // print to /dev/stderr
}
```

hac425xxx, 2021-08-24:
Then use another machine to capture the packet, you can find that there is no wifi frame

wireshark filter

```
wlan.sa == 08:10:7b:bf:cc:93
```

hac425xxx, 2021-08-24:
If in the 2.4 GHz frequency band （channel 6）, we can catch the beacon frame sent by pcap-test

morrownr, 2021-08-24:
Okay. This looks interesting. I'll reconfigure one system in my lab to make this a little easier and then I'll give it a try. I am a little busy today so it may take a couple of days.

morrownr, 2021-08-30:
Been a little too busy these last few days. I'll try to find time this week to take a look. Is this issue still active?

hac425xxx, 2021-08-31:
yes, it's still ative

morrownr, 2021-09-07:
Okay. Life keeps getting in the way. I'll try to work on this sometime this week.

morrownr, 2021-09-11:
I attempted to try to duplicate the problem this past Thursday as I had a little extra time in the afternoon of that day.

The code you posted would not compile. I chased around and found that my system needed a file so I installed it. Compile still failed. I changed out a function that was causing problems. It finally compiled but does not work.

Do you have an alternate way for me to try to duplicate the problem?

hac425xxx, 2021-09-12:
You need to manually set the network card to monitor mode first, you can use `sudo airmon-ng start wlan0.`

hac425xxx, 2021-09-12:
compile command
```
gcc pcap-test.c -o pcap-test -lpcap
```
you should install libpcap-dev in ubuntu or kali, I try this on kali.

morrownr, 2021-09-13:
I appreciate the info. I had already figured out the need to install libpcap-dev and what the compile command needed to be but it would have saved me some time if you had included that info commented out in the source code.

The best system I have to test with right now is a RasPi4b running RasPiOS: Here is what I get:

pi@rpi4b:~/src/pcap-test $ ./pcap-test wlan0mon
Segmentation fault

I do have an amd64 system that has Kali installed but it is a desktop system and it is not set up yet. I recently moved and my little lab is not fully setup due to life getting in the way. I will put a priority on getting the Kali system setup this week.

morrownr, 2021-09-14:
I tested on a 64 bit amd64 system running up to date Kali and I get the same thing:

Segmentation fault

hac425xxx, 2021-09-14:
could send the crash context for me, such as debug  with  gdb

morrownr, 2021-09-20:
It was my intent to use gdb. Sorry about the short reply above. 

Right now I am neck deep in high priority projects. I will get back to this when able. Keep me posted if new info is available.

morrownr, 2021-10-01:
I have some high priority projects that require my attention so I have limited time, for now, to chase issues like this. It could be a while before I can reengage on this issue. I do have some good news though...

Since your are using an in-kernel driver, you can report the issue directly to the kernel devs that work on the driver. Go to the following site for instructions:

https://wireless.wiki.kernel.org/en/users/documentation/reporting_bugs

Advice:

1. Follow the bug reporting guidance very carefully as you are submitting an issue to folks that are very busy.
2. I recommend you find a utility or app that is in the distro repo that you can use to duplicate the issue. Sending source without adequate information about prerequisites is not going to help you get this fixed.
3. I monitor linux-wireless. It is a busy list. It might be a good idea to subscribe and monitor while you put the bug report message together over a few days time.

Regards

morrownr, 2021-11-29:
Closing due to inactivity.

closed 2021-11-29 by morrownr