morrownr/USB-WiFi #8: Great Survey

gordboy, 2021-09-01:
This is not an issue, more of a thank you.

I was looking for an excuse to drop support for the rtl8812au and now I have it ... :)

Now I can get a cheapo MTK7612u dongle from Amazon for £12.99 and forget about all this Realtek Malarkey.

Thanks for all your Great Work.

morrownr, 2021-09-01:
Hey Gord, good to hear from you.

Malarkey... I like that word and it describes Realtek support perfectly. Moving over to an adapter with a mt7612u chipset gives us the ability to report problems and suggest features. Hey we can even submit changes that could end up in the Linux kernel. How cool is that?

Nick

morrownr, 2021-10-20:
You are welcome. How is that Alfa adapter working out for you?