morrownr/USB-WiFi #13: ROW02FD in master mode hostapd

cannonfodder728, 2021-11-07:
Can anyone confirm that they got this adapter working in master mode using hostapd for the 5g channel?  

I'm trying to migrate my raspberry pi access points away form  realtek adapters and wanted to confirm someone got it working as an address point before purchasing.  I plan on running Ubuntu on the pis

Thanks

morrownr, 2021-11-07:
> Can anyone confirm that they got this adapter working in master mode using hostapd for the 5g channel?

Yes, this adapter works well for 5 GHz AP mode with hostapd. Here is a repo that contains info that might be of help:

https://github.com/morrownr/7612u

> I'm trying to migrate my raspberry pi access points away form realtek adapters and wanted to confirm someone got it working as an address point before purchasing. I plan on running Ubuntu on the pis.

My own preference is to run AP mode with RasPiOS on Pi's. I have tried Ubuntu but have found RasPiOS to be lighter and easier to setup and maintain than either of the Ubuntu's (desktop or server.)

Good luck and let me know if you have any questions.

morrownr, 2021-11-16:
@cannonfodder728 

I just tested the very recently released RasPiOS. It still has the old usb_modeswitch so...

https://github.com/morrownr/USB-WiFi/blob/main/How_to_Modeswitch.md

I have no idea why this happened. I guess I will report it and see what happens.

closed 2022-05-21 by morrownr