morrownr/USB-WiFi #17: Need suggestion

vvkvas, 2021-11-28:
FOA thank you morrownr for detailed reviews of all these adapters and providing us drivers. I recently purchased Alfa AWUS1900 for $38.43(yes it is original), but after reading your reviews and also with my own experience of spending 1 whole day trying to make it up in a new kali installation I feel like to return it and buy new Alfa AWUS036ACM which will cost me almost same. I'm a new user both to linux and penetration, which one in your opinion should I go with, Alfa AWUS036ACM & Alfa AWUS1900 are costing me almost same and Alfa AWUS036ACHM will cost a little more but than that I have doubt about its speed. 
If I buy 9dbi separate antennas along with Alfa AWUS036ACM will its performance be equal or better as compared to AWUS036ACHM if signal is weak?
Please suggest any from these 3 considering me as a newbie, for a long term and cost effective purchase. Thank you

morrownr, 2021-11-29:
> thank you morrownr for detailed reviews of all these adapters and providing us drivers.

You are welcome. I try hard. I don't always do a good job but I give it my best shot and I get a lot of help.

> I recently purchased Alfa AWUS1900 for $38.43(yes it is original), but after reading your reviews and also with my own experience of spending 1 whole day trying to make it up in a new kali [...]

I remember my first Linux installation. I think it was in '95. The installation took some time.

> which one in your opinion should I go with, Alfa AWUS036ACM & Alfa AWUS1900 are costing me almost same and Alfa AWUS036ACHM will cost a little more but than that I have doubt about its speed.

I'll just throw out what I think I know and maybe that will help you make a decision:

I have access to several USB WiFi adapters as you might expect. The other day someone asked me "if you could only have one adapter, which one would it be?" I have thought about that question a lot. I have argued with myself. I use adapters for client mode, AP mode and security analysis. So, which adapter would keep me the happiest if I could only have one adapter? I am pretty sure the answer is the Alfa AWUS036ACHM. Here is some of my thinking on the topic:

What is important in client (managed) mode? Speed, reliability and ease of use perhaps? While that AC1900 looks good on paper, I cannot tell the difference between a good AC600 and AC1900 adapter when I am in client mode. The reason is that my internet access is 100 Mb/s cable. The AC600 adapters are very capable of twice that speed so it is hard to tell a difference in use. I moved earlier this year and used to have 1 Gb/s fiber. Same thing. Reliability and ease of use. The ACM and ACHM both use very good in-kernel drivers that are maintained with us looking on. Are they perfect drivers? No, but they are very good and getting better. The AC1900 is using a driver that Realtek seems to have forgotten about. There have been no new releases since 2019 and that was not their finest effort. My opinion is that if a person is determined to go with a Realtek chipset, go with the Alfa ACH. The driver for the 8812au is pretty good for an out-of-kernel driver. Yes, the ACH is expensive. For the price, the ACHM meets almost all speed requirements and will for a long time and it is super reliable and all you have to do is plug it in.

What is important in AP mode? Rock solid stability and 100% uptime. Also, the ability to push the amount of data needed at the speeds needed. The Realtek 8814au driver has a history of being problematic. We have patched it many times and tried many things. We wish Realtek would provide an updated version but we see nothing. On the other hand, the ACM and ACHM are rock solid. My ACHM is pulling AP duty here and has been for a couple of months. I forgot where I put it the other day and remember it was powering our Roku's (streaming TV) in FHD. You know how dependable something is when it has not had your attention for 2 months while working 24/7. The ACM also falls into this category as it pulls duty as a general purpose AP on a RasPi4B. It just works. The 8814au based adapter can't do this and it is because of the driver.

What is important in monitor mode?

There are 3 things that are important in monitor mode: Range, range and range. Okay, well, maybe compatibility with Aircrack-ng and Wireshark and the like is in there somewhere. The dual band adapters that have the longest range are the Alfa ACHM and ACH. Here is a head to head [test](https://github.com/morrownr/USB-WiFi/blob/main/Performance_Comparison.md). Compatibility with needed apps. The Mediatek drivers are simply better so the ACHM and ACM have to be considered.

For learning security analysis on Kali, I would rate adapters in this order with all 3 being very good:

1. Alfa ACHM (impressive range) (superb compatibility) 
2. Alfa ACM (good range)  (superb compatibility)
3. Alfa ACH (impressive range) (good compatibility)

 > If I buy 9dbi separate antennas along with Alfa AWUS036ACM will its performance be equal or better as compared to AWUS036ACHM if signal is weak?

I don't know the answer.

 > Please suggest any from these 3 considering me as a newbie, for a long term and cost effective purchase. Thank you.

As time passes and you have more experience, you will have a better idea of exactly what is important for what you are trying to do. These adapters are very flexible and are used in a lot of different ways. I don't know what will work best for you long term. If budget is a concern right now, look at something like the TEROW ROW02CD. The ANDDEAR - MTK7612U004 might be worth looking at as well.

I hope I have provided some information that helps you.

Regards

vvkvas, 2021-11-30:
Thanks a lot for your detailed explanation, this is really very informative!
A lot of things are clear now, such as, I need external adapter for pen-testing only, for speed my laptop already has ax adaptor. I should prefer adaptor which has better driver. Now I'm definitely going to return AC1900 and will choose from Alfa ACHM or Alfa ACM probably later one. I checked TEROW ROW02CD is not available in my country and shipping it from US will cost me almost same as I'm getting Alfa ACM for, so I believe Alfa ACM will be better comparatively. I now plan try to get Alfa ACM and will also get 9dbi antennas then will compare range, as they both will cost me equal to Alfa ACHM and as I now know that only thing better in Alfa ACHM from Alfa ACM is range and link quality, and getting 9dbi antennas should mitigate that, I plan to try with download speed test and check signal quality using my mobile at different ranges. I will try in next few days maybe this weekend and will let you know the results here. If you suggest anything else for me to test, then please let me know. And how did you check power usage? Using USB meter in between?

morrownr, 2021-11-30:
> Thanks a lot for your detailed explanation, this is really very informative!

Come back and let me know in a few months if it was helpful but thanks.

 >A lot of things are clear now, such as, I need external adapter for pen-testing only, for speed my laptop already has ax adaptor.

Okay. This is information that I was unaware of. Maybe my eyes are worse than I thought. If your usb wifi adapter is only for use in pen-testing and it will be used on a laptop... then I am going to push you toward the ACHM. Here is why: Remember the link in the previous message where I had performed what basically amounts to range testing? The two winners of the range testing were the ACHM and ACH. They are very close in capability as far as range goes. However, when it comes to quality of driver for pen-testing and the portability of the adapter, the ACHM speeds across the finish line in first place.

I like the ACM. It is a very high quality adapter and it is my first choice if you are looking for a 5 GHz AP or a really good general purpose adapter that is primarily going to be used in a static location. For pen-testing, where range and portability are 2 of the 3 most important things you are looking for, it is less portable than the ACHM... and while it has good range, it is not designed to be a Max Range product like the ACHM. You may see improvement with the antennas on the ACM but that is more crap to carry around.

> I should prefer adaptor which has better driver.

Absolutely. This quality of driver thing can make your life happy or sad. There is no question that the in-kernel Mediatek drivers are better for what you are doing and something else to consider... once drivers meet the standards to be included in the kernel, they tend to stay in the kernel for a long time. Linus was working on the floppy disk driver earlier this year... and yes, I still have a couple of operational system with floppy drives so I care. I also have an 8 bit FM radio card, that when last I checked, was still supported. My point here being that these adapters based on mt761xu chipsets will likely have good support for at least the next 10-20 years.

 > And how did you check power usage? Using USB meter in between?

Yes, I use a meter in between. 

Regards

vvkvas, 2021-11-30:
I agree with your almost every point, but being a newbie in penetration testing but an experienced networking professional, this point of yours I'm not able to digest:

> It is a very high quality adapter and it is my first choice if you are looking for a 5 GHz AP or a really good general purpose adapter that is primarily going to be used in a static location. For pen-testing, where range and portability are 2 of the 3 most important things you are looking for,

The reason is, if I'm pentesting an environment, for eg. in any company, then I will be testing their wireless security not their range or wireless performance, and I guess I will set up my laptop and tools at static point, please excuse my noobness but I can't figure out a reason that why will I be testing from far, yes if I'm doing something illegal then for sure I'd like to stay as far away as possible, but if I'm authorized to test then I'll most probably choose a better location with good signals, isn't it?

Also, please correct me if I'm wrong, isn't new infra moving to 5+Ghz such as ac/ax, and as per my belief won't 2.4 Ghz will be less used in coming future due to low speed and bandwidth? So, if I buy ACHM will it be future proof, not just driver prospective but usage prospective, I mean won't I be required to test more 5+Ghz network?

Best Regards

vvkvas, 2021-11-30:
> when it comes to quality of driver for pen-testing and the portability of the adapter, the ACHM speeds across the finish line in first place.
ACM also has mt7612u chip which has in-kernel drivers too, so how it's quality of driver will be inferior than ACHM?
I'm sorry, I'm not at all implying that what you are saying is wrong, it's just that I'm curious to know more about a few things. And thank you again for your time and explanations! 

Best Regards

morrownr, 2021-11-30:
> I agree with your almost every point, but being a newbie in penetration testing but an experienced networking professional, this point of yours I'm not able to digest:
>
>
> The reason is, if I'm [...]

I think we have two different visions of the size of the locations you will be working at. I shouldn't have made any assumptions but I did and I think I may have missed the mark. I was thinking that you would end up with jobs often in large facilities. The larger facilities would make it difficult to work from a static location and you would be left thinking to yourself, "darn, wish I had an adapter with longer range."

 >please excuse my noobness but I can't figure out a reason that why will I be testing from far or why I will be moving(in most scenarios), yes if I'm doing something illegal then for sure I'd like to stay as far away as possible, but if I'm authorized to test then I'll most probably choose a better location with good signals, isn't it?

I wasn't thinking of unauthorized testing, I was thinking about larger facilities than you but then you know your situation and I don't.

> Also, please correct me if I'm wrong, isn't new infra moving to 5+Ghz such as ac/ax, and as per my belief won't 2.4 Ghz will be less used in coming future due to low speed and bandwidth? So, if I [...]

I get the feeling that you think the ACHM is a single band adapter. The ACM, ACH and ACHM are all dual band adapters. All 3 are capable of 80211ac. The ACHM can do 5 GHz just as well as the other two. The difference is possible link speeds in 5 GHz. The ACHM can do a link speed of up to 434 Mb/s while the other two can do up to 867 Mb/s. Is the difference important in pen-testing? It may be but nothing that anyone in the security analysis business has said to me has indicated that it is a problem. On the other hand, lack of range and non-standards compliant drivers and adapters that get broken are things I hear a lot.

I have a couple of little scripts that I have been working on to help pen-testers start monitor mode and test monitor mode. Would you be interested in testing them?

Regards

vvkvas, 2021-12-01:
>  I was thinking that you would end up with jobs often in large facilities. The larger facilities would make it difficult to work from a static location and you would be left thinking to yourself, "darn, wish I had an adapter with longer range."

Even if I get to work in larger facilities, still I can't figure out why will I have to move, because even in large facilities network will be same, it will be multiple APs at a regular distance connected to most probably a single controller. Again I would have to test their wifi security not performance, please correct if I'm wrong here? And with my own experience as a networking professional whenever I have visited any client site it was their responsibility to provide me access, be it directly via console connection in DC, or a LAN port in their NOC, or a wireless connection in corridor. Similar way I think is if I will go to any site for testing they will have to give me a place where I can get good signals, and if not, that time I will be a professional and I would preferably keep a separate stronger antenna for those kind of scenario. Well, when I'm a pentesting professional I will have to be ready for multiple scenarios, signal issue I can most probably fix with a stronger antenna but maybe sometime I might need good speed which I can't for ACHM, maybe I will have to buy other hardware, and for portability, best scenario would be in-built wifi chip in a laptop which is rare, I will even have to carry ACHM so as ACM. But, for now I'm just starting and will mostly be working with my own wifi/network, and it maybe few months or few years even for me being a professional one and by that time maybe we have another adaptor which may even support wifi 6! There can be hundreds of scenario in future! :)

> I get the feeling that you think the ACHM is a single band adapter. The ACM, ACH and ACHM are all dual band adapters. All 3 are capable of 80211ac. The ACHM can do 5 GHz just as well as the other [...]

No, I know this, ACH is costliest and again Realtek chip, I'd like to stay away from it now. Now I have to choose from ACM and ACHM, I said this because as per you:

> I like the ACM. It is a very high quality adapter and it is my first choice if you are looking for a 5 GHz AP or a really good general purpose adapter that is primarily going to be used in a static location.

Shouldn't I get something which is better for 5 GHz?

If I talk about today, ACM is cheapest, also it has in-kernel driver, good speed and other than lesser range from ACHM I don't see any practical reason to not go with ACM. I believe there is no performance issue with ACM, for example if it is in same room as wifi? For long time usage and driver quality ACM is as trustworthy as ACHM, right? And I don't think range of ACM will be very bad, maybe if ACHM range is 10 of 10 then ACM would be maybe 8or9 of 10? Please confirm on these, and if you still think I should go with ACHM, I will go with ACHM for sure. :)

> I have a couple of little scripts that I have been working on to help pen-testers start monitor mode and test monitor mode. Would you be interested in testing them?

I would love to, please do share!

Best Regards

morrownr, 2021-12-02:
> Even if I get to work in larger facilities, still I can't figure out why will I have to move, because even in large facilities network will be same, it will be multiple APs at a regular distance connected to most probably a single controller.

That may be true if you designed and built the network but I have seen some really interesting networks.
 
> > I like the ACM. It is a very high quality adapter and it is my first choice if you are looking for a 5 GHz AP or a really good general purpose adapter that is primarily going to be used in a static location.
> 
> Shouldn't I get something which is better for 5 GHz?

You will have to define the word "better" before I can offer an answer.

> If I talk about today, ACM is cheapest, also it has in-kernel driver, good speed and other than lesser range from ACHM I don't see any practical reason to not go with ACM.

You are welcome to go with the ACM. You are doing what many folks do not do, you are investigating before buying. The problem is that there are so many factors to consider that it is hard to make a perfect decision. Alfa has been the choice of many pen-testers over the years for good reason. Their adapters are well made, tend to hold up while being toted around and have much better range than most. Those qualities do come with a price.

I have an ACM and an ACHM. If you want me to do any specific testing, please let me know.

> I believe there is no performance issue with ACM, for example if it is in same room as wifi?

If in the same room you can expect the ACM to handle 380-400 Mb/s in client or AP mode. An AC1200 Realtek chipset, such as the 8812au, will show 410-440 given the same conditions. In the same class, my experience is that Realtek chipsets are about 10% faster than Mediatek chipsets. For me, that is not enough difference to matter and there are other issues that end up leveling the playing field. Mediatek chipsets use less power and tend to run cooler. The more compatible and trouble free drivers also make Mediatek look better.

> For long time usage and driver quality ACM is as trustworthy as ACHM, right?

I have no evidence to the contrary.

> And I don't think range of ACM will be very bad, maybe if ACHM range is 10 of 10 then ACM would be maybe 8or9 of 10? 

The range of the ACM is not bad at all. When comparing the adapters, the thing that gets my attention is how impressive the [range](https://github.com/morrownr/USB-WiFi/blob/main/Performance_Comparison.md) of the ACHM is. The ACH is also impressive but it is very expensive and uses Realtek drivers. If I test injection on a channel with many systems on it around here, I will get good results with the ACM but due to distance many systems will come back showing 30% or 50% and then I pop the ACHM in and I see a nearly all reports at 100%. It is almost like some Mediatek engineers said "hey, let's make an adapter for the pen-testers." 

I don't have to estimate [8 or 9 out of 10](https://github.com/morrownr/USB-WiFi/blob/main/Performance_Comparison.md). The difference is 93 to 79. 

> Please confirm on these, and if you still think I should go with ACHM, I will go with ACHM for sure. :)

I recommend you do what feels right for you. Either adapter should work better than most other options out there. By the time you are good at the job and accidentally run over your adapter with your car, it happens, then you will know what is best for you. Maybe Alfa will have a good WiFi 6e adapter available at some point next year.

> > I have a couple of little scripts that I have been working on to help pen-testers start monitor mode and test monitor mode. Would you be interested in testing them?
> 
> I would love to, please do share!

[test-mon.sh.tar.gz](https://github.com/morrownr/USB-WiFi/files/7639247/test-mon.sh.tar.gz)

It is a work in progress. Open the file in a text editor to read the docs.

Regards

vvkvas, 2021-12-04:
> I recommend you do what feels right for you. Either adapter should work better than most other options out there. 

Thank you very much, now my first choice will be ACHM and second will be ACM, I've ordered ACHM will compare both once received, and then will keep either or both.

> If in the same room you can expect the ACM to handle 380-400 Mb/s in client or AP mode. 

What is the ideal way of testing the wifi speed? I try to check with iperf3 with either enabling hotspot on phone or W10 but in that case speed is not more that 110mbps with any adaptor, just now I checked with connecting both devices via my wifi router and speed reached about 150mbps but it is still less?

> test-mon.sh.tar.gz
It is a work in progress. Open the file in a text editor to read the docs.

Was busy so could not test it in deep, at the moment I can suggest to use macchanger -r to set random mac address as pasting given mac address is a bit confusing and I think it'll be better to use random mac every time. I have few other points but will test during the day today again and will let you know.

Best Regards

vvkvas, 2021-12-05:
> test-mon.sh.tar.gz

Issue# 2
When I run this script and cancel in between and want to run again, but it won't run:
Output before running:

> ❯ iwconfig
lo        no wireless extensions.
eth0      no wireless extensions.
wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
❯ iw dev
phy#3
        Interface wlan0
                ifindex 9
                wdev 0x300000001
                addr 00:00:00:00:00:00
                type managed
                txpower 20.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0

When I do Ctrl+C:

> test-mon.sh 20211130
    WiFi Interface:             wlan0
    name  -  wlan0mon
    type  -  monitor
    state -  UP
    addr  -  00:00:00:00:00:00
 airodump-ng can receive and interpret key strokes while running...
 a - select active area
 i - invert sorting order
 s - change sort column
 q - quit
 Do you want run airodump-ng to display a list of detected access points and clients? [y/N] ^C

> ❯ ./test-mon.sh
Cannot find device "wlan0"
ERROR: Please provide an existing interface as parameter!
Usage: $ sudo ./test-mon.sh [interface:wlan0]
Tip:   $ iw dev (displays available interfaces)
❯ ./test-mon.sh [interface:wlan0]
Not enough information: "dev" argument is required.
ERROR: Please provide an existing interface as parameter!
Usage: $ sudo ./test-mon.sh [interface:wlan0]
Tip:   $ iw dev (displays available interfaces)
❯ ./test-mon.sh [interface:wlan0mon]
Not enough information: "dev" argument is required.
ERROR: Please provide an existing interface as parameter!
Usage: $ sudo ./test-mon.sh [interface:wlan0]
Tip:   $ iw dev (displays available interfaces)

> ❯ iwconfig
lo        no wireless extensions.
eth0      no wireless extensions.
wlan0mon  IEEE 802.11  Mode:Monitor  Frequency:5.18 GHz  Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
❯ iw dev
phy#3
        Interface wlan0mon
                ifindex 9
                wdev 0x300000001
                addr 00:00:00:00:00:00
                type monitor
                channel 36 (5180 MHz), width: 20 MHz (no HT), center1: 5180 MHz
                txpower 20.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0

morrownr, 2021-12-05:
> > test-mon.sh.tar.gz
> 
> Issue# 2 When I run this script and cancel in between and want to run again, but it won't run: 

When you cancel before the script ends and resets the interface to the original name, the interface is left with the name `wlan0mon` so to start the script in this situation:

$ sudo ./test-mon.sh wlan0mon

If you only have one wireless interface, you can change the following inside the script:

# Activate option to set automatic or manual interface mode
#
# Option 1: if you only have one wlan interface (automatic detection)
#iface0=`iw dev | grep 'Interface' | sed 's/Interface //'`
#
# Option 2: if you have more than one wlan interface (default wlan0)
iface0=${1:-wlan0}

Comment out option 2 and uncomment option 1 and the script will automatically parse the interface name. There is probably a better way to do this. Ideas are welcome.

Let me clarify what I am trying to accomplish with this script:

I am not trying to write an app for general use. What I am trying to do is write a script that will speed up the testing process. When I work on the drivers, I have found that it takes a lot of time to go through the process of testing monitor mode operation to make sure things are working the way they need to work. I may be wrong but it seems that those doing pen-testing could use a good script to test their setups. I'm wondering if setting up a separate repo for this script would be as good idea as it would allow people to submit their proposed changes and additions.

Nick

vvkvas, 2021-12-07:
> $ sudo ./test-mon.sh wlan0mon

My bad I was using wrong syntax. 

> Comment out option 2 and uncomment option 1 and the script will automatically parse the interface name. There is probably a better way to do this. Ideas are welcome.

For now I use in vmware with wifi and bridged connection, will try again with few changes including using 2 adaptors and will see if with this I remain with Internet access.

> I may be wrong but it seems that those doing pen-testing could use a good script to test their setups. I'm wondering if setting up a separate repo for this script would be as good idea as it would allow people to submit their proposed changes and additions.

Script is a good idea but I think as a newbie that first I need to do manually then I can move to use script, and creating a separate repo is also a good idea, like you said features can be suggested and added, and it can also be useful to newbies like me who can see code and learn and may even modify/create own/suggest new features. 

Could you also please suggest best way to test speed of an adaptor?

Best Regards

morrownr, 2021-12-09:
> Could you also please suggest best way to test speed of an adaptor?

I generally use iperf3.

$ sudo apt install iperf3

You need one copy running in server mode on one system or router and another copy running on a client in client mode...

$ iperf3 -s

$ iperf3 -c 192.168.1.1

usama7628674, 2021-12-18:
@vvkvas Both, ACH and ACHM are high power adapters, which means both have in-built wifi-amplifier which is why both have better range compared to ACM. Another thing to notice is injection rate of ACH is crap, but ACHM has very strong injection rate. Same goes for ACM, it has less injection rate. You could argue ACH is high power adapter then why it has crap injection rate? The driver is bad. If I were you, I would choose ACHM.

vvkvas, 2021-12-18:
> When comparing the adapters, the thing that gets my attention is how impressive the range of the ACHM is. 

Just received ACHM and now I get it why you are so impressed by it! Currently I have ACHM, ACM & AWUS1900 and I tested all 3.
In my test speed and range of ACM & AWUS1900 are almost same. In my test speed of ACHM is almost half of other two but range is almost double!
In close speed test using iperf3 I get ~120mbps speed in ACM & AWUS1900 and I went to 2nd floor of my building and 2.4gh was connecting with fair speed and 5gh was not connecting, on 3rd floor there was no signal for both.
In ACM I get 60mbps U & 50mbps D speed at close range. I again tested on 3rd floor and I was able to connect and surf Internet as well with a speed of about 1-2mbps for 2.4gh! For 5gh there was weak signal and was unable to connect. I was able to connect 5gh on 2nd floor with fair speed.

> ACHM has very strong injection rate.

I'm sorry but what exactly does it means? Maybe could you please explain with an example?

> I again tested on 3rd floor and I was able to connect and surf Internet as well with a speed of about 1-2mbps for 2.4gh! For 5gh there was weak signal and was unable to connect.

Although I've made up my mind to keep ACHM and maybe ACM too, but I'm just curious to know that in this scenario when signal is not high enough to connect but I can still see it, will this be enough for say capturing handshake packets or something like that? Or maybe deauth attack or any other attack?
And is speed relevant for attacking or penetration testing? As on 3rd floor I was getting about 1mbps upload and <1mbps download speed will this be sufficient?

> I generally use iperf3.

I've been using the same but when I measure speed by enabling hotspot on my laptop using AWUS1900 or even on my mobile I get around 100mbps speed and when I try to connect both laptop and mobile via my wifi router I get around 150mbps speed. AWUS1900 has theoretical 800+mbps speed then why I'm getting max 150mbps on 5gh channel on USB 3 port, and on my mobile also I get 250mbps speed with my 300mbps wifi connection? So what wrong I'm doing in testing wifi adaptor speed?

Best Regards

morrownr, 2021-12-20:
Hi @vvkvas 

> > When comparing the adapters, the thing that gets my attention is how impressive the range of the ACHM is.
> 
> Just received ACHM and now I get it why you are so impressed by it!

It is one of those things where you can talk about it all day long but you really don't get it until you have and use the ACHM. When I first got mine, I put it to work and did not notice that it is really an exceptional little adapter. Later on, I decided to include it on some tests I was doing and I had to keep redoing the tests as I could not believe how it was smoking other adapters when it came to range. 

 > Currently I have ACHM, ACM & AWUS1900 and I tested all 3. In my test speed and range of ACM & AWUS1900 are almost same.

This is good info to know. I have an ACM and it is a good adapter. I don't have a 1900 but would expect its range to be about what the range of the ACM is.

> In my test speed of ACHM is almost half of other two but range is almost double!

Yeah. You sound like me. How does it do that? Good antennas and a good amp. In the days of single band adapters, there were several with good range but to find competitive range in a modern dual band adapter was a surprise.

> > ACHM has very strong injection rate.
> 
> I'm sorry but what exactly does it means? Maybe could you please explain with an example?

I'd prefer to let @usama7628674 jump back in and explain this but the short of it is:

If you are using Aircrack-ng to do frame injection, the ability to effectively do such depends on range and signal strength. I am not a professional security analyst but I do a lot of testing related to monitor mode so as to test the work that I do on the drivers so I see what usama is talking about. He is right, the ACHM really smokes the other adapters when it comes to frame injection.

> Although I've made up my mind to keep ACHM and maybe ACM too, but I'm just curious to know that in this scenario when signal is not high enough to connect but I can still see it, will this be enough [...]

@usama7628674 may be better able to answer this question.

> And is speed relevant for attacking or penetration testing?

I'm not the expert but my opinion, based on my testing, is that speed is not important for security analysis/pen testing.

>When I measure speed by enabling hotspot on my laptop using AWUS1900 or even on my mobile I get around 100mbps speed and when I try to connect both laptop and mobile via my wifi router I get around 150mbps speed.

Enabling hotspot on an laptop is usually not going to provide good speed as the settings will not be optimal.

> AWUS1900 has theoretical 800+mbps speed then why I'm getting max 150mbps on 5gh channel on USB 3 port

You could be getting much better speeds. I don't have a simple short answer as it depends on your driver, many settings, congestion and the AP/router you are connecting to. I can connect my 8814au adapter to my desktop right now and get over 600 Mb/s when connecting to my wifi router. I will connect to a 5 GHz DFS channel that has zero congestion because I am the only one on the channel and I am using a 80 MHz channel width and appropriate driver settings.

Regards

usama7628674, 2021-12-27:
> Although I've made up my mind to keep ACHM and maybe ACM too, but I'm just curious to know that in this scenario when signal is not high enough to connect but I can still see it, will this be enough [...]

@vvkvas I've had successful handshake capture when signal strength was weak enough for me that it was impossible to connect to network. ACHM will increase your chances to capture handshake even when signal strength is weak.

usama7628674, 2021-12-27:
> ACHM has very strong injection rate.
> 
> I'm sorry but what exactly does it means? Maybe could you please explain with an example?

Frame injection is useful when you want to disassociate clients from network. This feature has following use cases

1. Frame Injection is necessary to capture 4-way handshake.
2. It can be used to cause DoS attack.
3. It can be used to force the victim to connect to malicious or cloned access point.
4. We can recover a hidden SSID by performing deauthentication attack.
5. We can generate ARP frames for a WEP replay attack.

Rest is explained by Nick very well. Apart from aircrack-ng you can carry out deauth attack with mdk3 and mdk4 as well.

vvkvas, 2022-01-02:
> Rest is explained by Nick very well. Apart from aircrack-ng you can carry out deauth attack with mdk3 and mdk4 as well.

Thanks to both of you for amazing and very clear explanation!

> I will connect to a 5 GHz DFS channel that has zero congestion because I am the only one on the channel and I am using a 80 MHz channel width and appropriate driver settings.

I've tried a few settings as well but still my speed is not exceeding 160mbps. I think something in my laptop is throttling, will try with another system. 

> I don't have a 1900 but would expect its range to be about what the range of the ACM is.

If you want me to test anything on 1900 I can do it before I return it in next couple of days?

vvkvas, 2022-01-15:
Hi @morrownr 
[Here](https://github.com/morrownr/USB-WiFi/blob/main/Performance_Comparison.md) you have mentioned that WUS036ACHM support WPA3 but I'm trying to connect using this adaptor both in W10 & Kali but not able to connect in either, please suggest how do I enable that for both?

Best Regards

morrownr, 2022-01-15:
Hi @vvkvas 

Can i get you to start a new issue so that others can more readily see this issue?

Please add some details to the new issue when you post it. We will need to know the wifi router you are trying to connect to and what its WPA capabilities are.

vvkvas, 2022-01-15:
Thank you very much @morrownr !! I will open a new case.

closed 2022-01-15 by vvkvas

BlackstormCoder, 2023-11-29:
Hey @vvkvas, have you used the AWUS036ACHM adapter for wifi penetration testing? if yes then How does it performs? can I pentest the 5ghz network? whats your overall experience? I am planning to buy this in India which will cost me around 7k INR. Does it worth the penny? 

Best Regards.

morrownr, 2023-11-29:
Hi @vvkvas 

@ZerBea has tests at his site that includes the Alfa ACHM.

https://github.com/ZerBea/hcxdumptool/discussions/361

I have an ACHM. I am not a pen testing expert but my use of the ACHM for managed and AP modes tells me that it has exceptional range, runs cool and is basically very trouble-free.