morrownr/USB-WiFi #23: Realtek PCI drivers

bjlockie, 2021-12-07:
Why does Realtek have in-kernel drivers for their PCIe cards but not the USB cards?

I would think Realtek PCIe and USB drivers would either both use mac80211 or not.:-)

morrownr, 2021-12-09:
Hi @bjlockie 

My understanding is that Realtek has two separate wireless support teams. One handles the internal stuff like PCIe and the other handles USB. It is my understanding that these two teams are not located in the same country.

If you look at Realtek's support in the Linux kernel, you become aware that Realtek, as a company, knows how to support Linux because they appear to do a pretty good job with in-kernel drivers for audio and lan and even PCIe wireless but, as we know, USB wireless is simply BOTCHED. The Realtek USB Wireless team uses the wrong technology for its drivers and the support is terrible.

Linux Wireless standards call for drivers that are based on mac80211 technology with support and maintenance to be accomplished in-kernel. There is a BIG reason for this. Unlike Windows, Linux is is under constant development. This means changes at any time. Windows is developed by full time developers working full time for a single company. This allows for concrete plans and deadlines. Linux, on the other hand, is developed by people all over the world working for thousands of different companies and some people that simply use their spare time to work on Linux. The constant development model with releases of new kernels about every 2.5 months or so is the result of many years of figuring out how best to use the resources at hand. This development model has been adopted and supported by nearly all that work on the Linux kernel. A notable exception is the Realtek USB Wireless team. They do not push their working into the Linux kernel for development and maintenance and they refuse to use the designated and supported technology.

This all leads to problems for end users. Linux users need to understand that over the longer term, perhaps 1-2 years, they will have problems with the Realtek USB wireless drivers and there is nothing I can do about. Example:

Let's say that Jose has a computer that could benefit from a USB wifi adapter so he buys an adapter based on the 8812au chipset. This is a good solid chipset. He downloads the driver from here and everything is working fine. That is fine and dandy but the driver that he downloaded today is no longer getting maintenance since it is setting on his computer. In the meantime, Nick and others stay busy here in the repos updating the Realtek drivers for changes in the Linux kernel.

A few months or a year down the line, Jose says I want to upgrade my Ubuntu 20.04 setup to 22.04... so he clicks the Upgrade button and magic takes place. Once the upgrade is complete, Jose discovers that his usb wifi adapter does not work. Why? Because the driver on his system was only maintained up to kernel 5.4. He is now on kernel 5.16 and the unmaintained Realtek driver blows up when being compiled on the new kernel. (Remember: never install a Linux driver from a CD. The chances of it doing anything except making a mess of your system is slim to nil.)

How about if Jose had choosen an adapter with a Mediatek chipset instead. Let's say it is an adapter with a mt7612u chipset. The driver for that chipset is maintained in the kernel so Jose did not have to find and compile a driver. He simple plugged the adapter in and it worked. Now he wants to upgrade to Ubuntu 22.04. Is he going to be met with the same disaster that happened with the Realtek adapter? No. Since the Mediatek driver is in the kernel, it is maintained in the kernel by kernel developers and when the new version of Ubuntu comes into his system, a new version of the driver comes in with that new kernel because it is part of the kernel. Jose's adapter just keeps working after the upgrade.

There are more issues involved that just the having to find the right driver and learning how to compile it. There is also the issue of the lack of maintenance as just explained and there is the issue of lack of conforming to Linux Wireless Standards. This comes about when a driver just does not operate or provide results in accordance with a standard and I can show numerous examples of this. Monitor mode is an example. You can find a lot of guides to doing various things with monitor mode. Most of those guides shows things the way the official standard says it should be. Well, along comes Realtek and things don't conform to the standard and you have to do things a little differently and it is not always easy to figure out. I'm currently working on a guide to include with the Realtek drivers that I support.

The vast majority of Linux users that want or need to use a USB WiFi adapter would find that an adapter that uses an in-kernel driver will serve their needs better.

Nick

bjlockie, 2021-12-09:
Thanks for the explanation.

The OEM Realtek USB drivers are developed totally separate from the PCIe drivers?
That seems like a waste of effort. :-)

morrownr, 2021-12-09:
> Thanks for the explanation.

You are welcome.
 
> The OEM Realtek USB drivers are developed totally separate from the PCIe drivers?

That appears to be the case.

> That seems like a waste of effort. :-)

Yes it does.

Here is some speculation from me:

Realtek USB Wireless has no choice but to support Linux because of Android. If you look around inside the Makefile and in the code, it becomes clear that this is the driver that is used in most, if not all, smartphone uses that use Realtek chipsets. Realtek can get away with a non-standard dev model like this for Android because the kernel in Android is rarely, if ever, replaced. It is mostly a static environment that is controlled by the phone makers.

My guess is that someone in charge of the Realtek USB Wireless at some point got mad at someone that works for the Foundation and has taken a defiant stance. This could all be fixed rapidly by one company... Google. If Google were to mandate that all wireless drivers used in Android be in accordance with Linux Wireless Standards, poof, Realtek could either comply or leave the Android market... the Android market is billions and billions of installations.

Other things that might help:

- Getting the correct information out to Linux users worldwide so that they know buying adapters with in-kernel drivers is going to be far more reliable. If sales shift enough, it might shed some light on the issue at Realtek HQ.

- Leaving a message with your retailer that goes something like this:

Dear Sir/Ma'am,

I am requesting that you let Realtek know that I am requesting that their driver support for USB WiFi adapters be in accordance with Linux Wireless Standards which mandate mac80211 technology and in-kernel placement and maintenance. If this happens, as a seller of Realtek based USB WiFi adapters, you won't have to worry about Linux support because the driver will already be in Linux and support is provided by the Linux kernel devs.

Regards,

Jose Smith

-----
By having their driver in the kernel, they cannot avoid taking problem reports like they can now. Their current system of only releasing code to certain retailers at irregular intervals and providing no method for users to contact them to report bugs or otherwise is UNSATISFACTORY in my opinion.

Something to think about: Since creating this site, and the repos that are contained in it, not long after the pandemic got underway last year, I have seen a lot of growth in people visiting the site. Yesterday, I added the numbers up and this site is now over 16,000 visits for the last 2 weeks. This seems to indicate to me that there are far more Linux USB WiFi adapter users out there than I thought there would be. The numbers continue to grow and it isn't like I am spending a lot of money on advertising. If we work together, I think we can make a difference. Being treated like second class citizens by Realtek needs to come to an end.

Nick

bjlockie, 2021-12-10:
I sent feedback to dlink.
I'd be surprised if the email makes it to Realtek.

bjlockie, 2021-12-12:
Do you also think an email to Alfa Networks asking for a wifi6 USB device would be useful?

morrownr, 2021-12-12:
> Do you also think an email to Alfa Networks asking for a wifi6 USB device would be useful?

I think that is a good idea. Maybe an email to Panda requesting the same would be a good idea also.

Would you mind if I post an example letter to Alfa and Panda later today?

bjlockie, 2021-12-12:
> Would you mind if I post an example letter to Alfa and Panda later today?
I just said:

>I would like to see a wifi6 mediatek USB dongle similar to the AWUS036ACM. 

I should have mentioned the Linux kernel driver, etc.

bjlockie, 2021-12-14:
This is the response from Alfa Network. 

>Thank you for interest ALFA product
However, MediaTek doesn't have Wi-Fi 6 USB chipset, therefore, we cannot make it 
Thank you

morrownr, 2021-12-14:
Suggested follow up messages:

Dear ALFA,

Can the following Mediatek chipset be used to make a USB WiFi adapter?

MT7922U (N) (Filogic 330)
USB 3.0
abgn+ac+ax
2x2:2 (2.4Gbps)
Wi-Fi 6E, WPA3, OFDMA, Zero DFS, ESR
MU-MIMO, 1024QAM, HE160, LNA/PA
BT 5.2 • USB 3.0 [0e8d:7922]

Regards,

bjlockie

Also,

Dear Mediatek,

I have been asking USB WiFi adapter makers for a WiFi 6e capable USB WiFi adapter based on a Mediatek chipset and I am being told that Mediatek does not have a WiFi 6 chipset available for USB WiFi adapters. Is that true?

Regards,

bjlockie

bjlockie, 2021-12-14:
> My understanding is that Realtek has two separate wireless support teams. One handles the internal stuff like PCIe and the other handles USB. It is my understanding that these two teams are not located in the same country.

Let me guess the countries. :-)
PCIe is Taiwan and USB is China?
I'm guessing Mediatek is all Taiwan too. :-)

morrownr, 2021-12-14:
> Let me guess the countries. :-) PCIe is Taiwan and USB is China? I'm guessing Mediatek is all Taiwan too. :-)

I actually don't know where the PCIe and USB teams are located. If I had the time, I'll bet I could find out. I know Mediatek's HQ was located in Taiwan but I would have to research it to see if that is still the case.

I think sending versions of the two messages above is a good idea. If you send them today, I will send mine tomorrow and maybe others can join in.

Mediatek = https://www.mediatek.com/

This may be far more important that any of us realize at this point. I would love for Intel and/or Atheros to start providing USB WiFi chipsets to adapter makers but I'm not seeing anything on that front which leaves us with Realtek and Mediatek. Realtek recently released the 8852au/8832au chipset and makers are starting to release product. I and another person do have a copy of the Realtek source for this chipset. I have a single adapter I am using to test. The other person has multiple adapters. Unfortunately the driver is the same old Realtek out-of-kernel style that has been causing us Linux user a lot of pain for many years. Our testing so far has shown disappointing results. My current recommendation for Linux users is to avoid buying any adapter with the Realtek 8852au/8832au chipset.

We need Alfa and Mediatek to step up and provide us some WiFi 6(e) options. The more messages we send to both companies, the better.

Nick

bjlockie, 2021-12-15:
I sent this to sales@netgear.com

>I would like to see a wifi6 USB dongle that uses a Medistek chipset similar to the wifi5 A6210.
The Linux driver support for USB WiFi adapters based on Mediatek chipsets is in accordance with Linux Wireless Standards. Linux support is provided by the Linux kernel developers. 

As well as an Alfa Network one.
And Mediatek. :-)
Panda is next. :-)

morrownr, 2021-12-15:
Don't forget Panda!

bjlockie, 2021-12-15:
Thanks for the reminder.
Sent to sales@pandawireless.com

bjlockie, 2021-12-16:
>Panda Wireless Sales <sales@pandawireless.com>
Hi James, I will pass this to my boss. Thank you for your interests. Judy

bjlockie, 2021-12-20:
> Suggested follow up messages to Alfa and Mediatek:
> 
> Dear ALFA,
> 
> Can the following Mediatek chipset be used to make a USB WiFi adapter?
> 
> MT7922U (N) (Filogic 330)

Where did you see the 330?

https://www.mediatek.com/products/networking-and-connectivity/mediatek-filogic-wifi-6

Does the Filogic 130 only have a USB2 interface?

morrownr, 2021-12-22:
@bjlockie 

Check out Issue #28