morrownr/USB-WiFi #27: switch back and forth between builtin wifi and USB wifi

bjlockie, 2021-12-18:
Is there a recommended way to switch back and forth between builtin wifi and USB wifi?
I have built in iwlwifi.
`$ lsmod | grep wifi                                    
iwlwifi               380928  1 iwlmvm
cfg80211              892928  3 iwlmvm,iwlwifi,mac80211

and mt`76-usb
I can plug it in and sometimes it works for a while and then freezes the 5.13 kernel.
Sometimes it doesn't load.
I suspected the stand for it at first but now I am leaning toward conflicting drivers.

bjlockie, 2021-12-18:
Maybe rmmod?

```
#$ lsmod | grep iwl
#iwlmvm                425984  0
#mac80211             1024000  1 iwlmvm
#iwlwifi               380928  1 iwlmvm
#cfg80211              892928  3 iwlmvm,iwlwifi,mac80211

`sudo rmmod iwlmvm iwlwifi mac80211 cfg80211`;
```

```
# $ lsmod | grep mt76
#mt76x2u                28672  0
#mt76x2_common          24576  1 mt76x2u
#mt76x02_usb            24576  1 mt76x2u
#mt76_usb               36864  2 mt76x02_usb,mt76x2u
#mt76x02_lib            81920  3 mt76x02_usb,mt76x2u,mt76x2_common
#mt76                   77824  5 mt76_usb,mt76x02_lib,mt76x02_usb,mt76x2u,mt76x2_common
#mac80211             1024000  4 mt76,mt76x02_lib,mt76x02_usb,mt76x2u
#cfg80211              892928  4 mt76,mt76x02_lib,mac80211,mt76x02_usb
```

Is there any work trying to make the iwlwifi driver smaller?

morrownr, 2021-12-18:
> Is there a recommended way to switch back and forth between builtin wifi and USB wifi? I have built in iwlwifi. `$ lsmod | grep wifi iwlwifi 380928 1 iwlmvm cfg80211 892928 3 [...]

Maybe we can figure out what the cause is. Can you be more specific about what hardware and distro you are using?

bjlockie, 2021-12-18:
Is it supposed to be able to switch between wifi drivers?
That seems like it is rare someone would want to do that.

It is Lubuntu 21.10 and an Acer Aspire A515-55 (https://www.acer.com/ac/en/CA/content/support-product/8192?b=1).

> $ sudo lshw -c network
  *-network                 
       description: Wireless interface
       product: Wi-Fi 6 AX200
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:04:00.0
       logical name: wlp4s0
       version: 1a
       serial: 3
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=iwlwifi driverversion=5.13.0-22-generic firmware=63.c04f3485.0 cc-a0-63.ucode ip=192.168.68.65 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:29 memory:f7600000-f7603fff

It could be the video driver that locks up when I switch network cards.
I was streaming video when the desktop locks up (no mouse, nothing).
Could it be a conflict with the USB mouse I use?

bjlockie, 2021-12-18:
I've been using the USB and it hasn't crashed yet. :-)

morrownr, 2021-12-18:
> Is it supposed to be able to switch between wifi drivers?

I'm not sure how to answer as I am not sure what you are asking. Let me throw some things out and maybe it will answer the question.

I often operate with more than one usb wifi adapter on my systems. It is not uncommon for me to have up to 4 adapters going at the same time on my main dev box. Don't use two adapters with the same Realtek chipset at the same time as the Realtek drivers don't support it. Anyway, I can switch between the various adapters. I can turn individual adapters on or off. How this is done depends on your distro. I use Linux Mint and like Lubuntu, it is based on Ubuntu but Linux Mint uses its own way to turn interfaces on and off. On Ubuntu there is a button called "Forget connection." If you forget all connections for an adapter, you have basically turned it off.

> That seems like it is rare someone would want to do that.

If you are talking about people with more than one wifi device, that is actually very common but you have to learn how to turn them on and off and other things and how to do that depends on your distro.
 
> It is Lubuntu 21.10 and an Acer Aspire A515-55 (https://www.acer.com/ac/en/CA/content/support-product/8192?b=1).

Nice system.

> > product: Wi-Fi 6 AX200
> > vendor: Intel Corporation

Okay.

> It could be the video driver that locks up when I switch network cards. I was streaming video when the desktop locks up (no mouse, nothing). Could it be a conflict with the USB mouse I use?

It could be a lot of things. If it were me, I would turn off one wifi device and run with the other one for a long time to see what happens. Then I would turn off the other and activate the remain device and run with it for a long time. In other words, see what happens when you run with only one adapter connected at a time.

bjlockie, 2021-12-18:
I'm suspecting a bad cable.

closed 2021-12-19 by bjlockie