morrownr/USB-WiFi #28: Letter to ALFA Networks Inc

morrownr, 2021-12-22:
Dear ALFA Networks Inc
sales@alfa.com.tw

Ref: Future Products

I have been using and recommending your USB WiFi adapters for years. The quality
and range is appreciated. The reason I am writing today is to pass along some
information that I hope is useful.

I have a Github site that serves those who use USB WiFi adapters with Linux. The
site gets over 8,000 visits per week. What I have noticed is some recurring
themes in the conversations. A user will ask what adapter they should buy and
they spell out their requirements which usually goes something along these
lines of:

- Good range
- Very compatible with Kali Linux and the tools required for security analysis
- Uses a Linux in-kernel driver
- AC1200/1300 class or higher

Long conversations usually ensue. Your company provides answers to all of those
requirements but not all in the same product. Let's see:

The ALFA AWUS036ACHM and ALFA AWUS036ACH have the long range that users are
looking for.

The ALFA AWUS036ACHM and ALFA AWUS036ACM use excellent in-kernel, standards
compliant drivers which makes them very compatible with Kali Linux and the
needed tools.

The ALFA AWUS036ACM and AWUS036ACH are AC1200 class adapters

As you can see, you have all of the answers, just not all of the answers in one
adapter, so please allow me to suggest a new product that combines the answers:

1. Use the mt7612u chipset to provide AC1200/1300 class speed and capability
2. Use the antenna quality and amp from the ACHM/ACH to get the range
3. Use the case, including USB-C, from the ACH and the tan color from the ACHM

My opinion is that this would be a hot seller for many years to come.

Since I am suggesting recommended future products, let me throw 3 additional
requests out:

1. As it appears the ALFA AWUS036NEH and ALFA AWUS036NHA are at or near the end
of production, let me suggest the following product as it appears a good single
band adapter will sell for many years to come:

- Use the case and antenna from the ALFA AWUS036NEH but upgrade the chipset to
the rt5372 which will provide a N300 class adapter. I also like the tan color
from the ALFA AWUS036ACHM.

2. A version of the ALFA AWUS036ACS that uses the mt7610u chipset would provide
a plug and play, lower cost option to casual users.

3. Linux users are interested in a future WiFi 6(e) adapter and would like to
see it be based on a chipset that uses an in-kernel driver. While it is not
published at this time, it appears that Mediatek could provide a mt7921u or
mt7921ku chipset and that would likely work well.

Hopefully we hear good news about your future products soon.

Best regards,

Nick

morrownr, 2021-12-22:
Note: The name of the contact at ALFA has been x'ed as I do not have permission to publish the name.

-----

xxxx - ALFA Network Inc. <xxxx@alfa.com.tw>
8:40 AM (3 hours ago)
to me, sales

Dear Nick

Thank you for your great suggestions. We will do what you suggested.

I have one question,  you mention RT5372, is possible to use RT5370 instead of RT5372? Because due to chipset shortage, hard to get the RT5372 chipset, and we still can buy RT5370. And we can make tan color for it.

About MT7610U, no problem, we will make tiny dongle like AWUS036ACS.

For the WiFi 6 adapter, we still waiting for the chipset vendor’s information. Currently, WiFi 6 USB chipset only Realtek RTL8832AU and RTL8832BU, and Realtek announce discontinue RTL8832AU, therefore, only RTL8832BU might be available. We checked with the MTK distributor, they told us MT7921U or MT7921KU haven’t been released yet.

Thank you and Best regards

xxxx
ALFA NETWORK Inc.
Web: www.alfa.com.tw

morrownr, 2021-12-22:
Dear xxxx,

It is good to hear from you. Below are the answers to your questions.

Quote: "I have one question,  you mention RT5372, is it possible to use RT5370 instead of RT5372?
Because due to chipset shortage, hard to get the RT5372 chipset, and we still can buy RT5370
And we can make tan color for it."

Answer: I recommend that you do the work to get ready to start production but hold off until the RT5372 chipset is available in adequate amounts. I understand the chipset shortages we are experiencing but I also understand the work and expense of bringing a new product online. I think you will have better success with the RT5372 and here is why: The RT5372 can provide 2 streams for a total of 144 Mb/s throughput on a single 20 MHz channel width. The RT5370 can only provide 72 Mb/s throughput and the use cases that I am seeing actually need that extra throughput. 

Quote: "For the WiFi 6 adapter, we still waiting for the chipset vendor’s information. Currently, WiFi 6 USB chipset only Realtek RTL8832AU and RTL8832BU, and Realtek announce discontinue RTL8832AU, therefore, only RTL8832BU might be available. We checked with the MTK distributor, they told us MT7921U or MT7921KU haven’t been released yet.

Answer: Thank you for the information. Linux users are very interested in seeing a WiFi 6 adapter based on a Mediatek chipset as Mediatek does a great job supporting its USB WiFi chipsets with in-kernel standards compliant drivers. On the other hand, Realtek seems determined to continue supplying only non-standards compliant out-of-kernel drivers that are problematic.

Hopefully I was able to clarify the issues. Once again, your reply is greatly appreciated and I look forward to future ALfa product releases.

Nick

bjlockie, 2021-12-22:
Does the ALFA AWUS036ACH (RTL8812AU) have an in-kernel driver?
Realtek USB in general is scary. :-)

morrownr, 2021-12-23:
> Does the ALFA AWUS036ACH (RTL8812AU) have an in-kernel driver?

No. But the out-of-kernel driver that is available for that chipset is, in my opinion, the best Realtek out-of-kernel driver that has ever existed. And that is an exceptional adapter... it is also very expensive. I'm not recommending Linux users buy that adapter but you have to point out the good with the bad. Alfa really makes good quality adapters. My email was an effort to try to get Alfa to maintain at least 4-5 adapters that use in-kernel drivers so that we have options. Currently they sell 4 adapters that use in-kernel drivers but 2 of those adapters are at end of production.

> Realtek USB in general is scary. :-)

I have seen so many Linux systems messed up and users that are frustrated due to Realtek out-of-kernel usb wifi drivers that it is sad. Realtek has a strategy but I can't see where long term customer satisfaction is part of that strategy.

bjlockie, 2021-12-23:
Is the ALFA AWUS036ACH better than the Alfa AWUS036ACM?
I got Alfa AWUS036ACM  to use on my Raspberry Pi4 so an in-kernal driver was pretty important.

I had hope that Realtek drivers would get better with the wifi6 device.

morrownr, 2021-12-23:
> Is the ALFA AWUS036ACH better than the Alfa AWUS036ACM?

What is  "better" depends on the person making the judgement. If we judge by selling price, the ACH is better than the ACM or ACHM, The ACM and ACHM sell for about $40 USD. The ACH goes for $60 or more depending on where you buy it:

https://store.rokland.com/collections/wi-fi-usb-adapters/products/alfa-awus036ach-usb-c-802-11ac-ac1200-dual-band-high-power-wifi-usb-adapter

I have all three. My observations:

- ACM - it is generally the lowest cost of the three adapters as it usually sells for about $35 USD. It is a high quality adapter that has good range and power usage maxs out at about 380 mA. It is a super all around adapter. It works well with Raspberry Pi's. It spends most of its time here attached to a RasPi4B working as a 5 GHz AP. I classify it as a 24/7/365 device as it just works and it does not fail.

- ACHM - the ACHM is usually next higher in price coming in at around $40 USD. This little adapter is probably the best adapter available for folks that do security analysis/pen testing. It has exceptional range. It is also a good AP given the range it has and when I am not using it to test some monitor mode stuff that I work on, it is generally connected to my main router to act as an additional radio. My main router runs OpenWRT and has a couple of USB ports. The cool thing is that OpenWRT has drivers for the mt7612u and mt7610u chipsets which allows you to add extra usb adapters to your router. 

- ACH - it is the most expensive of the 3 adapters we are talking about. Is the quality good? Yes, very good. Is the range good? Yes, nearly as good as the ACHM. Is it a AC1200 class adapter? Yes. Does it use the best Realtek out-of-kernel driver available? Yes it does. Do I recommend it? Not really as I don't recommend any adapters based on Realtek out-of-kernel drivers because I know they cause problems.

What you see when you look at these 3 adapters is a situation where all 3 have good features but none of them have ALL of the features that some people would like to see which is why I detailed the issue in my letter to ALFA. ALFA has the ability to make what I will call an All-Star adapter if they combine the case, to include USB-C support, from the ACH, the mt7612u chipset from the ACM in order to have AC1200 class performance and the antenna/amp capability from the ACH and ACHM in order to have long range. I also happen to the think the tan color from the ACHM looks good. I guess we will see what ALFA comes up with.

Additional note: The ALFA rep noted the challenges they face getting the rt5372 chipset currently but he did not mention problems with the mt7612u chipset. From other sources, I can tell you that Mediatek has been having problems producing enough mt7612u chipsets but hopefully that all sorts itself out as we go forward. I really hope we see an announcement soon about the release of a mt7921u chipset (WiFi 6). The driver for the mt7921 is already in the Linux kernel and is in use. It does not have usb support yet as Mediatek has not announced the usb product.

> I got Alfa AWUS036ACM to use on my Raspberry Pi4 so an in-kernal driver was pretty important.

One of the big advantages of the ACM over the ACH on the RasPi is the far lower power requirement. The ACH can use around 800 mA.
 
> I had hope that Realtek drivers would get better with the wifi6 device.

So far that has not happened.

dmitttri, 2021-12-23:
This is a very interesting topic. It definitely seems we'll have to wait some decent amount of time until WiFi6 adapters become widely available. ACM, ACHM and ACH comparison is also very useful. Which made me think, I should definitely order couple of ACHM to try them out, mostly because of extended range and in-kernel drivers.

morrownr, 2021-12-27:
@dmitttri 

This is indeed an interesting topic. I am contemplating if my next letter should go to Panda. Panda has historically made adapters that work with in-kernel Linux drivers but they seem to have fallen behind the times as they don't have an AC1200 adapter available. Panda's quality and capability of their adapters is not at the level of Alfa but certain adapters they have produced have been good products. In fairness, there is no other adapter maker that is at the quality, durability and range capability that we see from Alfa products.

Separate issue: Santa brought me a new laptop. It had been many years since I have had a new laptop so I was pretty excited about it. What does this have to do with this thread? The internal wifi on this new laptop is based on the Mediatek mt7921e chipset. I installed Ubuntu 21.10 and the wifi worked very well. That is a good sign. I hope Mediatek releases a mt7921u chipset soon. For what it is worth: The laptop is an [Acer Aspire A514-54](https://www.acer.com/ac/en/US/content/support-product/8625?b=1). After installing Ubuntu I tested to see if everything was working. Everything worked fine...even little things like the buttons that adjust the volume, and the button that put the system in airplane mode and the buttons that adjust screen brightness. Santa, since he knows that Linux users can be picky, did consult with me back before the order went to the elves so there was a good probability that most things would work well but it seems Linux has come a long way in the last few years as far as working with many off the shelf laptops.

Ho Ho Ho

bjlockie, 2021-12-27:
I have an Acer Aspire 5 A515-55.
It has an Intel AX200.
I wish it had mediatek, they seem to produce smaller code. :-)

morrownr, 2021-12-28:
> I have an Acer Aspire 5 A515-55. It has an Intel AX200. I wish it had mediatek, they seem to produce smaller code. :-)

I actually thought I was getting the Intel AX200 but there it was, a mt7921. I monitor the linux-wireless list. It is interesting to see all that goes on with wireless dev in Linux. There are 3 groups that are really busy all the time... Intel, Atheros and Mediatek.

bjlockie, 2021-12-28:
I asked on the Acer forum what the wifi card was and they said one of two, Intel or Mediatek.
I figure either would be ok. :-)
I have an ax200 card in my desktop.
I also got a Dlink X1850.
Maybe it'll get a community driver.
I'll buy a Mediatek wifi6 one too if it ever comes out. :-)

usama7628674, 2021-12-28:
Indeed, panda does make quality adapters, just not that equal the performance of alfa. I like Panda PAU09. It is a solid adapter with rt5572 chipset.

morrownr, 2021-12-28:
> I asked on the Acer forum what the wifi card was and they said one of two, Intel or Mediatek. I figure either would be ok. :-)

That is what I figured also. I must add this though: I would not worry about getting an Atheros chipset either as they do a good job and believe it or not,  the Realtek 8852a driver is in the kernel:

https://github.com/torvalds/linux/tree/master/drivers/net/wireless/realtek/rtw89

I've even read messages that should be credible that Realtek intends to add usb support. 

> I have an ax200 card in my desktop. I also got a Dlink X1850. Maybe it'll get a community driver.

I have a copy of the out-of-kernel driver in a repo here. It is private so I have to give you access. Are you interested? The reason is it not public because the results from testing and working on it were not good.

 I'll buy a Mediatek wifi6 one too if it ever comes out. :-)

Me too. I'm hoping for a wifi 6e chipset as we need an adapter that uses the new 6 GHz channels.

bjlockie, 2021-12-28:
> the Realtek 8852a driver is in the kernel:
> I've even read messages that should be credible that Realtek intends to add usb support.

I'm not holding my breath.
They have PCI drivers in kernel for ethernet cards so they have potential. :-)

> I have a copy of the out-of-kernel driver in a repo here. It is private so I have to give you access. Are you interested? The reason is it not public because the results from testing and working on it were not good.

Is it fixed to run on kernel-5.13?

I have a copy of rtw89-usb which doesn't work but it seems like if anything gets fixed, it'll be that.

morrownr, 2021-12-28:
> Is it fixed to run on kernel-5.13?

Yes. It would be nice to have another opinion of the driver. The two of us that worked to make it public do not have a high opinion of the driver but a third opinion would be nice.

bjlockie, 2021-12-28:
You want me to tell you it locks up my machine? :-)
LOL
Ok, give me access to your repo and I'll try it.

morrownr, 2021-12-28:
> You want me to tell you it locks up my machine? :-) LOL

Just trying to spread good tidings and cheer. Ho Ho Ho LOL

> Ok, give me access to your repo and I'll try it.

Done. You should get an invitation.

bjlockie, 2022-01-01:
I sent email to Asus requesting a wifi 6 USB Mediatek chipset.