morrownr/USB-WiFi #29: How to increase txpower on comfash 785ac

Appleuy, 2022-01-02:
How to increase txpower on comfash 785ac  to 30dbm

morrownr, 2022-01-02:
My 785 AC is currently offline but I'll put it on my todo list to check this out. FYI: Many usb wifi adapters have their txpower locked in firmware these days. I don't know if that is the case here but will check it out as I have time.

Regards

morrownr, 2022-01-06:
Hi @Appleuy 

I had time this morning to connect my 785AC. It appears that this adapter has txpower locked in firmware.

Appleuy, 2022-01-07:
Is there any way to unlock it?

morrownr, 2022-01-07:
I'm not an expert in this area. I did do some testing on a couple of adapters at one point last year when a person asked a couple of questions but I don't even remember which adapters. What I did was run the txpower up as high and low as it would go 1 dBm at a time while tracking the signal with wavemon. I wrote the data down but am not sure where it is. What I remember is that running the power up from default did not do much for signal or for iperf3 measured throughput. 

The reason is likely that it takes 2 to tango. You can run txpower up but if the antenna sensitivity for receiving  is not there to support that extra power then it just doesn't matter. I generally do not mess with txpower. I'm sure results vary from adapter to adapter.

Do you mind telling me if you are seeing a problem? If there is a problem then I will be glad to add it to the information for this adapter so that others know. I add things people report to the links and reports on the listed adapters regularly. My own experience with this adapter has been exclusively in client mode and I haven't noticed a range problem.

usama7628674, 2022-01-07:
If you want to transmit at long range get an amplifier or high gain antenna. Almost all adapters these days have txpower locked in firmware.

usama7628674, 2022-01-07:
Some adapters from alfa come with in-built amplifier which helps a lot and that is why alfa adapters are excellent covering more range, like AWUS036ACHM

morrownr, 2022-01-07:
[injection-test-ACH.txt.tar.gz](https://github.com/morrownr/USB-WiFi/files/7826771/injection-test-ACH.txt.tar.gz)
[injection-test-ACHM.txt.tar.gz](https://github.com/morrownr/USB-WiFi/files/7826772/injection-test-ACHM.txt.tar.gz)

I happened to think about something I said above so I decided to stop by again. If I made it sound like there is not a big difference in the range of various adapters, I should not have. There are big differences.

The two files I included above include some output from some work I was doing yesterday. I was testing a script I was working on. I decided to test with an Alfa ACH and an Alfa ACHM. Those are the two adapters that Alfa advertises as High Power. They will smoke all other modern dual band adapters that I am aware of. The script first scanned and then tested injection. Txpower was not touched. Both adapters ran at default txpower. You can open and see the info. I then plugged two consumer grade adapters in to see what the results would be. What I saw was 12-16 clients and ap's scanned and reasonable injection on about a third. From the exact same location and channel, the ACHM scanned 43 clients and ap's. usama can probably put it into better words than I can but to me, the ACHM is an outlier because it is just so much better than anything else for the type of thing I was testing.

If you open the files, you will see the ACH was pushing 14 dBm and the ACHM was pushing 16 dBm. Those are the defaults for these adapters. There is more to range than txpower setting. These two adapters are exceptional, especially the ACHM, but they come at a price. The ACH sells for 60-70 USD and the ACHM for around 40 USD.

usama7628674, 2022-01-07:
> injection-test-ACH.txt.tar.gz
injection-test-ACHM.txt.tar.gz

Hmm, I've compared both files side-by-side and what I see is ACH is scanning less access points but injection rate is strong but ACHM is scanning far more APs but injection rate is less on some access points than that of ACH which is injecting at higher rate on those APs.

```
ACHM

16:00:52  F8:5E:42:0B:48:0A - channel: 44 - ''
16:00:55  Ping (min/avg/max): 0.171ms/7.133ms/17.273ms Power: -83.53
16:00:55  17/30:  56%
```
```
ACH

15:55:24  Trying directed probe requests...
15:55:24  F8:5E:42:0B:48:0A - channel: 44 - ''
15:55:26  Ping (min/avg/max): 0.680ms/11.099ms/23.801ms Power: -80.73
15:55:26  22/30:  73%
```

I can't make direct head-to-head comparison here, because I don't have ACH but a cheap [chinese adapter](https://www.netis-systems.com/Home/detail/id/96.html) which has same chipset as in ACH but it's not a high power adapter as Alfa awus036ach.

usama7628674, 2022-01-07:
My netis wf2190 with rtl8812au chipset gives less injection rate than my Alfa awus036achm.

usama7628674, 2022-01-07:
More comparison of injection rate between ACH and ACHM
```
ACH

15:55:26  F8:5E:42:0B:48:0E - channel: 44 - ''
15:55:27  Ping (min/avg/max): 0.945ms/10.918ms/25.166ms Power: -80.88
15:55:27  26/30:  86%
```

```
ACHM
16:00:55  F8:5E:42:0B:48:0E - channel: 44 - ''
16:00:58  Ping (min/avg/max): 0.156ms/6.933ms/22.885ms Power: -83.19
16:00:58  16/30:  53%
```

```
ACH

15:55:27  A0:FF:70:41:E1:34 - channel: 44 - ''
15:55:27  Ping (min/avg/max): 2.779ms/6.690ms/12.418ms Power: -61.30
15:55:27  30/30: 100%
```

```
ACHM

16:00:12  A0:FF:70:41:E1:34 - channel: 44 - ''
16:00:13  Ping (min/avg/max): 0.309ms/8.309ms/23.415ms Power: -66.96
16:00:13  25/30:  83%
```

usama7628674, 2022-01-07:
I think injection rate wise ACH is somewhat better and range wise ACHM is better than ACH

morrownr, 2022-01-07:
Hi @usama7628674 

It looks like you did what I was planning to do. The individual comparisons of injection information side by side does seem to be revealing. The comparison of ping and power numbers is also interesting. I have some time today to work on the script some more so maybe it would be interesting to do a 4 way test that includes the Comfast 785AC since that is the adapter that started this thread. I'm pondering what I should use for the 4th adapter?

Interestingly, there is a large difference in the power usage of the two adapters. The ACH can pull up to around 800 mA during heavy load while the ACHM will pull about 420 during heavy load. Given that information, we can see that the ACHM can be used in both USB2 and USB3 ports but the ACH should really be used only in a USB3 port given the 500 mA limitation of USB2.

I'm sure you saw the letter I sent to Alfa and their response. I think it will be interesting if they do make a "ACH" with a mt7612u chipset. One thing I run into regularly is users of the Realtek based adapters asking for support for various things like Virtual Interface (VIF) and set_wiphy_netns and other things that Realtek drivers don't support. I really wish that the folks out there that make the guides for various projects would emphasize that their guides are for adapters that use in-kernel drivers and what that means because many users run out and buy the first thing they see that they think will work but then they find out that things are missing.

We are doing a fine job of hijacking @Appleuy 's thread. Well, maybe not. Maybe this info will be of use to him.

Regards

usama7628674, 2022-01-07:
Yes, I track issues here almost everyday and I saw those letters. I would love to see Alfa launching a new adapter with 7612u chipset in high power mode. I really love this chipset. In fact, I've used an adapter from EDUP. It was [EP-AC1605 V1](http://en.techinfodepot.shoutwiki.com/wiki/EDUP_EP-AC1605). I used it for few days then returned it, but it was solid adapter with txpower locked at 18dbm. I returned it because I ordered V2 and got V1 instead. I wanted to test realtek adapter with 8812au chipset because many people were coming up with driver issues in one of the discord server dedicated for performing evil-twin attack with [fluxion](https://github.com/FluxionNetwork/fluxion). I suggest compatible adapters to newbie users who don't have experience with wifi pentesting. VIF is necessary for evil-twin attack, but realtek adapters do not support this feature.
VIF is not mentioned in any repository so people buy incompatible adapters then use fluxion, only to find out that adapter is not compatible and then come seeking for help on discord server where I've made a list of compatible adapters to point them to buy compatible adapter. I was googling one day and I found your repo which is well maintained with lots of compatible adapters.

usama7628674, 2022-01-07:
I've noticed mediatek chipsets are more sensitive and pick up more APs.

morrownr, 2022-01-07:
Hi @usama7628674 @Appleuy 

[injection.tar.gz](https://github.com/morrownr/USB-WiFi/files/7831334/injection.tar.gz)

More data for you gents to pick through. This batch includes 8 files. There are four adapters with one page for each adapter on each band. The adapters are the ALFA ACH, CF-785AC, ALFA ACHM and the ALFA ACS.

morrownr, 2022-01-07:
> I've noticed mediatek chipsets are more sensitive and pick up more APs.

That does seem to be the case. I've noticed the Mediatek chipsets tend to have a little shower throughput than the Realtek chipsets but within reason. The Mediatek chipsets use less power and run cooler in my experience.

morrownr, 2022-01-07:
> VIF is necessary for evil-twin attack, but realtek adapters do not support this feature. VIF is not mentioned in any repository so people buy incompatible adapters then use fluxion, only to find out [...]

I get really tired of messages from users of the Realtek drivers that I maintain where they are asking "where is VIF?" There are some other things missing as well. Do you mind if I take a look at your list of compatible adapters?

usama7628674, 2022-01-08:
> Do you mind if I take a look at your list of compatible adapters?

```
Atheros AR9271 <- present in TP-Link TL-WN722N v1 or Alfa AWUS036NHA (2.4Ghz - USB)
MediaTek MT7610U <- present in TP-Link Archer T2UH or AWUS 036ACHM(2.4Ghz/5Ghz - USB)
MediaTek MT7612U <- present in Alfa AWUS036ACM (2.4Ghz/5Ghz - USB) (not recommended to be used on Virtual Machines, but it works very well on native Linux. The recommended Linux kernel version is 5.3 or greater). For Rpi 2/3 run this command to get the card working echo "options mt76_usb disable_usb_sg=1" > /etc/modprobe.d/mt76_usb.conf
Ralink RT2770 <- present in Alfa AWUS051NH (2.4Ghz/5Ghz - USB)
Ralink RT3070 <- present in TP-Link TL-WN7200ND or Alfa AWUS036NH (2.4Ghz - USB)
Ralink RT3572 <- present in Alfa AWUS052NHS (2.4Ghz/5Ghz - USB)
Ralink RT5372 <- present in D-Link DWA-137 (2.4Ghz - USB)
Ralink RT5378 <- present in some some unbranded cheap chinese dongles (2.4Ghz - USB)
Ralink RT5572 <- present in Panda PAU07 or Panda PAU09 (2.4Ghz/5Ghz - USB)
Realtek RTL8187 <- present in Alfa AWUS036H (2.4Ghz - USB) (may require patched driver)
Realtek RTL8723BE <- present in Realtek RTL8723BE card (2.4Ghz - PCIe)
```

usama7628674, 2022-01-08:
I'm going through files and seeing very interesting results. Will post here once I'm finished.

usama7628674, 2022-01-08:
# 2 GHz

[image]

`All the adapters are injecting at 100% but power wise ACHM wins`

[image]

`All adapters failed to detect this AP except ACHM which not only is picking AP but also injecting at 40%. Another great result for ACHM` 

[image]

`ACH and ACS both are injecting at 90% and 96% respectively. Injection rate of ACHM is lagging behind but power wise it beats both of the adapters. Comfast fails to detect AP`

[image]

`Both comfast and ACH fail to detect this AP. ACHM and ACS both injecting at 6% and 3% respectively but ACHM is giving more power`

# 5 GHz

[image]

`ACH and ACS both fail and injecting at 0% while ACHM leaps ahead of them and injecting at whopping 93%. Comfast fails to detect AP`

[image]

`Same case here ACHM wins by large margin. ACH and ACS both failed the injection test. Comfast fails to detect AP`

[image]

`ACHM again wins here while ACH and ACS both failed miserably again. Comfast fails to detect AP`

[image]

`ACHM wins here with injection rate at 86% while ACH is injecting at 80%. ACHM has slightly better power here and ACS failed injection test. Comfast failed to detect AP`

usama7628674, 2022-01-08:
Comfast is worst among all 4 adapters. I've noticed one thing with ACH is it does not detect AP in 2 GHz with power less than -70 but it does detect APs with power less than -70 at 5 GHz. In some cases ACH gives better injection rate when power value is high. ACHM detects AP in both 2 GHz and 5 GHz with power less than -70. In some other cases ACHM injects and picks up AP but ACH either does not pick up or give 0% injection rate.

usama7628674, 2022-01-08:
ACHM is winner in my list. ACH finishes at second, ACS at third and Comfast at last.

morrownr, 2022-01-09:
@usama7628674 

I enjoyed your analysis. I probably should have included the Alfa ACM since it is always in conversations about "which adapter should I buy". If you think it is worthwhile to start a new Discussion/Issue to do a more inclusive analysis, I am will to gather the data and submit. I just need a list of adapters to include and I have adapters with most of the chipsets in your list.

Something that @Appleuy should be aware of is that the CF-785AC is a pretty good adapter. For normal client mode use I have not noticed any range related issues. While this test may lead folks to think the CF-785AC is not a good adapter, they need to consider that the CF-785AC was up against world class competition here. I had suspected that the Alfa ACS would do pretty good, but had not tested it previously. Given that the Alfa ACS  is not advertised as being a "long range" adapter as Alfa advertises the other 2, it does pretty good. My overall experience with Alfa adapters is that they are made of good materials, have long range and are durable. Of course, they are not cheap but it is good to have a high quality option.

usama7628674, 2022-01-09:
> I am will to gather the data and submit. I just need a list of adapters to include and I have adapters with most of the chipsets in your list.

I will happily analyse more data, especially ACM and a lot other adapters. I've heard good things about ACM. It'll be interesting to compare ACM with other adapters.

> Something that @Appleuy should be aware of is that the CF-785AC is a pretty good adapter. For normal client mode use I have not noticed any range related issues

It may not be excellent adapter for monitor mode operation but it is solid adapter for client mode. I've noticed its power numbers are not far off when compared with ACS and ACH.

> My overall experience with Alfa adapters is that they are made of good materials, have long range and are durable. Of course, they are not cheap but it is good to have a high quality option

I bought my first wifi usb adapter back in 2016, it was alfa awus036nha and it still works to this day. Now, I have another alfa adapter, awus036ACHM which is the best adapter of this time. They are durable and built to last long for years.

usama7628674, 2022-01-09:
I've one cheap [chinese adapter](https://www.aliexpress.com/item/32902795435.html) with `rt3070` chipset. It gives very good signal strength and beats by awus036nha. I'll pitch it against AWUS036ACHM. I'll post here for you to analyse.

morrownr, 2022-01-09:
> I've one cheap [chinese adapter](https://www.aliexpress.com/item/32902795435.html) with `rt3070` chipset. It gives very good signal strength and beats by awus036nha. I'll pitch it against AWUS036ACHM. I'll post here for you to analyse.

I would like to see this.

To continue this discussion, let me start a new thread in Discussions called Monitor Mode Performance.

usama7628674, 2022-01-15:
[ACHM injection test 2.4ghz.txt](https://github.com/morrownr/USB-WiFi/files/7875299/ACHM.injection.test.2.4ghz.txt)
[beini n9800 injection test 2.4ghz.txt](https://github.com/morrownr/USB-WiFi/files/7875300/beini.n9800.injection.test.2.4ghz.txt)

I've done some testing and leaving results here for you to give your verdict.

usama7628674, 2022-01-15:
beini n9800 was tested with 12bdi 2.4 GHz antenna while ACHM was tested with dual band 5 dbi antenna.

morrownr, 2022-01-15:
The Beini showed very fast ping times but the important item in this test is the injection rate and the results are close. The ACHM was able to handle what is likely the most distant system so I would rate the ACHM a little higher. This result is very consistent with what I have seen in various tests. The ACHM seems to have a superior ability to reach out over distance and through obstacles to get the job done.