morrownr/USB-WiFi #38: Need help optimizing hostpad.conf for Alfa AWUS036ACHM

amisix, 2022-01-12:
Hi Nick, I was hoping you could help me optimize the hostpad.conf for my AWUS036ACHM as we discussed earlier. Can you please tell me what the values you edit mean? Thanks!

--------------------- --------------- 2.4GHZ------------------------------------------------------
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
hw_mode=g
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 120 240
beacon_int=100
dtim_period=2
channel=9
chanlist=9
noscan=1

ieee80211n=1
ht_coex=0
ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40][RX-STBC1]

radio_config_id=d200e4fad5a7c9a41c19a1bb51e2fd3b
interface=wlan3
ctrl_interface=/var/run/hostapd
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=XXXXXX
wpa_psk_file=/var/run/hostapd-wlan3.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=XXXXXX
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wlan3.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
config_id=734edba6badac60dbb9ad3541828abc8
bssid=00:c0:ca:ae:67:fb

---------------------------------------------------- 5GHZ ---------------------------------------------------

driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
dtim_period=2
channel=48
chanlist=48
noscan=1
tx_queue_data2_burst=2.0
ieee80211n=1
ht_coex=0
ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40][RX-STBC1]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
vht_capab=[SHORT-GI-80][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-A-MPDU-LEN-EXP3]

radio_config_id=b9921f8c8f997ecd6fc45f6f3e2f880f
interface=wlan3
ctrl_interface=/var/run/hostapd
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=XXXXXXX
wpa_psk_file=/var/run/hostapd-wlan3.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=XXXXXXXX
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-wlan3.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
config_id=734edba6badac60dbb9ad3541828abc8

morrownr, 2022-01-12:
Hi @amisix 

> I was hoping you could help me optimize the hostpad.conf for my AWUS036ACHM as we discussed earlier.

I will do my best.

> Can you please tell me what the values you edit mean?

We can work on it one step at a time. This is an incredibly complex program that has a LOT of settings. Rome wasn't built in a day so let's go at it one brick at a time.

I looked at your hostapd.conf files and, well, I realized that I need to know your hardware and what you are trying to do. Here are some questions to get things started:

You are building an AP? Correct?

What computer are you using? RasPi4B?

What is the source of internet? An ethernet cable from your primary WiFi router?

I remember that you have at least one Alfa ACM and an Alfa ACHM. Correct?

Do you want the access point to be dual band? 2.4 GHz and 5 GHz?

What types of clients do you expect to use with this access point?

What guide, link if possible, did you use to make the hostapd.conf files you posted?

Have you analyzed the wifi bands to see if it appears one channel in each may be used less than the others? This will give us a place to start but doing tests is the only way to know which is really better. Once setup and optimized, it is easy to change channels and test. For 2.4 GHz, you basically have 3 choices: 1, 6, 11 and it is best if you stick with 20 MHz width. With 5 GHz you basically have the options of 36(42) and 149(155) and you might as well go with 80 MHz width (for the US). 

Regards

amisix, 2022-01-12:
Thank you, I really do appreciate your time & effort with this. 

>We can work on it one step at a time. This is an incredibly complex program that has a LOT of settings. Rome wasn't built in a day so let's go at it one brick at a time.

Understood. I've been using Luci (OpenWrt v.21.02) for configuration and at this time no settings within the .conf have been manually changed. I have made some modifications in the past when coming across relevant articles but they have long since been removed. Getting further into what some of these lines do would be most excellent. 

>You are building an AP? Correct?

Correct. Basically, a routed AP with multiple adapters/SSIDs that's double NATed for portability and so there's no configuration required on the hosting network. Like one of those RV wifi expanders with a couple extra adapters (and monitor mode).

>What computer are you using? RasPi4B?

A Raspberry Pi 3B v1.2. 

>What is the source of internet? An ethernet cable from your primary WiFi router?

Wifi bridge essentially. I have a gigabit ethernet adapter although it's only used for configuration in case I lose wifi connectivity or my last and only choice for connectivity is ethernet. Portability is important in this case. I've set a gateway metric for the lan (eth1) & wlan (wlanX) and it seems to work fine when switching between either source depending on accessibility. 

>I remember that you have at least one Alfa ACM and an Alfa ACHM. Correct?

Correct. I currently have 2 Alfa ACMs, an ACHM, and an NH. I mostly turn the other adapters off during testing/etc and sometimes mix/match the NH between managed or AP mode depending on my needs at the time (long distance uplink). Adapters are configured to be easily removable/reconfigured so I can have multiple APs on differing frequencies or physically remove pwr or data cables. The NH is not essential to what I'm trying to accomplish as the ACHM will replace it most of the time. 

>Do you want the access point to be dual band? 2.4 GHz and 5 GHz?

Yes. One AP for speed, one for a bit of distance. 

>What types of clients do you expect to use with this access point?

Relatively low impact users, web etc., except for some streaming at times. It will be a small network, probably half a dozen users (if possible) with only one that may be pretty heavy usage at times (me).  

>What guide, link if possible, did you use to make the hostapd.conf files you posted?

Luci, and I used this [OpenWrt relay configuration guide](https://openwrt.org/docs/guide-user/network/wifi/relay_configuration) but without using relayd and this [OpenWrt routed AP configuration guide](https://openwrt.org/docs/guide-user/network/wifi/routedap) along with a bunch of forum posts. 

>Have you analyzed the wifi bands to see if it appears one channel in each may be used less than the others? T

Yup. Everybody near me except my next door neighbor is using the default/auto channels provided by their internet provider. So basically 1, 6, 9, 11, 44 (my neighbor) and 157 (tons of neighbors). I'm using Wifi Analyzer on my android device and nirsoft WifiInfoView on my windows laptop. 

I'm currently running on channels 149 & 3 for my ASUS RT-ACRH13 home router (current source of internet) and channels 36 & 9 for the Raspberry Pi AP(s) (with the NH powered off for now). So, 1st ACM is channel 149 (Managed, wifi bridge), 2nd ACM is channel 36 (AP) & ACHM is channel 9 (AP). 

Thanks.

amisix, 2022-01-12:
>Whew! I'm going to have to ponder this. For now, I am not going to touch this because this is an incredibly complex setup.

I understand. That's one reason why I've made multiple iterations of system images. So that configurations can be swapped out quickly and any issues or mistakes made while working on it are easily resolved with the simple swap of an SD card. So, I guess, no biggie if something in the config doesn't work or bricks the OS, just swap in a new SD card and move on (I'm using Win32DiskImager, EaseUS Partition Manager (to clone), and gparted to extend the root partition) . Pi is perfect for this (even if it sometimes hates me for doing it).

>Can you elaborate on what you mean by "portability is important in this case?" Are we talking about something that you want to carry around with you like a phone? 

It's capable of running on a 268000mah battery for 6-10 hours and small enough to carry around in one hand. It will not be used at my residence except for testing/fun. Partly an educational toy (for myself). 

I hope that helps answer your questions, I'm sorry if it comes off as some hair brained attempt at making a Raspberry Pi/OpenWrt do things it shouldn't do, or that you'd waste your time with such a thing.

morrownr, 2022-01-12:
I am not about to say you are wasting your time. My comments were more to indicate that what you have is something very complex and I don't want to mess it up. I run OpenWRT on my main wifi router but I don't play with the internals much as it just works. My experience with hostapd and RasPi's is such that I am very good at setting up and optimizing hostapd manually but I have zero experience doing it in OpenWRT. 

What I will say is that you are putting 10 pounds worth of stuff in a 5 pound bag. Now that I know more about your setup, I can see why you are looking to make it go faster. A repeater setup automatically cuts throughput in half and then there is the double NAT thing and the RasPi3B usb subsystem has some serious limitations. This setup is not going to be fast.

If you are willing to use a spare sd card to try an alternative approach, I would feel comfortable doing that. If you don't like what we do then format c: and life is good.

If you want to start simply and then build it up to add what we can over time, here is where I would start:

- flash an sd card with the latest RasPiOS. The imager is located...

https://www.raspberrypi.com/software/

Use the default version of RasPiOS.

Use the Alfa ACHM as a client and the internal wifi as the ap.

Start with a simple 2.4 GHz setup.

I can forward some setup instructions if you want.

I know this is not what you wanted but it might turn into something more than you think.

Regards

amisix, 2022-01-12:
>I am not about to say you are wasting your time. My comments were more to indicate that what you have is something very complex and I don't want to mess it up. 

I understand where you're coming from. 

>My experience with hostapd and RasPi's is such that I am very good at setting up and optimizing hostapd manually but I have zero experience doing it in OpenWRT

That's why I've come to you. I'm somewhat comfortable now with the settings in Luci/OpenWrt but it's a GUI and it limits what my adapter settings are a bit. Any input is appreciated. 

>A repeater setup automatically cuts throughput in half and then there is the double NAT thing and the RasPi3B usb subsystem has some serious limitations. This setup is not going to be fast.

I did some iperf tests upon your recommendation yesterday in order to get a baseline to work with. I'm actually somewhat impressed with it. ~90Mbit-110Mbit at 5GHZ (80Mhz width) and 60-80Mbit at 2.4Ghz (40Mhz width) on my internal network & the internet (100Mbit down / 5Mbit out). This is with the ACHM in AP mode. I tried 2.4GHZ at 20Mhz and my speed was considerably slower. I'm getting close (enough) to the 150Mbit maximum of the ACHM at 40Mhz. Below are my results. 

```
Raspberry Pi Adapter: AWUS036ACHM (MT7612U) 
Laptop NIC: TP-Link Archer T4U Plus (RTL8812BU)
Desktop NIC: TP-Link Archer T4U Plus  

iperf3, AWUS036ACHM: 5GHZ/80MHZ

#Laptop --> OpenWrt Pi --> Asus RT-ACRH13 Router (Double NAT) --> Desktop
Connecting to host 10.10.10.100, port 5201
[  5] local 10.10.10.10 port 51388 connected to 10.10.10.100 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  13.8 MBytes   115 Mbits/sec    0    220 KBytes
[  5]   1.00-2.00   sec  13.1 MBytes   110 Mbits/sec    0    220 KBytes
[  5]   2.00-3.00   sec  12.9 MBytes   108 Mbits/sec    0    220 KBytes
[  5]   3.00-4.00   sec  12.9 MBytes   108 Mbits/sec    0    220 KBytes
[  5]   4.00-5.00   sec  13.1 MBytes   110 Mbits/sec    0    220 KBytes
[  5]   5.00-6.00   sec  13.7 MBytes   114 Mbits/sec    0    220 KBytes
[  5]   6.00-7.00   sec  12.7 MBytes   107 Mbits/sec    0    220 KBytes
[  5]   7.00-8.00   sec  13.5 MBytes   113 Mbits/sec    0    220 KBytes
[  5]   8.00-9.00   sec  12.9 MBytes   108 Mbits/sec    0    220 KBytes
[  5]   9.00-10.00  sec  12.6 MBytes   106 Mbits/sec    0    220 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   131 MBytes   110 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   130 MBytes   109 Mbits/sec                  receiver 

#OpenWrt Pi --> Laptop
Connecting to host 10.10.40.100, port 5201
[  5] local 10.10.40.1 port 52232 connected to 10.10.40.100 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  15.0 MBytes   126 Mbits/sec    1    151 KBytes
[  5]   1.00-2.00   sec  13.4 MBytes   112 Mbits/sec    4    106 KBytes
[  5]   2.00-3.00   sec  14.2 MBytes   119 Mbits/sec    0    180 KBytes
[  5]   3.00-4.00   sec  14.2 MBytes   119 Mbits/sec    0    210 KBytes
[  5]   4.00-5.00   sec  14.1 MBytes   118 Mbits/sec    0    210 KBytes
[  5]   5.00-6.00   sec  14.4 MBytes   121 Mbits/sec    0    210 KBytes
[  5]   6.00-7.00   sec  12.4 MBytes   104 Mbits/sec    0    210 KBytes
[  5]   7.00-8.00   sec  13.0 MBytes   109 Mbits/sec    0    212 KBytes
[  5]   8.00-9.00   sec  14.2 MBytes   119 Mbits/sec    0    212 KBytes
[  5]   9.00-10.00  sec  14.0 MBytes   118 Mbits/sec    0    212 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   139 MBytes   117 Mbits/sec    5             sender
[  5]   0.00-10.00  sec   138 MBytes   116 Mbits/sec                  receiver

#OpenWrt Pi --> Desktop
Connecting to host 10.10.10.100, port 5201
[  5] local 10.10.10.10 port 51380 connected to 10.10.10.100 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  13.4 MBytes   113 Mbits/sec    0    220 KBytes
[  5]   1.00-2.00   sec  12.9 MBytes   108 Mbits/sec    0    220 KBytes
[  5]   2.00-3.00   sec  13.7 MBytes   115 Mbits/sec    0    220 KBytes
[  5]   3.00-4.00   sec  12.6 MBytes   106 Mbits/sec    0    220 KBytes
[  5]   4.00-5.00   sec  13.4 MBytes   113 Mbits/sec    0    220 KBytes
[  5]   5.00-6.00   sec  13.5 MBytes   113 Mbits/sec    0    220 KBytes
[  5]   6.00-7.00   sec  13.1 MBytes   110 Mbits/sec    0    220 KBytes
[  5]   7.00-8.00   sec  12.8 MBytes   107 Mbits/sec    0    220 KBytes
[  5]   8.00-9.00   sec  13.7 MBytes   115 Mbits/sec    0    220 KBytes
[... 48 lines cut ...]
#Laptop --> OpenWrt Pi
Connecting to host 10.10.10.10, port 5201
[  4] local 10.10.40.100 port 63536 connected to 10.10.10.10 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  11.2 MBytes  94.2 Mbits/sec
[  4]   1.00-2.00   sec  12.2 MBytes   103 Mbits/sec
[  4]   2.00-3.00   sec  11.8 MBytes  98.2 Mbits/sec
[  4]   3.00-4.01   sec  12.2 MBytes   103 Mbits/sec
[  4]   4.01-5.00   sec  12.2 MBytes   103 Mbits/sec
[  4]   5.00-6.01   sec  12.5 MBytes   104 Mbits/sec
[  4]   6.01-7.00   sec  12.2 MBytes   103 Mbits/sec
[  4]   7.00-8.00   sec  12.4 MBytes   104 Mbits/sec
[  4]   8.00-9.00   sec  12.4 MBytes   104 Mbits/sec
[  4]   9.00-10.01  sec  12.2 MBytes   102 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[... cut ...]    sender
[  4]   0.00-10.01  sec   121 MBytes   102 Mbits/sec                  receiver
```

>If you are willing to use a spare sd card to try an alternative approach, I would feel comfortable doing that. If you don't like what we do then format c: and life is good.

>Use the Alfa ACHM as a client and the internal wifi as the ap.

Sure. I've played around a bit in the RasPiOS, I'll do that and see where it goes. Please link whatever you feel may be useful and I'll reply back when it's complete.

morrownr, 2022-01-13:
> I did some iperf tests upon your recommendation yesterday in order to get a baseline to work with. I'm actually somewhat impressed with it

Given the overall situation and setup, those are really good numbers. Can it be improved? Well, that is where the fun comes in.

> Sure. I've played around a bit in the RasPiOS, I'll do that and see where it goes. Please link whatever you feel may be useful and I'll reply back when it's complete.

Given me a some time to find a good guide for us to start with. I'd like to mirror your efforts on my Pi3B. I think starting with a simply manual configuration where we can slowly add features would be a good education exercise and would not destroy the work you have already done.

amisix, 2022-01-13:
>Given the overall situation and setup, those are really good numbers. Can it be improved? Well, that is where the fun comes in.

Yay! 

>Given me a some time to find a good guide for us to start with.

Great, no worries.

>I'd like to mirror your efforts on my Pi3B. I think starting with a simply manual configuration where we can slowly add features would be a good education exercise...

Works for me. 

>...and would not destroy the work you have already done.

I appreciate the consideration. I agree. There are too many variables with my current setup to be of use at this time.  Looking forward to the learning experience. 

Thanks.

morrownr, 2022-01-14:
[WiFi Repeater.tar.gz](https://github.com/morrownr/USB-WiFi/files/7872027/WiFi.Repeater.tar.gz)

See the above WiFi Repeater guide.

To use the guide, all you need is a Pi3 or Pi4, an sd card freshly burned with RasPiOS (a very with a gui is best), a power supply and an ALFA ACHM plugged into a USB2 port. I run headless but if you want, a keyboard, mouse and monitor can be attached as well.

The idea is to start off simply and optimize as we go along. This guide will NOT work with Realtek drivers so if anyone is watching and wants to try this with any of the Realtek driver maintained here, it won't work. The reason is that Realtek does not support adding an interface via `iw`. Is it possible to do this with the Realtek drivers that support concurrent mode? It may be but I have not tried it. If you have time to try it, please report the results.

Please note that the internal wifi capability is turned off so the Alfa adapter is serving as both the client and AP. Yes, these adapters can do more than one thing at a time.

```
$ iw dev
phy#0
	Interface ap0
		ifindex 4
		wdev 0x2
		addr 02:c0:ca:ad:4b:cc
		ssid myPi-Repeater
		type AP
		channel 11 (2462 MHz), width: 20 MHz (no HT), center1: 2462 MHz
		txpower 2.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 00:c0:ca:ad:4b:cc
		ssid Trout
		type managed
		channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
		txpower 2.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	0	0	0	0	0	00
```

I put this guide together rapidly with only minor testing so if you see ways to improve the guide, please speak up.

Regards

amisix, 2022-01-20:
I wanted to follow up and let you know that I've run through the steps and was not quite successful yet at getting ap0 created. 
All I have is Interface wlan0 after reboot (in managed mode) with no connection and ap0 is not listed. I ran through the instructions a second time although I didn't see anything I missed yet, but want to check again. I don't think it's the instructions, it's probably me. 

The only adapter(s) I have connected are the gigabit ethernet adapter (eth1) and the ACHM (wlan0 & ap0, eventually) 

Thanks.

Here's my **iw dev** (useless, really)
```
phy#0
                    Interface wlan0
                                ifindex 3
                                wdev 0x1
                                addr: XX:XX:XX:XX:XX:XX
                                type managed
                                txpower 4.00 dBm
                                multicast TXQ:
                                0     0     0     0     0     0
```

morrownr, 2022-01-20:
Not my finest effort. I'm take another look and see what I can come up with.

amisix, 2022-01-20:
Mine either, no worries. I'll do a fresh image again tomorrow when I have more time to see if I have better success. I appreciate your assistance very much.

closed 2022-05-21 by morrownr