morrownr/USB-WiFi #39: Unable to create AP on 5ghz TEROW ROW02CD RPI4 Alpine Linux

desultory, 2022-01-12:
I purchased this device: https://www.amazon.com/dp/B086L6TR6G?psc=1

Dmesg output:

`[ 1119.652447] usb 2-2: new SuperSpeed USB device number 4 using xhci_hcd
[ 1119.673827] usb 2-2: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[ 1119.673849] usb 2-2: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[ 1119.673863] usb 2-2: Product: Wireless
[ 1119.673875] usb 2-2: Manufacturer: MediaTek Inc.
[ 1119.673888] usb 2-2: SerialNumber: 000000000
[ 1119.809410] usb 2-2: reset SuperSpeed USB device number 4 using xhci_hcd
[ 1119.831002] mt76x2u 2-2:1.0: ASIC revision: 76120044
[ 1119.863848] mt76x2u 2-2:1.0: ROM patch build: 20141115060606a
[ 1120.028074] mt76x2u 2-2:1.0: Firmware Version: 0.0.00
[ 1120.028102] mt76x2u 2-2:1.0: Build: 1
[ 1120.028114] mt76x2u 2-2:1.0: Build Time: 201507311614____
[ 1120.840104] ieee80211 phy4: Selected rate control algorithm 'minstrel_ht'`

Iw list output:
`Wiphy phy4
        wiphy index: 4
        max # scan SSIDs: 4
        max scan IEs length: 2243 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                Bitrates (non-HT):
[... 204 lines cut ...]
country_code=US
wmm_enabled=1

ieee80211d=1
ieee80211h=1

hw_mode=a
ieee80211n=1
require_ht=1
ieee80211ac=1
require_vht=1

vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42
#channel=149
#vht_oper_centr_freq_seg0_idx=155

#ht_capab=[RX-LDPC][HT40-][HT40+][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]
#vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7][SU-BEAMFORMER][DSSS_CCK-40]

ht_capab=[RX-LDPC][HT40-][HT40+][MAX-AMSDU-7935][DSSS_CCK-40]
vht_capab=[MAX-MPDU-11454][RX-LDPC][SHORT-GI-80][TX-STBC][HTC-VHT]`

I can see in iw list that this device apparently does not support broadcasting on any 5ghz bands.  I was using this alfa card but I was getting really poor performance and the network would have to be reset once a day or so when the card crashed https://www.amazon.com/gp/product/B01MZD7Z76?psc=1

morrownr, 2022-01-12:
Hi @desultory 

> I can see in iw list that this device apparently does not support broadcasting on any 5ghz bands.

It supports broadcasting 5 GHz.

I have a repo for the 7612u chipset:

https://github.com/morrownr/7612u

At that repo is the following file which may contain a lot of info that might help:

https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point.md

I can see that the following lines in your hostapd.conf have problems: (these are for 5GHz and are ones that should work)

ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]

vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

hostapd can be very picky. In your `vht_capab` line, I see `[HTC-VHT]`. That is not a correct entry for this adapter. In fact, hostapd will not come up if that entry is included.

Let me know how it goes.

Regards

desultory, 2022-01-13:
I tried those hostapd config lines and got this output, I think I updated my config in an edit between when you last read, are you saying that I need to compile your drivers for full AP mode? That is fine but I was just under the impression it would be plug and play. 

```
ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]

vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
```

config file:

```
nterface=wlan2
driver=nl80211

country_code=US
wmm_enabled=1

ieee80211d=1
ieee80211h=1

hw_mode=a
ieee80211n=1
require_ht=1
ieee80211ac=1
require_vht=1

vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42
#channel=149
#vht_oper_centr_freq_seg0_idx=155

#ht_capab=[RX-LDPC][HT40-][HT40+][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]
#vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7][SU-BEAMFORMER][DSSS_CCK-40]

#ht_capab=[RX-LDPC][HT40-][HT40+][MAX-AMSDU-7935][DSSS_CCK-40]
#ht_capab=[RX-LDPC][HT40-][HT40+][MAX-AMSDU-3839]
#vht_capab=[MAX-MPDU-3895][RX-LDPC][SHORT-GI-80][TX-STBC]

ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
```

morrownr, 2022-01-13:
Hi again Zen

> are you saying that I need to compile your drivers for full AP mode? That is fine but I was just under the impression it would be plug and play.

Well, I don't have any drivers for 7612u. My repo is informational only.

Here is the repo for the 7612u:

https://github.com/torvalds/linux/tree/master/drivers/net/wireless/mediatek

It is run by a guy named Linus Torvalds. You may have heard of him.

Which distro and kernel version are you running? Unless it is very old, your adapter is plug and play.

If what you posted above for your hostapd.conf is all there is in the file, then we have a big problem because you are missing a lot of lines that are needed.

What I recommend is that you go to...

https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point.md

...and start looking at the details. That file provides an example hostapd.conf for both bands. You have said exactly what you are doing but that file should provide a lot of info.

Regards

desultory, 2022-01-13:
that is not my entire hostapd file, I have not included the bottom section which contains my ssid info and such.  I know that works.  I'm using the latest version of alpine linux.  I used your repo before with the rtl8814au chipset on an alfa card and it was very helpful.  I purchased the TEROW ROW02CD because you recommended it and it for whatever reason, it seems to not want to support 5ghz mode.

desultory, 2022-01-13:
For reference, I am also using a Asus USB N53 with this config and it works fine:

```
interface=wlan0
driver=nl80211

country_code=US
wmm_enabled=1

ieee80211d=1
#ieee80211h=1

hw_mode=g
require_ht=1
ieee80211n=1
channel=7
ht_capab=[HT40-][RX-STBC1][SHORT-GI-40][SHORT-GI-20]

ignore_broadcast_ssid=0
macaddr_acl=0
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
#wpa_pairwise=TKIP
rsn_pairwise=CCMP
```

morrownr, 2022-01-13:
> nl80211: Regulatory information - country=00

Is the country set in the OS? (I am not familiar with Alpine so will be not help there) However, I can tell you how to do it the RasPiOS.

To check:

$ iw reg get

Side note:
require_ht=1
require_vht=1

I don't use those lines. Just make sure understand what they do. I've seen folks have due to using those lines.

desultory, 2022-01-13:
I ran iw reg set US and it appears the same in iw list and is doing the same thing:

```
random: Trying to read entropy from /dev/random
Configuration file: hostapd_AC.conf
nl80211: TDLS supported
nl80211: TDLS external setup
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: interface wlan1 in phy phy15
nl80211: Set mode ifindex 21 iftype 3 (AP)
nl80211: Setup AP(wlan1) - device_ap_sme=0 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x7f8c3ef8a0
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x7f8c3ef8a0 match=
nl80211: Register frame type=0x0 (WLAN_FC_STYPE_ASSOC_REQ) nl_handle=0x7f8c3ef8a0 match=
nl80211: Register frame type=0x20 (WLAN_FC_STYPE_REASSOC_REQ) nl_handle=0x7f8c3ef8a0 match=
nl80211: Register frame type=0xa0 (WLAN_FC_STYPE_DISASSOC) nl_handle=0x7f8c3ef8a0 match=
nl80211: Register frame type=0xc0 (WLAN_FC_STYPE_DEAUTH) nl_handle=0x7f8c3ef8a0 match=
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x7f8c3ef8a0 match=
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=0501
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=0503
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=0504
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=06
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=08
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=0a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=11
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7f8c3ef8a0 match=7f
rfkill: initial event: idx=15 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 21 (ifidx_reason -1)
nl80211: if_indices[16]: 21(-1)
phy: phy15
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code 00, new country code US
Continue interface setup after channel list update
ctrl_iface not configured!
random: Got 20/20 bytes from /dev/random
RTM_NEWLINK: ifi_index=21 ifname=wlan1 operstate=0 linkmode=0 ifi_family=0 ifi_flags=0x1043 ([UP][RUNNING])
RTM_NEWLINK: ifi_index=21 ifname=wlan1 operstate=2 linkmode=0 ifi_family=0 ifi_flags=0x1003 ([UP])
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlan1
nl80211: RX frame da=ff:ff:ff:ff:ff:ff sa=a8:bb:50:86:29:00 bssid=ff:ff:ff:ff:ff:ff freq=2412 ssi_signal=-53 fc=0x40 seq_ctrl=0xa3a0 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=49
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlan1
nl80211: RX frame da=ff:ff:ff:ff:ff:ff sa=a8:bb:50:86:29:00 bssid=ff:ff:ff:ff:ff:ff freq=2412 ssi_signal=-56 fc=0x40 seq_ctrl=0xa3b0 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=49
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlan1
nl80211: RX frame da=ff:ff:ff:ff:ff:ff sa=50:e0:85:aa:88:0e bssid=ff:ff:ff:ff:ff:ff freq=2412 ssi_signal=-24 fc=0x40 seq_ctrl=0xfeb0 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=113
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlan1
nl80211: RX frame da=ff:ff:ff:ff:ff:ff sa=50:e0:85:aa:88:0e bssid=ff:ff:ff:ff:ff:ff freq=2412 ssi_signal=-26 fc=0x40 seq_ctrl=0xfec0 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=91
nl80211: send_mlme - da= 50:e0:85:aa:88:0e noack=1 freq=0 no_cck=0 offchanok=0 wait_time=0 fc=0x50 (WLAN_FC_STYPE_PROBE_RESP) nlmode=3
nl80211: send_mlme -> send_frame
nl80211: send_frame - Use bss->freq=0
nl80211: send_frame -> send_frame_cmd
nl80211: Frame command failed: ret=-22 (Invalid argument) (freq=0 wait=0)
handle_probe_req: send failed
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlan1
nl80211: RX frame da=ff:ff:ff:ff:ff:ff sa=50:e0:85:aa:88:0e bssid=ff:ff:ff:ff:ff:ff freq=2412 ssi_signal=-29 fc=0x40 seq_ctrl=0xff10 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=113
[... 153 lines cut ...]
ieee80211d=1
ieee80211h=1

hw_mode=a
ieee80211n=1
#require_ht=1
ieee80211ac=1
#require_vht=1

vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42

ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

ignore_broadcast_ssid=0
macaddr_acl=0
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
#wpa_pairwise=TKIP
rsn_pairwise=CCMP

```

morrownr, 2022-01-13:
This message is going through my ROW02CD. It is currently working as a 5 GHz AP on a RasPi4B running RasPiOS (32 bit). Setup is per the guide I posted earlier.

I am posting my `$ iw list` and the `hostapd.conf` that I am using. If you have a spare sd card, you might consider using the guide to make a setup like mine with the RasPiOS. If that works, that will rule out some issues and help confirm whether the adapter is working correctly. FYI: I set the guide up to where it is relatively easy to copy and paste things so it probably won't take as long as you think.

```
$ iw list
Wiphy phy1
	wiphy index: 1
	max # scan SSIDs: 4
	max scan IEs length: 2243 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		Bitrates (non-HT):
			* 1.0 Mbps (short preamble supported)
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
[... 304 lines cut ...]
# rtl8814au
#vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# rtl8811au - rtl8811cu
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# rtl8832au - hw vht capab: 0x03c070b1
#vht_capab=[RXLDPC][MAX-MPDU-7991][SHORT-GI-80][TX-STBC-2BY1][HTC-VHT]
#[SU-BEAMFORMEE][BF-ANTENNA-4][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#vht_capab=[MAX-MPDU-7991][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][BF-ANTENNA-4][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# Note: [TX-STBC-2BY1] may cause problems with some Realtek drivers

# IEEE 802.11ax
#ieee80211ax=0
#he_su_beamformer=1
#he_su_beamformee=1
#he_mu_beamformer=1
#he_bss_color=1
#he_default_pe_duration=0
#he_basic_mcs_nss_set=2

# end of hostapd-5g.conf

```

desultory, 2022-01-13:
I tried it with your conf and got the same error,

look at this portion in particular:

```
nl80211: Regulatory information - country=00
nl80211: 2402-2472 @ 40 MHz 20 mBm
nl80211: 2457-2482 @ 20 MHz 20 mBm (no IR)
nl80211: 2474-2494 @ 20 MHz 20 mBm (no OFDM) (no IR)
nl80211: 5170-5250 @ 80 MHz 20 mBm (no IR)
nl80211: 5250-5330 @ 80 MHz 20 mBm (DFS) (no IR)
nl80211: 5490-5730 @ 160 MHz 20 mBm (DFS) (no IR)
nl80211: 5735-5835 @ 80 MHz 20 mBm (no IR)
nl80211: 57240-63720 @ 2160 MHz 0 mBm
nl80211: Added 802.11b mode based on 802.11g information
nl80211: Mode IEEE 802.11g: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472[NO_IR] 2484[NO_IR]
nl80211: Mode IEEE 802.11a: 5180[NO_IR] 5200[NO_IR] 5220[NO_IR] 5240[NO_IR] 5260[NO_IR][RADAR] 5280[NO_IR][RADAR] 5300[NO_IR][RADAR] 5320[NO_IR][RADAR] 5500[NO_IR][RADAR] 5520[NO_IR][RADAR] 5540[NO_IR][RADAR]
nl80211: Mode IEEE 802.11b: 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467[NO_IR] 2472[NO_IR] 2484[NO_IR]
Channel 36 (primary) not allowed for AP mode, flags: 0x853 NO-IR
wlan1: IEEE 802.11 Configured channel (36) not found from the channel list of current mode (2) IEEE 802.11a
wlan1: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
wlan1: interface state COUNTRY_UPDATE->DISABLED
```

it is saying the card does not support IR mode on 5ghz, iw list says the same thing.  

Here is a relevant portion of my lsmod:

```
mt76x2u                28672  0
mt76x2_common          28672  1 mt76x2u
mt76x02_usb            20480  1 mt76x2u
mt76_usb               36864  2 mt76x02_usb,mt76x2u
mt76x02_lib            77824  3 mt76x02_usb,mt76x2u,mt76x2_common
mt76                   77824  5 mt76_usb,mt76x02_lib,mt76x02_usb,mt76x2u,mt76x2_common
rt2800usb              36864  0
rt2800lib             126976  1 rt2800usb
rt2x00usb              28672  1 rt2800usb
rt2x00lib              69632  3 rt2800usb,rt2x00usb,rt2800lib
mac80211              888832  7 mt76,mt76x02_lib,rt2x00lib,mt76x02_usb,rt2x00usb,mt76x2u,rt2800lib
libarc4                16384  1 mac80211
cfg80211              856064  6 mt76,8814au,mt76x02_lib,rt2x00lib,mac80211,mt76x02_usb
rfkill                 36864  3 rt2x00lib,cfg80211
crc_ccitt              16384  1 rt2800lib
```

IDK, everything points to this card not supporting 5ghz ap mode at all, I may try it on raspios at some point but this is not really that operating system dependent as this is built into the kernel and I'm using a very up to date version of the kernel.

morrownr, 2022-01-13:
> nl80211: Regulatory information - country=00

There is the problem. 

The same line in my RasPi hostapd.log says:

nl80211: Regulatory information - country=US (DFS-FCC)

The fundamental problem is that hostapd and the mt7612u driver must have regulatory information to determine what to support. Without that information, a very restrictive data set that is valid for anywhere in the world is used.

The problem you are seeing is not a problem with the driver.

The problem you are seeing is not a problem with hostapd.

It is highly unlikely that it is a problem with the wifi adapter.

It is nearly certain that the issue is that the regulatory information is not being properly set in your OS and I cannot help with that as I am unfamiliar with Alpine. I recommend you seek out the available support provided for Alpine... and if that is unsuccessful, the RasPiOS is available and there is a lot of support available.

Regards

desultory, 2022-01-13:
I guess I will need to try off of Alpine linux unless I can find a way to safely set it on boot.  It's just weird that setting it in hostapd doesn't seem to affect it at all, I've never had this issue with another card.

morrownr, 2022-01-14:
> It's just weird that setting it in hostapd doesn't seem to affect it at all

I agree but have not investigated it in detail. The setting in hostapd does no good if the proper information is not flowing from the OS. There are some legacy things still in hostapd so it could be that the hostapd.conf settings is from how things used to be done with other hostapd drivers (not to be confused with usb wifi adapter drivers).

I know this is frustrating. Been there, done many frustrating things. Playing with distros that are not mainline distros is fun but I always have an sd card with RasPiOS available. I also always have a good stable distro on a desktop system or two so I can check when issues like this come up. I don't think this is an issue with the adapter but if further testing shows that it is then I will be interested in the details so that I can post as much so other users know.

I know the guide I posted is long but it is well tested and it does something almost no other guides do... it doesn't just show you how to set up an ap, it shows you how to get the best performance from the commonly used adapters that we talk about around here.

desultory, 2022-01-14:
Yeah, I'll get around to this eventually, I thought this was gonna be a plug and play thing and I currently use that pi as a router among other things right now so I'm not ready for it to be offline for an extended period of time yet.  

This is a bit unrelated to this issue, but is this the right comfast adapter? I was having issues finding it: https://www.aliexpress.com/item/1005002599961851.html

jackloomen, 2022-01-15:
"Since Linux 4.15 there is support for loading regulatory domains directly from database file[0] (regulatory.db) without now deprecated 'crda' tool. The database is provided in 'wireless-regdb' package[1] since last month thus it's recommended to switch optional dependency in kernel package."

"The kernel will attempt to load regdb in a new extensible format from /lib/firmware/regulatory.db and
verify it with a detached signature lib/firmware/regulatory.db /lib/firmware/regulatory.db.p7s
This can replace the use of crda from linux 4.15 onwards provided the new files are packaged."

for some linux distros you can still (or may need to) install crda
in alpine, crda has been deprecated in favor of the kernel loading the regdb directly from /lib/firmware/regulatory.db

if that file it absent on your system or your kernel implements lockdown and the signature for that file fails to validate, it would result in highly restricted global reg domain

on alpine you need to install the wireless-regdb package

if you have regulatory.db but still fail to change reg domain with iw reg set US for example, check dmesg for errors pertaining regdb or cfg80211. Workaround may be to disable secure boot thereby disabling kernel lockdown.

morrownr, 2022-01-16:
Hi @jackloomen 

Good stuff. If he is running the system currently as a wired router or even if he just adds a 2.4 GHz AP, this can be overlooked as it mostly doesn't come into play but as soon as you try to bring up a 5 GHz AP, boom.

Hi @desultory 

> This is a bit unrelated to this issue, but is this the right comfast adapter? I was having issues finding it: https://www.aliexpress.com/item/1005002599961851.html

I am a little lost. I thought we were talking about a ROW02CD? If you are looking to add another adapter, the adapter in that link does work (it is a multi-state adapter so that you are aware). Comfast actually makes several adapters with Mediatek chipsets, actually several with the mt7612u. They do tend to make them multi-state so if I were picking between your ROW02CD and the adapter in the link, I would go with the ROW02CD as it is single state. I have both adapters. The performance is about the same.

Regarding the help mess above, have you considered testing your AP setup with 2.4 GHz? That could help determine if there is a problem with the adapter. If it works, the problem is likely with the reg stuff as we talking about.

desultory, 2022-01-18:
Simply running `apk get wireless-regdb` fixed it for me, using raspios also seemed to work but I don't want to be stuck with a 32 bit kernel

desultory, 2022-01-18:
I will note that the speeds don't seem to be great, for whatever reason the RX/TX speeds in windows are listed as "32/866 (Mbps)"and I can't get over 200mbps in iperf3 and the devices are right next to each other, I was able to get more than that with the alfa card, thought it would die after a day or so and needed to be manually unplugged, replugged in and services had to be restarted

morrownr, 2022-01-18:
> I will note that the speeds don't seem to be great, for whatever reason the RX/TX speeds in windows are listed as "32/866 (Mbps)"and I can't get over 200mbps in iperf3 and the devices are right next to each other

I'm going to have to make some assumptions here as it wasn't clear to me how things are set up.

I'll assume the Windows machine is the client and your TEROW adapter is the AP on your Pi. It appears your Windows machine is showing the link speed and 866 is indeed the link speed that the TEROW adapter is capable of in 5 GHz with an 80 MHz wide channel. That tells us a lot of things are set up properly.

> and I can't get over 200mbps in iperf3 and the devices are right next to each other

Being right next to each other could be hurting the iperf3 data. Closer is not always a better thing. Now, with a link speed of 866, the maximum iperf3 speed we should expect is around 400 Mb/s. It is a half duplex thing. We could get 400 Mb/s but we would need no interference and no congestion and all computers/adapters set up perfectly. Remember that wifi is designed to share. When other clients and ap's that are on a channel in the same block of channels that you are in, there will be sharing... which reduces your speed. Where I live these days, there are dozens of AP's on channel 44 and 157 (over 45 each in fact). They are set to 80 MHz channel width. That really hurts and I can't get 200 Mb/s with iperf3 if I try to compete with that depending on the time of day.

Here is what I do beat it and have very reliable wifi.

I don't compete on the highly congested channels/blocks. I go find cleaner air. I have a main router, ZyXEL NBG6817 running OpenWRT, that pulls in internet at about 100 Mb/s. The Zyxel uses channel 100 (DFS) for 5 GHz and channel 11 for band 1. A CAT6 cable connects the RasPi4B to the Zyxel. The RasPi is on channel 6 and channel 36 (set to 80211n, 20 MHz width). 

Given that we have some things that won't work on DFS, I think we all have that problem, I wanted to find some clear air for 5 GHz and I found a 20 MHz wide area of clean air at channel 36 because all the other AP's around here in the lower band are on 44. For whatever reason, it seems a lot of people are not using 2.4 GHz much. Channel 1 is heavily congested but 6 and 11 are relatively uncontested.

I can get 400 Mb/s on my main box here right now because it is connected to the DFS channel on my main router. That is serious overkill but cool to look at. Getting good throughput is a matter of knowing and taking advantage of what your location gives you.

Hope this helps.

desultory, 2022-01-18:
My setup is a rip4 with the wan on the built in port and lan on a usb to gigabit ethernet adapter.  I can get about 300/20mbps from my isp and I have no issue getting these speeds across the ethernet interfaces.  I was able to get about 400mbps on iperf with the alfa rtl8814au based card I had before which makes sense given what you said before.  
I've used a wifi analyzer to check the air and the only other network on the same bands as my network is my roommate's and it's on another floor and far away and shouldn't be interfering that much.  I should be getting DFS channels with 802.11h and 802.11d enabled in my conf, right?

jackloomen, 2022-01-18:
when in close proximity tuning txpower to be as low as possible often improves performance

`iw dev wlan0 set txpower fixed 500`

500 = 5dbm

morrownr, 2022-01-18:
> I should be getting DFS channels with 802.11h and 802.11d enabled in my conf, right?

Well, no. USB WiFi is considered a niche market by the chipset makers so we tend to end up at the end of the line when it comes to some features. Until recently I would have said that no USB WiFi adapters were supporting DFS channels in AP mode but low and behold, along comes an up dated driver for the rtl8812au that does support this:

https://github.com/morrownr/8812au-20210629

The following pull request over at the OpenWRT MT76 repo indicates that the code is in the 7612u adapter but it is not turned on. I am trying to push this issue and will continue.

 https://github.com/openwrt/mt76/pull/428

The Realtek 8814au driver has never supported this.

Overall, we are not in good shape in this regard. Hopefully we can get the appropriate changes made.

morrownr, 2022-01-18:
> I don't want to be stuck with a 32 bit kernel

Hey, don't be talkin' bad about my 32 bit kernel.

desultory, 2022-01-18:
Well regardless, your repo has made doing this a lot more accessible.  I for one prefer to be able to use my own wireless devices for networking equipment I use but I can see how that isn't very common. Hopefully more manufacturers will see that people care and that communities will be willing to do some of the work for them if they give them something to work with.

morrownr, 2022-01-19:
My opinion of the best path forward is increase the amount of information available to Linux users so that we can make good decisions. I think most Linux users would choose usb WiFi adapters that work with in-kernel drivers if that information is available to them. The Linux foundation has put a lot of effort into established standards that provide guidance to the many developers out there. Given the rolling dev model of the Linux kernel, the best way to support it with drivers is to do what Linux Wireless standards call for and that is mac80211 technology drivers that are maintained in-kernel. For usb WiFi, Mediatek does that and Realtek goes their own way which is very painful for Linux users.

I think as more Linux users move to Mediatek chipset based adapters, we will see more noise regarding things that users want to see. The MT76 drivers are very good but I there are 3 things that we collectively should be asking to have fixed:

- No DFS support in AP mode.
- Drivers default to LED off. They should default to on and give us an easy way to turn them off.
- Make the default for Scatter-Gather support off.

I just took a look at that list and thought to myself "that is the best you can come up with?" All 3 of those are basically configuration changes requiring no major work. WPA3 works. Monitor mode is great. AP mode is great. They support VIF. I could go on.

On the other hand, I could make a long list of things that are missing or don't work right or don't work the right way with the Realtek drivers.

We need Mediatek to announce a new WiFi 6 chipset.

jackloomen, 2022-01-20:
> * No DFS support in AP mode.

The chipset is likely not capable of radar detection.

I believe you can patch the driver to use DFS frequencies regardless:

https://github.com/openwrt/mt76/pull/428

morrownr, 2022-01-20:
I'm under the impression that the version of the mt7612 that is used in internal cards/boards is capable of radar detection. I don't know if the usb version is capable of radar detection.

I am aware of the patch but have not tested it. Have you tested it?

desultory, 2022-01-21:
I have only used the in kernel drivers with no extra configuration for the kernel modules

cristian-ciobanu, 2022-09-01:
I also have some hard time setting up an AP on 5 GHz using an EDUP EP-AC1605 USB WiFi card which has MT7612U chipset
I have followed the recommendations from this page https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md and created a new hostapd.conf file with the same content listed on the wiki page.
I also have set the **disable_usb_sg** parameter to 1

Whenever I start the hostapd using the created conf file I'm unable to see any SSID and receive beacon errors. Below is the debug output

`random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd-5g.conf
nl80211: TDLS supported
nl80211: TDLS external setup
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: key_mgmt=0x1ff0f enc=0xfef auth=0x7 flags=0x4000531db5bfae0 rrm_flags=0x10 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=4
nl80211: interface wlan1 in phy phy21
nl80211: Set mode ifindex 25 iftype 3 (AP)
nl80211: Setup AP(wlan1) - device_ap_sme=0 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x55a8e1934670
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x55a8e1934670 match= multicast=0
nl80211: Register frame type=0x0 (WLAN_FC_STYPE_ASSOC_REQ) nl_handle=0x55a8e1934670 match= multicast=0
nl80211: Register frame type=0x20 (WLAN_FC_STYPE_REASSOC_REQ) nl_handle=0x55a8e1934670 match= multicast=0
nl80211: Register frame type=0xa0 (WLAN_FC_STYPE_DISASSOC) nl_handle=0x55a8e1934670 match= multicast=0
nl80211: Register frame type=0xc0 (WLAN_FC_STYPE_DEAUTH) nl_handle=0x55a8e1934670 match= multicast=0
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x55a8e1934670 match= multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=04 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=0501 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=0503 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=0504 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=06 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=08 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=09 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=0a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=11 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55a8e1934670 match=7f multicast=0
rfkill: initial event: idx=21 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 3 (ifidx_reason 25)
nl80211: if_indices[16]: 3(25)
nl80211: Add own interface ifindex 25 (ifidx_reason -1)
nl80211: if_indices[16]: 3(25) 25(-1)
nl80211: Adding interface wlan1 into bridge br0
nl80211: Do not open EAPOL RX socket - using control port for RX
phy: phy21
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code US, new country code US 
nl80211: Regulatory information - country=US (DFS-FCC)
nl80211: 902-904 @ 2 MHz 30 mBm
nl80211: 904-920 @ 16 MHz 30 mBm
nl80211: 920-928 @ 8 MHz 30 mBm
nl80211: 2400-2472 @ 40 MHz 30 mBm
nl80211: 5150-5250 @ 80 MHz 23 mBm
nl80211: 5250-5350 @ 80 MHz 24 mBm (DFS)
nl80211: 5470-5730 @ 160 MHz 24 mBm (DFS)
nl80211: 5730-5850 @ 80 MHz 30 mBm
nl80211: 5850-5895 @ 40 MHz 27 mBm (no outdoor) (no IR)
nl80211: 5925-7125 @ 320 MHz 12 mBm (no outdoor) (no IR)
nl80211: 57240-71000 @ 2160 MHz 40 mBm
nl80211: Added 802.11b mode based on 802.11g information
[... 117 lines cut ...]
wlan1: Event TX_STATUS (16) received
mgmt::deauth cb
nl80211: Drv Event 16 (NL80211_CMD_STOP_AP) received for wlan1
wlan1: Event INTERFACE_UNAVAILABLE (28) received
Interface wlan1 is unavailable -- stopped
RTM_NEWLINK: ifi_index=25 ifname=wlan1 operstate=2 linkmode=0 master=3 ifi_family=0 ifi_flags=0x1002 ()
nl80211: Interface down (wlan1/wlan1)
wlan1: Event INTERFACE_DISABLED (26) received
wlan1: INTERFACE-DISABLED 
RTM_NEWLINK: ifi_index=25 ifname=wlan1 master=3 operstate=2 ifi_family=7 ifi_flags=0x1002 ()
nl80211: Add ifindex 3 for bridge br0
nl80211: Add own interface ifindex 3 (ifidx_reason 25)
nl80211: ifindex 3 already in the list
RTM_NEWLINK: ifi_index=25 ifname=wlan1 master=3 operstate=2 ifi_family=7 ifi_flags=0x1002 ()
nl80211: Interface wlan1 does not exist - ignore RTM_NEWLINK
nl80211: Add ifindex 3 for bridge br0
nl80211: Add own interface ifindex 3 (ifidx_reason 25)
nl80211: ifindex 3 already in the list
RTM_DELLINK: ifi_index=25 ifname=wlan1 master=3 operstate=2 ifi_family=7 ifi_flags=0x1002 ()
nl80211: Remove ifindex 3 for bridge br0
nl80211: if_indices[16]: 25(-1)
RTM_DELLINK: ifi_index=25 ifname=wlan1 operstate=2 linkmode=0 ifi_family=0 ifi_flags=0x1002 ()
RTM_DELLINK: Interface 'wlan1' removed - mark if_removed=1
wlan1: Event INTERFACE_STATUS (5) received
Unknown event 5`

morrownr, 2022-09-01:
Hi @cristian-ciobanu 

> I'm running the hostapd configuration on ArchLinux running standard kernel 5.19.4 and I have the mt7612u modules loaded.

The modules look good. I am not familiar with Arch so let me throw something out. Does Arch use Network Manager?

If it does, Network Manager will continue to try to run the show so it will have to be deactivated. The guide you are using is very heavily tested on RasPiOS 32 and 64 bit. The RasPiOS does not use Network Manager... Does this make sense?

Nick

cristian-ciobanu, 2022-09-02:
@morrownr I'm not using NetworkManager (not installed) but systemd-networkd instead.