morrownr/USB-WiFi #42: Enable WPA3 Support for AWUS036ACHM mt7610u

vvkvas, 2022-01-15:
Hello 

I'm trying to connect AWUS036ACHM on W10 and my WiFi router running OpenWrt has following available options:
[image]

When I choose selected option it connect using WPA2.

When I choose this option:
[image]

This is what appears in my list:
[image]
When I try to connect:
[image]

This is in Kali 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux
[image]
When I try to connect:
[image]

But WPA3 works fine with my internal wifi
[image]

What I need to do to fix it?

morrownr, 2022-01-15:
Hi @vvkvas 

I keep a system with Kali on it but I'll need to reconfigure a couple of things to move the ACHM over to it. I'll plan on testing as soon as I can.

morrownr, 2022-01-15:
I was able to work this faster than expected.

I plugged the ACHM into a system running Kali rolling. A check of the kernel shows 5.15. Kali is the 64 bit version for Intel/AMD.

ACHM showed up immediately. I cleared out previous connections.

AP is a RasPi4B setup in accordance with my guide at the following location:

https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point.md

The adapter I am connecting to is a Panda PAU08 (rt3070 chipset).

The OS is RasPiOS (32 bit standard version) running hostapd. WPA3 is set to full SAE, not mixed.

I was able to connect and the connection showed as WPA3. I was unable to do anything.

I then moved the adapter to my main dev box that runs Linux Mint with kernel 5.13. This time the connection was successful AND I was able to surf the net as usual.

I then moved the adapter to a system with Ubuntu 21.10. The connection was successful AND I was able to surf the net as usual.

My main router is running OpenWRT 21.02 so I can configure it to test and I have additional test partitians with various distros that I can test. I need to push the pause button on my testing for now but will continue as able. Right now it looks like you might want to test on a distro besides Kali. WPA3 support is a very complex thing and anything we can do to narrow down where the problem is is a good thing.

morrownr, 2022-01-16:
Further testing:

Connecting this time to my main wifi router: OpenWRT 21.02 ( Encryption: WPA3 SAE (CCMP) ). This time I am using band 2, 5GHz. Previously I used band 1.

Client system: x86_64, Linux Mint 20.02, Alfa ACHM. iperf3 results:

```
$ iperf3 -c 192.168.1.1

[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  23.3 MBytes   195 Mbits/sec    0    386 KBytes       
[  5]   1.00-2.00   sec  22.7 MBytes   191 Mbits/sec    0    436 KBytes       
[  5]   2.00-3.00   sec  22.7 MBytes   191 Mbits/sec    0    436 KBytes       
[  5]   3.00-4.00   sec  21.9 MBytes   184 Mbits/sec    0    436 KBytes       
[  5]   4.00-5.00   sec  21.8 MBytes   183 Mbits/sec    0    436 KBytes       
[  5]   5.00-6.00   sec  22.4 MBytes   188 Mbits/sec    0    477 KBytes       
[  5]   6.00-7.00   sec  22.9 MBytes   192 Mbits/sec    0    496 KBytes       
[  5]   7.00-8.00   sec  21.9 MBytes   184 Mbits/sec    0    496 KBytes       
[  5]   8.00-9.00   sec  21.9 MBytes   184 Mbits/sec    0    496 KBytes       
[  5]   9.00-10.00  sec  22.0 MBytes   185 Mbits/sec    0    496 KBytes       
```

I was able to surf the net, Ops normal with WPA3.

So, what we know so far is that the ACHM and the driver have supported OpenWRT and RasPi4B wifi routers set to WPA3 (not mixed). For clients, Ubuntu and Linux Mint along with the ACHM are working ops normal. The only failure so far is with Kali Linux.

Certainly this is not enough testing to have a firm conclusion but I can continue as I have time. I can test additional distros and various additional configurations. however, the only common point of failure between the two of us right now is Kali. Have you considered posting the issue in the appropriate Kali forum?

Regards

vvkvas, 2022-01-18:
Hi @morrownr 

Does your ACHM make any mechanical noise when downloading or uploading? I'm noticing a noise like HDD platter rotating, is it normal?

> I then moved the adapter to a system with Ubuntu 21.10. The connection was successful AND I was able to surf the net as usual.

Right, working for me as well!

> I can test additional distros and various additional configurations. however, the only common point of failure between the two of us right now is Kali.

Not working for me on W10, W11, ParrotOS.

But working for me on Kali-Linux-2021.4a which was recently released. I believe gradually all other OS will update as well for WPA3. Kali is best example, it was not working in Kali-Linux-2021.3 or Kali-Linux-2021.4 but now working in Kali-Linux-2021.4a.
I hope it works in Windows as well, but I think we can't do anything about that right?

Best Regards

morrownr, 2022-01-18:
Hi @vvkvas 

> Does your ACHM make any mechanical noise when downloading or uploading? I'm noticing a noise like HDD platter rotating, is it normal?

Mine is totally silent. I just used iperf3 to pound it for 2 minutes and I could not hear anything. You might contact Alfa or the place you bought it to see if they know anything.
 
> > I then moved the adapter to a system with Ubuntu 21.10. The connection was successful AND I was able to surf the net as usual.
> 
> Right, working for me as well!

Based on my testing and research, my opinion is that all the USB adapters that use in-kernel drivers are capable of WPA3. The issues are with the OS support and that support is `in progress` in many cases. Somewhere around here I made a forecast that I expect most mainline LTS distros to have good support for WPA3 by mid-2023. This is a very complicated thing but the support will get lined out.
 
> Not working for me on W10, W11, ParrotOS.
> 
> But working for me on Kali-Linux-2021.4a which was recently released.

Okay. I will mark this issue as solved and leave it up for others.

> I believe gradually all other OS will update as well for WPA3. Kali is best example, it was not working in Kali-Linux-2021.3 or Kali-Linux-2021.4 but now working in Kali-Linux-2021.4a.

Yes. This is happening with many distros as we speak.

> I hope it works in Windows as well, but I think we can't do anything about that right?

Well, I really don't know. I have never regularly used Windows and don't have a copy of it on anything around here. ... oh, wait. Forgot about the new laptop Santa brought me. It has Windows 11. I don't use it as I installed an extra ssd and installed Linux but I'll bet I can see if the ACHM works with it. Somebody else will probably read this and tell us what is going with Windows and WPA3.

Regards

vvkvas, 2022-01-23:
Again, thank you very much @morrownr 

Best Regards