morrownr/USB-WiFi #43: Mediatek wifi6

bjlockie, 2022-01-27:
I can't find any announcements of any wifi 6 USB Mediatek devices. :-(
Even the Mediatek website doesn't say much except they are partnering with AMD for some laptops.

There do seem to be some wifi 6 USB chips listed at:
https://wikidevi.wi-cat.ru/MediaTek
but they are probably affected by the chip shortage. :-(

morrownr, 2022-01-28:
I'm not seeing anything either. What I know is that there is a perfect storm going on. The problems go beyond chip shortages. Projects are having to be prioritized more so than normal. I'm really not too excited about wifi 6 but I am looking forward to wifi 6e. I think it will come.

bjlockie, 2022-01-28:
Wifi 7 will probably be out before any more wifi6 usb devices.

Malvineous, 2022-01-29:
There are quite a few WiFi 6 and 6E PCI-e devices though (e.g. MT7921K on [Aliexpress](https://www.aliexpress.com/item/1005003523988230.html)).  Are they out of scope for this repo?  It could be useful to include them somewhere given that there are some things to be aware of, like the Intel WiFi 6E cards not allowing transmission on the 5/6 GHz band until the firmware has somehow deduced via a proprietary method what region you're in, making them useless for running an AP outside of 2.4 GHz.

bjlockie, 2022-01-29:
This repo is mostly USB devices. :-)

Do the wireless developers know about the Intel problem?
If it was Mediatek it would get fixed. :-)

Malvineous, 2022-01-29:
Fair enough!  The wireless developers *are* Intel for these cards so they [implemented this feature on purpose](https://bugzilla.kernel.org/show_bug.cgi?id=205695#c6).

The problem has been [reported to the wireless developers](https://bugzilla.kernel.org/show_bug.cgi?id=206469) but the response was that this is by design.  A terrible design in most people's opinions, as the Linux kernel already has spectrum management built in, so ignoring it and doing your own implementation that doesn't work properly is just a terrible idea.

Let's hope Mediatek don't "fix" it in a similar way!

morrownr, 2022-01-30:
> There are quite a few WiFi 6 and 6E PCI-e devices though (e.g. MT7921K on Aliexpress). Are they out of scope for this repo?

No, PCIe and what do you call those mini cards inside laptops, they are on topic here. USB WiFi has been problematic for years so that was the main issue I was trying to help with. Can we create a section for other types of WiFi? Sure. Tell me what we should do?

FYI: I just got a new little laptop for XMas and it has a mt7921 card in it. I installed Ubuntu 21.10 and it just works.

I monitor the linux-wireless list so as to keep up with wifi work in the kernel. My opinion is that the Mediatek drivers are built with a goal of meeting Linux Wireless Standards as close as possible. In the end, that means the over best support for all supported modes such as Managed, AP, monitor and more plus the features that are expected.

bjlockie, 2022-01-30:
What do we need for a USB adapter to use the mini-PCIe cards?
I don't know if it would be too big.

morrownr, 2022-01-30:
> What do we need for a USB adapter to use the mini-PCIe cards?

What we need is to find or do some research about how to use miniPCIe or M.2 cards with USB and what would be required. Can you see a case like the Alfa ACH that has a pop open lid where you can install a mini card and the antenna connectors are already there?

bjlockie, 2022-01-30:
That sounds perfect.
M.2 would be really low power I think.
And small.

bjlockie, 2022-01-31:
Should I post to the linux wireless mailing list suggesting that a USB device that accepts a M2 card would be nice,
In case any of the manufacturers are watching. :-)

Malvineous, 2022-01-31:
M.2 devices have pins for both PCIe and USB connections, so you can use a PCIe or a USB chipset on an M.2 card.  The keying on the connectors tell you which protocols it supports (so you can't plug a PCIe M.2 card into an M.2 slot that only supplies USB signals.)

There is currently no way I'm aware of to allow PCIe data over a USB bus, as they are two entirely different protocols.  You can get adapters (aimed at crypto miners) that extend the PCIe bus by using USB3 cables, but these just put the PCIe x1 signals across the USB cable to keep costs low, they aren't actually USB-compatible.

The M.2 PCIe connection runs at 31 Gbps while USB3.2 is only 10 Gbps so quite a lot slower too.  So the only way to do it would be to have an adapter chip and a custom driver.  I don't think there'd be much interest from manufacturers, because with USB-C connectors appearing everywhere they can support Thunderbolt devices, which is basically a PCIe connection over the USB-C socket.  USB4 will also have a 40 Gbps PCIe mode, so once that becomes widely available very cheap adapters will be able to connect PCIe devices to any USB4 sockets available without needing special converter chips and drivers.

> Can we create a section for other types of WiFi? Sure. Tell me what we should do?

How about creating an additional file in the repo for PCIe devices, similar to what's already there for USB?  That's probably all we need.  I'd be happy to come up with something to get started if that would help.

bjlockie, 2022-01-31:
> How about creating an additional file in the repo for PCIe devices, similar to what's already there for USB? That's probably all we need.

I really like why the Intel driver isn't good for an AP.

morrownr, 2022-01-31:
@Malvineous 

> How about creating an additional file in the repo for PCIe devices, similar to what's already there for USB? That's probably all we need. I'd be happy to come up with something to get started if that would help.

That would be great. See `PCIe_WiFi_ Devices.md`

bjlockie, 2022-03-05:
https://patchwork.kernel.org/project/linux-wireless/patch/6df14f076220c0cbc1d32939cd8be8cb33d7b498.1646235785.git.lorenzo@kernel.org/

I want a mt7921u device :-)

amisix, 2022-03-05:
Oh.. that would be quite nice, hardware flow offloading in a USB adapter?

morrownr, 2022-03-05:
Hi @Malvineous @bjlockie @amisix 

It is great news to see that mt7921u driver going into the kernel. I created a new issue to discuss it ... #64 

Can I get you guys to move the discussion as most people won't see it at the end of an old thread.

Thanks