morrownr/USB-WiFi #44: maximum number of clients in AP mode

bdantas, 2022-01-27:
Hi, morrownr. Thank you for compiling such helpful information.

The wireless router in my home is a repurposed ThinkPad T400 running Tiny Core Linux. The laptop creates a wireless access point using hostapd, dnsmasq, and wireless USB adapter with the Ralink RT5370 chipset. I specifically chose this chipset because I found various reports that it is capable of supporting a high number of connected wireless clients (>20). (I previously used wireless USB adapters containing the Atheros AR9271 chipset, which had a maximum of 7-8 clients when in AP mode. The limited number of supported clients was a constant source of frustration.) The only downside to the wireless USB adapter that I use now is that the RT5370 chipset is 802.11n (Wi-Fi generation 4).

My goal is to upgrade to a 802.11ac (generation 5) or 802.11ax (generation 6) wireless USB adapter with good Linux support, which supports at least 10 wireless clients when in AP mode.

I hope you don't mind if I ask a few questions: 
 
1. You had good things to say about AP mode of Alfa AWUS036ACM, Terow ROW02CD, and Alfa AWUS036ACHM. Would you happen to know the maximum number of clients that these devices support when in AP mode?
 
2. I have tested a device with the mt7601u chipset. That particular device did not support AP mode (both per output of `iw list` as well as per my actual tests). Are you sure that the devices you reviewed that have this chipset actually support AP mode? 
 
3. To make your incredibly helpful reviews even more helpful, in the future would you consider testing and reporting on the maximum number of clients in AP mode?

morrownr, 2022-01-28:
Hi @bdantas 

> The newer adapter should support at least 802.11ac (Wi-Fi generation 5), should be supported by the linux kernel or one of its modules, and should support at least 10 wireless clients when in AP mode.

> Do you know off-hand a wireless USB adapter that would fulfill the above requirements?

Will the adapter be used for 5 GHz or 2.4 GHz? What level of internet service do you have?

> You had good things to say about AP mode of Alfa AWUS036ACM, Terow ROW02CD, and Alfa AWUS036ACHM. Would you happen to know the maximum number of clients that these devices support when in AP mode?

You can set a limit on the number of clients that can connect in hostpad and that can be useful in some situations but to answer your question: I'm not aware of a maximum numbers of clients that those adapters can handle. It depends on several factors. I've used them with one client and many clients successfully. The issues that I look at are:

- What is the capability of the internet service? 100 Mb/s, 1 Gb/s?

- What is the bandwidth that is needed by the clients collectively? You'll have to measure the need of each client and then add the numbers to see what your requirement is.

- What  are the congestion levels on the channels you plan on using? Wi-Fi is about sharing, not competing so  any systems on the same channel as your system will force sharing of the bandwidth. Here is a link to a good article:

https://metis.fi/en/2017/10/txpower/

You can use iperf3 to test how much bandwidth is available to you on the various channels. It can vary by time of day so you have to test a various times to get a good picture.

> I have tested a device with the mt7601u chipset. That particular device did not support AP mode (both per output of iw list as well as per my actual tests). Are you sure that the devices you reviewed that have this chipset actually support AP mode?

If I have written that adapters based on the mt7601u chipset support AP mode, please tell me where so that I can go erase it. I am fully aware that the 7601u does not support AP mode and I thought I had made that clear in all of the places that I address this chipset.

> To make your incredibly helpful reviews even more helpful, in the future would you consider testing and reporting on the maximum number of clients in AP mode?

I will ponder how to handle this. As you can tell from my answers above, there is no specific overall answer. It depends on your location and the situation at your location.

To maximize your enjoyment of wifi requires knowledge of how wifi works and what is available to you at your location. An example:

I do change adapters around as needed for testing but right now my Alfa ACM is pulling AP mode duty on a RasPi4B. It is providing 5 GHz service. Most folks will think that it is set to 80 MHz channel width 80211ac. However, that is not the case. It is set to 80211n, 20 MHz channel width on channel 36. Ask me why.

bdantas, 2022-01-28:
Hi, morrownr. Thank you for the interesting article and for the iperf3 tip. Decreasing the transmit power of my AP is not something I would have considered. I will do some experiments. 
 
Regarding your reviews and mt7601u, I was wrong in what I said. The Mediatek devices you said you use as an AP have the mt76**12**u chipset. My bad. 
 
I did several speed tests. It seems my internet service download speed maxes out at around 30 Mbps. Our service is more than enough for our needs. The 802.11n connection between my devices and my homebrewed AP seem to max out at 65 Mbps. So it seems I don't need to upgrade anything in my physical layer, after all, because it is not a bottleneck.

> my Alfa ACM is pulling AP mode duty on a RasPi4B. It is providing 5 GHz service. Most folks will think that it is set to 80 MHz channel width 80211ac. However, that is not the case. It is set to 80211n, 20 MHz channel width on channel 36. Ask me why.
Okay, I'll bite (I'm sure I will learn something interesting here). Why?

morrownr, 2022-01-30:
> my Alfa ACM is pulling AP mode duty on a RasPi4B. It is providing 5 GHz service. Most folks will think that it is set to 80 MHz channel width 80211ac. However, that is not the case. It is set to 80211n, 20 MHz channel width on channel 36. Ask me why.

> Okay, I'll bite (I'm sure I will learn something interesting here). Why?

Because I can get better throughput with a 20 MHz channel width given what my requirement is.

In my area, the 2 main sections of non-dfs channels are 149-165 and 36-48. My system can see around 44 AP's on channel 44 and more than that on 157. This makes for very heavy congestion. The systems are setup by the local internet provider and almost all people just use what is set. Both channels are set for 80 MHz channels width, Channel 44 ends up using the channels from 38 to 54. That leaves 20 MHz of totally open space at channel 36. I use it as the bandwidth is all mine. I need to because I have 2 Roku's and Roku does not support DFS channels. My main router uses DFA channel 100 which is totally unused space so basically my needs are met with 2 channels working entirely in clean air. Finding and using the cleanest (non-congested) air is the single most important factor is having happy wifi users. Strong txpower? No. Wide channels widths? No. Clean air? Yes. Learn how to determine what is going on around you and find the cleanest air...and use it to your benefit.

Regards

morrownr, 2022-01-31:
As a follow on to the above, I am not that excited about WiFi 6. Why? Because it uses the same congested channels that have been available for a long time.

What does excite me is WiFi 6e as it brings a lot of new channels. Those channels with have little congestion but just as important, the new 6e channels will help reduce the congestion on the 5 GHz and 2.4 Ghz bands. When you have little congestion things generally work well if things are setup properly. In fact, most people would be very happy with N150 adapters in most situations if the congestion is low enough to allow the N150 adapter to do what it is capable of doing. Buying faster adapters does not necessarily improve the real world experience. My wife's desktop system uses a little USB WiFi adapter based on the mt7601u chipset. It is a N150 adapter capable of 80211n and capacity of 72 MB/s on a 20 MHz width channel. It so happens that channel 11 here is very under utilized so we use it. My wife streams videos, music, surfs the web and I never hear a complaint. Could I put an AC1200 adapter on her system? Sure. Would it increase her happiness? No.

Regards

bdantas, 2022-01-31:
Hi, morrownr. Thank you for sharing your knowledge so generously. I am using 2.4 GHz. I scanned the air all around my house and found that channel 1 is quite congested. I changed hostapd's configuration so that it uses the least congested among channels 1, 6, and 11 (channel 6 is the winner) and internet connection on our wireless devices is humming now! 
 
Main things I've learned here is that a newer WiFi generation and higher tx power do not necessarily translate into a better user experience. On the other hand, avoiding congestion really pays off. I also realized that a link speed higher than 30 Mbps will do precisely zippo to help us because our limiting factors are ISP connection speed and congestion, not WiFi 4's link speed.

Happy hacking!

morrownr, 2022-02-01:
> Main things I've learned here is that a newer WiFi generation and higher tx power do not necessarily translate into a better user experience. On the other hand, avoiding congestion really pays off.

Something I am guilty of is suggesting users install a "wifi analyzer" app on their phone to check congestion. The problem is that "wifi analyzer" apps are only telling you how many radios are up on any particular channel. There is generally a correlation between number of radios on a channel and how congested it is but it is not always that way. What determines congestion is how many packets are traveling through the air, not how many radios are tuned in. You can have 50 radios on a wifi channel and almost no congestion if none of the radios are pushing many packets (you can see this when a lot of IoT devices are in use). On the other hand, you can have 3 radios on a channel and that channel can be so congested that it is almost unusable. So the real way to figure out how much congestion (sharing) is to test how much you can tx and rx at various times on the available channels and iperf3 is good to do that. If anyone is reading this and does not know how to set iperf3 up, start a new issue and ask the question.