morrownr/USB-WiFi #45: COMFAST CF-WU782AC is not supporing on RPI4B

Slmba, 2022-01-29:
Hello. I appriciate this topic, but WU782AC doesn't work by plug-in to device.

Raspberry Pi 4B, clean raspberry OS.
[image]

[image]

There is only my internal wi-fi module:
[image]

morrownr, 2022-01-30:
Hi @Slmba 

The RasPiOS devs work in mysterious ways.

When they rebased on Debian 11 last summer, the usbmodeswitch executable was updated but the data file that supports it was not. I reported it. It was fixed but it is possible for it to still be the old file depending on your installation. Here are some ways to fix the issue:

- Burn a new sd with the newly released RasPiOS 2021-01-28
- If running the previous release run:

$ sudo apt update
$ sudo apt full-upgrade
$ sudo reboot

- If neither of those work or they are not what you want to do, follow these instructions...

https://github.com/morrownr/USB-WiFi/blob/main/How_to_Modeswitch.md

Slmba, 2022-01-31:
Hi, @morrownr 
Upgrade doesn't work.
I will try Desktop and Legacy Raspberry to install. Now I use the latest Lite edition without desktops.
You are right it doesn't contains this driver. Hope there is lite version somewhere that contains this drivers or I waste money for this adapter.

soyersoyer, 2022-01-31:
Create /lib/udev/rules.d/41-usb_modeswitch-comfast.rules, and insert
```
# COMFAST CF-WU782AC dual WiFi
ATTR{idVendor}=="0e8d", ATTR{idProduct}=="2870", RUN+="usb_modeswitch '/%k'"
```
Create /usr/share/usb_modeswitch/0e8d\:2870, and insert
```
# COMFAST CF-WU782AC dual WiFi
TargetVendor=0x0e8d
TargetProductList=0x7612
StandardEject=1
```

Restart, it should work.

Slmba, 2022-01-31:
Hi, @morrownr 
After usbmodswitch command everything works fine.
Big thanks to you!

closed 2022-01-31 by Slmba

morrownr, 2022-02-01:
Hi @Slmba 

> This could be a problem specific to RasPiOS Lite so I will put it on my to-do list to check this.

I was able to test today. I downloaded RasPiOS Lite 2022-01-28 and burned a new sd. I tested with my CF-WU785AC which internally is the same as your 782AC. No problems noted. The adapter came right up even before updating the system. The only thing I can figure is that your sd might have been burned back before August of last year and somehow the usbmodeswitch datafile did not update for some reason as the system was upgraded.

Cheers

Slmba, 2022-02-02:
Hi, @morrownr 
May be my Raspberry Imager Cached old release of Raspbian OS. But anyway, its too difficult to change OS at my production solution - I will need to reconfigure many things there. Full upgrade doesn't help.
But as I mentioned above - usbmodeswitch solved my problem. You mentioned in your main topic, the WU782AC is multistate adapter and usbmodeswitch is required.