morrownr/USB-WiFi #48: WPA2-Enterprise

gitclint, 2022-02-02:
Is it possible to put the radios into WPA2-Enterprise/802.1x/radius auth mode? Thanks!

morrownr, 2022-02-02:
> Is it possible to put the radios into WPA2-Enterprise/802.1x/radius auth mode? Specifically the mt7612u based usb adapters.

I'll jump in here first. As you probably know, I work with a lot of different WiFi adapters in a lot of different configurations. I have not tested this capability but I have not run across infformation that indicates is does not work. In fact, given the difficulty of getting a WiFi driver into the Linux kernel, my bet is that this works fine. In-kernel drivers must be Linux Wireless Standards compliant and they must stay compliant while in the kernel.

With that said, I don't like to give conclusive answers unless I have a lot of evidence and first hand use is best. If you can give me a very simple setup to test, I will provide confirmed feedback. Please keep it simple as my to-do list is long right now.

FYI: I have several adapters based on the mt7612u so if you have one or more that you are looking at, tell me and I try to test the exact adapter you are looking at.

Regards

gitclint, 2022-02-03:
I also now have several mt7612u adapters, so which one I don't really care so much about as I would imagine they would probably all support it if you can get one of them to work. I found this other GitHub link that I think should do the trick for WPA-EAP:
https://gist.github.com/noahwilliamsson/f2714e63e5959ffa9c92

morrownr, 2022-02-13:
Hi @gitclint 

Sorry about not getting back to take a look at this. I have been on and off for the last week and my to-do list is very long. Hopefully I can take a look at this soon.