morrownr/USB-WiFi #50: how to set iperf3 up?

usama7628674, 2022-02-02:
What values do we need to monitor to check if there is too much congestion on particular channel during running of iperf3?

morrownr, 2022-02-08:
Hi @usama7628674 

I've been sick lately and still am. There is not a short answer to this question. I'll try to answer when able.

Regards

usama7628674, 2022-02-09:
@morrownr Ok, no worries. Take your time.

morrownr, 2022-03-02:
Hi @usama7628674 

I have some time this morning to come back and address this issue:

The information I am providing is how I do it and is subject to the better ideas of others.

Most articles and tutorials seem to advocate that wifi congestion can be measured by determining how many APs are on a specific channel. Basically, if there are a lot APs on channel x, it is likely highly congested. Well, this is not always the case. The number of APs can give us an idea of the amount of congestion but a channel with 15 APs that are lightly used can have less congestions than a channel with 5 heavily used APs.

What I do to check actual congestion is I first establish a baseline for the equipment. For 5 GHz, I set the AP to an unused DFS channel and run iperf3 for 30 seconds a few times to get a good number for what the equipment is capable of. I then move over to channels I want to check and run the same tests with iperf3. When checking channels you really need to run multiple tests at different times of the day and also tests on a weekday as well as weekend to make up for different levels of use during the day and day of the week. Given this data you can determine which is the least congested channel.

When I get baseline data for an AC1200 adapter in 5 GHz mode with a 80 MHz channel width, I generally see around 400 Mbps thoughput. When testing the non-DFS channels here I can get around 190 Mbps average on one channel and 220 Mbps on the other. This is a heavily congested area. At high usage times which happens here in the evening, throughput can get really low. If fact, what I do with my router, which supports 2 radios for 5 GHz, is set one for 80 GHz channel width on a totally clear DFS channel and most devices can use that. For the two devices that I have that don't support DFS, I use a 20 MHz width space that is available because of how most routers are setup here and it gives me clear air. So, I am running in totally clear air on 5 GHz here which makes for a very very reliable setup. However, if I did need to run on a congested channel, I know which one has the least congestion.

Hope that makes sense.

Regards