morrownr/USB-WiFi #51: Edimax AC600 USB Not Supported

alexdaversa, 2022-02-03:
The Edimax AC600 USB adapter (EW-7811UAC) does not appear to be supported by this driver though it is working with the aircrack-ng version. 
USB ID 7392:a812
Edimax has a driver released on their website https://www.edimax.com/edimax/download/download/data/edimax/us/download/wireless_adapters_ac600_dual-band/ew-7811uac

closed 2022-02-03 by alexdaversa

eighthave, 2023-08-01:
I also tried building 8812au-5.6.4.2_35491.20191025 and it didn't work with my 7392:a812 Edimax Technology Co., Ltd Edimax AC600 USB.  The aircrack-ng version also worked for me. 

@alexdaversa why did you close this?  Did you find a solution?

morrownr, 2023-08-01:
The correct driver for VID/PID `7392:a812` is:

https://github.com/morrownr/8821au-20210708

eighthave, 2023-08-02:
I believe I tried that one without luck.  8812au/5.6.4.2_35491.20191025 seems to 
be working for me.