morrownr/USB-WiFi #52: Alfa AWUS036ACM (MediaTek MT7612U Chipset) AP Crashing in OpenWrt

amisix, 2022-02-06:
Adapter: Alfa AWUS036ACM (mt7612u) 
System: Raspberry Pi 4B Rev. 1.5
OS: OpenWrt 21.02.1 (Linux RPi4 5.4.154 aarch64 GNU/Linux)
Driver(s): kmod-mt76x2u
kmod-mac80211.
kmod-mt76-core.
kmod-mt76-usb.
kmod-mt76x02-common.
kmod-mt76x2-common.
kmod-mt76x02-usb.

**Issue:** When connected via USB 3.0 adapter crashes after having any appreciable load applied (within a few seconds to a couple minutes). Issue persists after restart unless power is completely removed from the adapter/Raspberry Pi. The issue is repeatable. 

**Solution:** [Disable USB Scatter-Gather at startup.](https://github.com/openwrt/mt76/issues/405#issuecomment-1029636332)

Syslog message at time of crash:
```
kern.err kernel: [TIME] mt76x2u [USB BUS/PORT]: error: mt76x02u_mcu_wait_resp failed with -110
```

Kernel log after restart without full poweroff process completed:
```
usb 2-2.4: reset SuperSpeed Gen 1 USB device number 5 using xhci_hcd
mt76x2u 2-2.4:1.0: ASIC revision: 76120044
mt76x2u 2-2.4:1.0: ROM patch build: 20141115060606a
mt76x2u 2-2.4:1.0: firmware upload failed: -110
mt76x2u: probe of 2-2.4:1.0 failed with error -5
usbcore: registered new interface driver mt76x2u
```

**Solution: Disable USB Scatter-Gather at startup.**

Add "mt76-usb disable_usb_sg=1" to /etc/modprobe.d/mymoduleparam.conf then reboot

```
nano /etc/modprobe.d/mymoduleparam.conf
mt76-usb disable_usb_sg=1
```
Ctrl + O to save, Ctrl + X to quit
Reboot

Verify it worked:
```
cat /sys/module/mt76_usb/parameters/disable_usb_sg
```
Should be "Y"

morrownr, 2022-02-06:
Thank you very much for posting the solution as I am sure there will be others that run into this issue.

Regards

amisix, 2022-02-06:
You're welcome. You've put lots of effort into this site.
Thank you for your [write-up here](https://github.com/morrownr/7612u) regarding USB scatter-gather with additional [details here](https://github.com/morrownr/7612u/blob/main/mt76_usb.conf).

morrownr, 2022-02-09:
Hey @amisix 

Did you make a couple of typos in your post above?

-----

How to Disable USB Scatter-Gather (mt7612u driver)

```
sudo nano /etc/modprobe.d/mt76_usb.conf
```

add the following line:

```
options mt76_usb disable_usb_sg=1
```

`ctrl +  o` to save and `ctrl + x` to exit

```
sudo reboot
```

To check: 

```
grep [[:alnum:]] /sys/module/mt76_usb/parameters/*
```

amisix, 2022-02-09:
I thought so too, but for reference: https://github.com/openwrt/mt76/issues/405#issuecomment-1030740067

There is no sudo/no need for sudo in OpenWrt (depending if config is different than default I guess?)
There is no /etc/modprobe.d directory in OpenWrt. Apparently it's equivalent is /etc/modules.d ? 
Since there is no /etc/modprobe.d/ there is no m76_usb.conf. But I was kind of using @bjlockie 's instructions because I hadn't found yours yet. Where bjlockie said to use /etc/modprobe.d/mymoduleparam.conf. But there is no /etc/modprobe.d/ directory and no mymoduleparam.conf. 

So I modified it to /etc/modules.d/mymoduleparam.conf and it works fine.  I tried to follow the directions as written by others but that's what I ran into while configuring it. Am able to run the check command after restart and it's "Y". 

If you'd like me to modify some other way (knowing these instructions will be used by others) I'll edit as necessary.

Echo522-prog, 2022-03-31:
I need help, so I have the AWUS036ACM well I try and use the rtl8812au chipset but It doesn't work and even when I use the lsusb command the adapter Doesn't show up.

morrownr, 2022-03-31:
> I need help, so I have the AWUS036ACM well I try and use the rtl8812au chipset but It doesn't work and even when I use the lsusb command the adapter Doesn't show up.

Hi @Echo522-prog 

Can I get you to open a new issue and include some information such as a brief description of your computer hardware and operating system as well as what you are trying to do? Thanks.

> I try and use the rtl8812au chipset

I don't understand this comment. The Alfa ACM uses a mt7612u chipset.

Regards