morrownr/USB-WiFi #54: HOT! `usbmodeswitch` problem is the newly released 64 bit version of the Raspberry Pi OS (21-01-28)

morrownr, 2022-02-09:
HOT! The 64 bit version of the Raspberry Pi OS was finally recently (21-01-28) released. That is the good news. The bad news is that the data file for `usbmodeswitch` is the old version. What this means is that many of the multi-state adapters shown later in this document will not automatically work. The driver is there but cannot activate due to the adapter being stuck in CD-ROM or flash mode. To correct this:

https://github.com/morrownr/USB-WiFi/blob/main/How_to_Modeswitch.md

FYI: The 32 bit version works fine. Do not ask because I do not know.

morrownr, 2022-02-13:
FYI: I have found several bugs in the new 64 bit RasPiOS. Many have to do with networking.

soyersoyer, 2022-03-13:
There are good news too. They fixed the scatter gather usb bug in [5.10.103](https://github.com/raspberrypi/linux/commit/a538fd26f82b101cb6fb963042f3242768e628d4)
Now the mt7612u is working well without the disable_usb_sg=1 parameter.

morrownr, 2022-03-13:
Oh my! That is good news and it has been a long time coming. I am or was currently in the process of trying to track down this problem and have been in contact with the Mediatek devs about it. All evidence pointed to it being RasPi4b specific and it turns out to be an issue with a part used in the RasPi4b. It may not be RasPi4b specific but the problem part may be rare. I have also seen problems with the 8812bu and 8814au with the RasPi4b and I'll bet this is the problem. The amount of aggravation that this has caused has been big.

Now to see if this bug needs to be pushed to the Linux kernel (the one Linus maintains).

Thanks for the report and the link.