morrownr/USB-WiFi #55: Advice on a good range and speed USB WiFi dongle

Darkyere, 2022-02-09:
Hi all and ty for reading.

I found this page and is looking for a decent USB WiFi dongle.
I was hoping to get some good advices on my possibilities.

My setup is a PC in a basement.
It will have a floor and a wall between the PC and Router.
+ Around 10 meters in distance.

I need it for a ZFS send/receive from a RPI4 located in another city.
Because of the demanding ZFS filesystem i will transfer with between 40-55MB/s.

My hope is for a dongle with a good long range and a speed of up to 55MB/s with in kernel drivers. So I don't have to mess with failed Driver Updates.

I hope anyone out there can post a few recommendations on their experience.
And an explanation of why it's good. Etc. Signal quality and speed.

I hope there is someone that has good experience with similar cases.

Thank you and best regards,
Darkyere

Darkyere, 2022-02-12:
Thank you for answering both of u.
Sry for the slow return was not near the house with my BackUp server (My old folks house)

I checked all the questions and will post em now.

> Are we talking about concrete and rebar or wood and sheet rock?

The floor is wood and insulated with plasterboard
The wall is bricks, so i suppose kinda like concrete.

> Are you planning to use 2.4 GHz or 5 GHz?

yes definetly 5 GHz. Or the speed on 2.4Ghz will only be 150Mbps.
And i need to get up to the 55MB's range or something similar to 450Mbps.
Or something close to it.

> What channels are available to you? We don't know what country you are in.

I live in denmark.

On 5 GHz there is channel 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112
Currently using channel 56

> Do you live in an apartment build or a single family home?

The backup server is in a terraced house. Or a better description.
There is two house's direclty located on each site of the house.
So a max of 2 Routers can interfere with the router in the BackUp server.
Teoretically atleast. Depends i suppose on the AP's in the area.

> Have you checked to see how many APs are around you? I'm try to get a general impression of the level of congestion.

I have added two screenshots of a tablet scanning the 5GHz network in my area.

[image]
[image]

Unfortunately with the channels available to me i dont seem to be cable of changing it to something where i wont meet another AP.
Im the Yellow signal with C32A in its name.

amisix, 2022-02-12:
>...long range and a speed of up to 55MB/s

55MB/s is a requirement that I don't think you'll be able to meet without using Wifi 6 (AX). Using speedtest.net I'm only able to achieve ~100-120Mbps with Wifi 5 (AC), less than a quarter of the speed you're looking to achieve. I am in a heavily congested area though and you are not. 

55MB/s is ~1/2 the throughput of a consumer grade hard drive, that's a lot to push wirelessly imo. iperf numbers look very different when compared to my Speedtest.net runs, often showing 200 - 300 Mbps when I'm seeing nothing like that IRL. Doing a file copy from one machine to the other, I'm also only getting 100-120Mbps with Wifi 5.   

I'm not familiar with the AX USB offerings though so I'll defer to the experts. You could also try a [power line range extender](https://www.amazon.com/Powerline-Ethernet-Adapter-Extender-TP-Link/dp/B084CZMYNM/ref=pd_lpo_1?pd_rd_i=B084CZMYNM&th=1) that tops out at 1Gbps.

IMO it's AX (and even then..), cat-5/6, or a power line range extender.

morrownr, 2022-02-13:
Hi @Darkyere 

I think @amisix is right to express some skepticism as to whether this is possible given your requirements.

55 MBps is, well, a heavy load for a home networking setup. You really need 1 Gbps internet service and a router that is capable of pushing that load around. I'd be interested in the details of your wifi router so as to offer an opinion if it really can handle it. That is also a load that can be a challenge for WiFi.

If you have perfect WiFi conditions like no congestion and a straight line, no walls, link, with a distance of 5 or 6 meters, you might be able to get close with an ALFA ACM. That would probably give you around 400 Mbps but you are looking for up to 440 Mbps if my math is right. However, we don't know what your congestion levels are. Testing with iperf3 could give an idea of how much bandwidth is available. Remember that WiFi technology is based on sharing so any other AP on the same channel is going to cause sharing and reduce the bandwidth available to you. The biggest problem, in my opinion, is  the walls and the reduction in signal strength they are causing. In looking at the WiFi Analyzer report that you provided, I am skeptical.

 WiFi Analyzer is showing a signal strength of -69. I am going to assume this reading was taking from the area near your desktop where you want the WiFi adapter. I break down signal strength like this: <= -50 - you should be able to get full speed. From -51 to -65 - you should get a solid connect but probably less than full speed. > -66 - it is time figure out how to improve this or you may not get the results you are looking for. If you used a phone, that could help account for the low reading and things might be better with a good adapter... maybe, I would have to test.

If I had requirements like yours, here are some options I would look at:

1. Find a way to run CAT 6 ethernet cable from the router to the desktop. This is likely the most reliable solution.
2. Get a good PCIe card based on a mt7921k chipset to install in the desktop and upgrade your WiFi Router to WiFi 6e capability.  This would give you plenty of clean, uncongested, air and faster speeds. 
3. Give the ALFA ACM a try. It is always a handy adapter to have around.

Regards

Darkyere, 2022-02-13:
I did somehow realize that i was shooting for the sky when i posted this thread.

----

For the speed my house got 1000/1000Mbps and my parents got 1000/500Mbps. 
So on that part i got the upload and they got the download.

I got a netgear router X6 R8000. Connected to the origional fileserver with a 1 Gbps cable.

For my parents it's another story. The internet provider they got actually supply the Router.
Which is a router locked to their Internet service and usually not one a person can bye themself.

Its got 1Gbps ethernet and my Tablet can get 844Mbps of connection speed with WiFi 5.
This is not though the actual speed i get out of it obviously from a tablet.

----

I could'ten figure out which USB would provide me the best bandwidth but it seems like i should try out the ALFA ACM since it has been recommended twice.

Just to give a good and funny idea of my old folks faces, when i would have to try and explain to them we would need to route a cable through the living room, the hallway, kitchen, stars to the basement and then into a really small room where the backup server is. 

Believe they would either get a heart attack or look at me in utter disbelief ;)

----

For the 

> Get a good PCIe card based on a mt7921k

Can u recommend any? I ave allways had problems trying to find a USB or PCI wifi cart by its chip.
Maybe because i live in denmark none of the online shops write this info on their page.

And in the physical store they will just "convince" me that a TP-LINK PCI-E card will work perfectly.
Without having any clue of what im saying when i talk about linux and driver support to them...

I allready fell for that once before. 
Got an 844Mbps TP-LINK PCI-E card that gives the promised speed on windows.
But on linux i get 72Mbps on WiFi 5 :D

Searching on chipsets on google i havent figured out how to find the right card since it allways recommend PCI-E cards that end up having another chipset when i look up the actual cards chipset.

----

But for now im gonna have a look at the ALFA ACM

Thank you both of u for the help so fair.

Its been really educational and probably prevented me from bying a WiFi USB that would just have dissapointed me in the end.

amisix, 2022-02-15:
I know someone who just completed an install of a dozen Asus PCE-AX3000 PCI-E cards (Intel ax200 chipset) but unfortunately they're all Windows machines so the only thing I'll be able to attest to is the speed. It's a multi-story home with no neighboring wifi using a Calix BLAST router/mesh network with ~2 interior walls and sometimes a floor separating each machine from an AP. Maximum speed achieved without any optimization was 560Mbps down and 860Mbps out although that was not seen on every machine - I'd say the average was about 300-400Mbps down and 450-500Mbps out. There's some room for improvement there but coming from ~40Mbps wireless N I'd say they're quite happy. So, AX can get there (depending on how the drivers behave?) but I wouldn't rely on it for a 450Mbps file system if I also had to shoot the signal through a concrete wall. Sorry if this a repeat of what's been discussed, I just thought it was interesting getting a real world comparison.

Darkyere, 2022-02-15:
No that's completely fine with some more input.

Gonna look at the PCI card u mentioned.
Have heard before that Intel chipset is better supporter on Linux than some might be.

It gives me some good options with in kernel wifi (preferable alfa ACM)  and this PCI i am about to Google.

morrownr, 2022-02-19:
Hi @Darkyere 

I've been sick lately so have been slowed down.
 
> > Get a good PCIe card based on a mt7921k
> 
> Can u recommend any? I ave allways had problems trying to find a USB or PCI wifi cart by its chip. Maybe because i live in denmark none of the online shops write this info on their page.

Below is a PCIe card that looks interesting to me. 

https://www.aliexpress.com/item/1005001926230503.html

That card has the mt7921k chipset and is a full WiFi 6e Tri-Band card. I have a little laptop that has the mt7921 chip (WiFi 6) and it is pretty impressive. The in-kernel driver is very good. The driver has only been in the kernel since 5.12 so keep that in mind. My laptop has kernel 5.13.

Your problem regarding the ability to find adapters and cards by chipset is normal. I am pretty good at figuring out what chipset is in the various products but then I have been using Linux since 1994 so I have had practice. Feel free to ask questions about how I do it but start a new thread.

Regards

morrownr, 2022-02-19:
Here is another link to the AX181 Plus:

https://www.ebay.com/itm/304264114816

Disclaimer: I do not have this card and there is not adequate information available to know whether this is a good purchase.

Darkyere, 2022-02-20:
Im sorry u have been cought ill. Hope ur getting better.

I am very happy u send that PCIe card info from Aliexpress.

I was a bit cought off that most of the WiFI USB's like the ALFA ACM would send me down 300 DKK and another 200 DKK in logisitcs.
This from aliexpress is gonna take me down from between 200 and 300 DKK and that includes Logistics.

Its definetly a better attempt to make for my wallet 😉 

But ofcourse i understand that the PCIe card has a bit of a hit and miss chance.
I can allways return to ALFA ACM another time if this should fail.

morrownr, 2022-02-20:
> Hope ur getting better.

Starting to feel better. Thanks.

> But of course i understand that the PCIe card has a bit of a hit and miss chance.

I would not say that the AX181 Plus is hit and miss as far as whether it will work or not. I'm not sure of the quality. I do have two Comfast usb adapters and they are decent quality. My bar for quality is set high as I primarily use ALFA equipment which is like driving a Rolls Royce.

Since the AX181 plus is based on the mt7921k chipset, you can go to the following site:

 https://wireless.wiki.kernel.org/en/users/drivers/mediatek

See the names and emails at the bottom of the page. Those are the primary Mediatek wifi kernel devs. The page currently shows support for mt7921 since kernel 5.12. I have a laptop that uses a mt7921 chipset based M.2 card and it works very well. My kernel is v5.13.

What you could do is email one or more of them and ask if the mt7921k is supported in-kernel for PCIe yet.

morrownr, 2022-02-20:
To clarify:

mt7921 = WiFi 6 Dual Band
mt7921k = WiFi 6e Tri-Band

Darkyere, 2022-02-20:
this is annying.

My hotmail account gets bounced by the mail system in linux-wireless
And so does my  gmail

Hotmail is not eligeble for mail transfer.
And it keeps telling me that the gmail has html code in its body even though im writing in plain text.

Will have to figure this out later

morrownr, 2022-02-20:
I would not send it to linux-wireless because it is for all wireless to include Intel, Atheros, Broadcom, Mediatek and a few other cats and dogs. This is a Mediatek specific question.

morrownr, 2022-02-21:
After a little investigative IT work:

The mt7021 (aka RZ608) driver that is called mt7921e in the kernel does indeed also support the mt7921k chipset. However. there is more to the story than it currently works on current mainstream distros. The mt7921k chipset is new and adapters are still fairly rare.

It appears the mt7921k chipset is using different device IDs than the mt7921 so there was an update submitted:

https://patchwork.kernel.org/project/linux-wireless/patch/84ab45bf42f57fd0301c156ffc11d0fe330ff1f8.1636857817.git.deren.wu@mediatek.com/

Given the time frame of that patch, it probably went into kernel 5.17 which has not been released yet.

However, there is a work around as long long as you are using at least kernel 5.12. Ubuntu 21.10 uses kernel 5.13 for example.

```
sudo nano /etc/modprobe.d/rz608.conf 
```

add:

```
alias pci:v000014C3d00000608sv*sd*bc*sc*i* mt7921e
```
save

```
sudo nano /etc/udev/rules.d/99-rz608.rules
```

add:

```
SUBSYSTEM=="drivers", DEVPATH=="/bus/pci/drivers/mt7921e", ATTR{new_id}="14c3 0608"
```
save and reboot

The above might need to be slightly modified depending on your card.

If you get a PCIe card with the mt7921 chipset and you are running at least kernel 5.12, then the above is not necessary but my opinion is that the tri-band capability of the mt7921k chipset is where the real advantage comes in.

The branding of RZ608 is from AMD. AMD and Mediatek signed an agreement last year and AMD gets to use their branding on many of Mediatek's chipsets now. The RZ608 and the mt7921k are the same thing.

Darkyere, 2022-02-21:
This is great news.
Im definetly gonna buy this next month.

And now the potential diver issue is allso prepared for me so i wont have to fight through this myself, without the prober knowledge.

Am allso gonna use ubuntu 20.04 and with the HWE kernel i should be all set.
And if not i can just use the 21.10 release where userspace for ZFS and ZFS kmod actually matches.
In both cases i should be running with the 5.13 kernel with no problem.

I would just like u to know i really appreciate the time u spend on this.
I have failed myself to figure out how to find these things.
Have spend about a year trying to figure out where my money should go until i came across ur github repo.

This is really appreciated, i hope others will find this thread as well and maybe find it usefull.

morrownr, 2022-02-21:
Also, since you do appear to have much congestion:

https://www.aliexpress.com/item/1005003444714172.html

That is the CF-180Plus which uses the mt7921 (WiFi 6 dual band. I use the 5.13 kernel as is shipped with Ubuntu 21.10 on my laptop that has a M.2 card with the mt7921. It was plug and play, No need to use any workarounds. 

Note: I am not advocating Comfast products but they do seem to be the easiest to find. 

> I would just like u to know i really appreciate the time u spend on this.

You are welcome. You owe us a report.

> I have failed myself to figure out how to find these things.

Hang around and ask questions. I have an unfair advantage over most people. I've been using Linux since 1994.

Regards

Darkyere, 2022-02-28:
I have now ordered the AX181 Plus Antenna Card from Aliexpress.

Expected to arrive at 30 of March.

Now we/i must just have patience :)

morrownr, 2022-02-28:
Hi @Darkyere 

Cool. We need reports on that card. I might even get myself one at some point.

Now, it looks like we have a month to get you ready for the card to arrive. It appears that the device IDs for the new cards based on the mt7921k only went in during the kernel 5,17 cycle. The new 6 GHz capable stuff is pretty bleeding edge so we need to figure out how get you running at least a 5.17 kernel in a way that you like. What is your preferred distro?

Oh yeah, the 5.17 kernel has not been released yet.

Darkyere, 2022-03-01:
Preferred distro is Ubuntu LTS.
But i need a ZFS userspace and ZFS kernel module that matches.
So will most likely have to go with Ubuntu 21.10 since they match on newest release.

I must also add that my old folks router and mine doesn't have 6GHz WiFi.
But luckily enough according to AliExpress the AX181 Plus supports 6/5/2.4GHz WiFi.

I just picked the one with WiFi 6 capabilities since in the long run it's the future.

morrownr, 2022-03-01:
Hi @Darkyere 

I understand what you are saying. I cannot argue with your decision because the card you picked should get better with time as the rest of your network catches up to 6 GHz wifi. In the meantime you can run 5 GHz. Your driver support will get better as time passes also. Don't confuse `WiFi 6` and `6 GHz WiFi`. WiFi 6 supports the 5 GHz and 2.4 GHz bands. Your card is WiFi 6e meaning that it also supports the 6 GHz band. Cool.

While you wait, it is a good time to get prepared.  Since you are right on the bleeding edge and it appears you probably need kernel 5.17 which has not even been released yet, you need to look at your options which is why I ask what your preferred distro is. Initially your options are somewhat limited in that you will need to install the 5.17 when it is relased in the next 2-3 weeks:

https://itsfoss.com/upgrade-linux-kernel-ubuntu/

Note: I am not familiar with ZFS so if you have to have the exact same module on the RasPi, we need to figure out how to install 5.17 on the RasPi also. I'd have to search for a guide specific for RasPiOS but you can give it a go if you wish.

Now, once both systems are up and running with kernel 5.17, you are going to have to lock the kernel. The reason is that everytime new kernels flow into the systems, it will likely start using the new kernel which will be older and not support this card yet. We can find instructions for that also.

I know initially that is a little bit of a pain but as time passes and Ubuntu releases a version with at least kernel 5.17, everything should just work.

Regards

Darkyere, 2022-03-01:
It's good to know the difference on wifi 6 and 6e.
I really thought they where the same to be honest.

For ZFS one actually don't need the same version of kernel module on 2 different devices to use ZFS send/receive.

It's only on the individual device that the userspace and kernel module have to match.
By that i mean the command line part of ZFS and ZFS kernel module version on the same device must match.

I am currently transferring from a matching kernel 5.13 userspace and kernel module to a 5.4 matching userspace and kernel module without a hitch.

So we won't have to do anything on the pi.

The big question is (if we get the wifi working) how i get a matching version of ZFS kernel module for 5.17 and find the and userspace tools that match it.

A short search showed it's version 2.1.2 of ZFS in the kernel.
Maybe i will have to get the userspace tools from OpenZFS GitHub page.
Only time will tell.

It's kinda exiting. Never knew Linux WiFi was so difficult to figure out.
And know i might get a better wifi speed than on my windows laptop.
Funny how things turn out sometimes.

morrownr, 2022-03-02:
> It's good to know the difference on wifi 6 and 6e. I really thought they where the same to be honest.

You are not alone. The advertising for adapters and cards helps make this more confusing. You really have to know the chipsets and the little signs that point to whether an adapter or card is WiFi 6 or WiFi 6e. My opinion, which means nothing, is that WiFi 6 should probably be skipped by most people. The reason is that it provides no new channels which means that it is still using the crowded channels we are dealing with today. Anything else the makers say is just marketing BS. WiFi 6e adds a new band and many new channels so there will be less congestion. The adapter you ordered is 6e and it will be ready someday when you get a new router that supports 6e (don't buy a router that is WiFi 6 because it will not support the 6 GHz band.)

I stubbled onto something this morning that you might want to try. I'm thinking of trying it. The link below is an article where a group is making stable kernels available for Ubuntu as soon as they are released. Right now the stable kernel is 5.16. You are going to need 5.17 to support this new card as best I can tell so maybe if you try this everything will be ready when the card arrives.

https://www.theregister.com/2022/02/25/friday_foss_fest_liquorix_kernel/

I'm looking forward to your reports on the new card. Maybe you can start a new issue or discussion to let us know how it goes.

Regards