morrownr/USB-WiFi #57: Dual-band nano adapters

hackerb9, 2022-02-09:
First, I have to thank you for this amazing resource! In-kernel drivers are extremely important to me and thanks to you I now know how to buy the right chipset ("MediaTek Inside"). It's a great bonus that your list also marks the silly Windows-drivers-included adapters so I can avoid them.  

I have one request: would you please also recommend dual-band _nano_ adapters? There are bazillions for sale, but are there any that have in-kernel drivers?

The only nano adapter listed here (the Panda) is 2.4GHz, which is unusably saturated in my area. Also, the Panda is quite old: the [Amazon listing](https://www.amazon.com/Panda-Ultra-150Mbps-Wireless-Adapter/dp/B00762YNMG) for it says, "First available February 6, 2012".

If you could add a section for dual-band nano adapters, I would greatly appreciate it.

morrownr, 2022-02-10:
Check this out:

https://www.amazon.com/dp/B00HM0K61Y

hackerb9, 2022-02-12:
> Check this out:
> 
> https://www.amazon.com/dp/B00HM0K61Y

Interesting. Does the ASUS USB-AC51 have in-kernel drivers? Nobody in the reviews say it works with Linux and many of them warn strongly that it does not. One person said the ASUS claim of "works with Linux" is a tar file from 2013 that only compiles with Linux 2.x.

morrownr, 2022-02-12:
Hi @hackerb9 

> Interesting. Does the ASUS USB-AC51 have in-kernel drivers?

Yes

See: http://en.techinfodepot.shoutwiki.com/wiki/ASUS_USB-AC51

There is also this adapter though it may be bigger than you want:

https://www.ebay.com/itm/293268959565

There is a section in the README.md here for mt7610u chipsets and it has several links. Check it out.

FYI: The little Linksys adapter is the only one of the ones I have mentioned here that I own. I can tell you from first hand experience that it is very small and it works well.

> Nobody in the reviews say it works with Linux and many of them warn strongly that it does not. One person said the ASUS claim of "works with Linux" is a tar file from 2013 that only compiles with Linux 2.x.

What you are running into is old information, One of the reasons I started this repo is to try to make accurate, current information available to Linux users. The information you read was true when it was written but has not been true for several years now. This problem of old information that is no longer true and the big problem with FALSE information makes it hard on Linux users.

The background on this story: Ralink was bought by Mediatek around the 2013-2014 time frame. 80211ac was new at the time and initially Ralink/Mediatek provided out-of-kernel drivers. It can take a while for a company to integrate the purchase of a new company but by the 2017-2018 time frame we saw that Mediatek was headed toward in-kernel drivers and by 2018 the mt7610u and mt7612u drivers were in the kernel. It can take time to get a driver in the Linux kernel as there is a lot of quality control.

This is another example of how out-of-kernel drivers do not work well with Linux. Linux is basically a rolling OS. It is developed constantly with new kernels being released every 2-3 months and LTS versions released once per year or so. This system has developed over the years based on what works with the resources that have been available. Out-of-kernel drivers such as the ones from Realtek do not get any ongoing maintenance while they sit on your hd or sdd. If you decide to upgrade from Ubuntu 20.04 to 22.04, the Realtek driver will blow up on you. On the other hand, the Mediatek drivers will not blow up on you because they are in the kernel and get regular maintenance and new versions of the Mediatek drivers flow right into your system with new kernals so life is good.

Also, in digging deep into the quality and methods used by the makers of the WiFi chipsets, there are differences we all need to consider. While Intel supports its WiFi with good in-kernel drivers, their AP mode support is basically not there and many Linux users are looking for the full spectrum of capabilities. Mediatek does the best job of doing open source the right way. For those considering PCIe or M.2 cards, you really want to go with cards based on Mediatek chipsets. @Malvineous has recently started a new file here. It provides information about PCIe and M.2 WiFi chipsets. The file is called `PCIe_WiFi_Devices.md`. The file looks really good and hopefully we can add some links to specific cards soon. For Linux users wanting to add WiFi 6e capability to a desktop or laptop system, my recommendation is to seek out cards with the mt7921k chipset. WiFi 6e will come to USB at some point.

Regards

morrownr, 2022-02-14:
[Another very small adapter](https://www.walmart.com/ip/USB-5Ghz-Wireless-AC600M-Dual-Band-802-11ac-Wifi-Adapter-Wi-fi-Network/112794071?athcpid=112794071&athpgid=AthenaItempage&athcgid=null&athznid=PWSMT&athieid=v0&athstid=CS020&athguid=vL8wcMMXWiDPaoReDdfqS_GCjj-bnQssn7Up&athancid=null&athena=true)

jclark, 2022-06-03:
I bought one that looked exactly like the one https://github.com/morrownr/USB-WiFi/issues/57#issuecomment-1038630257 and was advertised at MT7610UN, but turned out to have an RTL chipset

https://s.lazada.co.th/s.h6hgp

morrownr, 2022-06-03:
Hi James,

> I bought one that looked exactly like the one #57 (comment) and was advertised at MT7610UN, but turned out to have an RTL chipset.

That happens. The reason I started this site is to help Linux users find information that helps them get right on what they need. I've had folks here say that I could just make a list of chipsets but that does not go far enough. You actually have to have good links to known good products because there is so much false information at resellers.

The closest adapter for what you are looking for is still the Linksys AE6000. There are many nano adapters with mt7601u chipsets but they are single band.

With that said, I need to point something out to you. About a week ago, the process of upstreaming support for the USB 8821cu/8811cu chipsets started. I can't say when this will be complete but it will result in a mac80211, in-kernel driver. I mention this because the 8811cu chipset seems to be the most popular dual band nano chipset. I have one and can provide details if interested. You can run my 8821cu out-of-kernel driver until the in-kernel driver is ready.

Regards

jclark, 2022-06-04:
I completely agree about the usefulness of including reseller links that work.

Great to hear the news about upstreaming the 8821cu/8811cu!

My guess is that the manufacturer decided to change the chipset, while keeping the same model number UNT-W01.  Several resellers on AliExpress are advertising MT7610UN nano adapters with that model number. I will try a few places on AliExpress and see if any of them deliver something with the MT7610.  I would still like to get a nano adapter with 5GHz support that works with Ubuntu 22.04 out of the box. Linksys AE6000 doesn't appear to be available locally.

hackerb9, 2022-06-11:
> the 8811cu chipset seems to be the most popular dual band nano chipset. I have one and can provide details if interested.

Details would be appreciated. Is this [Cudy AC 650](https://www.amazon.com/Cudy-650Mbps-Adapter-Wireless-Desktop/dp/B07RN44SHW/) the right thing?

[image]

morrownr, 2022-06-11:
The one I have follows:

https://www.amazon.com/gp/product/B0872VF2D8

I'm not going to recommend this Edup adapter as the only reason I have it is to test the 8821cu driver I maintain. I use it when I am working on the driver and it seems solid but that Cudy adapter may be solid also. If I were you, I would find and read reviews because there are lot of junk adapters based on the 8811cu.

The driver I use follows:

https://github.com/morrownr/8821cu-20210118

I am a little partial to that driver since I, and a few helpers, maintain that driver here on this site and it works wonderfully with Ubuntu 22.04. Hopefully I can shut down work on this driver when the properly written driver is upstreamed into the kernel.

Let us know how it goes.

hackerb9, 2022-06-16:
That worked. I'm typing this now over the Cudy WU-650. It seems great so far. I'll let you know if there are any problems.

Of course, this isn't yet a solution to this issue as I'm using your DKMS module. But, hopefully soon there will be an in kernel driver.  (Do you know if it'll be based off of your work?)

morrownr, 2022-06-16:
Glad you are making progress.

> Do you know if it'll be based off of your work?

It will not be based on my work... thank goodness! My work is based on the Realtek out-of-kernel driver so it is not standards compliant. It would never be allowed to be in the kernel.

https://www.phoronix.com/scan.php?page=news_item&px=Linux-Patches-RTW88-WiFi-USB

When we will actually see this work in the kernel is not known. I hope it is in the near future but we'll just have to see.