morrownr/USB-WiFi #59: COMFAST CF-WU782AC hostapd doesn't work

Slmba, 2022-02-11:
Hello. I appriciate this topic, but I cannot make WU782AC works with hsotapd.

Raspberry Pi 4B:
[image]
iwconfig:
[image]
ifconfig:
[image]
hostapd.conf:
[image]

So.
The problem - network exists, client connects to this network succesful. But no access to raspberry.
[image]

**- Before that the same configuration for wlan0 (default raspberry wi-fi) worked fine.**
**- Without hostapd wlan1 usb adapters works fine.**

Slmba, 2022-02-11:
Hi, @morrownr 
Could you please help? I'm sure I made like that before.

Slmba, 2022-02-11:
Never mind. Country_code "RU" fix this issue.

morrownr, 2022-02-11:
Hi @Slmba 

Glad you found a fix. hostapd requires a little time to understand all of the settings.

Let us know if anything else is a problem.

Regards

Slmba, 2022-02-11:
Hi, @morrownr 
Yeah, 2 more issues.

1. I cannot setup wlan0 to connect to my home network AND setup wlan1 (usb COMTEK) to share network. One of it works only.
2. USB wi-fi hostapd AP - too slow to connect. My smartphone drops the connection few times before connect.

Slmba, 2022-02-11:
No, doesn't work.

Slmba, 2022-02-11:
Hi, @morrownr 
Thank you for your help!

So, as I see now - very unstable solution.
When all interfaces are disabled - works fine at configuration above.

morrownr, 2022-02-12:
> I think - this is not good solution - hostapd + dnsmasq. No ideas.

I have no idea either because I don't know what you are trying to do.

We really need know what you are trying to accomplish and what hardware you have available.

What hardware? RasPi4B?
What OS?
What guide are you using?

Slmba, 2022-02-15:
Hi, @morrownr 
Somehow the problem is solved.

Slmba, 2022-02-15:
As I investigated when I connect COMTEK USB CARD - my wpa_supplicant crashes.
Internal wi-fi network also crashes.

Slmba, 2022-02-15:
Logs of USB CARD connection:
`
Feb 15 05:58:30 copter kernel: [  152.324775] usb 1-1.2: new high-speed USB device number 3 using xhci_hcd
Feb 15 05:58:30 copter kernel: [  152.456118] usb 1-1.2: New USB device found, idVendor=0e8d, idProduct=2870, bcdDevice= 0.01
Feb 15 05:58:30 copter kernel: [  152.456137] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 15 05:58:30 copter kernel: [  152.456153] usb 1-1.2: Product: Љ
Feb 15 05:58:30 copter kernel: [  152.456168] usb 1-1.2: Manufacturer: Љ
Feb 15 05:58:30 copter kernel: [  152.456183] usb 1-1.2: SerialNumber: Љ
Feb 15 05:58:30 copter kernel: [  152.461217] usb-storage 1-1.2:1.0: USB Mass Storage device detected
Feb 15 05:58:30 copter kernel: [  152.464505] scsi host0: usb-storage 1-1.2:1.0
Feb 15 05:58:30 copter mtp-probe: checking bus 1, device 3: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:
Feb 15 05:58:30 copter mtp-probe: bus: 1, device: 3 was not an MTP device
Feb 15 05:58:30 copter systemd[1]: Created slice system-usb_modeswitch.slice.
Feb 15 05:58:30 copter systemd[1]: Starting USB_ModeSwitch_1-1.2...
Feb 15 05:58:30 copter mtp-probe: checking bus 1, device 3: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:
Feb 15 05:58:30 copter mtp-probe: bus: 1, device: 3 was not an MTP device
Feb 15 05:58:31 copter usb_modeswitch: switch device 0e8d:2870 on 001/003
Feb 15 05:58:31 copter kernel: [  153.250466] usb 1-1.2: USB disconnect, device number 3
Feb 15 05:58:32 copter kernel: [  154.005120] usb 2-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
Feb 15 05:58:32 copter kernel: [  154.036989] usb 2-2: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
Feb 15 05:58:32 copter kernel: [  154.037008] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 15 05:58:32 copter kernel: [  154.037023] usb 2-2: Product: 802.11ac WLAN
Feb 15 05:58:32 copter kernel: [  154.037039] usb 2-2: Manufacturer: MediaTek Inc.
Feb 15 05:58:32 copter kernel: [  154.037053] usb 2-2: SerialNumber: 000000000
Feb 15 05:58:32 copter mtp-probe: checking bus 2, device 2: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:
Feb 15 05:58:32 copter mtp-probe: bus: 2, device: 2 was not an MTP device
Feb 15 05:58:32 copter kernel: [  154.325597] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
Feb 15 05:58:32 copter kernel: [  154.357090] mt76x2u 2-2:1.0: ASIC revision: 76120044
Feb 15 05:58:32 copter kernel: [  154.395120] mt76x2u 2-2:1.0: ROM patch build: 20141115060606a
Feb 15 05:58:32 copter kernel: [  154.514971] mt76x2u 2-2:1.0: Firmware Version: 0.0.00
Feb 15 05:58:32 copter kernel: [  154.514980] mt76x2u 2-2:1.0: Build: 1
Feb 15 05:58:32 copter kernel: [  154.514988] mt76x2u 2-2:1.0: Build Time: 201507311614____
Feb 15 05:58:33 copter kernel: [  155.418537] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
Feb 15 05:58:33 copter kernel: [  155.422647] usbcore: registered new interface driver mt76x2u
Feb 15 05:58:33 copter systemd[1]: Starting Load/Save RF Kill Switch Status...
Feb 15 05:58:33 copter mtp-probe: checking bus 2, device 2: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:
Feb 15 05:58:33 copter mtp-probe: bus: 2, device: 2 was not an MTP device
Feb 15 05:58:33 copter systemd-udevd[726]: Using default interface naming scheme 'v240'.
Feb 15 05:58:33 copter systemd-udevd[726]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable
Feb 15 05:58:33 copter kernel: [  155.469436] mt76x2u 2-2:1.0 wlxe0e1a915030c: renamed from wlan1
Feb 15 05:58:33 copter systemd[1]: Started Load/Save RF Kill Switch Status.
Feb 15 05:58:33 copter systemd-udevd[730]: Using default interface naming scheme 'v240'.
Feb 15 05:58:36 copter wpa_supplicant[428]: wlan0: CTRL-EVENT-DISCONNECTED bssid=9c:9d:7e:16:cc:ff reason=0 locally_generated=1
Feb 15 05:58:36 copter wpa_supplicant[428]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Feb 15 05:58:36 copter wpa_supplicant[428]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=RU
Feb 15 05:58:38 copter systemd[1]: systemd-rfkill.service: Succeeded.
Feb 15 05:58:40 copter kernel: [  161.895017] usb 2-2: USB disconnect, device number 2
Feb 15 05:58:40 copter systemd[1]: Starting Load/Save RF Kill Switch Status...
Feb 15 05:58:40 copter systemd[1]: Started Load/Save RF Kill Switch Status.
Feb 15 05:58:44 copter wpa_supplicant[428]: wlan0: Trying to associate with SSID 'ONLYROG'
Feb 15 05:58:45 copter systemd[1]: systemd-rfkill.service: Succeeded.
Feb 15 05:58:46 copter wpa_supplicant[428]: wlan0: Associated with 9c:9d:7e:16:cc:ff
Feb 15 05:58:46 copter wpa_supplicant[428]: wlan0: CTRL-EVENT-CONNECTED - Connection to 9c:9d:7e:16:cc:ff completed [id=0 id_str=]
Feb 15 05:58:46 copter wpa_supplicant[428]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Feb 15 05:58:46 copter wpa_supplicant[428]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=RU
Feb 15 05:58:51 copter systemd[1]: usb_modeswitch@1-1.2.service: Succeeded.
Feb 15 05:58:51 copter systemd[1]: Started USB_ModeSwitch_1-1.2.
Feb 15 05:58:51 copter systemd[1]: session-1.scope: Succeeded.
Feb 15 05:58:52 copter kernel: [  173.990017] w1_master_driver w1_bus_master1: Attaching one wire slave 00.c00000000000 crc ca
Feb 15 05:58:52 copter kernel: [  173.999117] w1_master_driver w1_bus_master1: Family 0 for 00.c00000000000.ca is not registered.
[... cut ...]
Feb 15 06:01:12 copter kernel: [  313.999437] w1_master_driver w1_bus_master1: Attaching one wire slave 00.a00000000000 crc af
Feb 15 06:01:12 copter kernel: [  314.007900] w1_master_driver w1_bus_master1: Family 0 for 00.a00000000000.af is not registered.
Feb 15 06:01:50 copter kernel: [  351.919183] w1_master_driver w1_bus_master1: Attaching one wire slave 00.600000000000 crc 65
Feb 15 06:01:50 copter kernel: [  351.924518] w1_master_driver w1_bus_master1: Family 0 for 00.600000000000.65 is not registered.
`

Slmba, 2022-02-15:
I think this is my bad.
I've found that when I configure static ip for eth0 and wlan0 there is one gateway for both interfaces.
And it cause big problems.
Thats why when connect COMTEK all network crashes. Same gateway at wlan1 interface, after connection network is down.
Thats why when I use eth0 connection - wi-fi connections crashes too.

So, the solution - to use dhcp for eth0 and hotplug, as connection back door if wlan networks would crash somehow.
Topic closed. Big thanks!

closed 2022-02-15 by Slmba