morrownr/USB-WiFi #64: Possibly GREAT News ...

morrownr, 2022-03-02:
I monitor `linux-wireless` in an effort keep up with what is going on with wireless in Linux. Today a message caught my attention:

```
This is a preliminary patch to add mt7921u driver support.

---
 drivers/net/wireless/mediatek/mt76/mt7921/mac.c      | 12 ++++++++++++
 drivers/net/wireless/mediatek/mt76/mt7921/mt7921.h   |  2 +-
 drivers/net/wireless/mediatek/mt76/mt7921/sdio.c     |  2 +-
 drivers/net/wireless/mediatek/mt76/mt7921/sdio_mac.c | 11 -----------
 4 files changed, 14 insertions(+), 13 deletions(-)
...
```

Lorenzo is one of the Mediatek kernel devs that works on Mediatek wireless drivers. I and some of the rest of you have been hoping for news that Mediatek is going to release a new WiFi 6/e chipset for USB adapters. It appears that work to support USB adapters is now underway in the kernel which probably means that a new mt7921u chipset will be released to adapter makers in the not so distant future and we may actually have a driver ready in the kernel before products ship. The mt7921e driver that has been in the kernel since v5.12 is a very good driver as I have a new little laptop that has an internal wifi card based on this chipset. When I installed Ubuntu 21.10 on the laptop, the mt7921 based wifi card just worked. Good stuff!

Regards

morrownr, 2022-03-05:
Update: As I continue to watch the new mt7921u driver flow into the kernel, I wish I had a usb wifi adapter to test. Hopefully new adapters with this chipset will be available soon.

It is good to see this happening. Mediatek is fast becoming the best provider of wifi chipsets and in-kernel drivers for Linux. Intel provides drivers for their cards but no usb chipsets are made available and Intel does a poor job of AP mode support. Atheros no longer provides usb chipsets and and their in-kernel Linux drivers, while being pretty good, are just not as good as Mediatek. Realtek's release of their 8832/52 WiFi 6 chipset for usb last year is just subpar. Instead of moving to provide a standards compliant in-kernel driver, Realtek continued to provide an out-of-kernel driver and it is a terrible driver.

My recommendation for those needing a USB WiFi adapter now is to go ahead and get one of the ones shown here in this repo. The drivers for all of the adapters shown here will be in the kernel and continue to improve for many years. The mt7612u, mt7610u, mt7601u and rt5572 chipsets will likely continue to be produced for several more years giving Linux users a lot of options and price levels depending on need.

One last thing. Mediatek signed a big agreement with AMD last year. Mediatek is now providing wifi chipsets to AMD for use in AMD processor based systems... and, it appears, to be sold as separate products as well. I am already seeing products and systems that list their WiFi capability with an AMD model number. All of the ones I have seen so far, AMD RZ6xx, are based on the Mediatek mt7921 chipset and are supported by the existing Mediatek mt7921 driver that has been in the kernel since v5.12. To clarify, it appears that AMD rebrands Mediatek products so don't start looking for an AMD driver if you get wifi based on a RZ6xx product, it will use the Mediatek drivers.

Your thoughts on this new driver?

amisix, 2022-03-06:
What will NAT hardware flow offloading do for me in OpenWrt since the mt7621u will support it? Can it increase the throughput of my double NAT configuration?

Malvineous, 2022-03-06:
This is indeed excellent news.  I submitted PR #65 to include a bit more information about it, and also discovered there is an MT7922 chipset that directly competes with the Intel AX210 (i.e. it also supports 160 MHz channels) but it also has USB3 support (as well as PCIe).

Looking at the patch it's interesting - they don't add support for specific devices, they add support for the new features and whether they are used or not depends on whether the device reports them as supported.  This means everything is backwards compatible, and an MT7621U USB device would've worked before this patch, just on 2.4 and 5 GHz only.  Now the patch has added support for 6 GHz to MediaTek USB devices, it should mean future devices (like the MT7922) work without special treatment.

So hopefully it won't be too long until we start getting some decent USB devices with 6 GHz support!

morrownr, 2022-03-07:
> I submitted PR #65 to include a bit more information about it

Thanks. I pulled it. I also did some work on the chart. 

> also discovered there is an MT7922 chipset that directly competes with the Intel AX210

I have missed this. Link?

> an MT7621U USB device would've worked before this patch

Yes, the base chipset would have worked but USB support is not trivial and the only way to have a USB adapter is to have USB support.

> Now the patch has added support for 6 GHz to MediaTek USB devices

Yes. My opinion at this point is that Mediatek very well could blow Realtek out of the water as far as Linux and the USB WiFi adapter world is concerned.

Regards

Malvineous, 2022-03-07:
> > also discovered there is an MT7922 chipset that directly competes with the Intel AX210
> 
> I have missed this. Link?

There are a couple in the newly merged docs :)

> Yes, the base chipset would have worked but USB support is not trivial and the only way to have a USB adapter is to have USB support.

Yes, to be clear, it looks like the core chipset is programmed the same way, and those commands just travel over the USB or PCIe bus.  So this patch only really had to add basic USB support to carry the messages as the MT76 chipset programming was already done.  It's quite an elegant way to design hardware as it means you get to reuse a lot of code, which is why the `mt76` driver supports a whole range of chips.

It also means once your driver is merged into the kernel, each new chipset that comes out only requires minor additional changes to add just the new features, you don't need a whole separate driver *cough*Realtek*cough*

> Yes. My opinion at this point is that Mediatek very well could blow Realtek out of the water as far as Linux and the USB WiFi adapter world is concerned.

I must admit I didn't really know much about Mediatek before this, but Ralink had a good reputation when they were around and my opinion of Mediatek is improving all the time thanks to their good Linux support.

amisix, 2022-03-08:
@morrownr Maybe I should rephrase my previous question. 

What does NAT hardware flow offloading on the mt7621 chipset do for us? From what I've read it significantly increases throughput - do you think that would be the case with a mt7621u adapter in AP mode?

morrownr, 2022-03-08:
> > > also discovered there is an MT7922 chipset that directly competes with the Intel AX210
> >  
> > I have missed this. Link?
> 
> There are a couple in the newly merged docs :)

Whoa horse!

I did a little investigating this morning. It appears that the mt7922 chipset is for use in wifi routers and similar equipment. It is not appropriate for USB adapters or PCIe.M.2 cards so it really needs to be pulled out of our charts here.

Malvineous, 2022-03-09:
No problem at all if you want to remove it.  My reason for including it was twofold.  One is that one day adapters may show up including it, and it would be nice to know straight away that there is no driver support for that chipset yet, and the other is that if you want to buy an access point running Linux it can be useful to know whether there is mainline kernel support for it, since projects like OpenWRT often don't make it clear that they are including weird vendor drivers in their builds.

If you only want to focus on known devices already seen in the wild then that makes sense (and might be worth explicitly mentioning this), but then in that case you'll have to remove your MT7921U until the first device shows up on the market 😛

Malvineous, 2022-03-09:
@amisix Where are you seeing NAT hardware flow offloading in the Linux driver?  I didn't think Linux made use of that.  I imagine it would increase speeds if you had a really underpowered CPU, but modern machines that can do gigabit routing without breaking a sweat would hardly notice a difference.  I have 10 year old machines doing 10 gigabit routing so NAT offloading probably only makes sense in a router with an underpowered CPU, but then you probably need a modified kernel as I am not sure that iptables or its replacement supports hardware offloading.

morrownr, 2022-03-09:
Hi @Malvineous 

> If you only want to focus on known devices already seen in the wild then that makes sense (and might be worth explicitly mentioning this), but then in that case you'll have to remove your MT7921U until the first device shows up on the market stuck_out_tongue

I have already thought of this last point you make. I need to either explain or remove mt7921u.

What are my thoughts up to this point? As USB-WiFi has grown, I have had to ponder many things. I have had to respond to many issues and I have tried to always follow one basic rule: What is best for the overall Linux USB WiFi user? While there are many technically proficient Linux users, first and foremost this site must take care of the nubies because nubies is how Linux use grows.

Nubies may not know what a chipset is. Most probably don't care. Most nubies only want to something to work and I think charts of chipsets and capabilities are for experienced users. Nubies need links to products where all they have to do is click on `Buy`. If the truth is known, many experienced users do not know where they can find nice adapters with in-kernel drivers because the internet has been so full of bullshit. There are hundreds of `The best usb adapters for Linux  ... ` articles and almost all of them are taking unsuspecting users on a bad ride. One of the missions of this site is help correct that and I have written to some article authors and they are using the info here to rework their articles. This is a challenge but I appear to be making progress. This repo has grown to average over 500 views per day. It isn't just home users, I am seeing information that tells me IT and gov accounts are stopping by often and I get feedback from a couple of vendors that sales of adapters with in-kernel drivers has taken off.

So, yes, I lean toward certain things that I think are needed. I've watched since I started using Linux in 1994 and I think a pragmatic approach that is limited to things that matter to Linux users is important. The mt7668u chipset is irrelevant. It was not designed for usb adapters or PCI cards. It is AC1200 capable but only supports USB2 so it would not sell as a USB adapter.

I do like your idea about moving USB-WiFi to a better site and will look toward doing that as I have time to explore doing so.

Thanks,

Nick

bjlockie, 2022-03-12:
> I did a little investigating this morning. It appears that the mt7922 chipset is for use in wifi routers and similar equipment. It is not appropriate for USB adapters or PCIe.M.2 cards so it really needs to be pulled out of our charts here.

It'd be super nice to see a USB device that takes an M.2 card. :-)
That should fit. :-)

Malvineous, 2022-03-12:
Totally understand wanting to cater to newbies.  But surely we can also cater to experts as well?  It seems people are either in two camps - they aren't hardware people so they just want to know which devices they can buy and have it work out of the box, or they are enthusiasts that want specific features and don't mind digging through datasheets to figure it out.  My thinking was that with your "short list" of devices plus the chipset lists, it means we can satisfy both audiences.

Since there is nothing stopping one of the chipsets designed for AP use from being stuck on a PCIe card or USB stick, my thoughts were to include them in case it happens one day, but equally so if we just add them once the first device appears, that's fine too, it makes a lot of sense doing it that way.  As I am learning more about it, I see the PCIe/USB devices tend to be 2x2 and lower speed, whereas the AP chipsets are 4x4 and faster speed.  Ideally in my case wanting to turn an existing server into an AP, the 4x4 chipsets are what I am after.  But you are right, these rarely if ever seem to make it into PCIe or USB devices for some reason.  It seems like the best way to get a high performance access point is still to buy a physical AP running Linux.

@bjlockie What do you mean by a USB device that takes an M.2 card?  You can get these depending on what M.2 features you need.

bjlockie, 2022-03-12:
> they aren't hardware people so they just want to know which devices they can buy and have it work out of the box

People should be asking what is guaranteed to work out of the box on any version of Linux.
Most people don't know that an in kernel driver is future proof.

> What do you mean by a USB device that takes an M.2 card? You can get these depending on what M.2 features you need.

Some kind of USB adapter that you can put in a M.2 wifi card in and have it work.

Malvineous, 2022-03-13:
Yes it would be good to have a kernel version listed as the starting point from when the device was first supported, but then at present the list seems to be targeted at the "latest stable release", so if your distro is running behind it's a matter you'd have to take up with them.  Which, to be fair, is something newbies probably would know little about.

The M.2 connector provides a bunch of different transports - SATA, USB, PCIe, etc.  So to get an M.2 WiFi card to work, depends on what transport it uses.  If it's USB-over-M.2 then the enclosure is basically passthrough so a cheap one (that doesn't do SATA-to-USB conversion) will work.  If the WiFi card is PCIe then you'll need an enclosure that's NVMe compatible (since NVMe uses PCIe).  I don't believe at this time that PCIe can be wrapped inside USB, but it can be wrapped inside Thunderbolt.  So if you have any USB ports that are Thunderbolt compatible, then an M.2 to Thunderbold adapter like [this one](https://www.aliexpress.com/item/1005003085755470.html?spm=a2g0o.productlist.0.0.34f147450gRCwy) should allow you to run a PCIe M.2 card into a Thunderbolt-enabled USB-C port.

morrownr, 2022-03-21:
hi @amisix 

Sorry about the slow reply. I have busy.
 
> What does NAT hardware flow offloading on the mt7621 chipset do for us? From what I've read it significantly increases throughput - do you think that would be the case with a mt7621u adapter in AP mode?

I thought hardware flow ofloading only applied to wifi routers. Which applies to several Mediatek chipsets but I'm not sure that it applies to the mt7921. It would apply to the mt7621 and mt7922 chipsets as those chipsets are used in routers..

Yes, Mediatek has a lot of wifi related chipsets but, unlike Realtek, they do not seem to flood the market with many similar chipsets that are all for one use...and all require different drivers which ends up being very confusing for users.

Regards

morrownr, 2022-03-21:
@Malvineous @bjlockie @amisix 

Hey, should we have a contest to see who can report the first availability of a USB WiFi adapter with a 7921 chipset?

amisix, 2022-03-22:
@morrownr lol, I'm just going to sit back and watch you guys talk about numbered chipsets that are so similar I forget the differences. Give me a "u" and I'll be happy for a new toy.

Malvineous, 2022-03-22:
That's what the [chipset list](USB_Chipsets.md) is for, so you don't have to remember the differences 😛

Malvineous, 2022-04-27:
The drivers would support all the router chipsets as most routers run Linux, so on a router either they ship a hacky proprietary driver or they just use the normal kernel one.  But I think your point stands - we could end up with hundreds of chipsets in the list, none of which have ever appeared in a device, unless we say we'll only add entries to the list once they appear in a USB/PCIe device you can actually buy.  If we do that then it avoids the problem of finding a great chipset in the list but then wasting time trying to find a USB device that uses said chipset, because none exist.

Technically any of those chipsets could be used in a PCIe or USB device, since that's how they are connected in a router, but the only difference between those and other "normal" chipsets is that they contain features only usable by a WiFi access point, like additional streams for multiple clients.  Since normal WiFi clients couldn't use those features as they're not running in AP mode, it'd be unnecessary extra expense when making a USB device.

So it wouldn't surprise me if those router-focused chipsets never appear in standalone USB or PCIe devices.

morrownr, 2022-04-28:
I'm looking forward to seeing what we get. I am not a fan the way Realtek does it as far as how many chipsets are available. The great number of Realtek chipsets over the years has made life a challenge for Linux users. Give me one darn good chipset at each level of capability and do the driver right!

I think what I would like to see out of Mediatek's WiFi chipsets is just the mt7921 and mt7922. And hopefully those chipsets are flexible enough to support the needed form factors for usb wifi adapters. I want simple, reliable, low power, runs cool and has good range.

Now please get those chipsets out there.

Regards

objelf, 2022-07-22:
> @Malvineous @bjlockie @amisix
> 
> Hey, should we have a contest to see who can report the first availability of a USB WiFi adapter with a 7921 chipset?

I saw MT7921KU USB dongle sample can be ordered there at
https://m.alibaba.com/product/1600531490895/COMFAST-NEW-WIFI6-WIRELESS-USB-WIFI.html
It might be it is helpful for the people who wanna have a fresh try with USB Wifi adapter with a 7921 chipset

bjlockie, 2022-07-22:
> I saw MT7921KU USB dongle sample can be ordered there at https://m.alibaba.com/product/1600531490895/COMFAST-NEW-WIFI6-WIRELESS-USB-WIFI.html It might be it is helpful for the people who wanna have [...]

Except they have the same model with a Realtek chipset:
https://www.alibaba.com/product-detail/COMFAST-CF-951AX-Long-Range-5GHz_1600502608546.html
Maybe they just copied the description, if someone in North America wants to order 2, I'll buy one off you.

sjtmbg, 2023-11-02:
Sorry for commenting on such an old thread but I just upgraded to xubuntu 23.10, 6.5 kernel, and my mt7921u USB adapter was plug and play! (this is the Comfast CF-953AX with the weird 3574:6211 vid/pid).  Sadly, I don't have a 6/e router to test it with.

morrownr, 2023-11-02:
Hi @sjtmbg 

Good to hear this. I think that vid/pid went into kernel 6.3. As of right now, I am not aware of any company specific vid/pids that are not in kernels 6.4 and later. Bug fixes and tweaks continue to go into this driver as well so going forward, things should be smooth. How is your adapter working for you?

> Sadly, I don't have a 6/e router to test it with.

Don't worry about it. Last summer I did pick up a low cost wifi router that is WiFi 6, not 6e, so that I can test some of the things I work on. It has worked well. I always watch OpenWRT as it helps me decide what to buy and I always want something that works well with OpenWRT so as to move over to it at some point:

https://www.amazon.com/dp/B0BRK3CYY3?psc=1&ref=ppx_yo2ov_dt_b_product_details

WiFi 7 products will be shipping soon and that should bring the prices on WiFi 6e wifi routers down over time. 

FYI: Mediatek has a new WiFi 7 chipset called the mt7925. The driver for usb and PCIe is going into kernel 6.7 right now but I have no idea how long before adapters will be available. Hopefully sometime in 2014.

Stop by anytime.

@morrownr

sjtmbg, 2023-11-02:
> Hi @sjtmbg
> 
> How is your adapter working for you?

Download speeds are so-so but this is on a *very* old laptop.  I will eventually test on a machine where some side-by-side comparisons with other adapters are more valid.

> 
>  I always watch OpenWRT as it helps me decide what to buy and I always want something that works well with OpenWRT

Thanks for that advice - very helpful!

Cheers,

Steve