morrownr/USB-WiFi #66: Mediatek MT7921K "Driver does not support AP mode"

craiganderson, 2022-03-07:
Using the `mt7921e` driver on a `5.15.12` kernel for a `MT7921K` device.

When attempting to bring up an AP hosted with wpa_supplicant `v2.11` I get the following error:

`wlp3s0: Driver does not support AP mode`

Would you be able to confirm (or add to your table) the last kernel version that was successfully tested for the devices listed there? And meanwhile if anyone has successfully used this card in AP mode and can supply their kernel version and environment details, that would be helpful.

Thanks!

morrownr, 2022-03-07:
Hi Craig,

As hard as i try to keep with all of the details related to the new mt7921 chipsets and drivers, I am not where I would like to be.

I am under the impression that the mt7921k supports 6 GHz and I saw the new `k` device IDs go in about 3 months ago which means they would not be in kernel 5.15 yet. Are you sure your card is not based on the mt7921?

Can you post the results of the following so I can get a better handle on what you have?

iw list
lsusb

My laptop has a mt7921 based card. It is dual band, not triple band like you would get from a mt7921k based card. I just ran `iw list` and it certainly shows support for AP mode. I'd test but my to-do list for this coming week is overflowing.

Nick

craiganderson, 2022-03-07:
I updated to `5.17.0-rc3` for good measure but no change in behavior. I'm for sure using the `MT7921K`, the 6GHz bands show up in `iw list` but I can't create AP on any band (not even 2.4GHz), but it looks like this is expected since  AP mode does **not** show up in the listed of Supported interface modes reported by `iw list`.

Can you confirm that your card uses the `mt7921e` driver? You can find this under `Kernel driver in use` for your Mediatek device after running `lspci -vv`

```
$ lsusb | grep -i media
Bus 005 Device 003: ID 0e8d:0608 MediaTek Inc. Wireless_Device
Bus 002 Device 003: ID 0e8d:7612 MediaTek Inc. USB3.0 Hub
```

```
$ iw list
Wiphy phy0
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * monitor
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: Station
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce120dc0b306423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
				HE SU PPDU with 1x HE-LTF and 0.8us GI
				NDP with 4x HE-LTF and 3.2us GI
				STBC Tx <= 80MHz
				STBC Rx <= 80MHz
[... 408 lines cut ...]
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * mesh point: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
```

morrownr, 2022-03-07:
My laptop with mt7921:

$ lspci

0000:02:00.0 Network controller: MEDIATEK Corp. Device 7961

$ iw list
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: Station
			HE MAC Capabilities (0x08031a000040):
				+HTC HE Supported
				TWT Requester
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x22700c120000b006423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
[... 268 lines cut ...]
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support

craiganderson, 2022-03-07:
Can you run `lspci -vv` to get the driver you're using? If your card were using a different driver, that could explain the difference in behavior. (Note last two lines)

```
$ lspci -vv
03:00.0 Network controller: MEDIATEK Corp. Device 0608
	Subsystem: MEDIATEK Corp. Device 0608
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 82
	Region 0: Memory at e0300000 (64-bit, prefetchable) [size=1M]
	Region 2: Memory at e0400000 (64-bit, prefetchable) [size=16K]
	Region 4: Memory at e0404000 (64-bit, prefetchable) [size=4K]
	Capabilities: [80] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 75.000W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <8us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (ok), Width x1 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, NROPrPrP-, LTR+
			 10BitTagComp-, 10BitTagReq-, OBFF Not Supported, ExtFmt+, EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS-, TPHComp-, ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
			 AtomicOpsCtl: ReqEn-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [e0] MSI: Enable+ Count=1/32 Maskable+ 64bit+
		Address: 00000000fee00000  Data: 0000
		Masking: fffffffe  Pending: 00000000
	Capabilities: [f8] Power Management version 3
		Flags: PMEClk- DSI+ D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108 v1] Latency Tolerance Reporting
		Max snoop latency: 1048576ns
		Max no snoop latency: 1048576ns
	Capabilities: [110 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=3us PortTPowerOnTime=28us
		L1SubCtl1: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+
			   T_CommonMode=0us LTR1.2_Threshold=163840ns
		L1SubCtl2: T_PwrOn=150us
	Capabilities: [200 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Kernel driver in use: mt7921e
	Kernel modules: mt7921e
```

morrownr, 2022-03-07:
$ sudo lspci -vv

0000:02:00.0 Network controller: MEDIATEK Corp. Device 7961
	Subsystem: Lite-On Communications Inc Device 3802
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 158
	Region 0: Memory at 6003000000 (64-bit, prefetchable) [size=1M]
	Region 2: Memory at 6003100000 (64-bit, prefetchable) [size=16K]
	Region 4: Memory at 6003104000 (64-bit, prefetchable) [size=4K]
	Capabilities: [80] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 10.000W
		DevCtl:	CorrErr- NonFatalErr- FatalErr- UnsupReq-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <8us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s (ok), Width x1 (ok)
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR+ OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [e0] MSI: Enable+ Count=1/32 Maskable+ 64bit+
		Address: 00000000fee005b8  Data: 0000
		Masking: fffffffe  Pending: 00000000
	Capabilities: [f8] Power Management version 3
		Flags: PMEClk- DSI+ D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108 v1] Latency Tolerance Reporting
		Max snoop latency: 3145728ns
		Max no snoop latency: 3145728ns
	Capabilities: [110 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=3us PortTPowerOnTime=28us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=65536ns
		L1SubCtl2: T_PwrOn=44us
	Capabilities: [200 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Kernel driver in use: mt7921e
	Kernel modules: mt7921e

craiganderson, 2022-03-07:
The principal dev for this driver got back to me quickly and confirmed that "mt7921 only supports sta (client) and monitor mode at the moment." I'm not sure why your card would like AP Mode support in this case since it sounds like it won't work for any mt7921 devices. Anyway, feel free to update your table with these findings to remove indication of AP mode support.

morrownr, 2022-03-07:
I changed AP mode in the PCIe chart to show `in progress` for the mt7921k.

There is very heavy development going on with the mt7921 driver right now.

craiganderson, 2022-03-07:
Sounds good! Hopefully we get something soon.

One other small nit - technically the AX200 and AX210 drivers do support AP mode, it's just handicapped for LAR affected bands. So in the AP Mode row for these, you could say something like "2.4 GHz only" to be more precise.

craiganderson, 2022-03-07:
Also I just happened to have a MediaTek MT7921 lying around (same card as you), so I tried it:

But it gives the same results, no AP supported with the mt7921e driver:

```
	Supported interface modes:
		 * managed
		 * monitor
```

I actually had two 7961 cards with different sub-models, `7961:4680` which didn't match yours, and `7961:3802` which did match. However the results were the same for both, and neither showed up with AP mode. I have no explanation for how AP mode shows up on your MT7921 when the devs are telling me this driver doesn't support AP mode in any way.

Are you certain the `iw list` output you pasted is coming from your `MT7921` and not another WiFi card? Otherwise I'm stumped...

morrownr, 2022-03-07:
> Are you certain the iw list output you pasted above showing AP mode support is coming from your MT7921?

I am sure. No other internal wifi in that laptop and no usb adapter plugged in.  You are just going to have to face the fact that my driver is better than yours. ... he he he. Seriously, I'm running kernel 5.13. I've not checked to see if AP mode is working. I have checked monitor and managed modes. Just because my driver says AP mode is working does not make it so. This site is mostly about USB adapters and there are no USB adapters based on the mt7921 yet so there is very little testing going on at this point.

gitclint, 2022-03-08:
> The principal dev for this driver got back to me quickly and confirmed that "mt7921 only supports sta (client) and monitor mode at the moment." I'm not sure why your card would like AP Mode support [...]

Sorry for the dumb question, but where is the table with the AP Mode support? Does it also show how many SSIDs are supported? I've been looking everywhere for a document like this and it sounds like it may have been under my nose the whole time! xD

craiganderson, 2022-03-08:
> Sorry for the dumb question, but where is the table with the AP Mode support? Does it also show how many SSIDs are supported? I've been looking everywhere for a document like this and it sounds like [...]

https://github.com/morrownr/USB-WiFi/blob/main/PCIe_WiFi_Devices.md

morrownr, 2022-03-15:
Hi @craiganderson 
 
> But it gives the same results, no AP supported with the mt7921e driver:
> 
> ```
> 	Supported interface modes:
> 		 * managed
> 		 * monitor
> ```

Word from a source that would know is that the AP mode support should be in the firmware very soon. It appears to me that Mediatek is going as fast as they can and they decided to go ahead and release the firmware and driver with support as you see it so that Linux users would have at least good managed mode support in the short term but are working very hard to get the rest of the parts in as fast as they can. These WiFi 6 chipsets and drivers are incredibly complex. It astounds me how well the 7921 based card in my laptop works given this new generation of wifi.

All indications are that the 7921e and 7921u will be full featured once everything is complete.

> Are you certain the `iw list` output you pasted above showing AP mode support is coming from your `MT7921` and not another WiFi card? Otherwise I'm stumped...

Hopefully the above info helped. I still haven't had time to use AP mode but managed mode works great. I  am busy. Golden Rule that I learned working on the Realtek drivers: Just because `iw list` or `iw dev` says something does not make it so. They probably had forgotten to remove the AP support statement in the version of the driver I have.

So, a little patience and we will have what we seek.

Regards

morrownr, 2022-03-27:
Hi @craiganderson 

> Word from a source that would know is that the AP mode support should be in the firmware very soon.

The following was submitted last week:

```
Update binary firmware for MT7921 WiFi devices

File: mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
Version: 20220311230842a
File: mediatek/WIFI_RAM_CODE_MT7961_1.bin
Version: 20220311230931
```

I suspect that these new firmware files include the AP support for the mt7921e/u. You can keep an eye on the log at the following site:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/mediatek

Once you see a new entry for those files then you can go to `tree` and download them and replace your current versions and see what the results are. Please let us know.

Regards

morrownr, 2022-04-05:
Hi @craiganderson 

Today the follow patch was submitted. It appears that more than the new firmware was needed to turn AP Mode on:

```
From: Sean Wang <sean.wang@mediatek.com>

add AP mode support to mt7921 that can work for mt7921[e,s,u]
with the common code.

Tested-by: Deren Wu <deren.wu@mediatek.com>

---
 .../net/wireless/mediatek/mt76/mt7921/init.c  | 11 ++-
 .../net/wireless/mediatek/mt76/mt7921/mac.c   |  9 +++
 .../net/wireless/mediatek/mt76/mt7921/main.c  | 46 +++++++++++
 .../net/wireless/mediatek/mt76/mt7921/mcu.c   | 79 ++++++++++++++++++-
 .../wireless/mediatek/mt76/mt7921/mt7921.h    |  4 +
 5 files changed, 147 insertions(+), 2 deletions(-)
```

As you know, I keep a watch on the linux-wireless mailing list. The amount of work required for the WiFi 6 drivers is incredible. As best I can tell Mediatek has all hands on desk in an effort to finish work on the remaining features for the mt7921[e,s,u].

FWIW: I've used Linux since before WiFi, as we know it, existed. If my memory is correct, the mt7921u will be the first USB WiFi driver that will be in the kernel and fully ready to go BEFORE adapters are available.

zzxxtt, 2022-05-03:
Hello, Although their's a patch on mt7921 ap mode, It seems it not merged to mainline kernel yet? I installed the latest 5.17.5 kernel and download the latest formware but still not have ap mode in iw list. Any idea how to build it by ourselve, like dkms?

morrownr, 2022-05-03:
hi @zzxxtt 

After I looked at the date on my message above about the latest commit, I can see the date is only 29 days ago. It won't be in 5.17. It may not have made it into 5.18 but for sure will be in 5.19. It is possible to compile and use a module that is in-kernel, however, that is on my to-do list. I need to learn how to do it. Let me know if you find a good guide.

FYI: I updated my guide about Mediatek firmware:

https://github.com/morrownr/USB-WiFi/blob/main/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

morrownr, 2022-05-12:
Hi @zzxxtt @craiganderson @gitclint @Malvineous 

Regarding the mt7921 AP mode issue:

A message came through linux-wireless today and I am posting it below. Driver support for AP mode is going into kernel 5.19. New firmware is also required and has been uploaded to the normal place and I have updated the instructions that I keep in the following document...

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

----------------------------------------------------------------
mt76 patches for 5.19

- tx locking improvements
- wireless ethernet dispatch support for flow offload
- non-standard VHT MCS10-11 support
- fixes
- runtime PM improvements
- mt7921 AP mode support
- mt7921 ipv6 NS offload support

----------------------------------------------------------------
Bo Jiao (2):
       mt76: mt7915: disable RX_HDR_TRANS_SHORT
       mt76: mt7615/mt7915: do reset_work with mt76's work queue

Christophe JAILLET (1):
       mt76: mt7921: Fix the error handling path of mt7921_pci_probe()

Colin Ian King (2):
       mt76: mt7915: make read-only array ppet16_ppet8_ru3_ru0 static const
       mt76: mt7921: make read-only array ppet16_ppet8_ru3_ru0 static const

Deren Wu (2):
       mt76: fix antenna config missing in 6G cap
       mt76: mt7921: add ipv6 NS offload support

Evelyn Tsai (1):
       mt76: fix MBSS index condition in DBDC mode

Felix Fietkau (15):
       mt76: mt7915: fix DBDC default band selection on MT7915D
       mt76: mt7915: rework hardware/phy initialization
       mt76: reduce tx queue lock hold time
       mt76: dma: use kzalloc instead of devm_kzalloc for txwi
       mt76: mt7915: accept rx frames with non-standard VHT MCS10-11
       mt76: mt7921: accept rx frames with non-standard VHT MCS10-11
       mt76: fix use-after-free by removing a non-RCU wcid pointer
       mt76: fix rx reordering with non explicit / psmp ack policy
       mt76: do not attempt to reorder received 802.3 packets without agg session
       mt76: fix encap offload ethernet type check
       mt76: fix tx status related use-after-free race on station removal
       mt76: dma: add wrapper macro for accessing queue registers
       mt76: add support for overriding the device used for DMA mapping
       mt76: make number of tokens configurable dynamically
       mt76: mt7915: add Wireless Ethernet Dispatch support

Lorenzo Bianconi (11):
       mt76: mt7921u: add suspend/resume support
       mt76: mt7921: rely on mt76_dev rxfilter in mt7921_configure_filter
       mt76: mt7921: honor pm user configuration in mt7921_sniffer_interface_iter
       mt76: mt7915: fix unbounded shift in mt7915_mcu_beacon_mbss
       mt76: mt7915: fix possible uninitialized pointer dereference in mt7986_wmac_gpio_setup
       mt76: mt7915: fix possible NULL pointer dereference in mt7915_mac_fill_rx_vector
       mt76: mt7915: do not pass data pointer to mt7915_mcu_muru_debug_set
       mt76: mt7915: report rx mode value in mt7915_mac_fill_rx_rate
       mt76: mt7915: use 0xff to initialize bitrate_mask in mt7915_init_bitrate_mask
       mt76: mt7915: configure soc clocks in mt7986_wmac_init
       mt76: add gfp to mt76_mcu_msg_alloc signature

MeiChia Chiu (1):
       mt76: mt7915: add support for 6G in-band discovery

Peter Chiu (4):
       mt76: mt7915: update mt7986 patch in mt7986_wmac_adie_patch_7976()
       mt76: mt7915: fix twt table_mask to u16 in mt7915_dev
       mt76: mt7915: reject duplicated twt flows
       mt76: mt7915: limit minimum twt duration

Ryder Lee (7):
       mt76: mt7915: always call mt7915_wfsys_reset() during init
       mt76: mt7915: remove SCS feature
       mt76: mt7915: rework SER debugfs knob
       mt76: mt7915: introduce mt7915_mac_severe_check()
       mt76: mt7915: move MT_INT_MASK_CSR to init.c
       mt76: mt7915: improve error handling for fw_debug knobs
       mt76: mt7915: add more statistics from fw_util debugfs knobs

Sean Wang (3):
       mt76: mt7921: Add AP mode support
       mt76: mt7921: fix kernel crash at mt7921_pci_remove
       mt76: connac: use skb_put_data instead of open coding

Shayne Chen (1):
       mt76: mt7915: add debugfs knob for RF registers read/write

Yunbo Yu (1):
       mt76: mt7603: move spin_lock_bh() to spin_lock()

-----

zeddit, 2024-02-22:
hi all, I am using RZ608 wifi card on ubuntu 22.04 to set up an wifi access point.
I am using netplan and NetworkManager to set the wifi in ap mode.
However, devices could not connect to the ap and `sudo wpa_cli -i wlp2s0 status` shows
```
bssid=10:6f:d9:af:56:c5
freq=2412
ssid=test
id=0
mode=AP
pairwise_cipher=CCMP
group_cipher=CCMP
key_mgmt=UNKNOWN <- this problem. 
wpa_state=COMPLETED
ip_address=192.168.18.1
``` 
key_mgmt should be `WPA2-PSK` or something else. so is the issue related to the driver `mt7921e`.
how could I fix it, great thanks.

morrownr, 2024-02-23:
Hi @zeddit 

Recommend that you start a new issue. Very few people will see this issue in this location.

I'd help if I could but using netplan and NM for AP mode is not something I know anything about. I do use AP mode a lot. In fact, I have an AP mode guide called Bridged Wireless Access Point on the Main Menu.

@morrownr

zeddit, 2024-02-23:
@morrownr great thanks for your kindness and reply.
The problem solved after I changed to use `hostapd` rather than NetworkManager.
It's a problem of NetworkManager. 
thank you again.

rohithhackthlon, 2024-05-23:
any updates about mt7921 ap mode support

morrownr, 2024-05-23:
@craiganderson 

This is a very old thread that has drifted toward various topics. I have various adapters/cards that use the mt7921u and mt7921e drivers and AP works fine for me.

Recommend you start a new issue with details on your adapter/card, distro and what you are trying to do.

dreirund, 2024-11-05:
[@craiganderson wrote](#issuecomment-1061385080):

> [https://github.com/morrownr/USB-WiFi/blob/main/PCIe_WiFi_Devices.md](https://github.com/morrownr/USB-WiFi/blob/main/PCIe_WiFi_Devices.md)

Ahoj (@morrownr, too),  
I just wanted to see that site, but it gives a HTTP error 404 "Not Found".

It now seems to be at [https://github.com/morrownr/USB-WiFi/blob/main/home/PCIe_WiFi_Devices.md](https://github.com/morrownr/USB-WiFi/blob/main/home/PCIe_WiFi_Devices.md).

So, maybe you update [your comment](#issuecomment-1061385080), @craiganderson?

Regards!

morrownr, 2024-11-06:
@dreirund 

Where is the location of the bad link?

dreirund, 2024-11-06:
> *Where is the location of the bad link?*

In the comment in this thread which I linked in my citation: [#issuecomment-1061385080](#issuecomment-1061385080).