morrownr/USB-WiFi #67: Alfa ACHM not scanning in monitor mode in Kali

amisix, 2022-03-22:
Hi again @morrownr. Now that I had a chance to play with airodump-ng and the Alfa ACHM in Raspbian I figured I'd try it in Kali (32 bit) but it's not scanning any APs. None are displayed on either band despite it changing channels as if it's scanning. Per airmon-ng the adapter is going into monitor mode successfully. I also tried manually setting monitor mode with iwconfig and the issue persists. The adapter is recognized and works fine in station mode and these commands worked fine in Raspbian. What am I doing incorrectly? 

**airmon-ng**
>airmon-ng start wlan1

>phy1    wlan1           mt76x0u         MediaTek Inc. WiFi
                (mac80211 monitor mode vif enabled for [phy1]wlan1 on [phy1]wlan1mon)
                (mac80211 station mode vif disabled for [phy1]wlan1)

**Manually set monitor mode with iwconfig:**
>sudo ifconfig wlan1 down
sudo iwconfig wlan1 mode monitor
sudo ifconfig wlan1 up

Thanks much..

amisix, 2022-03-26:
>I see you are using ifconfig and iwconfig. Those are old and have been depreciated. Use iw and ip

Will do, thanks. Easier to remember anyway.

I was eventually able to get the ACHM to scan after changing USB ports, which I thought was a bit odd given the adapter functioned fine in managed mode on that port. Oh well. 

Then I used your script and it worked great, thanks!! I'd prefer to be able to restart processes without reboot, that's helpful. More so... being able to stay connected while working in monitor mode, that was fun with airmon-ng the first couple times I got booted. Very neat stuff - this adapter has so much reach, I'm loving it,

Thanks.

morrownr, 2022-03-29:
Hi @amisix 

> Then I used your script and it worked great,

I decided to make a repo for the script today as that will allow folks to work on it and submit PRs. It is a good way to learn BASH and there are things that can be added.

> this adapter has so much reach

Interestingly enough, I did not notice the range of my ACHM until after I had it for a while. Someone had requested a head to head performance test using some usb wifi adapters so I put together a test and my eyes were opened while doing the test.

amisix, 2022-04-07:
Hi @morrownr - I apologize for the delayed response. 

>I decided to make a repo for the script today as that will allow folks to work on it and submit PRs. It is a good way to learn BASH and there are things that can be added.

Awesome, thank you for continuing to create these resources. I would like to learn more after going through the script a bit - very useful stuff. I have a question - when an adapter is placed in monitor mode the name it's given is always wlan0mon no matter what wlan adapter you choose. It's only an issue if you're placing two adapters in monitor mode which I'm sure is a rare use case..

>Interestingly enough, I did not notice the range of my ACHM until after I had it for a while...

I'm enjoying it more and more every day. With a magnetic base and a 10dBi antenna sitting on my desk it's a monster. Running the ACHM against my TP-Link T4U in airodump-ng really opened my eyes (stock antennas of course). Thanks for that rtl8812bu driver!!

morrownr, 2022-04-08:
> when an adapter is placed in monitor mode the name it's given is always wlan0mon no matter what wlan adapter you choose

Open start-mon.sh in a text editor and search for the following:

```
# Assign default monitor mode interface name
iface0mon='wlan0mon'
```

You can then edit the document to assign the name as you wish:

iface0mon='amisix0mon'

Regards

amisix, 2022-04-08:
Cool, got it, thank you. Doing that and just having one script for each adapter works great - as suggested, looks like I should learn about BASH scripting so I can modify as I'd like. Can you recommend a good resource to start off with?