morrownr/USB-WiFi #68: Help with MediaTek MT7603U

haevalencia, 2022-03-22:
Hello, I am a rookie in using USB WiFi adapters on gnu/linux. I always read that it was recommended to buy MediaTek adapters and review the list of compatible in-kernel drivers so that they work out-of-the-box. My goal is to have a simple backup, since many times laptops come with network cards not native compatible with gnu/linux and not always having a wired network, having a compatible USB adapter is a good idea.

I searched for a trustworthy store on AliExpress where they sell Comfast network components. This company has versions of the same model of WiFi usb adapter (usually V2) with a MediaTek chip instead of Realtek. I purchased the CF-WU825N V2 model with MT7603U chip (WiFi 4). I had previously checked that MT7603 appeared in the list of in-kernel drivers, however, when I tried it with various distributions it did not work (fedora, Ubuntu, Manjaro and OpenSuse). Using `lsusb` it appears listed.

I honestly don't know what steps to follow. I haven't found much information on the internet regarding that particular chip. Maybe I just made the wrong choice for my purposes.

Regardless of the above, do you think it is a good idea to opt for the CF-928AC WiFi 5 (MT7612U+flash) model? since for now I don't see WiFi 6 USB adapters with Mediatek chip adapters, only with Realtek chip.
My goal is to get it to work on a modern gnu/linux distribution out-of-the-box.

Note: I repeat, I am very rookie to this topic. If there is a guide or place where I can contribute with the adapter that I acquired to make it work, I am willing to collaborate.

morrownr, 2022-03-23:
Hi @haevalencia 

> I had previously checked that MT7603 appeared in the list of in-kernel drivers

The list I use for `mt` chipsets is found at the following site:

https://wireless.wiki.kernel.org/en/users/drivers/mediatek

The mt7603e is supported by the in-kernel mt76 driver. However no support for a mt7603u is listed. If you read all of the very long README.md, you will see numerous chipsets and links to specific adapters that are known to work, I am pretty knowledgeable when it comes to usb wifi adapters and Linux and I had no idea that a usb adapter with a mt7603u even existed.

> having a compatible USB adapter is a good idea.

Agree. It is a very good idea.

> I honestly don't know what steps to follow. 

Recommend you send it back and ask them to refund your money. Then we can get down to the business of helping you find an adapter that works well.

> I am very rookie to this topic.

I've been using Linux since '94 and I am a rookie with some topics. 

You said you are looking for a good dependable adapter to use as a backup. Go through the many links in README.md and pick a few and then we can discuss what might work best.

Regards

haevalencia, 2022-03-23:
> The mt7603e is supported by the in-kernel mt76 driver. However no support for a mt7603u is listed. If you read all of the very long README.md, you will see numerous chipsets and links to specific [...]

It was my mistake, I didn't know how to read that information. But I am happy to see that the MT7612U chip is supported.

> 
> Recommend you send it back and ask them to refund your money. Then we can get down to the business of helping you find an adapter that works well.
> 

The seller offered me to keep it for free if I buy another one of their adapters, like the CF-928AC (MT7612U) I mentioned earlier. I read the Readme.md and that chip is listed so I am a little more confident to buy it.

On the other hand, the seller told me that they (Comfast) are working on a model with an alternative MediaTek chip to CF-957AX WiFi 6 (RTL8832AU) for the second half or early next year. He didn't tell me which one, but maybe it's the MT7921u I read in the Readme.md

> 
> I've been using Linux since '94 and I am a rookie with some topics.
> 
> You said you are looking for a good dependable adapter to use as a backup. Go through the many links in README.md and pick a few and then we can discuss what might work best.
> 
> Regards

Thank you very much for answering. You were very kind.

morrownr, 2022-03-24:
Hi @haevalencia 

> CF-928AC

Would you mind sending me the link to the seller website for this product. I would like to look it over before you order as sellers often provide bad information.

> On the other hand, the seller told me that they (Comfast) are working on a model with an alternative MediaTek chip to CF-957AX WiFi 6 (RTL8832AU) for the second half or early next year. He didn't tell me which one,

The chipset that the seller is probably talking about is the mt7921u. We have seen no evidence that Mediatek has released the mt7921u chipset yet but the driver is currently going into the Linux kernel. Products starting to be available in the second half of this year is consistence with what we know. 

I worked on the driver source released by Realtek for the 8832/52 chipset. It is the worst driver I have ever seen. My advice to Linux users is to avoid any USB WiFi adapters based on that Realtek chipset. It is best if you want WiFi 6 to wait for the adapters based on the mt7921u.

Regards

haevalencia, 2022-03-24:
Hello again @morrownr 

> > CF-928AC
> 
> Would you mind sending me the link to the seller website for this product. I would like to look it over before you order as sellers often provide bad information.
> 

Of course. For this particular product I only found the link on their official Comfast website in Chinese [here](http://www.comfast.cn/index.php?m=content&c=index&a=show&catid=13&id=135), however, on the AliExpress site (and others that sell that product like comfastwifi.us) there is the same information, but in English, like [here](https://www.aliexpress.com/item/1005003468052833.html?gatewayAdapt=glo2esp&spm=a2g0o.cart.0.0.30643c000JLIa4&mp=1).

[image]

> The chipset that the seller is probably talking about is the mt7921u. We have seen no evidence that Mediatek has released the mt7921u chipset yet but the driver is currently going into the Linux [...]

I would like to wait for this product, but when it is released it will be expensive and it will be low in stock for a while. But I like to know that there is already an in-kernel driver for this MediaTek chip.

> I worked on the driver source released by Realtek for the 8832/52 chipset. It is the worst driver I have ever seen. My advice to Linux users is to avoid any USB WiFi adapters based on that Realtek chipset. It is best if you want WiFi 6 to wait for the adapters based on the mt7921u.
> 

I agree. I already had bad experiences with Realtek chip in WiFi adapters like Ugreen CM492 (RTL8812AU) which I was able to get working, but it was not what I expected. Unfortunately, for another type of device, I have to trust Realtek because it is what I find the most on the market, such as Bluetooth USB, RJ-45 or SD cards adapters.

This same vendor prevented me from buying model CF-B03 (RTL8761BUV) BT 5.1 and told me to safely buy model CF-B02 (RTL8761B) BT 5.0, since he had verified that it worked out-of-the-box in gnu/linux.

Cheers

morrownr, 2022-03-26:
Hi @haevalencia 

Regarding the CF-928AC: I found a link on AliExpress. Interesting adapter. It is based on the mt7612u so should use the in-kernel driver. Note that it says `MT7612U + Flash`. The `Flash` is a dead giveaway that this adapter is a multi-state adapter. It probably won't be an issue but you might want to read the following for a better understanding as some distros mess this up:

https://github.com/morrownr/USB-WiFi/blob/main/How_to_Modeswitch.md

If you order this adapter, I have 2 requests:

1. Ask the person you order from to ask Comfast to start putting a small slider switch on their multi-state adapters that will allow us Linux users to turn multi-state off.

2. Please give us a good report on the adapter so that I can add it to the list of adapters.

Thanks

haevalencia, 2022-03-27:
> Hi @haevalencia
>
> Regarding the CF-928AC: I found a link on AliExpress. Interesting adapter. It is based on the mt7612u so should use the in-kernel driver. Note that it says `MT7612U + Flash`. The [...]

I understand, thank you for clarifying this point. I'm really looking for something that doesn't involve any complications. I talked to the seller and he told me that he actually uses the "zeroCD" system and that they began to implement that system due to user criticism who said that including a CD with drivers is from the past decade. He told me that the first production run of the model CF-928AC didn't use this system and there was an alternative non-clamshell model similar to this new model's (http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=182) casing that also didn't use the zeroCD system. It's a shame.

Anyway he told me he'd stock up if he found one (there's a special Comfast anniversary sale this week)

He pointed me to other models available with a MediaTek chip, but they use the same one that doesn't have in-kernel drivers. He kindly told me that he would let me know if he found an older model with WiFi 5 support that didn't use the zeroCD system (He mentioned the MT7663 chip).

See:
https://es.aliexpress.com/item/1005003951477208.html?gatewayAdapt=glo2esp&spm=a2g0s.imconversation.0.0.1d813e5f73ySQU

This is the one I had bought:
https://es.aliexpress.com/item/4000256060286.html?gatewayAdapt=glo2esp&spm=a2g0o.order_list.0.0.21ef194dV6IvJC

> If you order this adapter, I have 2 requests:
>
>     1. Ask the person you order from to ask Comfast to start putting a small slider switch on their multi-state adapters that will allow us Linux users [...]

I am currently hesitating to buy it. I think I'll go with the generic model I saw mentioned in the Readme.md which uses a case similar to the "Panda Wireless PAU07" model. However, as I have a good relationship with the vendor, I'll push for the products to be more open, in in the sense that although they do not offer direct support to gnu/linux, they do not include methods that hinder its operation, such as the ZeroCD tool, or as you indicate, that they include a switch to be able to deactivate it.

morrownr, 2022-03-27:
Hi @haevalencia 

> He kindly told me that he would let me know if he found an older model with WiFi 5 support that didn't use the zeroCD system (He mentioned the MT7663 chip).

You don't want the mt7663. While that is an AC1200 chipset, it was only designed as USB2 capable. Sometimes chipsets that show USB support, actually don't mean USB as in USB WiFi adapter. USB support can be used internally as well. I think that was the primary purpose of the mt7663.

This is all very complicated which is why I don't just throw out a list of supported chipsets. People actually needs links to specific products that are known to work well. I don't just throw links to adapters up in the README.md. The links I add have to pass my quality assurance measures that are aimed at helping Linux users find adapters that work well and are as trouble free as possible.

> I think I'll go with the generic model I saw mentioned in the Readme.md which uses a case similar to the "Panda Wireless PAU07" model.

Are you talking about the following adapter?

https://www.aliexpress.com/item/4000048659616.html

I don't have first hand experience with it but do have a link to an article posted:

https://wlan-pi.github.io/wlanpi-documentation/admin/cf912_issues/

I can't find any negative comments anywhere and the above article spoke highly of this little `New Arrival Black MT7612U Adapter". If you decide to go with this adapter, can you write up a report that I can post to help others?

Regards

haevalencia, 2022-03-28:
> You don't want the mt7663. While that is an AC1200 chipset, it was only designed as USB2 capable. Sometimes chipsets that show USB support, actually don't mean USB as in USB WiFi adapter. USB support can be used internally as well. I think that was the primary purpose of the mt7663.

Ok, thank you very much for the information.

> Are you talking about the following adapter?
>
> https://www.aliexpress.com/item/4000048659616.html
>
> I don't have first hand experience with it but do have a link to an article posted: [...]

Yes, I will buy that model. I saw several posts and checked the comments. Several say it works fine on gnu/linux and even macOS. I think it is the most sensible thing to do and thus avoid any additional complications. I have my doubts about the quality, but I'm going to try it anyway.

Of course when I have it, I will test and write about the product.

ChalesYu, 2023-04-07:
about mt7603u , can try a modified vendor driver.
https://github.com/ChalesYu/buildroot_platform_hardware_wifi_mtk_drivers_mt7603

And , let `mt76`  driver support mt7603u is possible ,  but it need some people to read the vendor code , find the different between `USB` and `PCI` interface. Then patch mt76 driver (add usb interface init code , modify some code , like load firmware part. but handle mt7603 part seems can use the same code  .) , anyway it seems need lot's of time.