morrownr/USB-WiFi #70: Need Your Help: We need to let Comfast know what we think about Multi-state adapters...

morrownr, 2022-03-26:
ALL,

Comfast has been making many adapters with mt7612u chipsets recently and it appears that they have plans to release new adapters based on the new mt7921u during the second half of this year. Almost all Comfast adapters that use the mt7612u chipset are released as multi-state adapters. See the following link for more info:

https://github.com/morrownr/USB-WiFi/blob/main/How_to_Modeswitch.md

While usb_modeswitch is a good utility, I continue to see distros messing up upgrading its data file which causes probablems and users do not know what to do. They assume they need to install a driver but that is not the problem.

My opinion is that we need Comfast and other usb wifi adapter makers to add a small slider switch that allows us Linux users to turn the multi-state support off.

I am asking everyone that see this message to spread the message and send emails to Comfast. Here are a couple of email addresses but I am sure others can be found. If you find other Comfast support or sales email addresses. Please post them.

Comfast Wifi
Retail & wholesale: sale@comfastwifi.us
Technical Support: info@comfastwifi.us

My suggested message goes like this:

Dear Comfast,

As a Linux user, I appreciate Comfast selling adapters based on Mediatek chipsets as the modern Mediatek chipset are well supported with Linux in-kernel drivers. However, the reason I am writing is to request that any future usb wifi adapters products include a small slider switch that will allow users to deactivate the onboard FLASH that contains the Windows driver. Multi-state (the ones with FLASH) adapters may be good for Windows users but are not good for non-Windows users. Non-Windows users need a simple way to deactivate the FLASH!

Linux users are looking forward to future usb wifi products that use the mt7921u chipset but please, give us the capability to turn the FLASH off.

Regards

Darkyere, 2022-03-28:
I have send the mail for Comfast as requested.
This is of interest for all linux user's.

amisix, 2022-03-29:
>This is of interest for all linux users.
Agreed, I've also e-mailed them.

sagar-chandarana, 2022-04-03:
sent!

morrownr, 2022-04-18:
Thanks for those emailing Comfast. Comfast seems to be the primary maker that has decided multi-state adapters are a good thing. This post is going to stay up for the long term and hopefully, over the long term we can get enough folks to email Comfast that we can get some changes made. 

In my opinion, if multi-state is working for their Windows users, great but they need to know that we in the Linux community are being hurt by what they are doing. A simple slider switch to turn multi-state off would work well for us.

Regards

haevalencia, 2022-07-09:
They sent me the link late. I don't know when it was officially released, but Comfast already has -at least- 2 models of adapters with the new MT7921AU chip. See https://es.aliexpress.com/item/1005004428702222.html or http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=209

The information is not clear, but it does not indicate that it uses the ZeroCD system like previous MediaTek models.

morrownr, 2022-07-09:
Hi @haevalencia 

Thanks for the info. I will be making this info available as these 2 adapters appear to be the first available.

> They sent me the link late. I don't know when it was officially released, but Comfast already has -at least- 2 models of adapters with the new MT7921AU chip. See [...]

I took a look at your links and also dug into the main ComFast site for more info. It may not say `ZeroCD` but I will give your about a 99% chance of both being multi-state as there is no mention of a link to download the driver, only Windows 10/11 is officially supported and no cd is included. I would like to be wrong. The good part is that ComFast seems to use the same ZeroCD method on all of their adapters which means modern mainline Linux distros are already setup to handle this.

I am hoping that we see additional adapters based on the mt7921au (and possibly others) from other makers soon. ALFA has never made a multi-state adapter and I specifically asked them not to start now.

Thanks for the info.

Regards

haevalencia, 2022-07-10:
> I took a look at your links and also dug into the main ComFast site for more info. It may not say ZeroCD but I will give you about a 99% chance of both being multi-state as there is no mention of a link to download the driver, only Windows 10/11 is officially supported and no cd is included. I would like to be wrong. The good part is that ComFast seems to use the same ZeroCD method on all of their adapters which means modern mainline Linux distros are already setup to handle this.

They always offer download links for all of their products regardless of whether they use the ZeroCD system or not. Curiously, they only use that system for MediaTek chips, since for Realtek chips they include a CD with the drivers.

I found out that they use the ZeroCD system by reading the manual (in the "Product related" tab in the product information). In that same section they offer driver downloads. Apparently this product exists since March, but it was released recently.

[image]

> I am hoping that we see additional adapters based on the mt7921au (and possibly others) from other makers soon. ALFA has never made a multi-state adapter and I specifically asked them not to start now.

I was able to verify that there are several Chinese brands that will start working on adapters with that chip. I also hope for a good adapter that is not multi-state.

morrownr, 2022-07-10:
> I was able to verify that there are several Chinese brands that will start working on adapters with that chip. I also hope for a good adapter that is not multi-state.

Thanks for the info. I'm hoping to upgrade my main wifi router to WiFi 6 with a good mt7921xu usb adapter so I will keep an eye open. I sure would like to know what ALFA is up to as they have a long track record of good quality, long range, single state adapters. If you see anything else, let us know.

haevalencia, 2022-07-11:
> Thanks for the info. I'm hoping to upgrade my main wifi router to WiFi 6 with a good mt7921xu usb adapter so I will keep an eye open. I sure would like to know what ALFA is up to as they have a long [...]

I have seen that many mention the Alfa brand, however, the type of adapters that I usually use (indoor/portable) only use Realtek chips and not Mediatek, such as Alfa AWUS036EAC model and it is a shame. 

For now, Lenovo support confirmed me that it no longer uses wireless cards whitelist in Bios settings and I will be able to change my broken wireless card for a new one (CF-AX210-M based on Intel AX210NGW) and I will be able to enjoy the speed of WiFi 6 in gnu/linux. Anyway I will wait for a good USB WiFi 6 adapter.

Thank you for all of your work.

morrownr, 2022-07-11:
@haevalencia 

> I will be able to change my broken wireless card for a new one (CF-AX210-M based on Intel AX210NGW) and I will be able to enjoy the speed of WiFi 6 in gnu/linux. 

My opinion of Intel is that they usually do a good job with drivers in Linux but they are not doing a good job of supporting AP mode with their AX-210 drivers. If you are at all interesting in 5 or 6 GHz AP mode, you really want to stay away from Intel, at least for now.

There is one thread here where a guy was looking to increase his throughput and he ended up going with the following PCIe card and was very happy with it.

http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=188

It supports 6 GHz via the mt7921k chipset. He seemed to be very pleased. 

I have a new laptop that has a little card using the mt7921 chipset. I use Linux Lite 6 with kernel 5.15 and that card just works in client and monitor mode. AP mode support was just added in kernel 5.18. I am not up to speed on what those little laptop cards are called... M.2 or so but my point is that Mediatek seems to be kicking Intel's backside when it comes to full feature wifi drivers these days and the little cards that use the mt7921 and mt7921k cards are available at low prices.

haevalencia, 2022-07-27:
> I have a new laptop that has a little card using the mt7921 chipset. I use Linux Lite 6 with kernel 5.15 and that card just works in client and monitor mode. AP mode support was just added in kernel [...]

Hello. I really didn't have many options. On the one hand, Lenovo's ThinkPad computers used to have a white list in the bios of network cards that could be used for security reasons, so going to a newer generation of Intel was safer, also in the Lenovo forums other users had already tried it successfully, on the other hand, in my company it is a requirement to have support for Intel VPro.

The truth is that in the first place I was going to buy a [MediaTek MT7921K card from Comfast](http://www.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=160), but I read that there were several problems with the Kernerl drivers and Linux users in forums did not recommend buying it. Maybe that changed quickly, but the impression I got was that the support was not very good.

morrownr, 2022-07-27:
> The truth is that in the first place I was going to buy a MediaTek MT7921K card from Comfast, but I read that there were several problems with the Kernerl drivers and Linux users in forums did not [...]

Mediatek bought Ralink around 2014 in an effort increase their presence in wifi. I can take a while to digest the purchase of a company and Mediatek was not that large of a company anyway. If you read messages about adapters that use the mt7612u chipset from 2015-2017, you see complaints. The in-kernel driver went in during 2018 and continues to be improved. It is a good chipset with plug and play performance and has been for some time now. My opinion of how things have transpired with Mediatek is that it took time to determine what to do and how to best do it. One thing they figured out is that you can't support Linux with driver like you can windows. It won't work because the development model of Linux is such that development is continuous. Drivers have to be regularly updated and the only way for that to happen in way that works for users is the the changes to happen in-kernel. This process and the standards that need to be met are followed very closely by Mediatek in this day in time. It takes time to build the expertise. New usb adapters based on the mt7921au chipset, WiFi 6, were just released earlier this month. For the first time in the history of usb wifi in Linux, driver support, mt7921u, was in the mainline kernel before the hardware was available to buy. That does not mean everyone could just plug a new adapter in see plug and play results. Being in a mainline kernel does not mean folks are using the kernel yet. Ubuntu 22.10, due in Oct., will be the first Ubuntu release where users won't have to upgrade kernel and firmware. If you look at my list of usb wifi adapters, you will see that I go into detail as to what the status is.

Now, support for internal wifi cards based on the mt7921 and mt7921k has been in-kernel since 5.12. Were there Linux users that had problems when the hardware first came out? Probably. If you new laptop was using kernel 5.10, you had to upgrade to kernel 5.12 for the adapter to work. Distro maintainers have instructions for upgrading to more modern kernels. Some people are not technical in nature and probably should stick to their iPhones. I mostly keep up with USB WiFi so my expertise is there. My laptop with the mt7921 chipset is only a few months old. I installed Ubuntu 22.04 on it and managed mode/monitor mode both worked fine. I don't use AP mode on that computer but if I did, I would know that I need to get a later kernel and firmware... which will be in Ubuntu 22.10.

I would rank overall quality of wifi drivers for Linux in this order in this day in time:

Mediatek
Qualcomm
Intel
Everybody else
Realtek

If you have any specific questions for a guy that has a mt7921 card and knows what he is doing, let me know.

Regards

haevalencia, 2022-07-27:
> 
> If you have any specific questions for a guy that has a mt7921 card and knows what he is doing, let me know.
> 
> Regards

Thank you for always sharing your knowledge on this topic. After a quick search, I saw that Intel engineers limit the use of the 5 and 6 Ghz bands based on whether or not they are allowed in certain countries (In my country the use of these bands for WiFi is approved), using a somewhat obscure method. I don't usually use AP mode from the laptop, but it seems absurd to me. I think I will buy a Mediatek mt7921K based M.2 NGFF WiFi module. It only remains for me to test if it will work or not on my ThinkPad.