morrownr/USB-WiFi #71: The promised report on AX181 Plus WiFi PCI-E Card

Darkyere, 2022-03-28:
This as promised is the report on the "AX181 Plus" WiFi PCE-E card as discussed in

[Advice on a good range and speed USB WiFi dongle](https://github.com/morrownr/USB-WiFi/issues/55)

This is only the first attempt at making it work, in short it worked out of the box.
But i will continue the test's when Ubuntu 22.04 comes out. 
The kernel 5.17 had to many unavailable dependencies for Ubuntu 20.04.
And i might not be the best to make these drasting changes to the system to make kernel 5.17 work.

More info about the process down below.

If u like u can skip to the bottom part called

----
# The last attempt
----

To skip the long chatter and just look at the working part

----

I got the WiFi Card AX181 Plus on the 21-03-2022
But at first i didnt have time to look at it. 

When i eventually got down to look at it. I unpackaget it and put it in the PC i wanted for my BackUp server.
Please note that this PC is still located at my own home since i got no way to route Internet down into my parent's basement.

At a first i noticed a WiFi device from the `iwconfig` command called `wlp4s0`
I decided that i wouldten try to connect it to the internet without the 5.17 kernel.
Since i didn't see any reason to try it out before i got a kernel that could run it proberly.

So at first i tried to install the kernel, 
but wasnt documenting it at the first because i didn't think it was nessecary.

Later i tried to retract my steps and document the past and the present process.
So i might have missed some steps or info in the progress.

----

I first followed the part from

[How to Install the Latest Mainline Linux Kernel Version in Ubuntu [GUI and Terminal Methods]](https://itsfoss.com/upgrade-linux-kernel-ubuntu/)

About during it in commandline

I downloaded the four described kernel headers, image and modules

Then proceeded with

`dpkg install *.deb`

It failed with dependency missing for kernel headers,
and lead to an unbootable system

I then followed the uninstall of the kernel 5.17

----

Proceeded to install ukuu, the GUI way to install an ubuntu kernel.

But it only allowed up to kernel 5.7 since its end of life and cost's money for the newer version

----

I found a ppa for an Ubuntu kernel installer, that is a fork of "ukuu" called "mainline"

`sudo add-apt-repository ppa:cappelikan/ppa`

But the first attempt failed because i have an nvidia gtx 645,
and the proprietary nvidia driver does not support kernel 5.17 "yet".

So again i was left with an unbootable system and had to remove kernel 5.17

----

I then decided that it was worth an attempt to use the opensource nouveau driver
So i used the ubuntu "Software & Updates" to revert to the nouveau driver.

Reboot and then follow up on the "mainline" app and install 5.17

This time i was met with a dependency issue again for kernel headers that it needed 

`libssl3`

i took the chance and rebooted, got in on kernel 5.17

But now i funny enough had a problem that my browser Brave (Chromium based) and firefox ESR,
froze the entire system every time i tried to open it

I figured it was because of the missing nvidia driver, and some problem with the nvidia nouveau driver.

At the same time the "AX181 Plus" allso frooze the entire system upon attempting to connect to my Router.

----

I rebooted went back to 5.13 removed kernel 5.17 and proceeded to install the missing dependencies for the 5.17 kernel headers. Figuring this was the culbrit for it all.

I found a page on stackexchange explaining how to install libssl3 and its dependencies

[How to use 5.16 kernel with Ubuntu 21.10?](https://unix.stackexchange.com/questions/685830/how-to-use-5-16-kernel-with-ubuntu-21-10)

So after the install i went through the logs and found that (believe it or not it continues) that dkms doesnt support the kernel :D
So a lot of dkms errors popped up

I probably should not have reboot at this time but took the chance and rebooted again but this time the initramfs was not available and i was met with an unbootable system again.

----

Booting back in kernel 5.13 i learned i couldten remove, install or make any changes to the system in any way 
because of how i installed the `libssl3` and its dependencies

So this time i felt i was left with no other choice than revert my changes because of the install of `libssl3` and its dependencies, that had left me with to many added and removed changes to the system.

Luckely enough i am on BTRFS with TimeShift installed.
So i reverted back to the snapshot i made before i started messing around.

With the system up and running again,
and the knowledge that all dependecies for kernel 5.17 and `libssl3` etc. is in Ubuntu 22.04

I decided that i'm gonna wait to try and install kernel 5.17 again until 22.04 LTS is official released and give it a go then.

----
# The last attempt
----

Now comes the best part, i decided heck i cant make this much worse just by trying out the "wlp4s0" on kernel 5.13.
And during my many reboots i had noticed in dmesg that kmod mt7921e was loaded on kernel 5.13.

So i connected to the 5 Ghz network on my home router,
Got a connection speed from "iwconfig" of "780mbps"

Ran the speedtest.net cli tool and got

```
Speedtest by Ookla

     Server:
        ISP:
    Latency:     2.62 ms   (0.08 ms jitter)
   Download:   371.28 Mbps (data used: 397.3 MB )                               
     Upload:   600.90 Mbps (data used: 680.3 MB )                               
Packet Loss:     0.0%
```

----

So until now i suppose this is what i got.

It's not the worst start to actually get a connection on kernel 5.13.
And to get a speed of 371 down on an unsupported kernel wasnt the worst scenario either.
But please note the PC is located 2 meters unopstructed from the routher atm.
This wont be the same scenario at my parents place since it will be a floor of wood and a brick wall in between.

I am now waiting eagerly for Ubuntu 22.04 to be released.
Then i will upgrade to 22.04 and make sure to make a BTRFS snapshot.

And proceed with the kernel 5.17 install.

I just have to check if it is possible for me to get a better connection and/or speed with kernel 5.17 than on 5.13.
Especially since it loaded "mt7921e" and not "mt7921k" in kernel 5.13.

Best regards,
Darkyere

morrownr, 2022-03-28:
Hi @Darkyere 

Thanks for the report and please accept my apology.

In my enthusiasm to help, I think I ended up causing you a lot of unnecessary work. Networking cards and adapters have pci and usb IDs that are used by drivers to see if they support the hardware. I incorrectly assumed that this AX181 used a pci ID that only went into the kernel somewhat recently which would have meant that your 5.13 kernel driver would not have recognized and supported the AX181. We now know that is not the case and the AX181 uses a pci ID that is supported in kernels 5.12 and later. When Ubuntu 22.04 is released with kernel 5.15, the AX181 should just work (for your purposes.)

Now, when I say `for your purposes` above, what I mean is dual band client mode operation. Remember that the AX181 is actually a tri-band card but that doesn't matter until you have a 6 GHz capable wifi router/AP and I would not get in a hurry to buy one until things have matured. After seeing your performance numbers, you might be close enough already that we can tweak it to get what you need.

Can I get you to post the results of:

```
$ sudo iw phy phy0 info
```

...assuming that your new card is phy0.

We can take a look at signal strength with

```
wavemon
```

Regards

Darkyere, 2022-04-01:
Sry for my delay havent been at home until now.

This is the output of the 2 commands.

----

The output of `sudo iw phy phy0 info` is 

```
Wiphy phy0
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: Station
			HE MAC Capabilities (0x08031a000040):
				+HTC HE Supported
				TWT Requester
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x22700c120000b006423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
[... 300 lines cut ...]
│link quality: 100%  (70/70)                                                                                                                                                                │
│===========================================================================================================================================================================================│
│                                                                                                                                                                                           │
│                                                                                                                                                                                           │
│signal level: -38 dBm (0.16 uW)                                                                                                                                                            │
│================================================================================================================================                                                           │
│                                                                                                                                                                                           │
├─Statistics────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│RX: 7,357 (1.62 MiB), drop: 4 (0.1%)                                                                                                                                                       │
│TX: 11,410 (1.47 MiB)                                                                                                                                                                      │
├─Info──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│mode: Managed, connected to: A0:63:91:C9:3A:61, time: 7:56m, inactive: 0.6s                                                                                                                │
│freq: 5500 MHz, ctr1: 5530 MHz, channel: 100 (width: 80 MHz)                                                                                                                               │
│rx rate: 866.7 Mbit/s VHT-MCS 9 80MHz short GI VHT-NSS 2, tx rate: 780.0 Mbit/s VHT-MCS 9 80MHz VHT-NSS 2VHT-NSS 2                                                                         │
│beacons: 2,319, avg sig: -38 dBm, interval: 0.2s, DTIM: 2                                                                                                                                  │
│power mgt: on,  tx-power: 3 dBm (2.00 mW)                                                                                                                                                  │
│retry: short limit 7,  rts/cts: off,  frag: off                                                                                                                                            │
├─Network───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│wlp4s0 (UP RUNNING BROADCAST MULTICAST)                                                                                                                                                    │
│mac: A8:93:4A:DF:E4:25, qlen: 1000                                                                                                                                                         │
│ip: 192.168.1.11/24                                                                                                                                                                        │
└───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```

Hope this is what is needed :)

Darkyere, 2022-04-12:
A little update on the AX181 Plus WiFi PCI-E Card.

I have now moved the PC to my old folks house.
And set it up in the basement.

Thought i just wanted to add the latest test of this device with +-10 meters distance with a floor and a brick wall between.

----

This is from wavemon

```
┌─Interface────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
│wlp7s0 (IEEE 802.11), phy 0, reg: DK (DFS-ETSI), SSID: WiFimodem-C32A                                                                                                                                                                     │
├─Levels───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│                                                                                                                                                                                                                                          │
│link quality: 60%  (42/70)                                                                                                                                                                                                                │
│============================================================================================================================================                                                                                              │
│                                                                                                                                                                                                                                          │
│                                                                                                                                                                                                                                          │
│signal level: -68 dBm (0.16 nW)                                                                                                                                                                                                           │
│===================================================================================                                                                                                                                                       │
│                                                                                                                                                                                                                                          │
├─Statistics───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│RX: 11,532 (3.24 MiB), drop: 11 (0.1%)                                                                                                                                                                                                    │
│TX: 145,819 (63.16 MiB)                                                                                                                                                                                                                   │
├─Info─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│mode: Managed, connected to: 34:53:D2:9B:C3:30, time: 7:11m, inactive: 2.7s                                                                                                                                                               │
│freq: 5280 MHz, ctr1: 5290 MHz, channel: 56 (width: 80 MHz)                                                                                                                                                                               │
│rx rate: 780.0 Mbit/s VHT-MCS 9 80MHz VHT-NSS 2, tx rate: 585.0 Mbit/s VHT-MCS 7 80MHz VHT-NSS 2VHT-NSS 2                                                                                                                                 │
│beacons: 4,080, avg sig: -67 dBm, interval: 0.1s, DTIM: 1                                                                                                                                                                                 │
│power mgt: on,  tx-power: 3 dBm (2.00 mW)                                                                                                                                                                                                 │
│retry: short limit 7,  rts/cts: off,  frag: off                                                                                                                                                                                           │
├─Network───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────── [... cut ...]
     Upload:    57.90 Mbps (data used: 50.3 MB )                               
Packet Loss: Not available.
```

----

Of course i was hoping for better.

But with all the obstructions and without having a choice in the matter of the router.
I dont think this is too bad a speed after all :)

I probably wont in anyway get a better result no matter how many WiFi USB's or PCI-E Cards i should end up testing.

All in all i am pretty satisfied  with the result.

Hope all fares well.
Best Regards,
Darkyere

morrownr, 2022-04-12:
Hi @Darkyere 

Thanks for the report.

> signal level: -68 dBm (0.16 nW)

This tells me the wall is taking a toll on your signal. You can usually get full speed with a reading no higher than -50/=52. A reading higher than -70 will usually give poor results and you are right on the edge. I'm actually impressed with how well it is doing with a reading of -68.

Comments: Antenna position can be very important. While running wavemon, you might want to reposition your antenna and the wifi router anteena to different locations to see if you can improve results.

Also, you are a prime candidate to use a directional antenna. I can help you explore what might be best for your setup. Directional antennas focus the signal in one direction. They are not as expensive as we might think.

Overall, we knew the wall might present a challenge but the battle is not over.

Nick

Darkyere, 2022-04-13:
This sounds like some good choices.

I did think about trying to move the anteanna around but setting it all up and getting it to work i decided that i would wait playing around with that.

A directional antenna might be a good idea its worth a shot atleast.

I wont be home at my parents again until Friday.
So i cant test the location of the antenna until then.

Darkyere, 2022-04-17:
Tried moving the antenna around to the locations it was possible to reach.

It seemed to change between 68 and 75.

The original placement seemed to stay between 68 and 72.

It's not because it's the best location to place it that i choose it.
It was just the highest location i could reach, that almost touch the ceiling.

Those directional antennas.
Do u have an idea of which might be a good option ?

morrownr, 2022-04-18:
As funny as it sounds, I would first do an internet search for `using aluminum foil to improve wifi signal` and see if you come across an idea that helps.

amisix, 2022-04-19:
@Darkyere [Alfa makes a 10dBi flat panel antenna](https://www.amazon.com/Alfa-APA-M25-directional-connector-WL-ANT-157/dp/B00R1PA9EO/ref=sr_1_1_sspa?keywords=alfa+directional+antenna&qid=1650330427&sprefix=alfa+direction%2Caps%2C140&sr=8-1-spons&psc=1&smid=A2LM6ZPY06LT1N&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExTU9DS01OSDFEMllWJmVuY3J5cHRlZElkPUExMDA2ODM4MURWRTVMSEczVEtUSSZlbmNyeXB0ZWRBZElkPUEwMTkyMjgxMzA1R0U5RzRTQ1JCUSZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=). I haven't used it but I own their older 2.4ghz model and it works great.

I like @morrownr's suggestion... it's practically free, free solutions are good. Do you like Pringles?

Darkyere, 2022-05-21:
I got some time to sit down at play with the Aluminium Foil wall enhancer.
And it worked like a charm.

----

```
┌─Interface────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐
│wlp7s0 (IEEE 802.11), phy 0, reg: DK (DFS-ETSI), SSID: WiFimodem-C32A                                                                                                                                                                     │
├─Levels───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│                                                                                                                                                                                                                                          │
│link quality: 63%  (44/70)                                                                                                                                                                                                                │
│===================================================================================================================================================                                                                                       │
│                                                                                                                                                                                                                                          │
│                                                                                                                                                                                                                                          │
│signal level: -66 dBm (0.25 nW)                                                                                                                                                                                                           │
│========================================================================================                                                                                                                                                  │
│                                                                                                                                                                                                                                          │
├─Statistics───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│RX: 23,467 (12.89 MiB), drop: 6 (0.0%)                                                                                                                                                                                                    │
│TX: 634,233 (253.74 MiB)                                                                                                                                                                                                                  │
├─Info─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
│mode: Managed, connected to: 34:53:D2:9B:C3:30, time: 11:46m, inactive: 2.1s                                                                                                                                                              │
│freq: 5280 MHz, ctr1: 5290 MHz, channel: 56 (width: 80 MHz)                                                                                                                                                                               │
│rx rate: 650.0 Mbit/s VHT-MCS 7 80MHz short GI VHT-NSS 2, tx rate: 650.0 Mbit/s VHT-MCS 7 80MHz short GI VHT-NSS 2                                                                                                                        │
│beacons: 6,477, avg sig: -66 dBm, interval: 0.1s, DTIM: 1                                                                                                                                                                                 │
│power mgt: on,  tx-power: 3 dBm (2.00 mW)                                                                                                                                                                                                 │
│retry: short limit 7,  rts/cts: off,  frag: off                                                                                                                                                                                           │
├─Network──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┤
[... cut ...]
   Speedtest by Ookla

     Server: Kviknet - Taastrup (id = 9891)
        ISP: TDC Danmark
    Latency:    20.15 ms   (2.34 ms jitter)
   Download:   495.24 Mbps (data used: 442.3 MB )                               
     Upload:    57.91 Mbps (data used: 55.5 MB )                               
Packet Loss:     0.0%
```
----

This is so darn perfect.
Ty all for helping me get to this.

Now i should have better speed than what my pi can actually send, so it cant get any better.

Best regards,
Darkyere

morrownr, 2022-05-22:
Hi @Darkyere 

> Download:   495.24 Mbps (data used: 442.3 MB )

That is impressive. I was skeptical given the wall but you put it all together and made it happen. Good stuff.

Nick