morrownr/USB-WiFi #73: Great work man !

SoniaMalki, 2022-04-25:
I am so happy I found your GitHub today ! And I am so happy and thrilled because you just updated it two hours ago ! 

I am intensively searching for an wifi card that supports active monitor mode, but I have the entire internet and no response at all ! I have thought this feature don’t exist because I couldn’t find it anywhere. And then you updated your GitHub and you gave me hope.

I was modifying  kernel code just in a hope to understand of this sorcery works, it was my first time going this deep into Linux. 

Can we chat via discord ? An issue is not the best thing to do for this. 
I will read your GitHub carefully and I hope I will find an adapter that finally works. Greats hope there !

SoniaMalki, 2022-04-25:
Hi @morrownr !

"active monitor mode" is a functionnaly that is not just monitor mode. For OWL, it is mentionned in git that the adaptater must have this functionnality. (see there: https://github.com/seemoo-lab/owl/issues/9). To see if it supports it, when we do "iw list", there will be a listing, and a line mentionning it. 
For example in your repo: 7212u, in the output of "iw list", we can see this : 

```
Alfa AWUS036ACM Technical Information

$ iw list
Wiphy phy0
	max # scan SSIDs: 4
	max scan IEs length: 2243 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
   	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
     		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		Bitrates (non-HT):
			* 1.0 Mbps (short preamble supported)
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
[... 178 lines cut ...]
* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
* [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
* [ DEL_IBSS_STA ]: deletion of IBSS station support
* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

> Please tell me what distro and kernel you are using.
Ubuntu 20.04 with kernel 5.13.0-40-generic

> Can you provide the link to opendrop that you are using so that I can read about?

https://github.com/seemoo-lab/opendrop
OpenDrop uses OWL to do the transfers, and I am stuck in the OWL part so :
https://github.com/seemoo-lab/owl

> chipset - Atheros ar9271 [2] - N150 - USB 2
I didn't know it is equivalent to ath9k, I was searching in this page : https://wireless.wiki.kernel.org/en/users/drivers/ath9k/products/external but couldn't find a product available in my country.

> Did you look in the folder called iw_list? I can add more if need be

Sorry I missed it ! Yep I am interested to have a more complete list if possible

Thank you !

SoniaMalki, 2022-04-25:
More info on "active monitor mode", I saw that in monitor mode we can put flag with the command 
`sudo iw <device> set monitor <flag> `
The different flag available are: 
- none
- fcsfail
- plcpfail
- control
- otherbss
- cook
- active
(source: https://www.kernel.org/doc/html/v4.12/driver-api/80211/cfg80211.html#c.monitor_flags & https://wireless.wiki.kernel.org/en/users/documentation/iw)

Devices that don't support "active monitor" have an error when trying to put the active flag "iw command failed operation not supported (-95)".

So that's why I am searching for the support for this active monitor mode, hope it is more clear!

morrownr, 2022-04-25:
Now things are clearing up. Can't say that I have used that.

You might want to go to the following repo:

https://github.com/morrownr/Monitor_Mode

The `start-mon.sh` script could be used to speed up testing. You can probably understand and learn bash fast. Look for the part that starts monitor mode. There is documentation in the script. I will play with it later.

You did not pass on the `lsusb` of your adapters?

Regards

SoniaMalki, 2022-04-25:
I have to get the usb device, I have 2 pic device and 4 usb one. Are you interested for the pci too ?

SoniaMalki, 2022-04-25:
And also those are all TP-Link devices with Realtek chipsets

morrownr, 2022-04-25:
> Are you interested for the pci too ?

No

I have modified the start-mon.sh script:
[start-mon.sh.tar.gz](https://github.com/morrownr/USB-WiFi/files/8558182/start-mon.sh.tar.gz)

This modified version attempts to set `active` and if unsuccessful, it kicks out with the error code. I tried it on an old adapter and it kicked out with an error. I then tried it with an adapter that uses a mt7612u chipset and it worked like a charm. If you go to README.md and scroll down to the section that has information and links to adapters that use the mt7612u chipset, you can see some examples of usb wifi adapters that use the mt7612u chipset.

I can also test the mt7610u chipset. Look around and tell me what chipsets you want tested.

morrownr, 2022-04-26:
> And also those are all TP-Link devices with Realtek chipsets

Yes, with TP-Link I expect Realtek chipsets. If I know what chipsets they have, I can them. I have an extensive collection of adapters.

Testing so far:

Good tests for `active`: mt7612u, mt7610u, mt7601u

Bad tests for `active`: rt5370, ar9271

morrownr, 2022-04-26:
Additional tests: rtl8812bu and rtl8811cu tested bad. I suspect all Realtek drivers will test bad if those two did. I'm not sure if you are aware but I have repos for Realtek drivers here on this site. These are about as good as it gets but they are not standards compliant and as a result there is a lot of stuff that does not work or does not work right. Linux users are almost always better off getting usb wifi adapters that use in-kernel drivers. If you start at the top of README.md and read all the way to the end, you will learn a lot.

Status:

Good tests for active: mt7612u, mt7610u and mt7601u

Bad tests for active: rt5370, ar9271, rtl8812bu and rtl8811cu

Regards

SoniaMalki, 2022-04-26:
Thank you ! Yeah I read your github and it was very informative on the matter, I will not buy any Realtek chip from now, you framed all the things that annoyed me from my one week experience.

I suspect that the drivers have been written not respecting the standars as you say, and that they ignore some functions like the active monitor for example. This is just a guess on my part, but for the same device in V1 version that has an Atheros chipset, and the device in V2 version with the Realtek chipset, the active monitor disappears. To me this seems strange, I don't think it's the hardware that is limited, but simply the implementation of the drivers that don't include this support. I could be totally wrong, I don't know.

So I searched in the backport site (source: https://backports.wiki.kernel.org/index.php/Main_Page), to read the implementation of the drivers with the backport project which is very interesting, and I started to read the code of the drivers to find where it was wrong. Unfortunately after hours of research, I had trouble finding what test was performed to see if the function was supported or not. But here is my guess, and what I could find. 
In my opinion, I think that the "NL80211_FEATURE_ACTIVE_MONITOR" which is a flag used in the code was imported but never used. There are other names like "nl80211_mntr_flag_active", "MONITOR_FLAG_ACTIVE", "NL80211_MNTR_FLAG_ACTIVE". I think they have either never been used, or not even imported or correctly written in the code. I was reading the drivers from the backport project because they seem to be interesting and up to date. They work for my devices, but they don't include support for active mode. I would really like to understand where this comes from, mostly out of curiosity. 

Unfortunately I have a lot of work to do at the moment so I can't dig any further into this, but I would really like to do it in the future so it can be activated for more drivers. Hopefully I understand where the limitation comes from, is it the driver that doesn't respect the standards or is it a hardware limitation? 

What is your opinion? 

Regards

SoniaMalki, 2022-04-26:
Here is the lsusb for the device TL-WN725N (EU) V3
`Bus 002 Device 004: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter`
for the device Archer T2U PLUS (EU) V1
`Bus 002 Device 005: ID 2357:0120 TP-Link 802.11ac WLAN Adapter `
for the device TL-WN722N (EU) V3
`Bus 002 Device 006: ID 2357:010c TP-Link TL-WN722N v2`
for the device TL-WN821HN (EU) V6.0
Bus 002 Device 007: ID 2357:0107 TP-Link TL-WN821N Version 5 RTL8192EU

SoniaMalki, 2022-04-26:
I have also another question, how can I find a device based on the chipset
For example, if I have this chipset "AR5416 Wireless", that is with the athk9 driver, how can I find a device that have this chipset in it? It is just for learning purpose

morrownr, 2022-04-26:
> for the device Archer T2U PLUS (EU) V1
> Bus 002 Device 005: ID 2357:0120 TP-Link 802.11ac WLAN Adapter

That adapter has the rtl8811au chipset. Good chipset. Below is the driver I maintain for this chipset:

https://github.com/morrownr/8821au-20210708

I can check if `active` works if you want. I doubt it. My experience is that many of the lesser used features just don't work or are not fully implemented on the Realtek out-of-kernel drivers.

Based on my testing, it appears `active` monitor mode support is working for the more modern in-kernel drivers. My recommendation would be for you to look at using an adapter based on a mt7612u chipset if you need something now. We are anticipating new USB WiFi adapters based on the mt7921u (WiFi 6) soon and I suspect you will find `active` monitor mode will work in that driver as well.

>  I would really like to do it in the future so it can be activated for more drivers. Hopefully I understand where the limitation comes from, is it the driver that doesn't respect the standards or is it a hardware limitation?

> What is your opinion?

If you are talking about trying to activate the `active` support in the Realtek out-of-kernel drivers, I think you are wasting your time. Adding or activating lesser used features on the Realtek drivers is problematic. I have had very limited success. Much of the information needed seems to be hidden in the coded firmware files and there is basically no documentation and no way to contact Realtek. I think a far better use of your time is work on in-kernel drivers as there is documentation and people to contact about issues and your can even get your work included in the kernel.

Regards

morrownr, 2022-04-26:
> I have also another question, how can I find a device based on the chipset For example, if I have this chipset "AR5416 Wireless", that is with the athk9 driver, how can I find a device that have this chipset in it? It is just for learning purpose

The same way I maintain the list of adapters and what chipsets they use in the README.md. You have to search. It takes time and you get better with practice.

morrownr, 2022-04-26:
> Thank you

Hopefully I was able to help.
 
> I bought an adapter from your list yesterday and it will arrive tomorrow, I will test it and keep you updated.

Please do let us know how it goes and I am always looking for user reviews that can be included with the information about specific adapters. Do you mind telling me which one you ordered?

Regards

morrownr, 2022-06-08:
Please do share.

Regards

qqm55, 2022-06-13:
Got the adapter and tried it out with opendrop/owl.
Used the script and confirmed said wifi card was able to go into monitoring mode it. Reverted back, used the script to kill the processes so that it would not report a busy error when initializing owl. After that owl was running, then moved onto open drop. For the results which had a future warning can be seen here on this thread #https://github.com/seemoo-lab/opendrop/issues/89

morrownr, 2022-06-15:
It appears that you are running Kali. Is that correct?

Is Kali running on bare metal or in a VM?

I have an ALFA ACM. It has been a solid, very compatible adapter. With that said, you are doing some things I am not familiar with yet.

qqm55, 2022-06-15:
I am on Kali linux but i just checked and its kali-linux-2022.2-live-amd64.iso, which i believe is the live boot version for running on a usb (which i am doing). I am very much a linux newbie but do you think it the live version could be causing that error and bare metal would be better suited?

On second thought I should probably try it with the mentioned distributions on the page, will try either fedora or debian linux.

morrownr, 2022-06-15:
> I am on Kali linux but i just checked and its kali-linux-2022.2-live-amd64.iso, which i believe is the live boot version for running on a usb (which i am doing). I am very much a linux newbie but do you think the live version could be causing that error and bare metal would be better suited?

The best advice for a Linux newbie that I can give is that you find someone that has experience to talk you through some of the things you are trying to do. My recommendation right now is for you to start a new issue and describe what you are trying to do.
 
> On second thought I should probably try it with the mentioned distributions on the page, will try either fedora or debian linux.

I, and others,  can make suggestions but it would help if we better understood what computer you are using and what you are trying to do. Kali is a fine Linux distro but it is probably not the greatest as a starter distro. Something like Linux Mint would probably work better as a starter distro and then, over time, you can try other distros to see which best meets your need:

https://linuxmint.com/

BernH4, 2022-07-21:
Sadly all adapters listed in the repo are external ones.
I searching for a card i can put into my laptop (Mini PCI Express) like the Atheros AR9280.
The atheros chip is very hard to get in my country, does someone know an alternative?

morrownr, 2022-07-22:
I have new laptop I bought about 6 months ago. The little internal wifi card is based on the mt7921 chipset. It works well with Ubuntu 22.04. The driver has been in the Linux kernel since kernel 5.12. I'm no expert on the various types of little internal cards that are available so maybe you can find the one that works for you.

Regards

ulist, 2022-09-15:
I can't walk around with a usb pluged to my phone..
Is there any one to introduce this active monitor mode feature in kernel or modify owl source code to use just monitor mode

morrownr, 2022-09-26:
Hi @SoniaMalki 

I thought of this thread when doing some testing on the new mt7921u driver today. From the `$ iw list`:

> Device supports active monitor (which will ACK incoming frames)

Sabuhi1984, 2022-10-18:
Hello, I have a problem. İ have usb wifi model rtl8811cu. How can I add it to kali linux installed on ssd without internet? please help me . Thank you

morrownr, 2022-10-20:
You did not read the README in the  https://github.com/morrownr/8821cu-20210118  repo.

It is much easier to get internet access to your computer than you might think. Do you have a smartphone and a usb cable?

https://www.makeuseof.com/tag/tethering-use-mobile-internet-pc/

morrownr, 2022-10-20:
opps... the above link was for Windows... here is a better link...

https://www.makeuseof.com/tag/how-to-tether-your-smartphone-in-linux/

PhilDevProg, 2023-02-06:
@SoniaMalki Did it support active monitor mode?

v74863, 2023-11-17:
ZerBea, the author of hcxdumptool, knows ALL about active monitor mode.
Ask him about the best adapters.
https://github.com/ZerBea/hcxdumptool#adapters

c0des1ayr, 2024-09-26:
> Additional tests: rtl8812bu and rtl8811cu tested bad. I suspect all Realtek drivers will test bad if those two did. I'm not sure if you are aware but I have repos for Realtek drivers here on this [...]

FYI, RTL8821CE works with active monitor mode if you use either the `rtw88_8821ce` module or the `rtw_8821ce` module from [the wifi-next repo of the kernel](https://github.com/lwfinger/rtw88).

bmwalters, 2024-11-02:
@c0des1ayr looks like that isn't actually the case: https://github.com/lwfinger/rtw88/issues/236#issuecomment-2425490660

c0des1ayr, 2024-11-06:
> @c0des1ayr looks like that isn't actually the case: [lwfinger/rtw88#236 (comment)](https://github.com/lwfinger/rtw88/issues/236#issuecomment-2425490660)

Yes, I realized that a while ago...seeing as that is the issue I made on the repo.

flame-kaiser007, 2024-11-29:
Hi I'm currently having a Qualcomm QCA9565 intergarted laptop modem on my dell Inspiron 3595 powered by amd a9,
My doubt is can i force it to run monitor mode for pentest?

Please reply