morrownr/USB-WiFi #74: Reorganization of USB-WiFi

morrownr, 2022-04-30:
All,

As this site continues to grow with the addition of more information, it has become obvious that the repo needs to be reorganized in a way to keep things simple and easy to use. To that end, I am currently working toward changing README.md to be a MAIN MENU that provides links to files that provide the information that users are looking for.

If you have suggestions, please let me know.

@morrownr

Bartholomew42, 2022-05-05:
Hi Nick, 
I finished proofreading of the first section, _USB_WiFi_Adapter_Information_for_Linux.md_ . I consider the text well-written. 

I have only two suggestions: 
1.) `If you are going to be traveling with the adapter, the adapters with long antennas may not not work well as the adapter may be broken.` You could add "... as long as one does not want to unscrew and rescrew the antennas, which is inconvenient." But your point is delivered without this, so it's not really needed. 
2.) `WPA-3 SAE support: 2120-09-15` Probably a typing error and you mean 2021-09-15. 

I can proofread the other sections next week. 
Greetings!

morrownr, 2022-05-05:
@Bartholomew42 

I really appreciate you proof reading the material. Don't worry about the other files for now as I am doing some cleaning as I add things to the main menu. I figured I needed to do something before the new mt7921u adapters are available as hopefully there will be a lot to add when they are available.

Cheers