morrownr/USB-WiFi #75: ANDDEAR MT7612U004 Adapter Issue on Raspberry Pi 4B with USB 3.0

amisix, 2022-05-01:
@morrownr - I'm having a strange issue with the ANDDDEAR MT7612U004 adapter that I recently picked up where it won't initialize properly on the USB 3.0 bus of my Raspberry Pi 4B but it will work fine on the USB 2.0 bus (Kali 2022.2). I originally evaluated the adapter with a CM4 and had no issues because, USB 2.0. The issue occurs whether I have a single adapter or multiple adapters connected. 

I think this may impact other users of this adapter if they choose to use it on a Raspberry Pi 4B with USB 3.0. We may need to modify or remove my review depending on the solution.

lsusb freezes/locks up & there is no output 
iw dev output does not show the adapter whatsoever

Error:
[    6.911467] usb 2-1: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[    6.956844] mt76x2u 2-1:1.0: ASIC revision: 76120044
[    7.004924] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.029016] mt76x2u 2-1:1.0: ROM patch build: 20141115060606a
[    7.029426] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Oct  8 2018 09:18:41 version 7.45.154 (nexmon.org: 2.2.2-195-g01e0-1) FWID 01-4fbe0b04
[    7.166674] **proc: Bad value for 'hidepid'**
[    7.225078] mt76x2u 2-1:1.0: Firmware Version: 0.0.00
[    7.230139] mt76x2u 2-1:1.0: Build: 1
[    7.233831] mt76x2u 2-1:1.0: Build Time: 201507311614____
[    7.318155] bcmgenet: Skipping UMAC reset
[    7.323965] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[    7.331680] bcmgenet fd580000.ethernet eth0: Link is Down
[    7.342322] systemd-journald[161]: Oldest entry in /var/log/journal/b919d2f2382e4da0969c18397e43b317/system.journal is older than the configured file retention duration (1month), suggesting rotation.
[    7.360193] systemd-journald[161]: /var/log/journal/b919d2f2382e4da0969c18397e43b317/system.journal: Journal header limits reached or header out-of-date, rotating.
[    8.637094] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[    8.640220] usbcore: registered new interface driver mt76x2u
[    8.663682] **proc: Bad value for 'hidepid'**
[   10.533815] **usb 2-1: USB disconnect, device number 2**
[   10.540887] **mt76x2u 2-1:1.0: error: mt76x02u_mcu_wait_resp failed with -108**
[   10.551564] **mt76x2u 2-1:1.0: mac specific condition occurred**
[   11.524055] usb 2-1: Device not responding to setup address.
[   11.744123] usb 2-1: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[   12.554000] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   12.562068] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.150032] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   16.795804] usb 2-1: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[   16.804094] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   16.811250] usb 2-1: Product: 802.11ac WLAN
[   16.815500] usb 2-1: Manufacturer: MediaTek Inc.
[   16.820151] usb 2-1: SerialNumber: 000000000
[   16.831324] **usb 2-1: Enable of device-initiated U1 failed.**
[   16.995144] **usb 2-1: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd**
[   17.025775] mt76x2u 2-1:1.0: ASIC revision: 76120044
[   20.243763] **mt76x2u 2-1:1.0: vendor request req:06 off:0080 failed:-110**
[   23.443779] **mt76x2u 2-1:1.0: vendor request req:07 off:0080 failed:-110**
[   26.644337] **mt76x2u 2-1:1.0: vendor request req:06 off:0080 failed:-110**
[   29.844558] **mt76x2u 2-1:1.0: vendor request req:47 off:0148 failed:-110**
[   33.043913] **mt76x2u 2-1:1.0: vendor request req:46 off:0148 failed:-110**
** vendor request errors repeat indefinitely**

morrownr, 2022-05-01:
Hi @amisix 

Hmmm... so Kali 2022.2 on a RasPi4B only while using a USB3 port. My first thought is to say bad words about the selection of the USB3 hardward in the RasPi4B. There are probably thousands of Pi4B users that have had issues with this. 

Let's do some troubleshooting:

Do you have an extra sd card that you could do a clean installation of the RasPiOS v 2022-04-04? If so, make it happen and see if you get the same problem.

Would you mind also try the ALFA ACM with the USB3 port with Kali 2022.2 to see if you have the same problem?

Have you tested the adapter in a x86/amd64 based system with Linux installed to see if you see the problem?

Hopefully that will help us narrow things down. Not long ago a patch was applied to the RasPiOS USB3 driver due to a hardware incompatibility. The problem was not in the Linux driver but was a hardware bug. I can be more specific if depending on what we find.

Regards

amisix, 2022-05-02:
>My first thought is to say bad words about the selection of the USB3 hardware in the RasPi4B. There are probably thousands of Pi4B users that have had issues with this.

Fun times. Hopefully whatever we find out can be useful.. In a previous post you had mentioned bugs in the Raspberry Pi 4B USB controller, I was thinking we might end up there with this but I'm not so sure now after testing it with the Alfa ACM. 

>Do you have an extra sd card that you could do a clean installation of the RasPiOS v 2022-04-04? If so, make it happen and see if you get the same problem.

Sure do - I'll do that. 

>Would you mind also try the ALFA ACM with the USB3 port with Kali 2022.2...

Done. I do not have the same problem with the ALFA ACM in Kali 2022.2. It initializes properly without error. 

>Have you tested the adapter in a x86/amd64 based system with Linux installed...

Not yet but I have a live boot Kali image ready to go. I'll run it up to see what comes of it. 

Thanks.

morrownr, 2022-05-02:
> Fun times. Hopefully whatever we find out can be useful.. In a previous post you had mentioned bugs in the Raspberry Pi 4B USB controller, I was thinking we might end up there with this but I'm not so sure now after testing it with the Alfa ACM.

That the ALFA ACM is working is just a data point. It does not rule out the Pi4B USB3 hardware being the problem imho. The 8812au driver has worked fine on the Pi4B while the 8812bu driaver has had ongoing problems for a long time. It seems the fix that went into the Pi4B has fixed the problem with the 8812bu driver but I am not aware of that fix being upstreamed. 

We will just have to search and since I don't have that adapter, all I can do forward suggestions.

Regards
> Sure do - I'll do that.
>
>
> Done. I do not have the same problem with the ALFA ACM in Kali 2022.2. It initializes properly without error.
>
>
> Not yet but I have a live boot Kali image ready to go. I'll [...]

morrownr, 2022-05-02:
Let me be clear: The 2022-04-04 version of the RasPiOS should and does appear to have the RasPi4b USB3 fix but the version of Kali you are using most likely does not as the fix would need to work its way up into the kernel Linus maintains and then downward. That does not mean the problem you are seeing is the result of the problem that the fix addressed but the fact that it is happening only on a USB3 port makes one ponder the issue.

amisix, 2022-05-02:
^ Understood. 

The issue does not occur whatsoever in RasPiOS - the adapter works fine on USB 3.0 - able to see and connect to access points with no errors in the logs. 

There is an issue with the ANDDEAR adapter in Kali on x86 hardware although it is not identical it still leads to the adapter being non-functional. I can see the adapter but it won't connect to anything then it says "network disconnected" while no longer allowing me to even attempt a connection. Is it loading the incorrect firmware - ? There's a reference to mt7662.bin (not 7612)? Line 2 and Line 5. 

Errors on x86 hardware: 
>[   15.069700] mt76x2u 2-3:1.0: ASIC revision: 76120044
[   15.069796] mt76x2u 2-3:1.0: firmware: direct-loading firmware mt7662_rom_patch.bin
[   15.069807] mt76x2u 2-3:1.0: ROM patch build: 20141115060606a
[   15.126913] Bluetooth: hci1: RTL: fw version 0x0999646b
[   15.201949] mt76x2u 2-3:1.0: firmware: direct-loading firmware mt7662.bin
[   15.201958] mt76x2u 2-3:1.0: Firmware Version: 0.0.00
[   15.201962] mt76x2u 2-3:1.0: Build: 1
[   15.201964] mt76x2u 2-3:1.0: Build Time: 201507311614____
[   15.928821] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   15.930020] usbcore: registered new interface driver mt76x2u
[   18.503647] r8169 0000:03:00.0: firmware: direct-loading firmware rtl_nic/rtl8168h-2.fw
[   18.533693] Generic FE-GE Realtek PHY r8169-0-300:00: attached PHY driver (mii_bus:phy_addr=r8169-0-300:00, irq=MAC)
[   18.725739] r8169 0000:03:00.0 eth0: Link is Down
[   18.741103] iwlwifi 0000:02:00.0: Applying debug destination EXTERNAL_DRAM
[   18.818936] iwlwifi 0000:02:00.0: Applying debug destination EXTERNAL_DRAM
[   18.820194] iwlwifi 0000:02:00.0: FW already configured (0) - re-configuring
[   18.855815] iwlwifi 0000:02:00.0: Applying debug destination EXTERNAL_DRAM
[   18.933961] iwlwifi 0000:02:00.0: Applying debug destination EXTERNAL_DRAM
[   18.935387] iwlwifi 0000:02:00.0: FW already configured (0) - re-configuring
[   20.885724] **mt76x2u 2-3:1.0: mac specific condition occurred**
[   21.005747] **usb 2-3: USB disconnect, device number 3**
[   21.140248] **mt76x2u 2-3:1.0: mac specific condition occurred**
[   21.349695] **mt76x2u 2-3:1.0: mac specific condition occurred**
[   21.453699] **mt76x2u 2-3:1.0: mac specific condition occurred**
[   21.553693] **mt76x2u 2-3:1.0: timed out waiting for pending tx**
[   21.885886] usb 2-3: new SuperSpeed USB device number 4 using xhci_hcd
[   21.906677] usb 2-3: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[   21.906681] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   21.906682] usb 2-3: Product: 802.11ac WLAN
[   21.906683] usb 2-3: Manufacturer: MediaTek Inc.
[   21.906684] usb 2-3: SerialNumber: 000000000
[   22.038055] usb 2-3: reset SuperSpeed USB device number 4 using xhci_hcd

morrownr, 2022-05-02:
> The issue does not occur whatsoever in RasPiOS - the adapter works fine on USB 3.0 - able to see and connect to access points with no errors in the logs.

Hmmm... Kali is starting to look guilty.

> There is an issue with the ANDDEAR adapter in Kali on x86 hardware although it is not identical it still leads to the adapter being non-functional.

Kali is really starting to look guilty.

Are you going to be able to test a non-Kali x86 Linux distro such as Ubuntu 22.04?

> I can see the adapter but it won't connect to anything then it says "network disconnected" while no longer allowing me to even attempt a connection. Is it loading the incorrect firmware - ?

It is possible that the firmware is somehow messed up in Kali.

> There's a reference to mt7662.bin (not 7612)? Line 2 and Line 5.

The mt7612u chipset requires two firmware files:

mt7662u.bin
mt7662u_rom_patch.bin

My recommendation is that you see my guide:

https://github.com/morrownr/USB-WiFi/blob/main/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

What I think you should do at this point is download both files as instructed in the above document and then compare the files to what is in Kali. The files in Kali should be identical to what you download byte per byte. There is one thing to note, the mt7612u firmware files are usually installed in two different directories and there is a history to understand.

At one time, all firmware files were just dumped into `lib/firmware` but as time passed and more files were showing up, it was decided that each company have its own directory. All Mediatek firmware files are in `/lib/firmware/mediatek' however due to old code in the mt7612u driver, it looks in the old location which is `lib/firmware". What most distros do is they put the two files in both locations so you should check the files in both locations.

The adventure continues...

amisix, 2022-05-02:
>Kali is really starting to look guilty.

We're narrowing it down... yay

>Are you going to be able to test a non-Kali x86 Linux distro such as Ubuntu 22.04?

Yeah, I'll just create another live boot usb with Rufus or balena.

>My recommendation is that you see my guide...

It could be the firmware? I'll follow your instructions and do a diff/hash comparison as suggested.

We shall see. Thanks for breaking it all out.

morrownr, 2022-05-03:
> It could be the firmware?

It could. These files get sent all over the place. One little bit gets flipped and now the file does operate as designed. Is it likely this is the problem? No, but firmware errors are what is showing up so we need to be 100% that firmware file integrity is not the problem. I am looking forward to your report from a test on an x86/amd64 system. So far, the only time we are seeing the problem is with Kali.

morrownr, 2022-05-04:
Well. Interesting.

Let me dive into the code in the mt7612u driver so that we know exactly which firmware files the driver is looking for and what their location without a doubt.

amisix, 2022-05-04:
>Let me dive into the code in the mt7612u driver so that we know exactly which firmware files the driver is looking for and what their location without a doubt.

Awesome, thanks.

amisix, 2022-05-05:
@morrownr RE: Kali Firmware

I removed the firmware from /lib/firmware/ then copied the firmware from /lib/firmware/mediatek in its place. I renamed the two .bin files to match what was already in /lib/firmware (removed the "u" after mt7662, Example: mt7662u.bin = mt7662.bin & mt7662u_rom_patch.bin = mt7662_rom_patch.bin)

Weirdest thing, the adapter now works on one USB 3.0 port (top) without errors, but not the other USB 3.0 port (bottom). The error I receive on the bottom USB 3.0 port is the same as the previous error: 

>mt76x2u 2-1:1.0: error: mt76x02u_mcu_wait_resp failed with -108
[ 10.551564] mt76x2u 2-1:1.0: mac specific condition occurred

BUT, one port is now working with the firmware swap.. thoughts?

morrownr, 2022-05-05:
> EDIT: After additional testing both USB 3.0 ports are now working without errors. Apparently swapping the firmware worked.. ?

I have busy since we worked on thiss yesterday. Give me a chance to read your reports and then I will try to sit aside the next hour to look at this.

morrownr, 2022-05-05:
While I am working, here is something for you to test:

Delete `mt7662.bin` and `mt7662_rom_patch.bin` from `/lib/firmware` and then reboot so as to have a clean boot and then see what happens.

morrownr, 2022-05-05:
A little grepping later and this is what I have:

```
mt76/mt76x0/mt76x0.h:#define MT7610E_FIRMWARE		"mediatek/mt7610e.bin"
mt76/mt76x0/mt76x0.h:#define MT7650E_FIRMWARE		"mediatek/mt7650e.bin"
mt76/mt76x0/mt76x0.h:#define MT7610U_FIRMWARE	"mediatek/mt7610u.bin"
mt76/mt76x2/mt76x2.h:#define MT7662_FIRMWARE		"mt7662.bin"
mt76/mt76x2/pci.c:MODULE_FIRMWARE(MT7662_FIRMWARE);
mt76/mt76x2/pci.c:MODULE_FIRMWARE(MT7662_ROM_PATCH);
mt76/mt76x2/pci_mcu.c:	ret = request_firmware(&fw, MT7662_FIRMWARE, dev->mt76.dev);
mt76/mt76x2/usb_mcu.c:	err = request_firmware(&fw, MT7662_FIRMWARE, dev->mt76.dev);
mt76/mt76x2/usb.c:MODULE_FIRMWARE(MT7662_FIRMWARE);
mt76/mt76x2/usb.c:MODULE_FIRMWARE(MT7662_ROM_PATCH);
```

It appears mt7662.bin and its mate are for the PCI version of the chipset and should be in /lib/firmware whick should answer the below question you had:

> Why the discrepancy between the firmware in /lib/firmware compared to /lib/firmware/mediatek (and git)?

One is for the PCI chipset and one is for the USB chipset. It is interesting how they ended up to two different places.

I am not seeing anything in the code that tells me when mt7662u.bin and its mate should be. Interesting. We will have to assume it is /lib/firmware/mediatek as I have checked multiple distros now and that it where they are.

It is interesting that you copied the files from /mediatek to /lib/firmware and removed the `u` and things started working. I am being to suspect a faulty part in the adapter as what you did should not have solved the problem. Is it possible that there is some dust in the ports or in the adapter?

This may be a case where putting things down for a couple of days and then start over.

morrownr, 2022-05-06:
> I removed mt7662.bin and mate from /lib/firmware and the issue reappears. The adapter fails to initialize.
> 
> Error:
> 
> [ 7.329300 ] **mt76x2u 2-2:1.0: Direct firmware load for mt7662_rom_patch.bin failed with error -2**

This makes no sense to me. Why would a USB adapter load `mt7662_rom_patch.bin`?

> [ 8.040892] **mt76x2u: probe of 2-2:1.0 failed with error -2**

?

> [ 8.048231] usbcore: registered new interface driver mt76x2u

Okay

> The adapters and plugs are clean. I have a second MT7612U004 adapter I've tried it with, same behavior.
> 
> EDIT: I put mt7662.bin and mate back in /lib/firmware and the adapter starts working again.

Is this mt7662.bin and mate the originals or the ones that are actually mt7662u.bin and mate but were renamed?

> EDIT EDIT: It fails when I connect a second adapter, whether it be an ANDDEAR or Alfa MT7612U. The first adapter that initialized fine stays functioning but the second adapter immediately starts throwing disconnect/reconnect errors repetitively every few seconds. Oh well, one works great for now.

When I put two 7612u adapters in any system here, they both work fine but I don't have a ANDDEAR MT7612U004.

This is starting to hurt my head.

This testing needs to be done in a scientific manner. We really need a sheet where we write the exact details. I'm getting lost on which distro and which hardware and how many or which adapters you are testing.

Are you using a usb hub or direct into the port?
 
> So, there's a temporary fix - but, why? This can totally wait if you want to try again another day.

I'll try to hang in there.

amisix, 2022-05-06:
>This makes no sense to me. Why would a USB adapter load mt7662_rom_patch.bin?

Unknown. Agreed, why?  

>Is this mt7662.bin and mate the originals or the ones that are actually mt7662u.bin and mate but were renamed?

Correct, yes. Pulled from /lib/firmware/mediatek and renamed. 

>This is starting to hurt my head.
This testing needs to be done in a scientific manner. We really need a sheet where we write the exact details. I'm getting lost on which distro and which hardware and how many or which adapters you are testing.

Same. I'm trying my best. I can go back through the thread and compile things a bit better if need be. I only used the ALFA in place of the ANDDEAR as requested in one of your tests, and once as a secondary adapter (recently) on the USB 3.0 bus - past that it's all the ANDDEAR-MT7612U004 that's throwing a fit (I have two of them on hand).

No hub, direct to the port(s). Single adapter connected unless specifically testing dual adapters as requested. 

Thanks..

amisix, 2022-05-06:
>How about we do a short review and come up with a plan? We can contact the Mediatek kernel devs if need be but right now all we have is kind of a mess as far as something to present.

Sure, Ok. 

>We zeroed in on checking the firmware as a possible problem. Here is how we think the firmware should be setup for mt7612u/mt7612e chipsets:...

Yes, on the same page. 

>Assumption: If test systems have the above 4 files exactly as they are shown above, the firmware files should be the right files and in the right place.

Ok, will run through requested tests to confirm. 

>Can I get you to run the following tests while booting in between tests?

Of course. It will take a bit but I will get it. All previous tests have been completed with a clean, freshly imaged copy of whichever OS was requested, I will continue to do that for these tests. 

Thanks much for all your efforts regarding this.

morrownr, 2022-05-06:
Take your time. I enjoy solving a good mystery. It often takes time.

You may want to start a text document to put the results of your testing and you can zip it and post it here instead of having a very long report. It is also easier to add to a text document than to edit and keep up with results here. Below is an example like I might use:

```
RasPi4B 
RasPiOS 2022-04-04
one ANDDEAR 004 adapter in usb3 port
$ dmesg | grep -i mt76
[   11.289928] mt76x2u 2-2:1.0: ASIC revision: 76120044
[   11.332857] mt76x2u 2-2:1.0: ROM patch build: 20141115060606a
[   11.497666] mt76x2u 2-2:1.0: Firmware Version: 0.0.00
[   11.497688] mt76x2u 2-2:1.0: Build: 1
[   11.497694] mt76x2u 2-2:1.0: Build Time: 201507311614____
[   12.319504] usbcore: registered new interface driver mt76x2u
[   12.363881] mt76x2u 2-2:1.0 wlx0013ef5f0c7c: renamed from wlan0
````

The multiple tests with the RasPiOS hopefully will give us a baseline of tests that are successful, then the same tests with Kali on the RasPi4B can be compared.

Regards

morrownr, 2022-05-06:
Can I get you to add one test to the previously posted list?

one ALFA ACM adapter in usb3 port

amisix, 2022-05-06:
[Here's what I've completed so far - all the tests in RasPiOS.](https://github.com/amisix/USB-Wifi/blob/63401a508bd970ed3063c3b3933b79f5e1430b83/ANDDEAR-MT7612U004%20Testing%20-%20RasPiOS.txt) It was not as friendly as when I ran the tests last time, all the ANDDEAR tests for USB 3.0 failed, just like they did in Kali.. But the Alfa ACM worked fine. (table, meet forehead)

amisix, 2022-05-07:
OK, 50th time is the charm. Github sure doesn't make it easy to simply upload a text file without trying to tie it to some code or a branch or some such sh*t. [Here's a .zip file with the RasPiOS tests](https://github.com/amisix/USB-Wifi/blob/f3e2c44dbb4270338b533bafc4a5f26440d0ab75/ANDDEAR_TEST_RPiOS.zip)

morrownr, 2022-05-07:
Thanks for giving it 50 times. I think I can show you an easier way to upload the compressed file. What OS are your trying to upload it with?

I read through the file. I saw consistent results. I took the primary error line and did some searching. Try this:

Open a terminal (Ctrl + Alt + t)

```
sudo nano /etc/modprobe.d/mt76_usb.conf
```

add:

```
options mt76_usb disable_usb_sg=1
```

Save the file: Ctrl + Alt + o, Enter, Ctrl + Alt + x

Reboot

Redo one of the tests that failed.

Results?

amisix, 2022-05-07:
>Thanks for giving it 50 times. I think I can show you an easier way to upload the compressed file. What OS are your trying to upload it with?

Please do. Windows 10 (chrome) - I thought it should have been easier, don't know what I was missing. Can run basic tests in CLI, can't upload .txt file. Hopeless. 

>I read through the file. I saw consistent results. I took the primary error line and did some searching. Try this (disable SG):

Consistent results, good. 

I disabled USB Scatter Gather, rebooted, then confirmed it was off with the command below. I tested with a single ANDDEAR adapter on USB3 port. Unfortunately the issue persists with no change in error messages.

>cat /sys/module/mt76_usb/parameters/disable_usb_sg

Thanks for searching the error further. I also completed some searching for similar error messages but almost all referred to the error "mt76x02u_mcu_wait_resp failed with **-110**", not the "mt76x02u_mcu_wait_resp failed with **-108**" I've been receiving. Although I don't know how relevant that difference is at this time.

Thanks.

amisix, 2022-05-08:
Quick test (re-run a dozen times): Kali 32bit 2022.1 
Single ANDDEAR adapter, USB3 Port 1 - FAIL (same errors as in RasPiOS)
Single ANDDEAR adapter, USB3 Port 2 - SUCCESS

Why would the adapter work on one USB3 port and not the other? 

I also tested USB3 Port 1 with an Alfa ACM and the Alfa adapter works fine on that port, no errors. 
Then I ran the test on a different Raspberry Pi 4 and USB3 Port 1 with the ANDDEAR adapter failed just like previous tests. 

(USB Port numbers, count from top to bottom. Ex: USB3 Port 1, top port)

TurfJakkals, 2022-05-09:
Greetings! I've read through the thread of progress made regarding this issue. Now the thing is, I'm using the MT7612UN and having no luck whatsoever, either. Not sure what makes the "UN" different from the "U" or "BU" versions?

morrownr, 2022-05-09:
> Please do. Windows 10 (chrome) - I thought it should have been easier, don't know what I was missing. Can run basic tests in CLI, can't upload .txt file. Hopeless.

Zip the text file and the drag the zip and drop it here (actually in the window where you are replying)

> I disabled USB Scatter Gather, rebooted, then confirmed it was off with the command below. I tested with a single ANDDEAR adapter on USB3 port. Unfortunately the issue persists with no change in error messages.

Well, okay. While searching I saw similar reports. I had previously only seen issues requiring that parameter in AP mode.

morrownr, 2022-05-09:
> Quick test (re-run a dozen times): Kali 32bit 2022.1 Single ANDDEAR adapter, USB3 Port 1 (top) - FAIL (same errors as in RasPiOS) Single ANDDEAR adapter, USB3 Port 2 (bottom) - SUCCESS
> 
> Why would the adapter work on one USB3 port and not the other?

Very good question. I don't have an answer yet.

Which revision of the RasPi4B do you have?

At this point, I'd like to see the results of testing on x86/amd64 hardware. Yes, again. The expanded version of testing like you have been doing and documenting. Please continue documenting your testing as it is the best we have right now to see something to help and if we figure out what the problem is, it will be something to send in with a report.

I have 4 different adapters based on the mt7612u chipset and I have never run into such as this. I have a RasPi4B. It is a v1.2.

> I also tested USB3 Port 1 with an Alfa ACM and the Alfa adapter works fine on that port, no errors. Then I ran the test on a different Raspberry Pi 4 and USB3 Port 1 with the ANDDEAR adapter failed just like previous tests.

Are both Pi's the same version?

> I also found another error - something related to the xhci controller itself?
> 
> > **xhci_hcd 0000:01:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.**
> > mt76x2u 2-2:1.0: mac specific condition occurred
> 
> USB bug?

Need to do some searching.

morrownr, 2022-05-09:
> Greetings! I've read through the thread of progress made regarding this issue. Now the thing is, I'm using the MT7612UN and having no luck whatsoever, either. Not sure what makes the "UN" different from the "U" or "BU" versions?

Hi @TurfJakkals 

I don't have an answer to your question at this time.

Can you be specific exactly which adapter that  you have? And the hardware that you are plugging the adapter into?

Regards

morrownr, 2022-05-09:
Hi @amisix 

I have moved the listing and review of the ANDDEAR 04 until such time as we resolve this issue. If I am not comfortable with a listing, I can't keep it up. Have you seen any issues with the ANDDEAR 03?

amisix, 2022-05-09:
>Well, okay. While searching I saw similar reports. I had previously only seen issues requiring that parameter in AP mode.

It's a variable we needed to account for, now it's done.

>Which revision of the RasPi4B do you have?

Both boards are revision 1.5 (2GB version). 

>At this point, I'd like to see the results of testing on x86/amd64 hardware. Yes, again. 

I can do that. I'll be able to complete it within the next couple of days. I'll include completed ARM Kali tests as well. 

>I have 4 different adapters based on the mt7612u chipset and I have never run into such as this. 

Odd, I know, but if the issue was going to affect anybody.. it'd be me. My opinion of Mediatek is not impacted by this issue. Apparently this is currently impacting other users as well.. I was stoked to learn it's quite possibly an actual issue as opposed to raw incompetence on my part. I almost said F* it, how many people will have this hardware combination (very few)? Who needs to do all this effort troubleshooting an issue with an off label USB wifi adapter? I mean, there is a "fix" - don't put the adapter in that port. But it's a lazy solution that just passes the buck to the next poor SOB that happens to stumble upon this or something similar. And I want an answer. 

>I have moved the listing and review of the ANDDEAR 04 until such time as we resolve this issue. If I am not comfortable with a listing, I can't keep it up. Have you seen any issues with the ANDDEAR 03?

Thank you - I feel the same way. I legit had anxiety that someone was going to purchase one of these based on my review only to find out it is the complete opposite of "it just works" (for now). I haven't had any issues with the ANDDEAR 03. I've swapped them in/out of USB2 & USB3 ports without errors many times. They've been working great so far although I may test them more rigorously due to this experience.

amisix, 2022-05-09:
@TurfJakkals From what I've read the U stands for USB (we knew that) an the N stands for Nano. 
I could be wrong. Is it a nano adapter? Thanks for letting us know this impacts others, hopefully that will help us to determine a solution. The Mediatek chipset in the adapter is great.. Are you also on a Raspberry Pi or another platform?

TurfJakkals, 2022-05-09:
> Can you be specific exactly which adapter that you have? And the hardware that you are plugging the adapter into?

I got it working on my Win11 system by assigning it a driver manually.
Driver: MediaTek, Inc. 802.11ac Wireless LAN Card Version:5.1.22.0 (updated to 5.1.25.0 after installation)
Driver node strong name: oem15.---:efa2156a6b29dd80:OS61_RTWLANR_7650.ndi:5.1.25.0:usb\vid_0e8d&pid_7650
Plugged into USB3.0. Works perfectly on my Win11 system after assigning it that driver.

> What distro are you using?

Tried to get it working on 3 different distros over on my other system. Latest updates on all of them. Plugged into USB3.0, oh and no, it isn't a Nano adapter.

Kali: Doesn't pick up, not showing when lsusb.
Parrot: Shows when lsusb, nothing further.
Zorin: Shows when lsusb, nothing further.

morrownr, 2022-05-09:
Hi @TurfJakkals 

> I got it working on my Win11 system by assigning it a driver manually.

You are talking about an `ANDDEAR MT7612U004` adapter?

> Driver: MediaTek, Inc. 802.11ac Wireless LAN Card Version:5.1.22.0 (updated to 5.1.25.0 after installation) Driver node strong name: [...]

Interesting. I have a laptop that has Win11 on it these days. I don't use the Win11 but it is there. I booted it and plugged my ALFA ACM into a USB3 port. The ALFA ACM (mt7612u chipset) was detected and driver v5.1.22 was autoinstalled. I turned off the internal wifi and logged into my AP. It worked fine. It is not clear to me why it would have been necessary to install a driver.
 
> > What distro are you using?
> 
> Tried to get it working on 3 different distros over on my other system. Latest updates on all of them. Plugged into USB3.0, oh and no, it isn't a Nano adapter.

I am not aware of any nano adapters ever having been built with a mt7612u chipset. It may not be suitable for use in a nano adaoter.

> Kali: Doesn't pick up, not showing when lsusb. Parrot: Shows when lsusb, nothing further. Zorin: Shows when lsusb, nothing further.

Request that both of you post the output of $ lsusb . Thanks.

@amisix @TurfJakkals 

A recap of where my thinking is at this point: amisix has done a good job of testing and creating a document with results that can help us identify or at least narrow down where the problem is.

So far, the results on a RasPi4B with the RasPiOS and Kali for RasPi4B show the `ANDDEAR MT7612U004 Adapter` autodetected and working fine if plugged into a USB2 port and not working at all if plugged into a USB3 port. A comparison with the ALFA ACM shows the ALFA ACM autodeteched and working fine on both USB2 and USB3 ports.. I see what appears to be some testing earlier in this thread that appears to be with Linux distros on x86/amd64 bit is not clear as it i is not a clear as it could be.

> Both boards are revision 1.5 (2GB version).

Details like this need to documented in the master document that amisix is creating IMHO.

Let add something to the conversation that you may not be aware of:

In addition to the USB-WiFi repo, I also have 5 repos specifically for modern Realtek based WiFi adapters. One of them is:

https://github.com/morrownr/88x2bu-20210702

As your read the README.md for the 88x2bu driver, you will see:

```
[1] Overall this driver does a good job with AP mode. During testing and work prior to making this driver available, the team working on this driver noticed some problems in AP mode if used with a Raspberry Pi 4B. We were unable to discover or fix the exact cause of the problem but the workaround is to keep the driver in USB2 mode. This workaround only applies to AP mode with Raspberry Pi 4B. No problems were noted with systems that use x86 or amd64 processors.
```

The USB3 problem specifically with the RasPi4B is so severe that we had to publish a specific workaround for users. Interestingly enough, the following driver does not suffer the same problem, or at least not to the same degree:

https://github.com/morrownr/8812au-20210629

Both are Realtek AC1200 chipsets for USB WiFi adapters and are widely available. Those repos are VERY heavily used. What is the difference between the 2 Realtek chipsets? The two drivers are very different and the large coded firmware files make it a challenge to understand many details but testing has shown the the 88x2bu is a faster chipset than the 8821au. However, the 8812au is an overall better chipset based over compatibility with AP mode and monitor mode.

My point being that USB3 problems are certainly not unique to the subject Mediatek based adapter. The problem on the 88x2bu based adapter shows in 5 MHz AP mode and the symptoms range from reduced performance to  total connection shutdown and is 100% reproducible. The problem is ONLY found with the Raspi4B, well, probably with the RasPi 400 as well but no testing has been completed due to lack of hardware.

In my mind, we are at the point where it we get solid evidence that the `ANDDEAR MT7612U004 Adapter` does not work in a USB3 x86/amd64 port, then we likely are looking at a `ANDDEAR MT7612U004 Adapter` bug. We need more testing to confirm or disprove that theory. I can't help due to lack of having that specific adapter but I will stay engaged to help to to get the problem.

Regards

amisix, 2022-05-10:
Ubuntu 22.04 LTS
x86 laptop / Asus

Test A (USB3): Failed 
>mt76x2u: vendor request req: 42 off:0236 failed: -110
firmware upload failed: -110

Test B (USB2): Success - Able to see and connect to access points

I'll upload the test data when I'm done compiling it.

amisix, 2022-05-10:
>When you say disconnect/reconnect without restart, do you mean that you pulling the adapter out of the port and then put it back in? 

Pulling the adapter out of the port and plugging it back in while the machine is on. 

>Are you checking with lsusb -t to see if the individual tests are resulting in USB3 (5000) or USB2 (480)?

Yes. USB3 connectivity displayed 5000Mbps and USB2 displayed 480Mbps as expected.

>It might be a good idea to add USB3 chipset to the header 

I'll do that. The x86 laptop has an Intel 100 Series/C230 Series Chipset

> I think I am going to setup to run detailed testing with the four 7612u based adapters that I have. I have a small lab with 4 systems that I use for testing and said systems span a wide range of capabilities

Well, now it's a party. I bet your lab is fun (more details?), I'm very much looking forward to the results. 

>Is this an issue with the adapter? Is this an issue with the 7612u driver? Is this an issue with the USB stack in Linux? Is this issue specific to certain chipsets? 

Well, we're ruling out pebkac so I'm happy. 

>Something you may want to do is contact the folks behind the below article and ask them what they are seeing:

Oh, that's neat. I'll reach out to them.
Thanks.

amisix, 2022-05-10:
Quick testing with Intel 7/210 Series and ASMedia 1042 USB3 chipsets.

ANDDEAR-MT7612U004
Intel x86 desktop / Asus P8Z77 v-pro Thunderbolt motherboard
USB Controller(s): Intel 7/210 Series & ASMedia ASM1042

Ubuntu 22.04

Test A: FAILED
ASMedia 1042, USB3 Port 1

Test B: SUCCESS
ASMedia 1042, USB3 Port 2

Test C: FAILED
Intel 7/C210 Series, USB3 Port 1

Test D: SUCCESS
Intel 7/C210 Series, USB3 Port 2

We've tried 3 USB chipsets and all 3 behave in the same manner - the adapter works with one USB port and not the other. Same error messages too. The Intel 7/210 Series and ASMedia chipset is 10 years old..

amisix, 2022-05-11:
TL;DR 

Issue: ANDDEAR-MT7612U004 Wireless USB 3.0 Adapter does not function in a particular USB 3.0 port on several different platforms and linux distros. 

Linux + ANDDEAR-MT7612U004 + USB 3.0 Port 1 = Failure
Linux + ANDDEAR-MT7612U004 + USB 3.0 Port 2 = Success

Temporary solution: Plug the adapter into USB3 port 2 (counting from top to bottom) or a USB 2.0 port.

Platforms tested: Raspberry Pi 4B (ARM), Asus laptop (x86), Asus desktop (x86)
Linux distros tested: Kali 22.01, Ubuntu 22.04, RasPiOS 2022-04-04 
USB3 chipsets tested: VL805, Intel 7/210 Series, Intel 100/230 Series, ASMedia 1042
This issue does not occur with other mt7612u adapters such as the Alfa AWUS036ACM

Errors encountered during testing:

>xhci_hcd 0000:01:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
mt76x2u 2-2:1.0: mac specific condition occurred

>usb 2-1: USB disconnect, device number 2
mt76x2u 2-1:1.0: error: mt76x02u_mcu_wait_resp failed with -108
mt76x2u 2-1:1.0: mac specific condition occurred
usb 2-1: Device not responding to setup address.

>usb 2-1: Enable of device-initiated U1 failed.
usb 2-1: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
mt76x2u 2-1:1.0: ASIC revision: 76120044
mt76x2u 2-1:1.0: vendor request req:06 off:0080 failed:-110
mt76x2u 2-1:1.0: vendor request req:07 off:0080 failed:-110
mt76x2u 2-1:1.0: vendor request req:06 off:0080 failed:-110

>mt76x2u 2-3:1.0: mac specific condition occurred
usb 2-3: USB disconnect, device number 3
mt76x2u 2-3:1.0: mac specific condition occurred
mt76x2u 2-3:1.0: mac specific condition occurred
mt76x2u 2-3:1.0: mac specific condition occurred
mt76x2u 2-3:1.0: timed out waiting for pending tx

morrownr, 2022-05-11:
Good report. It could use a little editing before we get to the point that we submit it. After a few years of taking bug reports from users of the Realtek drivers I have up, I have a good feel for what should be included and what not. I can do a little editing as an example if you do not mind?

Status from my perspective:

Problem: USB WiFi adapter will not successfully come up on hot or cold boot if plugged into USB3, port 1.

Operates normally if plugged into USB3 ports 2+ or if plugged into any USB2 port.

Hardware: ANDDEAR MT7612U004 (URL:  ? )

Adapter chipset: MT7612u

USB3 chipsets tested: 

VL805 (RasPi4B)
Intel 7/210 Series
Intel 100/230 Series
ASMedia 1042 (powered USB3 external hub)

Linux distros tested:

Ubuntu 22.04 (kernel 5.15) (x86_64)
Kali 22.1 (kernel ?) (32 bit ARM)
RasPiOS 2022-04-04 (kernel 5.15) (32 bit ARM)

Log: to be added.

-----

Thoughts:

I took a look at the bug you pointed me to regarding xhci. Wow. That thread is longer than this one. I need time to digest.

In looking at the distros and hardware you have tested with, two things in common stick out: All distros are Debian based and all use systemd. I'm pondering if testing with Manjaro (Arch based) or Fedora would show anything. I'm also wondering if finding a distro that does not use systemd would be of any help.

I started setting up yesterday to test my adapters to see what I see in the logs. I have four USB3, mt7612u based adapters and various test systems so I will take a look as I have time. Hopefully today.

What do I think the problem is at this point?

We are closer than we were. Why would this only show up on USB3 port 1? Maybe something during the boot process is happening in the wrong order initially? Maybe it is a timing issue and something is not ready for port 1? The xhci usb3 driver could be the guilty party. Time for more research and reading. I'll report back as able.

Regards

amisix, 2022-05-11:
I will edit as you've suggested and change the verbiage regarding port numbers. I need a short break tho, too. much. data. 

>I took a look at the bug you pointed me to regarding xhci. Wow. That thread is longer than this one. I need time to digest.

Yeah, I dunno - it's not an exact fit but it's close. I found a couple others but need to read up more on them. 

>In looking at the distros and hardware you have tested with, two things in common stick out: All distros are Debian based and all use systemd. I'm pondering if testing with Manjaro (Arch based) or Fedora would show anything.

I'll test with fedora. 

>I started setting up yesterday to test my adapters to see what I see in the logs. I have four USB3, mt7612u based adapters and various test systems so I will take a look as I have time. 

Thanks, I know it can be tedious, I appreciate the effort. I don't think you're going to have a single issue. Can you get your hands on one of these ANDDEAR adapters? I'll buy you a few pints of beer to cover things.

I'm glad we're closer, maybe? So much data to parse that its become convoluted and difficult to follow, making me feel we're farther away. The damn thing keeps throwing different, obscure errors with behavior differing from port to port, what a mess. 

Thanks.

amisix, 2022-05-12:
Fedora does not work either (Workstation 36, x86 64bit). Same exact errors as on the other distros. 
Also, while troubleshooting one time (in Ubuntu, x86) the adapter was somehow set to USB2 protocol while plugged into a USB3 port. And it worked fine as if it was in a USB2 port.

morrownr, 2022-05-12:
> Can you get your hands on one of these ANDDEAR adapters?

I have investigated this. I can't get one currently without international shipping being involved. I have tried international shipped twice in the last two years and it did not turn out well either time so I would like to avoid that which leaves me without a good option.

> Fedora does not work either (Workstation 36, x86 64bit). Same exact errors as on the other distros.

Understand.

> I need a short break tho, too. much. data.

I very much understand this. I need a break also. When troubleshooting, it can be good to take a few days off and then come back to the master document to start fresh. I've had to do this hundreds of times over the last years as I try to support those that use the Realtek drivers. I've grown to very much dislike the company Realtek in that their drivers are missing many features you see in the Mediatek drivers and because there is no way to work issues with Realtek. I am very much looking forward to the new generation of Mediatek USB WiFi adapters based on the mt7921u driver.

> I don't think I can recommend this adapter. lol

I've already pulled the listing and review. Before writing this replay, I did a quick review of this thread and while we are making progress that could eventually lead to code modifications that could help with this issue and others that are similar, it is beginning to look like there is something specific to this adapter that is problematic.

> https://wlan-pi.github.io/wlanpi-documentation/admin/cf912_issues/

Have you had time to contact the folks from the above article? There are several folks there that have this adapter and can test to see if they have any luck with USB3 port 1.

> I know it can be tedious,

I started testing my adapters with mt7612u chipsets yesterday. I started testing on a very modern laptop that has three USB 3.1 (3.0) ports and one USB 3.2 port. The system is running Ubuntu 22.04 (not my favorite distro but it will do until the next release of Mint is available). Anyway, I test my ALFA ACM, Netgear A6210 and TEROW ROW02CD. I also have a COMFAST CF-WU785AC and I need to include it as well. I tested each on all 4 ports. I have a little converter for the USB 3,2 USB-C port. I decided to start with WPA3, DFS channel on my router and anything else I could think of to make it hard. Result: No operational problems at all. I tested and pushed them with iperf3 and all I could see was smooth sailing. I had never tested USB-C 3.2 before so a good test was a good thing to see. I did capture mt76 and usb log entries to files and I will look at them today to see what came up.

At this point, I am beginning to suspect a manufacturing abnormality in the ANDDEAR adapter could be the cause. Have you tested with Windows or OSX?

I think we probably have enough data at this point and maybe we should slow down some and look at doing more research.

Regards

amisix, 2022-05-12:
>I have tried international shipping twice in the last two years and it did not turn out well either time so I would like to avoid that...

Yeah, that's no good. I have an "extra" one that I can mail to you if you'd be amenable to that. 

>I very much understand this. I need a break also. When troubleshooting, it can be good to take a few days off and then come back...

Agreed. Start seeing double and dyslexia kicks in, not too good for things like mt7612u & mt7621u etc.  

>I've grown to very much dislike the company Realtek in that their drivers are missing many features you see in the Mediatek drivers and because there is no way to work issues with Realtek.

Realtek sounds like monsters, flooding the market with sh*t chipsets & code. But... consumers would never know because: Windows. Seems like their primary market is naive users who never look under the hood because they wouldn't know how to fix it if they tried. Then a bunch of linux aficionados come along and are like... hey... that's a _nice_ chipset you have there, it would be a shame if we made it do things you didn't design it to do... and it just spirals from there. 

I too am looking very much forward to the upcoming release of the mt7621u chipset despite not having an AX capable rout... oh, yeah, "If you build it, wifi will come". 

>I've already pulled the listing and review. (of the ANDDEAR adapter)

Thank you. This thread is more like an anti-review, unfortunately. 

>Before writing this reply, I did a quick review of this thread and while we are making progress that could eventually lead to code modifications that could help with this issue and others that are similar, it is beginning to look like there is something specific to this adapter that is problematic.

Awesome, you work fast. Having some eyes on it that know what they're doing is very much appreciated. Looking forward to hopefully determining a resolution that doesn't involve percussive maintenance. and a shot glass.  

>Have you had time to contact the folks from the above article?

Not yet as their contact link goes to an error 404 page. But I found their repository and am still trying to determine the best way to reach out (It's not really geared towards Q&A, more like feedback). Their input may be invaluable though imo. Interestingly... they have a system image that runs on Raspberry Pi 4 that supposedly uses this adapter - is a functioning driver baked into their image or...? We shall see when I download and try it out. 

>Result: No operational problems at all. I tested and pushed them with iperf3 and all I could see was smooth sailing...

Yeah, there we go, a bunch of testing to show the ANDDEAR adapter does not function in the same environment as other mt7612u adapters. I mean.. it's a USB adapter based on an exceptional chipset - is there a hardware difference compared to other mt7612u adapters? Is there some kind of memory addressing issue or is it a power management issue that resets the adapter? Or is it a multi-layered issue that's compounding the problem? Just postulating a bit - I'll wait for more info from you while I do a bit of searching. 

>Have you tested with Windows or OSX

Yup, Windows 10, It works - I've actually been using one of them as my 2nd wifi adapter part of the time.

Thanks.

morrownr, 2022-05-12:
> I too am looking very much forward to the upcoming release of the mt7621u chipset despite not having an AX capable rout... oh, yeah, "If you build it, wifi will come".

Ah, you don't need an AX capable router. You just need 2 mt7921u wifi adapters. My router is an AC2600 device with a USB3 port and a USB2 port. I have tested my ALFA ACM in the USB3 port as a second 5 GHz radio and it works very well. I use OpenWRT. You can run OpenWRT on many routers AND on a RasPi4B. I can show you how with one of your Pi's if interested. You could learn with the ALFA ACM and then you are ready to go when we actually have adapters that we can buy.

> Yup, Windows 10, It works...

Well, it seems this problem is specific to Linux but not specific to Debian which makes it look like something between power on and interface creation is fouled up. Since we are looking at many other mt7612u adapters that are not having the problem, it very well could be something in this specific adapter. Adapter makers buy the chipset from the chipset maker and then add antennas, amps and the various other things that make up an adapter. This really does not seem to be a chipset problem but rather something ANDDEAR has added may have messed up the timing. 

> Regarding the WLAN Pi...

I think the default hardware they use only has one USB2 port so they may not be aware of the USB3 problem. But, of course, they should have adapters that they can test in USB3 porrts so if you can get in touch with of them and ask them to test, that could be very useful and it is something they probably need to know and warn people about.

>  I have an "extra" one that I can mail to you if you'd be amenable to that.

I'm fine with that. Now if we had a good way to exchange private email addresses so that we aren't broadcasting them to the world. I'm sure there is a way, I just haven't run into this situation before.

Regards

amisix, 2022-05-13:
>Ah, you don't need an AX capable router. You just need 2 mt7921u wifi adapters...

Exactly where I was going with that. Just an excuse to build another WiPi hotspot! I built one that's been running nicely for a couple months now as my only router. It's a RPi 3B with OpenWrt and an Alfa ACM along with two Realtek 8153 USB ethernet adapters and a 5 port TP-Link managed switch w/VLAN capability. Learning OpenWrt while at the same time figuring out the Pi atmosphere has been challenging and fun. 

>I can show you how with one of your Pi's if interested. You could learn with the ALFA ACM and then you are ready to go when we actually have adapters that we can buy.

lol, similar thinking, I like it. If it's not wasting your time I'll take you up on that offer. I'd like to do antenna and system optimization (why irqbalance? who cares about jitter?) with what I have now but it's my edge router so let's build something I can break. 

>I think the default hardware they use only has one USB2 port so they may not be aware of the USB3 problem. But, of course, they should have adapters that they can test in USB3 ports

Yeah, USB 2.0 on the CM4 version.. I downloaded the wlanpi system image for the Raspberry Pi 4B and the ANDDEAR adapter initialized fine with no errors (dmesg is clear, ifconfig shows adapter). So I put another one in the other USB3 port and it initialized fine with no errors. Then I checked it with iwlist  and both adapters function. lsusb shows both adapters at 5000Mbps on the 3.0 bus. Rebooted a few times and repeated. It seems to work but I don't know enough to navigate their build yet (uname = unknown, kernel 5.10) and documentation is practically non-existent. 

Also, I reached out to ANDDEAR and [they sent me these drivers](https://github.com/morrownr/USB-WiFi/files/8683561/Anddear.mt7612u.drivers.zip) that should come on a CD with the adapter. They're 6 years old. The documentation makes reference to the Ralink/Mediatek RT2870 chipset? But there are .bin files related to mt7612 and mt7662. This looks like a nightmare to implement given their "instructions" - what to? 

Are these drivers included in the wlanpi image? I'll reach out to one of their maintainers for more info. 

>I'm fine with that. Now if we had a good way to exchange private email addresses so that we aren't broadcasting them to the world. I'm sure there is a way, I just haven't run into this situation before.

Neither have I. I'll try to think of a way too. Thanks for all your help

morrownr, 2022-05-14:
Hi @amisix 

I've been sick so am behind on everything I am working on.

I tried the temp email in your link but it came back saying no address exists.

>Also, I reached out to ANDDEAR and they sent me these drivers that should come on a CD with the adapter. They're 6 years old. The documentation makes reference to the Ralink/Mediatek RT2870 chipset? But there are .bin files related to mt7612 and mt7662. This looks like a nightmare to implement given their "instructions" - what to do?

Those drivers are a waste of time. They are from a time right after Mediatek bought Ralink and they seemed to have thought doing what Realtek was doing was a good idea. They then learned it was not a good idea and started working with the kernel devs to make good in-kernel drivers.

I'll have to defer answering other issues until I feel a little better.

Regards

amisix, 2022-05-14:
>I'll have to defer answering other issues until I feel a little better.

Hope you feel better. We kinda need you 

Google nuked my temp e-mail address in less than 24hrs. [I made another one with a different provider.](mailto:githubalfa@yahoo.com)

>Those drivers are a waste of time. They are from a time right after Mediatek bought Ralink and they seemed to have thought doing what Realtek was doing was a good idea. 

Yeah... For ANDDEAR to even suggest that path was ridiculous imo. So, no real support path, the only path I had to them was through the chat on Ali Express and there was clearly a language barrier which meant no troubleshooting. 

I'd like to hear more about Ralink & Mediatek history, please share when able. I have an old Alfa AWUS036NH that kinda got me into this stuff. 

Thanks.

jclark, 2022-06-03:
I tried the ANDEAR adapter on a x86 mini PC using an N5105 (not sure about MB chipset) and I got similar results: works fine on USB 3.0 port #2 but is unreliable on USB 3.0 port #1.

Disappointing because apart from that it's a nice small adapter.

amisix, 2022-06-08:
@jclark Thanks for testing that. So odd, that it doesn't function in a specific port despite differing OS and USB chipsets. I do also really like the adapter because of its size. I would have thought that the adapter only functioning on 50% of onboard USB3 ports would be considered faulty but apparently the vendor thinks otherwise. The guys over at WLAN-Pi also confirmed the issue recently and have updated their documentation accordingly.

Snuupy, 2022-08-28:
Hi, I have these adapters. I ran into a similar problem as you, then I upgraded to kernel 5.15 on the rpi and no longer had these issues.

amisix, 2022-08-28:
@Snuupy Your efforts are very much appreciated - looks like you've gotten real far with this adapter. Yes, anything you can compile regarding your solution and necessary resources would be awesome. I think the most common test @morrownr runs is sustained iperf3 tests (15+ minutes). It usually drops out after 5 minutes if you get it to function at all. The WLAN-Pi guys recommended this to show the adapter is plugged into the problematic port (USB3 Port1). iperf3 tests on both USB3 ports would be great though because of how it functions/fails on Port 2 also. Thanks!

>grep '7612' /sys/bus/usb/devices/2-1/idProduct

Snuupy, 2022-08-28:
It's possible the adapter overheats too. I ran some benchmarks on it a few months ago and it got really hot. Do you have a preferred iperf command you run?

Snuupy, 2022-08-28:
[openwrt-bcm27xx-bcm2711-rpi-4.zip](https://github.com/morrownr/USB-WiFi/files/9439093/openwrt-bcm27xx-bcm2711-rpi-4.zip)

Try this

Snuupy, 2022-08-28:
Actually, I think you're all right and I misremembered, I get these in my logs:

```
Thu Jun  2 10:12:14 2022 kern.err kernel: [   77.122591] mt76x2u 2-2:1.0: MAC error detected
Thu Jun  2 10:12:15 2022 kern.err kernel: [   78.238714] mt76x2u 2-2:1.0: MAC error detected
Thu Jun  2 10:12:16 2022 kern.err kernel: [   79.488679] mt76x2u 2-2:1.0: MAC error detected
Thu Jun  2 10:12:17 2022 kern.err kernel: [   80.718591] mt76x2u 2-2:1.0: MAC error detected
```

maybe that's why I had the other adapter in the usb2 port. My bad.

Looks like it does not work on usb3 port 1.

Snuupy, 2022-08-30:
Ok I made a bunch of edits and notes in the previous comment, please test and let me know.

Snuupy, 2022-08-30:
Ok I made a bunch of edits and notes in the previous comment, please test and let me know. @amisix

morrownr, 2022-08-30:
@Snuupy 

My to-do list is too full to take another look at this right now. I'm not aware of any adapters that have flashable firmware. I'm pretty sure the firmware inside the adapters is small and not something that would be subject to errors for the most part. The external firmware is changable but that would be the Mediatek kernel devs that would work it but they may not have the adapter. This could be a hard nut to crack.

I tested at length against other adapters with mt7812u chipsets. This adapter is the only one showing the problem so it seems there is something specific to this adapter. Exactly what is causing the problem is still a mystery. Depending on how a person uses the adapter, putting it in a USB2 port works everytime and is fast enough for most uses.

Nick

Snuupy, 2022-08-30:
No worries, thank you for setting up this repo @morrownr. We wouldn't be here if it weren't for your docs!

One limitation of the usb2 bus is if want to set up a wireless repeater with 2 adapters. For instance, 300 send +300 receive = 600mbps, which is already greater than the 480mbps limit on the usb2 bus (as it is shared on the rpi4) so one would be limited to 240mbps instead. And sometimes I can even hit 400mbps on a single adapter.

This adapter is also the cheapest (even shipping included) so it'd be really nice to find a way to get this working. I think I bought 2 for 36usd (so $18 each) and the others on the list are slightly more, none with the same slim form factor where I can comfortably fit 2 of these adapters right next to each other. With the other adapters it's likely I would also need to buy right angled (90 degree) usb3 adapters as well.

I have yet to test it on usb3 port 2 as I was going to run some benchmarks today to test stability after keeping it running for 1-2 days but I suspect results will be the same: unplug/replug to get a new wlanID as a workaround temporarily.  I'll report back when I have tested that.

morrownr, 2022-08-30:
@Snuupy 

I get what you are saying about USB2 limiting the stream. I wish was not a problem. @amisix is actually the one who discovered the problem. I'm glad he did as I try to do a good job of only putting good adapters on the in-kernel list. Sometimes something slips by for a while.

Give it a try in port 2 and see what you get. This is a strange problem. Let us know what you find out.

Nick

amisix, 2022-08-30:
@Snuupy 
I hear ya - I also like this adapter very much (when it works). Perfect size, awesome chipset... It's a shame.
Thanks much for uploading the OpenWRT image and doing your troubleshooting. I too am quite busy this week so will not have a chance to do anything until the weekend. Yeah, try it in port 2 with an extended iperf3 test (default settings). It failed after roughly 5 minutes for @morrownr (if I remember correctly). Others have had it work but may not/have not run extended iperf3 tests. 

>Has anyone reached out to ANDDEAR?

I contacted them via AliExpress online chat. They were responsive but uncaring. There was a language barrier which added an additional layer of complexity. I referenced this thread. I was told to use the drivers that come on CD with the adapter (that are 6+ years old). I tried looking around for a contact outside of AliExpress but was unable to find anything.

morrownr, 2022-08-30:
> I was told to use the drivers that come on CD with the adapter (that are 6+ years old).

When you hear or read this line, you know you are dealing with someone who knows absolutely nothing about Linux. To help correct the situation we really need to support those who support us. That will channel money to those who are doing the better job and it will move the market and make things better for all of us.

Right now, the only retailer that I am aware of that does a good job with their Linux customers is Rokland. The only maker that does a reasonably good job is ALFA. Panda used to do a good job of making adapters that use in-kernel drivers available but they are not making any modern adapters in this day in time. I don't know what is going on there.