morrownr/USB-WiFi #76: Suspend/resume error with AWUS036ACM

gonsolo, 2022-05-07:
After a resume I get the following error:

 +0,000007] UBSAN: invalid-load in
/build/linux-lowlatency-fSdu7c/linux-lowlatency-5.15.0/net/mac80211/status.c:1164:21
[  +0,000004] load of value 255 is not a valid value for type '_Bool'
[  +0,000004] CPU: 22 PID: 387 Comm: kworker/u256:5 Tainted: P
  OE     5.15.0-27-lowlatency #28-Ubuntu
[  +0,000004] Hardware name: To Be Filled By O.E.M. To Be Filled By
O.E.M./X399 Phantom Gaming 6, BIOS P1.31 01/14/2021
[  +0,000003] Workqueue: phy0 mt76x02_mac_work [mt76x02_lib]
[  +0,000014] Call Trace:
[  +0,000003]  <TASK>
[  +0,000003]  show_stack+0x52/0x58
[  +0,000006]  dump_stack_lvl+0x4a/0x5f
[  +0,000007]  dump_stack+0x10/0x12
[  +0,000003]  ubsan_epilogue+0x9/0x45
[  +0,000002]  __ubsan_handle_load_invalid_value.cold+0x44/0x49
[  +0,000004]  ieee80211_tx_status_ext.cold+0x4e/0x5f [mac80211]
[  +0,000068]  mt76_tx_status_unlock+0x111/0x160 [mt76]
[  +0,000010]  mt76_tx_status_check+0x68/0x90 [mt76]
[  +0,000009]  mt76x02_mac_work+0x14b/0x240 [mt76x02_lib]
[  +0,000008]  process_one_work+0x222/0x3f0
[  +0,000006]  worker_thread+0x50/0x3f0
[  +0,000004]  ? process_one_work+0x3f0/0x3f0
[  +0,000004]  kthread+0x13b/0x160
[  +0,000003]  ? set_kthread_struct+0x50/0x50
[  +0,000004]  ret_from_fork+0x22/0x30
[  +0,000006]  </TASK>
[  +0,000001] ================================================================================

This is a stock Ubuntu Jammy lowlatency kernel (with Nvidia drivers).

I would be happy to provide all the needed information to resolve this.

I also want to note that the device is running quite unstable (lots of
deauthenticating, network lost, etc...).

morrownr, 2022-05-07:
Hi @gonsolo 

Welcome.

> I also want to note that the device is running quite unstable (lots of
deauthenticating, network lost, etc...).

How about we look at this issue first. If you do not have `wavemon` installed, install and run it like this:

```
$ sudo apt install wavemon
$ wavemon
```

What `link quality` and `signal level` readings do you see?

Regards

gonsolo, 2022-05-07:
uHi!

> How about we look at this issue first. If you do not have wavemon
> installed, install and run it like this:
>
> $ sudo apt install wavemon
> $ wavemon
>
> What link quality and signal level readings do you see?
>

Link quality: 71% (50/70)
Signal level: -49dBm (0.02uW)

-- 
g

morrownr, 2022-05-07:
> Link quality: 71% (50/70) Signal level: -49dBm (0.02uW)

Those are good. Generally speaking, if I see someone with the symptoms you are describing and there is not a problem with the link quality, it has to do with AP/router settings. My short guide to AP settings is below:

https://github.com/morrownr/USB-WiFi/blob/main/Recommended_WiFi_Router_Settings.md

I haven't added it to the main menu yet as I am currently reworking the site.

The AP/router setting that is often problematic is when both networks (2.4 and 5) are both set to the same name. We can discuss your settings and make sure the AP is not a problem.

FYI: Regarding the other problem. I have an ALFA ACM. I also have a modern Laptop with Ubuntu 22.04 installed. It is not the low latency kernel but we can probably make some progress in figure out what the issue is.

Regards

gonsolo, 2022-05-07:
> Those are good. Generally speaking, if I see someone with the symptoms you
> are describing and there is not a problem with the link quality, it has to
> do with AP/router settings. My short guide to AP [...]

Thanks, I will have a look at this. I'm thinking it's a USB problem because
I remember changing the stick from an USB3 port to a USB2 one made things
better. But I can be wrong.

> FYI: Regarding the other problem. I have an ALFA ACM. I also have a modern
> Laptop with Ubuntu 22.04 installed. It is not the low latency kernel but we
> can probably make some progress in figure out what the issue is.
>

I will try things out and keep you up to date about my findings. You might
have also noticed that I sent another email to lkml with the same error
message with a pristine kernel.

Thanks and best regards,
g

morrownr, 2022-05-07:
> Thanks, I will have a look at this. I'm thinking it's a USB problem because I remember changing the stick from an USB3 port to a USB2 one made things better. But I can be wrong.

Ah. This is good info. Sometimes it takes a conversation to find the right info. Some USB3 hardware has some bugs and it is not just an issue with Mediatek based adapters. It can be a problem with Realtek based adapters as well.

I have a repo specific to the mt7612u chipset:

https://github.com/morrownr/7612u

Go down to the `KNOWN ISSUES` section.
Go to `Known Issue 2`
Recommend you use Method 2 to apply the module parameter `disable_usb_sg=1`
Remember to reboot after saving the file.

Let me know if that stabilizes things.

gonsolo, 2022-05-08:
> Let me know if that stabilizes things.

I tried something different yesterday: I noticed that there was constant
deauthenticating/reauthenticating with BSSID 8C:6A:8D:9E:2A:90 and
8C:6A:8D:9E:2A:88. I found out that I could set the BSSID in the wireless
configuration panel of Ubuntu under "Identity". I suspect that ...:88 is
2.4Ghz and ...:90 is 5Ghz. I set it to ...:90 and most of the annoying
stalls are gone.

-- 
g

morrownr, 2022-05-09:
> I suspect that ...:88 is 2.4Ghz and ...:90 is 5Ghz. I set it to ...:90 and most of the annoying
stalls are gone.

You should be able to confirm which band you are connected to with `wavemon`.

Overall, your above message seems to confirm what I was talking about when I said having both networks using the same name. I realize changing the name can present issues if care is not used. Many IoT devices can use only the 2.4 GHz band so changing the name of the 2.4 Ghz band is more problematic, So what usually works out is to simply add `-5` to the end of the name of the 5 GHz network. That way the names are different and it usually causes minimal problems.

Regards

gonsolo, 2022-05-11:
> What is the status with the suspend/resume issue? I need to do some
> testing with my ACM and other adapters based on mt7612u chipsets later
> today or tomorrow so I could look at this issue. I have not noticed any
> suspend/resume problems here but there is a lot of different hardware out
> there.
>

There is no response from lkml. I sent a second message with a pristine
kernel with the same symptoms.

-- 
g

morrownr, 2022-05-14:
Hi @gonsolo 

Regarding the suspend/resume issue: I had a couple of thoughts:

- I have not seen a single problem with suspend/resume with the mt7612u driver and I do have a laptop that is really modern. It runs on a Intel Gen 11 processor. I suspect the problem could be in the ACPI subsystem given that your system appears to be bleeding edge. Have you checked for ACPI reports with `dmesg`?

- Generally if a problem is picked up, the first question is whether you have tested the lastest available kernel to see if the problem is still there. Since you are using Ubuntu:

https://itsfoss.com/upgrade-linux-kernel-ubuntu/

FYI: I saw where Nvidia is finally going open source like AMD and Intel. This bodes well for your Nvidia card.

> I tried something different yesterday: I noticed that there was constant
deauthenticating/reauthenticating with BSSID 8C:6A:8D:9E:2A:90 and
8C:6A:8D:9E:2A:88. I found out that I could set the BSSID in the wireless
configuration panel of Ubuntu under "Identity". I suspect that ...:88 is
2.4Ghz and ...:90 is 5Ghz. I set it to ...:90 and most of the annoying
stalls are gone.

You can check your signal level and channel with `wavemon`:

$ sudo apt install wavemon
$ wavemon

Regards

gonsolo, 2022-05-16:
Hi!

> I have not seen a single problem with suspend/resume with the mt7612u
driver and I do have a laptop that is really modern. It runs on a Intel Gen
11 processor. I suspect the problem could be in the ACPI subsystem given
that your system appears to be bleeding edge. Have you checked for ACPI
reports with dmesg?

I took you advice and changed the router configuration to separate BSSIDs
for 2.4 and 5Ghz. Since then I had no troubles.

Best regards
-- 
g

morrownr, 2022-05-16:
> I took you advice and changed the router configuration to separate BSSIDs for 2.4 and 5Ghz. Since then I had no troubles.

Glad you are having a more pleasant experience now. I have never thought using the same name for 2.4 and 5 GHz was a good idea and I certainly don't do it here. 

Regards