morrownr/USB-WiFi #79: Another source for TEROW MT7612U

jclark, 2022-05-28:
I ordered one from here:

https://www.aliexpress.com/item/33060439389.html

This is nice source in that it has a bunch of options, including a choice of "CD Driver" or "Free Driver".  I'm guessing "CD Driver" corresponds to the ROW02CD, and "Free Driver" corresponds to ROW02FD  ("Free Driver" I guess means "Driver Free" means "no extra driver needed" means "driver stored in adapter").  So I ordered the "CD Driver" version.

AliExpress has a number of sellers of this - just search for MT7612U.

morrownr, 2022-05-28:
Hi @jclark 

Good link. I will get this up. That link shows 6 variations. You are correct in your assessment above. The versions that say "CD Driver" are the ones Linux users should prefer since there will be no need for usbmodeswitch. Of the 6 listed, I'd say the following 2 should be preferred:

- Basic CD Driver with no holder/cable
- CD Driver with USB 3 holder/cable

I own both a ROW02CD and a ROW02FD. They are good adapters at a good price.

morrownr, 2022-05-30:
I updated the section on this adapter. If you want to take a look and see if things look okay:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

jclark, 2022-06-17:
I received the one I ordered a couple of days to go: it was reasonably quick by AliExpress standards.  I got the CD driver with USB3 holder/cable. I think the USB3 holder/cable is a useful extra: helps when there is not enough space around the USB3 ports.

morrownr, 2022-06-17:
>  I think the USB3 holder/cable is a useful extra: helps when there is not enough space around the USB3 ports.

Agree. I make mention of this in the last paragraph of the information in Main Menu item 1 but it could probably be a better writeup. One thing I have noticed is that some cables are not well made as I have had to return a few over the years. I'm curious to hear if you see good results from the one included with your new adapter.

Regards

user723045, 2023-06-07:
the link is no longer relevant, the seller replaced the product

morrownr, 2023-06-07:
Hi @user723045 

It has been a challenge keeping a good link for the TEROW adapter. I just checked and the link in the Plug and Play list for the TEROW ROW02CD adapter appears to be current.

Go to the Main Menu (README), select menu item 2 and scroll down to the TEROW ROW02CD adapter.

Cheers

morrownr, 2023-06-08:
I'm confused. I see your examples from a product from AliExpress but the link I have up is for newegg. I just checked the newegg link and it appears to still be for the appropriate TEROW adapter.

What am I missing?

jclark, 2023-06-09:
I tried to find a good source in Asia for the TEROW adapter. I tried one place on AliExpress and two on Lazada. In both cases, they were clearly advertising as MT7612U and the photo looked exactly like the TEROW MT7612U one. But in every case, what has the RTL chipset and looked slightly different.  My guess is that if you try to buy this adapter on AliExpress, you are going to be disappointed.

morrownr, 2023-06-09:
Now I understand. I used to have a link for this adapter on Aliexpress but removed it when it appeared to no longer be a solid link to the right product. The TEROW ROW02CD is a solid adapter and I think there is a lot of rebranding going on with the product but that can make it less than certain when you go t buy. It is hard to keep a good link on the list these days.

For what it is worth, Alfa just released similar looking adapter today that has the mt7921 chipset and the price is lower than their other WiFi 6 adapter. I posted links in a newly posted issue today.