morrownr/USB-WiFi #80: Chenyu M-1300Q

jclark, 2022-05-28:
I came across another MT7612U adapter

http://chenyu-wifi.com/product/en/Product-20190905-123313-2020030713510249-2020060911115183-2020060916194860.html

Available on AliExpress

https://www.aliexpress.com/item/1005004313413963.html

I ordered one.

morrownr, 2022-05-28:
Hi @jclark 

Hey, I appreciate the information and will include it soon.

Did you order the version with the external antennas or the version with internal antennas?

Sure would appreciate a review after you get it.

Nick

jclark, 2022-05-29:
I ordered the one with the external antennas.

Happy to do a review. Is there a check-list of points to cover? Normally all I do is plug it in and see if it works.

morrownr, 2022-05-30:
> I ordered the one with the external antennas.

That is probably what I would have done as well. 
 
> Happy to do a review. Is there a check-list of points to cover? Normally all I do is plug it in and see if it works.

I don't really have a checklist but that might be a good idea. Here is a quick and dirty start to a checklist:

- Check function with all ports, both usb2 and usb3 on your system and check to see that the adapter is running in usb2 or usb3 mode as appropriate with the following command:

$ lsusb -t

480 = usb2
5000 = usb3

- install and use `wavemon` to see what signal strength and connection details are:

$ sudo apt install wavemon
$ wavemon

- Use your fingers to see if the device is running hot or not.

- What hardware and distro are you testing with.

- What modes are you testing? Client, AP, Monitor or other.

- Any observations that you feel like mentioning.

I will wait to include the adapter until after you have it and have tested enough to be satisfied that it is a good adapter. Let me know.

Regards