morrownr/USB-WiFi #82: For help, COMFAST CF-wU782AC cannot work in Kali 2022.

lszhwxa, 2022-06-10:
[image]
[image]
[image]

I'm using vmware,Kali kernel 5.16.0-kali7-amd64

I have already installed "https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md", but is unable to work, I don't know what to do.

I'm new to Kali, and I don't speak English. I use Google Translate.

I hope I can get help. Thank you.

morrownr, 2022-06-10:
Hello @lszhwxa 

> I'm using vmware,Kali kernel 5.16.0-kali7-amd64

I understand that you are using vmware and Kali Linux but are you using the version of Kali that is pre-made for a vmware installation? More info:

https://www.kali.org/docs/virtualization/install-vmware-guest-vm/

https://www.kali.org/get-kali/#kali-virtual-machines

If you are using a recent version of Kali, you likely do not need to do anything regarding usb_modeswitch because it should already be setup for you. A simple test to see if it is active and working is to run:

$ lsusb

If the results include the following then usb_modeswitch has done its thing and all is good:

ID 0e8d:7612 MediaTek Inc.

Your problem is not with usb_modeswitch. The issue is with vmware. I am not familiar with vmware. I use VirtualBox. Here is an example article pointing out how to setup a usb wifi adapter when using VirtualBox.

https://null-byte.wonderhowto.com/forum/wifi-hacking-attach-usb-wireless-adapter-with-virtual-box-0324433/

There may be others with more experience with vmware that come along here over the next few days so stay tuned. Hopefully we can help resolve your issue.

Nick

lszhwxa, 2022-06-10:
> Hello @lszhwxa
>
>
> I understand that you are using vmware and Kali Linux but are you using the version of Kali that is pre-made for a vmware installation? More info: [...]

Thank you, now I am also using VirtualBox and now mt7612 is finally working. Gosh, I've been using vmware. thanks for the heads up. (using translation software)

morrownr, 2022-06-12:
Glad to help. Give us a report on that 782AC when you are able.

Nick

lszhwxa, 2022-06-12:
(Use translation software, Chinese - English)

@morrownr

Now I have a new problem. When I use VirtualBox-version 6.1.34a and kali-linux-2022.2-virtualbox-amd64.ova. at first 782AC works fine, after using it normally for a while, suddenly it doesn't work anymore. I don't know what's the reason, it's really strange.

Then after some more time, the 782AC worked again, during which I didn't do anything and it got up properly on its own.

Then after some time, I restarted my Win10 system, opened VirtualBox again, and opened kali again, and 782AC didn't work again, which was really strange. I tried to reinstall VirtualBox and re-import kali-linux-2022.2-virtualbox-amd64.ova, but the problem is still not solved.

$ lsusb
It keeps getting stuck and not responding to anything.

$dmesg (here is part of the information I found)

[ 159.987781] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[ 160.010677] usb 2-1: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[ 160.010681] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 160.010682] usb 2-1: Product: 802.11ac WLAN
[ 160.010683] usb 2-1: Manufacturer: MediaTek Inc.
[ 160.010685] usb 2-1: SerialNumber: 000000000
[ 160.101248] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[ 160.101412] cfg80211: Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[ 160.101595] cfg80211: Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[ 160.101925] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[ 160.103120] platform regulatory.0: firmware: direct-loading firmware regulatory.db
[ 160.103419] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
[ 160.475657] usb 2-1: reset SuperSpeed USB device number 2 using xhci_hcd
[ 160.498763] mt76x2u 2-1:1.0: ASIC revision: 76120044
[ 160.545900] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13
[ 160.545907] xhci_hcd 0000:00:0c.0: Looking for event-dma 0000000002d4b020 trb-start 0000000002d4b040 trb-end 0000000002d4b070 seg-start 0000000002d4b000 seg-end 0000000002d4bff0
[ 302.696837] mt76x2u 2-1:1.0: vendor request req:47 off:9018 failed:-32
[ 302.776191] mt76x2u 2-1:1.0: vendor request req:47 off:9018 failed:-32
[ 302.856309] mt76x2u 2-1:1.0: vendor request req:47 off:9018 failed:-32
[ 302.940423] mt76x2u 2-1:1.0: vendor request req:47 off:9018 failed:-32

Above, thank you for your patience and professional support.

lszhwxa, 2022-06-13:
Too many problems encountered, distracted.

I have returned the 782AC, in my opinion it is not a very good product.

closed 2022-06-13 by lszhwxa