morrownr/USB-WiFi #83: about the AWUS036ACHM

XZCendence, 2022-06-23:
Hi there Nick! Thank you so much for this indispensable repository. You convinced me to buy an AWUS036ACHM and I couldn't be happier with it. It was plug n play with parrot OS on my pi 4 and running a quick airodump blew my mind and I was actually insanely impressed with it. The only issue I've had with it is this strange coil whine noise when I run mdk4 in beacon mode with a high pps speed on it. Most of my other cards don't do this so I wonder what exactly is causing it. I'm going to try different power sources to see if this changes so I'll keep you posted.
I own another MT7610UN based card in the form of the asus USB-AC51, it works pretty well but the range is incomparable to the ACHM, and I initially suspected some badass frontend circuitry to be the culprit. 
**A quick teardown pretty much explains it all; I did a quick reverse engineering of this thing that kinda breaks down the frontend circuitry.**
[image]

[image]
 
Compared to my AC51 with a single frontend module: they're really pushing every drop of perf out of that chipset.
[image]

Thanks again for the recommendation!

XZCendence, 2022-06-23:
I'm a amateur EE at best, and no RF engineer at all, so these drawings may not be 100% accurate, but the definitely not confidential datasheet on the MT7610U (definitely don't google it) doesn't seem to disagree with me.

morrownr, 2022-06-26:
Hi @XZCendence 

Thanks for the post and welcome.

> Hi there Nick! Thank you so much for this indispensable repository. You convinced me to buy an AWUS036ACHM and I couldn't be happier with it. It was plug n play with parrot OS on my pi 4 and running [...]

Glad that this site can help. I had known a site like this that provides the most truthful that we can provide would be helpful to USB WiFi users and the pardemic finally provided time for me to make it happen.

The `whine noise` is something that I remember one other user reporting a couple of years ago and it was never determined what the issue was and I wondered if it might be a one off from a manufacturing issue. The adapter did work fine other than the noise. I have never experienced the noise and there are several ACHM users here so I would expect me reports if it were a common problem. I am willing to try to duplicate but I would need a lot of details of your setup if you have the time to provide it.

You might want to contact Alfa to see what they have to say. I'd like to know what they say.

I changed the title as some companies are not fond of the terminology `reverse engineering" and Alfa does monitor this site.

Let me know what you find out about the noise and if you can provide enough detail regarding hardware, software and configuration, I will attempt to duplicate.

Regards

bdantas, 2022-06-30:
Hi, morrownr. I'm looking for a wireless USB adapter to use on my laptop-turned-wireless-router. I currently use an adapter with the Ralink RT5370 chipset, but I want to explore other options. I see that the ALFA AWUS036ACHM is on your shortlist and is listed as "Excellent for 5 GHz AP mode" and "Outstanding for 2.4 GHz AP mode". I am thinking of getting one and trying it out.

Just one question: Do you know the maximum number of clients that can be connected to the ALFA AWUS036ACHM at the same time? If you don't know the hard maximum, how many devices have you managed to connect at the same time?

morrownr, 2022-06-30:
Hi @bdantas 

> Do you know the maximum number of clients that can be connected to the ALFA AWUS036ACHM at the same time?

If you need a lot of bandwidth, then 5 Ghz is probably the way to go as the ACHM is capable of 80 Mhz channel width and can push 433 Mbp/s. A realistic total throughput is around 200 Mbp/s second as this is half duplex technology. However, that is a lot. That can handle multiple FHD streaming video feeds. I think it is safe to say that most people in the world still have 100 Mbp/s or less internet access so the ACHM will certainly not be the bottleneck.

> If you don't know the hard maximum, how many devices have you managed to connect at the same time?

You can set maximum clients that can connect depending on the hosting software you use. Certainly hostapd lets you set a maximum if you want. I am not aware of a maximum limit imposed by the ACHM. I think the maximum numbers of clients I have had connected when using the ACHM as a 5 Ghz AP has been around 8-10.

Hope this helps.

bdantas, 2022-07-01:
That helps a lot, thank you! I will give the ACHM a try. 
Thank you very much for the wealth of information that you have collected here. This repository is the absolute best resource for USB WiFi adapters on Linux. Kudos.

bdantas, 2022-07-01:
I will be using this as a Wireless N access point, 5 GHz. My ISP speed is 100 Mbps, so 40 MHz channel width should give me enough throughput that my ISP speed will be the bottleneck. Can you help me with the relevant hostapd settings? Specifically, how do I specify the 40 MHz channel width?

Here are the relevant settings I have so far:

```
channel=150
# for 5 GHz:
hw_mode=a
# for Wireless N:
ieee80211n=1
wmm_enabled=1
```

morrownr, 2022-07-01:
> channel=56

When last I checked the in-kernel drivers for the 7610u and 7612u chipsets, they do not support DFS channels in AP mode. It is not clear why. I have seen work that might indicate it is coming but have not checked for sure so you should stay with the regular channels for your country, not the DFS.

> # for 5 GHz:
> hw_mode=a

Good.

# IEEE 802.11n
ieee80211n=1
wmm_enabled=1
#
# mt7612u - mt7610u
ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]

Those will work.

Much more info is available in item 9 in the Main Menu here.

bdantas, 2022-07-01:
Thank you for that. Last question: I'm in the USA. By "regular channels" do you mean 36, 40, 44, and 48?

morrownr, 2022-07-01:
>  I'm in the USA. By "regular channels" do you mean 36, 40, 44, and 48?

Yes but there are more: 149, 153, 157, 161 and 165.

bdantas, 2022-07-01:
Thank you so much. I ordered one of these. As soon as it arrives I think I my home wireless network will undergo a massive upgrade.

morrownr, 2022-07-04:
Give us a report when able. Hope you enjoy it as much as I have mine.

I didn't notice the range for a while after I had it because it was in a place where range was not an issue. At some point, someone requested that I do a performance test and the ACHM just smoked everything I threw at it in the range test.

bdantas, 2022-07-07:
I've been really enjoying this adapter. Definitely much better range than the adapter that I was previously using (one with Ralink RT5370 chipset) on my homebrewed linux-laptop-turned-wireless-router. Wireless client download speeds have been around 90 Mbps when AP is Wireless N using 5 GHz band and 40 Mhz channel width.
 
The one fly in the ointment, which is not the adapter's fault, is that I have three devices in my home network that are 2.4 GHz only. It seems safe to assume that the only way to have a dual band homebrewed router is to use two USB wireless adapters, which seems more complicated than it's worth.

So I'm stuck with using the ACHM to create a Wireless N 2.4 GHz AP. Client download speeds are in 30-35 Mbps range. Are there any HT, UHT, or other settings that could help me squeeze out better speeds from this adapter if I'm locked into 2.4 GHz channels?

morrownr, 2022-07-07:
> I've been really enjoying this adapter. Definitely much better range than the adapter that I was previously using (Ralink RT5370 chipset) on my homebrewed linux-laptop-turned-wireless-router. Wireless client download speeds have been around 90 Mbps when AP is Wireless N using 5 GHz band and 40 Mhz channel width.

Not a bad speed at all. When you need it one of these days, you can bump up to Wireless AC and 80 Mhz channel width.
 
> But I found a fly in the ointment, which is not the adapter's fault: Three devices in my home network are 2.4 GHz only. I don't want to complicate things and connect a second adapter to my wireless router. It seems I'm stuck with 2.4 GHz :'-(

Hmmm... are you using hostapd? If so, using two adapters is no problem. I can show you how. You can run the ACHM in 5 Ghz mode and the rt5370 in 2.4 Ghz mode.
 
> Client download speeds when connected to the ACHM's Wireless N 2.4 GHz hotspot are 30-35 Mbps. What settings would you recommend for me to get the fastest possible speeds with this device while limited to 2.4 GHz?

The ACHM and the rt5370 are both capable of 40 Mhz channel width in 2.4 Ghz mode. If you live where there are not many other wifi signals around, there is no good reason not to do it.

> This is my current hostapd configuration:
> 
> ```
> beacon_int=100
> ssid=myssid                        
> interface=wlan1
> driver=nl80211
> channel=6
> ignore_broadcast_ssid=0
> ap_isolate=0
> hw_mode=g
> wpa=2
> wpa_passphrase=topsecret
> wpa_key_mgmt=WPA-PSK
> wpa_pairwise=TKIP CCMP
> rsn_pairwise=CCMP
> ieee80211n=1
> wmm_enabled=1
> ```

I'm out of time for now but can give you the details needed to go to 40 Mhz width so you can test. Ping me with your thoughts on the above as you have time.

bdantas, 2022-07-08:
> Hmmm... are you using hostapd? If so, using two adapters is no problem. I can show you how. You can run the ACHM in 5 Ghz mode and the rt5370 in 2.4 Ghz mode.
> 
Yes, I am using hostapd. With dnsmasq as DHCP server. I like the idea of running the ACHM in 5 GHz mode and my RT5370 in 2.4 GHz mode. I'd love to see your ideas on making that work. In the meantime I'll do some of my own tests.

If the above doesn't work, then we could try running the ACHM in 2.4 GHz with 40 MHz channel width. 

Thank you for being so generous with your knowledge and time.

bdantas, 2022-07-08:
Never mind. I'm going to solve this by migrating to an all-5 GHz network (I will either upgrade the wireless cards of the old 2.4 GHz devices or, for the devices where that's not possible, upgrade the whole device).

morrownr, 2022-07-08:
Are you sure you want to do that? Most of these IoT things are 2.4 GHz only...and you will buy some at some point.

It is easier than you may thing to run both bands with hostapd.

bdantas, 2022-07-08:
You make very good points. But I have a ton of questions about such a setup: Where can I find a sample hostapd configurations? Would the two adapters be on same subnet or on different ones? Would I need to run two DHCP servers (in my specific case, I use dnsmasq), one for each AP?

morrownr, 2022-07-08:
> You make very good points. But I have a ton of questions about such a setup: Where can I find a sample hostapd configurations?

Go to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Item 9 is a long detailed guide on how to set up a wireless access point. It gives examples of both 5 GHz and 2.4 Ghz hostapd.conf files. You will need two files if you run two adapters. Note that the names are not hostapd.conf.

> Would the two adapters be on same subnet or on different ones?

My guide runs both on the same subnet.

 Would I need to run two DHCP servers (e.g., dnsmasq), one for each AP?

No.

Read the guide and bring your questions back here. I don't know how you access point is setup so it is best if you read my guide and see how it is different than what you have.

morrownr, 2022-07-08:
On second thought, can you start a new issue about `Setting up a dual band access point` or thereabouts since this thread has drifted a lot?

Andy-45, 2024-09-15:
@XZCendence

this inspired me to put my AWUS036ACHM behind an acrylic (Arduino Uno) case:

[image]

XZCendence, 2024-09-15:
> @XZCendence
> 
> 
> 
> this inspired me to put my AWUS036ACHM behind an acrylic (Arduino Uno) case:
> 
> 
> 
> 
> 
> [image]
> 
> 

@Andy-45 This definitely looks pretty cool! I wish they made a fully translucent blue or red case for this.

morrownr, 2024-09-17:
@Andy-45 

That does look cool. It looks like the case on my RasPi4B. Maybe we should leave some messages on Alfa's website? I like the idea of the translucent blue or red cases... maybe even green.