morrownr/USB-WiFi #84: can't create access point with AWUS036ACHM: frequency not allowed

bdantas, 2022-07-02:
I just received my AWUS036ACHM but was not successful in upgrading my home wireless router to 5 GHz.

This is my hostapd.conf:
```
beacon_int=100
ssid=HomeSweetHome
interface=wlan1
driver=nl80211
country_code=US
channel=44
ignore_broadcast_ssid=0
ap_isolate=0
# g means 2.4 GHz, a means 5 GHz:
hw_mode=a
wpa=2 
wpa_passphrase=topsecret
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP CCMP
rsn_pairwise=CCMP

# for 802.11n support
ieee80211n=1
wmm_enabled=1

# 802.11 high throughput options
ht_capab=[HT40-][SHORT-GI-20][SHORT-GI-40]
```

This is what shows up in the terminal I get when I try to run hostapd:
```
wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Frequency 5805 (primary) not allowed for AP mode, flags: 0x6073 NO-IR
Primary frequency not allowed
wlan1: IEEE 802.11 Configured channel (161) or frequency (5805) (secondary_channel=-1) not found from the channel list of the current mode (2) IEEE 802.11a
wlan1: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
wlan1: interface state COUNTRY_UPDATE->DISABLED
wlan1: AP-DISABLED 
wlan1: interface state DISABLED->DISABLED
wlan1: AP-DISABLED 
wlan1: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan1 wasn't started
nl80211: deinit ifname=wlan1 disabled_11b_rates=0
```
I have tried all the regular channels (36, 40, 44, 48, 149, 153, 157, 161, and 165) and result was the same.

Any ideas how to get hostapd to create a 5 GHz hotspot with this device?

morrownr, 2022-07-02:
> Any ideas how to get hostapd to create a 5 GHz hotspot with this device?

This is not the fault of the ACHM.

To investigate, I need to know what distro and hardware you are using. Please be specific. I have noticed an increase of issues having to do with country information not getting where it needs to be and this appears to be the case in this situation. This is the first case with an in-kernel driver. I have a work around for out-of-kernel drivers but am not sure what has happened to cause this so it is time to investigate.

Regards

bdantas, 2022-07-02:
Here is the information you requested:
- Distro: Tiny Core Linux 13.1 x86_64
- Hardware: Thinkpad X200 with Libreboot
- in case it's relevant: hostapd is version 2.10
 
If hostapd.conf's country_code needs to match the country the OS thinks it's in, then I wouldn't be surprised if TCL is to blame. TCL only comes out of the box with what's absolutely required for the OS to work. I don't think it has any concept of country. I like TCL's philosophy but sometimes it does create some snags.

bdantas, 2022-07-02:
Kernel and in-kernel driver configuration is here:
http://repo.tinycorelinux.net/13.x/x86_64/release/src/kernel/config-5.15.10-tinycore64

I can recompile the kernel and/or drivers if a configuration option needs to be tweaked.

For now I'm using this device to create a 2.4 GHz hotspot and it is working well. 
 
P.S. With 2.4 GHz AP, I'm getting download speed of 30-35 Mbps. Wired ISP speed is ~100 Mbps. If it's realistic to get 50 Mbps or higher with 5 GHz and 40 MHz channel width, I'm definitely interested in making it work.

morrownr, 2022-07-02:
> For now I'm using this device to create a 2.4 GHz hotspot and it is working well.

Yes, 2.4 Ghz should work well without country data because the lowest common denominator, channels 1-11, is the same worldwide. Some countries allow channels 12-14 so they would probably not work given the situation. This tends to confirm that this is a country data situation.

The following would normally tell us the country data:

$ sudo iw reg get

...but I don't know if you have `iw` installed.

Does TCL have forums where you can ask about making the country setting available to a wifi driver?

bdantas, 2022-07-02:
I do have iw installed and it does seem that there is no country data:
```
$ sudo iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#1
country 99: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 40), (N/A, 20), (N/A), PASSIVE-SCAN
        (5140 - 5360 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN
        (5460 - 5860 @ 80), (N/A, 30), (N/A), PASSIVE-SCAN
```
I will ask on the TCL forum how to make the country setting available to wifi drivers. I'll report back here.

bdantas, 2022-07-03:
TCL guru Rich helped me figure out the solution to setting the country on TCL. Solution was basically to add two files (/lib/firmware/regulatory.db and /lib/firmware/regulatory.db.p7s) to the system. Gory details here: http://forum.tinycorelinux.net/index.php/topic,25840.0.html

The last gotcha was that `hostapd` kept complaining about "HT40 channel pair not allowed" and was very unstable, dropping connections once every minute or so. 

I had to try different channels and different `ht_capab` settings. Eventually I found that using channel 44 and adding `[HT40+]` (so that the relevant line looks like this: `ht_capab=[HT40-][HT40+][SHORT-GI-20][SHORT-GI-40]`) made the complaints and stability issues go away.

Thank you for all your help, morrownr. Download speeds for wireless clients on my home network increased from 30-35 Mbps to 80-90 Mbps!

closed 2022-07-03 by bdantas

morrownr, 2022-07-03:
I can help you get AC, 80 Mhz channel width going if you want.

bdantas, 2022-07-04:
I'm intrigued. Do you think it would be advantageous for me? My wired ISP download speed is ~100 Mbps, all my devices are WiFi N only, and I'm getting all the range I need with this adapter on N with 5 GHz.

morrownr, 2022-07-04:
Well, life being what it is, now that you have your setup optimized for what you are currently doing, you will mess around and add some storage or a web cam feeding video or something else to your LAN. When you do that, you know where to get me.

Regards