morrownr/USB-WiFi #86: Setting up a dual band access point

bdantas, 2022-07-09:
Thank you for all your help so far. As you know, I got a AWUS036ACHM and set up a nice 5 GHz AP with it. Now I'm trying to add a 2.4 GHz AP on the same machine's Ralink RT5370 adapter (wlan2).

After machine is up and 5 GHz AP is working beautifully (on wlan1, which has local ip of 192.168.10.1), I create this script as /opt/hotspot2G.sh:

```
#!/bin/sh   
              
lan_if=wlan2      
ssid=home_2G    
password=topsecret
ip_stem=192.168.10
channel=6
      
main()                                                     
{                                                          
ip addr add ${ip_stem}.2/24 dev $lan_if
setup_hostapd
}              
               
setup_hostapd()
{             
echo "        
beacon_int=100   
ssid=$ssid       
interface=$lan_if
driver=nl80211         
channel=$channel       
ignore_broadcast_ssid=0
ap_isolate=0
hw_mode=g               
wpa=2                   
wpa_passphrase=$password
wpa_key_mgmt=WPA-PSK  
wpa_pairwise=TKIP CCMP
rsn_pairwise=CCMP
ieee80211n=1                  
wmm_enabled=1                 
" >/tmp/hostapd_${lan_if}.conf                       
                                                     
hostapd /tmp/hostapd_${lan_if}.conf # >/dev/null 2>&1
}   
    
main
```
Then run `sudo /opt/hotspot2G.sh &`

wlan2 comes alive. wireless clients are able to authenticate and get an ip address from the dnsmasq instance that's already running, but there is no internet connectivity.

Maybe router's routing table is ambiguous and packets that should be leaving the router through wlan2 are actually leaving it through wlan1?
```
my_router$ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         c-73-198-144-1. 0.0.0.0         UG    0      0        0 eth0
73.198.144.0    *               255.255.248.0   U     0      0        0 eth0
127.0.0.1       *               255.255.255.255 UH    0      0        0 lo
192.168.10.0    *               255.255.255.0   U     0      0        0 wlan1
192.168.10.0    *               255.255.255.0   U     0      0        0 wlan2
```
I'm stuck. It seems that `hostapd` and `dnsmasq` are both doing their jobs but there is a problem with routing. Is some kind of bridge needed? Routing is not my forte.

bdantas, 2022-07-09:
Is the fact that both wlan1 (ACHM 5GHz) and wlan2 (Ralink 2.4GHz) are on the same subnet (192.168.10.x) the issue? I would prefer everything on the same subnet if it is workable.

bdantas, 2022-07-09:
P.S. I did look at item 9 of your main page here, but I cannot follow many of the steps because my router is not a Raspberry Pi, does not have a persistent filesystem, does not use systemd, does not use apt for package  management, etc. It is Tiny Core Linux, which uses Busybox init. So it's hard for me to know how much of the instructions are implementation-specific and how much are things that need to be accomplished regardless of implementation.

bdantas, 2022-07-09:
Hi, morrownr. I think all my problems will disappear if I put each adapter on a different subnet. Please don't spend any time on this. I'm going to do some experiments and will post my solution here if I'm successful.

bdantas, 2022-07-09:
Success! As I suspected, putting each adapter on a different subnet made all problems go away.

In case it helps anyone, this is how I solved this:

1. You need **two USB wireless adapters**. One needs to support 5GHz AP mode. The other one can be more primitive, needs only to support 2.4GHz AP mode. In my case, I'm using AWUS036ACHM to create the 5GHz AP and am using a cheaper adapter with Ralink RT5370 to create the 2.4GHz AP.

2. **Figure out which adapter has which logical name**. Use whatever tools you need (`iw dev`, `lsusb`, etc.). In my case, using the particular USB ports I chose, the ACHM is wlan1 and the RT5370 is wlan2. There's probably a more correct way to ensure that the same logical name always refers to the same adapter (e.g., predictable names or udev rules), but I'm happy to just rely on the fact that, after multiple reboots, wlan1 and wlan2 always refer to the expected adapter.

3. **Create these three shell scripts and make them executable**:

This one is `/opt/setup-dnsmasq`:
```
#!/bin/sh

lan_if1=wlan1
ip_stem1=192.168.10
lan_if2=wlan2
ip_stem2=192.168.20

echo "
dhcp-leasefile=/tmp/dnsmasq.leases
dhcp-range=$lan_if1,$ip_stem1.100,$ip_stem1.159,255.255.255.0,24h
dhcp-range=$lan_if2,$ip_stem2.100,$ip_stem2.159,255.255.255.0,24h
" >/tmp/dnsmasq.conf

dnsmasq -C /tmp/dnsmasq.conf >/dev/null 2>&1 &
```
 
This one is `/opt/setup-5GHz-hotspot`:
```
#!/bin/sh

lan_if=wlan1
ssid=home_5G
password=topsecret
channel=44

echo "
beacon_int=100
ssid=$ssid
interface=$lan_if
driver=nl80211
country_code=US
channel=$channel
ignore_broadcast_ssid=0
ap_isolate=0
hw_mode=a
wpa=2
wpa_passphrase=$password
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP CCMP
rsn_pairwise=CCMP
ieee80211n=1
wmm_enabled=1
ht_capab=[HT40-][HT40+][SHORT-GI-20][SHORT-GI-40]
" >/tmp/hostapd_${lan_if}.conf

hostapd /tmp/hostapd_${lan_if}.conf >/dev/null 2>&1 
```
 
This one is `/opt/setup-2GHz-hotspot`:
```
#!/bin/sh

lan_if=wlan2
ssid=home_2G
password=topsecret
channel=6

echo "
beacon_int=100
ssid=$ssid
interface=$lan_if
driver=nl80211
channel=$channel
ignore_broadcast_ssid=0
ap_isolate=0
hw_mode=g
wpa=2
wpa_passphrase=$password
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP CCMP
rsn_pairwise=CCMP
ieee80211n=1
wmm_enabled=1
" >/tmp/hostapd_${lan_if}.conf

hostapd /tmp/hostapd_${lan_if}.conf >/dev/null 2>&1 &
```
 
4. Run these commands at each boot, as root:
```
echo 1 >/proc/sys/net/ipv4/ip_forward
ip addr add 192.168.10.1/24 dev wlan1
ip addr add 192.168.20.1/24 dev wlan2
/opt/setup-dnsmasq
/opt/setup-5GHz-hotspot
/opt/setup-2GHz-hotspot
```
 
Now I have created a dual-band wireless router out of my GNU/Linux laptop. wlan1 provides a 5GHz AP on the 192.168.10.x subnet, wlan2 provides a 2.4GHz AP on the 192.168.20.x subnet. Password for the two APs is the same. There are two instances of `hostapd` running (one per AP) but only a single instance of `dnsmasq`, which handles DHCP requests on both APs.

I hope this helps someone.

morrownr, thank you for your support and for encouraging me to pursue creating a dual-band router. I couldn't be happier with the result!

closed 2022-07-09 by bdantas

morrownr, 2022-07-09:
> Hi, morrownr. I think all my problems will disappear if I put each adapter on a different subnet. Please don't spend any time on this. I'm going to do some experiments and will post my solution here if I'm successful.

I did not expect my guide to be exactly what you needed, My guide expects systemd. I was really trying to point you to the hostapd.conf files as they should give you a good idea of the contents needed to optimize setting up hostapd with 2 or more adapters and the speific information for your adapters is in them.

Something that might be of use during your work to get this dual band setup going:

https://www.raspberrypi.com/software/raspberry-pi-desktop/

I think you mentioned that you are using an x86 based computer. Well, there is an x86 based version of the RasPiOS. It is called the Raspberry Pi Desktop for MAC and PC. It would make my guide easy to use. There is also an x86 version of OpenWRT. Both of these distros required limited resources. 

Keep us posted on your progress.

bdantas, 2022-07-09:
> Keep us posted on your progress.
It's 100% solved :) See above.
Thanks again for all your support!

morrownr, 2022-07-09:
I think we posted at the same time.

Glad it is working. Sorry that I could not be specific help but I currently have no experience with Tiny Core and I had very limited visibility on what you were doing. There are many ways to set this up. If you are interested in fully developing your guide, I will make space for it on the site.

Regards