morrownr/USB-WiFi #87: (hot info) New USB WiFi Adapters based on the mt7921au chipset are now available

morrownr, 2022-07-09:
Incoming data thanks to @haevalencia

CF-953AX - chipset: mt7921au
http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=182

CF- 951AX - chipset: mt7921au
http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=209

AliExpress
https://es.aliexpress.com/item/3256804242387470.html?gatewayAdapt=4itemAdapt
There are 5 adapters shown on the above page. The 2 that are powered by the new mt7921au chipset are the CF-953AX and CF-951AX.

Important info: The mt7921 base driver has been in the mainline Linux kernel since 5.12 so internal cards have been working well for some time with the exception of AP mode. The above devices are usb adapters and usb support was not added to the mt7921 driver until kernel 5.18. Most mainline distros provide instructions for upgrading. I can provide instructions for Ubuntu based distros on request. AP mode support is being adding right now in kernel 5.19. It requires a firmware upgrade, see Main Menu 8.

I suspect both of the above adapters are multi-state being that they are from ComFast. See Main Menu 7 for more information.

FYI: I see from the ad that the chipset is called mt7921au. This is the first time I have seen that specific name. Maybe the `au` means will be see a `bu` that is tri-band and uses the mt7921k chipset or the mt7922 chipset.

Disclaimer: I am not recommending the above adapters as I have no experience with them nor do I have any Linux user reports As I am able to buy and test the above adapters or as I get user reports, I will then look at adding the adapters to the list in Main Menu 2 list as recommended adapters.

Regards

haevalencia, 2022-07-10:
I confirm that these adapters are multi-state. The vendor confirmed to me that they have received "suggestions from users" to sell adapters with a MediaTek chip including a CD with drivers, just as they do with Realtek chips. The truth is that I don't understand why Comfast limit the support that MediaTek chips already have using that system that is designed only for Windows. Anyway.

Some stores on AliExpress are already preparing devices with this new MediaTek chip like Edup Official Store which will released it soon. The [Anddear Store](https://es.aliexpress.com/store/3624001), which makes the current unbranded adapter with the MT7612U chip, said that if there was interest from users, it would work on a new model with the MT7921au chip. So go show your interest wink wink.

amisix, 2022-07-10:
Thanks for contacting ANDDEAR. Be wary - they are in my opinion currently selling non-functional mt7612u adapters. My hope would be they don't repeat the same mistake(s) with their mt7621au adapters.

bcdonadio, 2022-07-10:
I bought two units of the `CF-953AX` dongle with `mt7921au` from AliExpress, supposedly to be delivered until August 06. Let's see if they work at all when they arrive.

Using your really useful, but obviously lacking in features, `8814au` module in a `TP-Link Archer T9UH` has been not so comfortable. Looks like even range is severely reduced when comparing with my (impressively good, featureful and stable on Fedora 35 with kernel 5.18.6, but not really up to date with the standards) onboard `Qualcomm Atheros QCA6174` using the `ath10k_pci` module.

Will report when they arrive.

morrownr, 2022-07-10:
> The vendor confirmed to me that they have received "suggestions from users" to sell adapters with a MediaTek chip including a CD with drivers, just as they do with Realtek chips. 

Hopefully we can spread the word to Linux users to be careful what they ask for. We should be asking for single-state adapters and if the maker wants to include a small, cheap flash drive for the Windows and MAC drivers, that would be great. The reason I word it this way is because out-of-kernel drivers, such as those that Realtek makes, do not work well because of the development model of Linux. Unlike Windows or the MAC OS, Linux does not lock apis for long periods. Linux is under constant development so the only way to keep device drivers up to date and working is for them to be located in the mainstream kernel. Linux users do not want Linux drivers on a CD...it will not work. Period.

The 5 out-of-kernel Realtek drivers that I and other maintain here make for a lot of work and we do it with no help from Realtek. Realtek is doing things wrong. Linux device drivers need to be developed in-kernel and they need to meet the applicable standards. In the case of the Realtek out-of-kernel drivers, the driver are not in-kernel and they do not meet published stanards. Thank goodness for Mediatek. They are doing it right.

> Some stores on AliExpress are already preparing devices with this new MediaTek chip like Edup Official Store which will released it soon. The [Anddear [...]

I will be showing my interest by talking directly to makers and resellers. Come on ALFA!

morrownr, 2022-07-10:
> Thanks for contacting ANDDEAR. Be wary - they are in my opinion currently selling non-functional mt7612u adapters. My hope would be they don't repeat the same mistake(s) with their mt7621au adapters.

@haevalencia 

@amisix speaks words of wisdom. The unbranded mt7612u adapter in the Anddear store is to be avoided. I did have it in the list of recommended adapters for a short period but a serious bug was found. We researched and tested the issue and can only conclude that there is an engineering error somewhere in that adapter. Try to use it in USB3 port 1 and see what happens. The unbranded mt7610u adapter works well and is still listed.

Regards

morrownr, 2022-07-10:
Hi @bcdonadio 

> I bought two units of the `Comfast CF-953AX` dongle with this `mt7921au` chipset from AliExpress, supposedly to be delivered until August 06. Let's see if they work at all when (and if) they arrive.

We are waiting on your review. Cool.

> Using your really useful, but obviously lacking in features, `8814au` module in a `TP-Link Archer T9UH` has been not so comfortable. Looks like even range is severely reduced when comparing with my [...]

I wish the 8814au driver was better because a lot of Linux users have adapters based on that chipset. At one point last year, a post doctoral student contacted me as he was doing research on beamforming. He was doing a deep dive on the beamforming support in 8814au and he reported that things are not good in 8814au. It is hard to work on the Realtek drivers and we get zero help from Realtek so it is a minor miracle that we have that driver as stable as it is. I really think Linux users that want to work on code should concentrate on the in-kernel Mediatek MT76 and MT7601u drivers.
 
> Will report back when I get my hands on them. 

Did I mention that we are waiting? He he.

bcdonadio, 2022-07-10:
@morrownr drop a link here to an AliExpress, Amazon wish list or even a Paypal account so I can send this adapter (or another one you may want) your way. It's only fair with all this awesome work you've been doing for the community. My hardware would be an expensive brick without your effort. I would have sent you something already trough Github Sponsors if it weren't a pain to setup. You can send the link [to my email](mailto:bcdonadio@bcdonadio.com) too if you prefer.

Thanks for all your effort and be safe!

haevalencia, 2022-07-11:
> @haevalencia
>
> @amisix speaks words of wisdom. The unbranded mt7612u adapter in the Anddear store is to be avoided. I did have it in the list of recommended adapters for a short period but a serious [...]

Oh. What is the bug? I have been using it on several laptops and even my ThinkPad T495 which has a 3.x and 3.1 gen 2 (10Gbps) USB-A ports and no visible issues.

morrownr, 2022-07-11:
Use `$ lsusb -t` to move the adapter around until it is in a USB3 port 1.

Not a port 2 or port 3 or port 4.

@amisix and I have not been able to get it to get it to work or at least work well in a port 1. If it does work and you put iperf3 on a run of 300 seconds or more it will slow down and eventually stop. Sometimes it will show the same symptoms on port 2. This has happened on a variety of hardware. I have some test results around here somewhere. I'm currently in major maintenance mode so I'll need to to local the file.

amisix, 2022-07-11:
It doesn't initialize/work in certain USB3 ports, issue was originally identified on a Raspberry Pi 4B then replicated on several other x86 platforms. USB3 Port 1 "Top" USB port fails, USB3 Port 2 "Bottom" USB Port functions. It will initialize in a USB2 port intermittently but ultimately fail while running iperf3 as referenced by @morrownr.

Original Thread: 
https://github.com/morrownr/USB-WiFi/issues/75#issuecomment-1123143847 

Discussion with WLAN-PI team regarding this issue (replicated on their Pi setup and x86 hardware). 
https://github.com/WLAN-Pi/feedback/discussions/21#discussioncomment-2906912

morrownr, 2022-07-12:
> @morrownr drop a link here to an AliExpress, Amazon wish list or even a Paypal account so I can send this adapter (or another one you may want) your way. It's only fair with all this awesome work [...]

@bcdonadio 

I appreciate the kind words and offer. I might set something up someday but for now let's just say that I am paying forward what others did to help me over the years. 

Nick

croissantpetrichor, 2022-07-13:
>Thank goodness for Mediatek. They are doing it right.

Partly, I mean sure their hardware has good specs and they have great drivers but how much is that really worth when you can't even acquire their products. Let's be honest either they don't care or they've got a little Nvidia market manipulation "chipset shortage" thing going on.

> I will be showing my interest by talking directly to makers and resellers. Come on ALFA!

When's the last time those guys manufactured anything new and interesting? Couple of years ago, yeah, they don't care...

Not sure if this one's been spotted yet but here you are, [Gigabyte PCIe MT7921K](https://www.gigabyte.com/Motherboard/GC-WBAX1200-rev-10) (MT7921K (RZ608)) 40$.

morrownr, 2022-07-13:
> Sure to some extend there is a chipset shortage but RealTek among others seem to be having no issues pushing out there products. Nothing generates profits like a global conflict, am I right?

@croissantpetrichor 

I am all for having a good discussion about economics as it has to do with USB WiFi in some way. Your message is taking the original topic off course so I am requesting that you start a new topic in the Discussion section and I am requesting that you refrain from using foul language is I would like to keep this site G rated.

Would I enjoy a discussion about economics and business practices around the world? Certainly. I don't give much information about my background usually but I will here so that you know: I have undergraduate and graduate degrees in Economics. I have taught economics at the university level. I learned programming using FORTRAN on an IBM mainframe. That should give you an idea of my age. I have little tolerance for conspiracy theories. I do my best to make sure the information on this site is useful and factual.

I understand a great deal about how businesses work and what factors are at work in the world economy. In the world of IT related products, some companies own their own fabs. Some do not. Fab plants are extremely expensive. Whether a company owns their own fab can have a great deal of influence on how they are able to conduct business.

I look forward to you starting a new discussion.

Regards,

Nick

croissantpetrichor, 2022-07-16:
>I am all for having a good discussion about economics as it has to do with USB WiFi in some way.

Doesn't everything have to do something with something?

>Your message is taking the original topic off course 

You're right and GitHub isn't exactly the appropriate platform for this.

>so I am requesting that you start a new topic in the Discussion section

Do you really want to discuss economics on GitHub? I mean ultimately we can beat around the bush all we want. The new MediaTek chipsets are currently just not available. Blame it on economics, blame it on Admin Smith it doesn't matter, there's nothing we can do in order to change that.

>I am requesting that you refrain from using foul language is I would like to keep this site G rated.

I think it's important to note that I normally don't swear at all, more on this later.

>Would I enjoy a discussion about economics and business practices around the world? I don't give much information about my background usually but I will here so that you know: I have undergraduate and graduate degrees in Economics.

Given your mention about having studied economics I doubt we'd come to agreement about how things work.

>I have taught economics at the university level. I learned programming using FORTRAN on an IBM mainframe. That should give you an idea of my age.

That's a good thing, I personally think the best programmers are born due to their understanding of "low level" languages because grasping them really helps you understand on a deeper level how a computer works. 

>I have little tolerance for conspiracy theories. I do my best to make sure the information on this site is useful and factual.

I probably should've worded that differently, it's not necessarily a conspiracy theory if there's some truth in it an well I think we all can establish that all markets in the world are in one way or another manipulated.

I don't mean to come across as someone who believes in crazy conspiracy theories and is actively spreading them, that's not at all the case. More so it was a moment of weakness, desperation, because I've been actively searching for new hardware for about a month now and I lost my patience.

I also should probably note, and not as a reason to excuse my actions but I struggle from severe health issues which sometimes cause me to act irrationally.

> I understand a great deal about how businesses work and what factors are at work in the world economy. In the world of IT related products, some companies own their own fabs. Some do not. Fab plants [...]

Both RealTek and MediaTek are fabless according to Wikipedia. Undoubtedly these factories are incredibly expensive, but the profit some of these companies turn over, or are able to turn over must make such a expense worth it don't you think?

bcdonadio, 2022-07-16:
This one seems pretty humble. I think I want it.
https://www.mediatek.com/products/broadband-wifi/mediatek-filogic-880

Also, I'm gonna have to agree with @morrownr about Mediatek support on Linux. Look at the sheer amount of work on these chipsets that we have coming in to the mainline tree, with a good chunk of those patches authored by Mediatek employees.

[image]

morrownr, 2022-07-17:
> This one seems pretty humble. I think I want it. https://www.mediatek.com/products/broadband-wifi/mediatek-filogic-880
>
> Also, I'm gonna have to agree with @morrownr about Mediatek support on Linux. [...]

I monitor linux-wireless so I see the flow on a daily basis:

http://vger.kernel.org/vger-lists.html#linux-wireless

While it is easy to pick out employees of a company since their email address will be something like @mediatek.com or @intel.com, what may not be obvious unless you monitor the list is that individuals such as Lorenzo Bianconi and Felix Fietkau are likely under contract to Mediatek as they do a LOT of work that they could not do unless they were under the direction of Mediatek and were supported by Mediatek with information that would be needed. Both individuals also appear to have contracts with the Foundation as they work on wireless parts that are maintained by the Foundation.

My opinion is that Mediatek is all in. When is the last time you can remember a USB WiFi adapter driver being in the mainline kernel BEFORE the hardware is available to buy? I'm pretty sure the answer is never. I certainly can't this happening. I will not surprise me to see Mediatek grab a lot of market share from several of their competitors over the next few years.

croissantpetrichor, 2022-07-17:
>My opinion is that Mediatek is all in.

All in is a bold move. I must say however, you're optimism and continual updates are very uplifting and help em remain hope. 

>When is the last time you can remember a USB WiFi adapter driver being in the mainline kernel BEFORE the hardware is available to buy? I'm pretty sure the answer is never. I certainly can't this happening.

You're very much so right.

> While it is easy to pick out employees of a company since their email address will be something like @mediatek.com or @intel.com, what may not be obvious unless you monitor the list is that [...]

That's great, I think this is them @nbd168 and @LorenzoBianconi. Both of them seem to have a lot of experience and primarily devote their attention to wireless communication, they've both developed drivers and worked on OpenWRT.

Felix seems a bit more high profile and has a bigger social media presence, he's even got a speech about [open source WiFi drivers](https://www.youtube.com/watch?v=hiUosbhR0Wo), ironically I think I've come across this video on [hackaday](hackaday.com/) or a similar site in the past already.

I hope MediaTek can continue supporting developers as such, and considering they made [4 766 832.82 USD](https://corp.mediatek.com/investor-relations/financial-information/monthly-revenue) this year in Q1 already I doubt it'll be much of an issue for them, financially speaking anyway. 

If this deems true however I fully agree with you, I think MediaTek will absolutely dominate the Linux  market.

I doubt they're allow to say anything but maybe these gentlemen can tell us a little more about what the future holds?

bcdonadio, 2022-07-17:
Hmm, let's try to take a more "companies are made to be profitable" approach.

USB WiFi devices today are a niche market. Everything is supposed to connect "to the internets" via WiFi or cellular. There's virtually no device today you can buy that does not have adapters for one, the other, or both. Unless you're in a tech conference or something of the sorts, you don't see anyone with a WiFi USB dongle on their notebooks, desktops, home routers, not even mentioning phones. That's where the money is: embedded into all those devices.

Since a real big market share of these kind of devices - namely Android phones, home routers and Chromebooks - run Linux, having drivers for them is a no-brainer. Mainlining that support as GPL was refrained a lot in the past because of licensed intellectual property from third-parties and fears of competitors copying their ideas.

This is changing now because of two main reasons: the most persuasive of those is that a lot of the "real workhorse logic" is in the firmware that you have to load as blobs you have no idea what's in. The second argument is that finally companies are realizing that the open source development model is cheaper than the alternatives in many more scenarios than previously thought, specially if you're targeting a market with a really diverse hardware landscape. Phones and home routers have completely different compromises to be made (power utilization, size, throughput...) and therefore have big different architectural decisions. It really costs less to simply develop a good FOSS interface that gets into mainline (reducing their efforts to maintain the code up to date with ABI changes) than keeping a variety of closed code bases that need to be kept in sync with the mainline and as diverse as customers need.

Having all that, USB support is just a little bit more effort. Of course it's not free, but costs orders of magnitude less than the closed model I defined above. It probably makes sense financially to develop that support even versus the small amount of USB adapters sold. Mediatek is going that route.

Other companies, specially the really big ones, are behemots. Hard to change course and have lots of money to spend inefficiently yet still turn a profit. To mention the cliché tech punch-bag, they're the IBMs. Some may be able to overcome this inefficiency, like Microsoft did, others will perish soon, and a few will linger on supporting themselves by income from other market segments.

My 2 cents, at least.

yaslama, 2022-07-21:
> I suspect both of the above adapters are multi-state being that they are from ComFast. See Main Menu 7 for more information.

I ordered and received both devices (CF-951AX & CF-953AX) and they are **not** multi-state! The box contains a small cdrom with the drivers for Windows.
Alas, the mt7921 module is not enabled in the debian kernel (5.18.0-2-amd64). I opened a bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1015808 for the kernel package. I had to recompile the kernel with the module enabled and I also had to upgrade the firmware files from the linux-firmware git repository, but the device works now.

morrownr, 2022-07-21:
Hi @yaslama 

Thanks for the information. We would appreciate a performance report from you as you are able. Something like the reviews for certain adapters in the following document:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

> I ordered and received both devices (CF-951AX & CF-953AX) and they are not multi-state!

This is great news. Just to be sure, can I get you to disable usb_modeswitch?

Look at the last section at the below link:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md

If they fire up and work with usb_modeswitch deactivated, then we have confirmation that they are single-state.

> Difference:
CF-955AX   1800Mbps  RTL8832AU with PA   2*3dbi Antenna  support Win 10/11
CF-959AX   1800Mbps  RTL8832AU with PA  4*3dbi Antenna  support Win 10/11
CF-957AX   1800Mbps  RTL8832AU with PA 2*2dbi Antenna  support  Win 10/11
CF-951AX   1800Mbps  MT7921AU    Built in 2*3dbi Antenna  support  Win 10/11
CF-953AX   1800Mbps  MT7921AU    2*3dbi Antenna  support  Win /10/11

I was doing a little research the other day and ran across the above. When it did not mention `with PA` I thought to myself, could they be single state?

I realize that I am going to need to include instructions for users since adapters based on the mt7921au are going to require some assistance for now:

- need to have at least kernel 5.18 and the mt7921u module has to have been set to compile by the distro maintainers.

Probably should give instructions for at least one popular distro.

- need to have updated firmware installed. Some distros, like Debian, unless you use an unofficial iso, don't include the firmware anyway. I have instructions and could sure use some folks to test the instructions:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

- for AP mode support, need kernel 5.19 and the most recent firm

Regards

yaslama, 2022-07-22:
> Thanks for the information. We would appreciate a performance report from you as you are able. Something like the reviews for certain adapters in the following document:
> 
> https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

I ran iperf3 with an Turris Omia (not upgraded to wifi 6) and a minipci at a distance of 50cm from the Omnia:

- CF-951AX
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   606 MBytes   508 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   605 MBytes   508 Mbits/sec                  receiver
```
 - CF-953AX
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   659 MBytes   553 Mbits/sec    0             sender
[  5]   0.00-10.09  sec   656 MBytes   545 Mbits/sec                  receiver
```
- AX200
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   498 MBytes   418 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   496 MBytes   416 Mbits/sec                  receiver
```
-- Wired using an ethernet cable
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.08 GBytes   932 Mbits/sec  220             sender
[  5]   0.00-10.00  sec  1.08 GBytes   930 Mbits/sec                  receiver
```
> 
> This is great news. Just to be sure, can I get you to disable usb_modeswitch?

I made sure that no usb-modeswitch is installed in my test system

```
# dpkg -l '*modeswitch*'
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
un  usb-modeswitch <none>       <none>       (no description available)

# lsusb -t -v
...
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Wireless, Driver=btusb, 5000M
        ID 0e8d:7961 MediaTek Inc.
    |__ Port 1: Dev 2, If 1, Class=Wireless, Driver=btusb, 5000M
        ID 0e8d:7961 MediaTek Inc.
    |__ Port 1: Dev 2, If 2, Class=Wireless, Driver=, 5000M
        ID 0e8d:7961 MediaTek Inc.
    |__ Port 1: Dev 2, If 3, Class=Vendor Specific Class, Driver=mt7921u, 5000M
        ID 0e8d:7961 MediaTek Inc.
...
```

>    Some distros, like Debian, unless you use an unofficial iso, don't include the firmware anyway. I have instructions and could sure use some folks to test the instructions:

I think that your instructions are OK. I copied the same 3 files to /lib/firmware/mediatek/
In debian, the firmware is in the package: firmware-misc-nonfree but it's old and there is a bug to upgrade the firmware files (https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1015728)

morrownr, 2022-07-22:
@yaslama 

>I made sure that no usb-modeswitch is installed in my test system.

Thanks for checking again. I am also seeing the same from @foen73 in discussion #88 so I am convinced the CF-953AX is indeed a single state adapter and giving the performance reports so far I think I will add it to the in-kernel list soon.

@foen73 sent along the `iw list` for the CF-953AX. See below. Interestingly enough, it shows Band 4 and the 6 GHz channels. 

```
phy7
	wiphy index: 7
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * monitor
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: managed
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce120dc0b306423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
				HE SU PPDU with 1x HE-LTF and 0.8us GI
				NDP with 4x HE-LTF and 3.2us GI
				STBC Tx <= 80MHz
				STBC Rx <= 80MHz
				Full Bandwidth UL MU-MIMO
				Partial Bandwidth UL MU-MIMO
[... 410 lines cut ...]
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-device: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

morrownr, 2022-07-22:
I'm curious as to why the 6 GHz channels are shown?

objelf, 2022-07-23:
> I'm curious as to why the 6 GHz channels are shown?

Based on the press release it said: MT7921AU supports Wi-Fi 6E and Bluetooth 5.2 from a single, highly integrated platform. 
... Wi-Fi 6E is designed to make use of uncongested bandwidth in the **6GHz** spectrum .... so should it be reasonable to see 6GHz to be supported in the device?

morrownr, 2022-07-23:
Hi @objelf 

I am aware of that press release. I found it yesterday when trying to get to the bottom of this 6 GHz issue. Is the MT7921AU chipset that is used in the new CF-953AX the same as the chipset used in the Samsung TV? That is not clear to me.

If we select `Product Information` at the COMFAST CF-953AX site, I see specs for dual band capability:

http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=182

I would love it if the CF-953AX really is a tri-band adapter. That is what we want. 

Regards

yaslama, 2022-07-24:
> I'm curious as to why the 6 GHz channels are shown?

@morrownr It seems that indeed CF-953AX  (and CF-951AX) supports 6 GHz channels. I tried to run hostapd (from git because the debian version is too old) with a configuration using channel 93 from Band 4 with the CF-953AX and tried to connect to this AP with the CF-951AX in another PC and the connection didn't work (I need to work more on that), **but** I saw that data was exchanged in the 6415 MHz frequency.

morrownr, 2022-07-24:
> I saw that data was exchanged in the 6415 MHz frequency.

Oh wow! 

I have not set up hostapd with 6 GHz due to the lack of a usb wifi adapter that supports 6 GHz. Reading the hostapd.conf docs is probably needed. Would you like for me to read the docs and make suggestions to your hostapd.conf?

yaslama, 2022-07-24:
The hostapd configuration file I used is:
```
interface=wlxe0e1a934a6d1
ssid=MySSID
driver=nl80211
country_code=FR
ieee80211d=1
ieee80211h=1
ieee80211n=1
ieee80211ac=1
ieee80211w=2
beacon_prot=1
hw_mode=a
#channel=36
channel=93
op_class=131
wmm_enabled=1
ht_capab=[HT40]
vht_oper_chwidth=1
ieee80211ax=1
he_6ghz_max_mpdu=1
he_su_beamformer=1
he_su_beamformee=1
he_mu_beamformer=1
he_bss_color=1
he_default_pe_duration=8
he_twt_required=1
he_rts_threshold=2
he_oper_chwidth=1
wpa=2
group_cipher=CCMP
wpa_pairwise=CCMP
wpa_key_mgmt=SAE
sae_password=secret
sae_pwe=2
macaddr_acl=0
```
In order to compile hostapd, I had to create a .config file from defconfig with the following modifications:
```
--- defconfig   2022-07-24 13:38:40.037014896 +0300
+++ .config     2022-07-24 23:07:11.264384645 +0300
@@ -153,10 +153,10 @@
 #CONFIG_WNM=y

 # IEEE 802.11ac (Very High Throughput) support
-#CONFIG_IEEE80211AC=y
+CONFIG_IEEE80211AC=y

 # IEEE 802.11ax HE support
-#CONFIG_IEEE80211AX=y
+CONFIG_IEEE80211AX=y

 # IEEE 802.11be EHT support
 # CONFIG_IEEE80211AX is mandatory for setting CONFIG_IEEE80211BE.
@@ -166,10 +166,10 @@
 #CONFIG_IEEE80211BE=y

 # Simultaneous Authentication of Equals (SAE), WPA3-Personal
-#CONFIG_SAE=y
+CONFIG_SAE=y

 # SAE Public Key, WPA3-Personal
-#CONFIG_SAE_PK=y
+CONFIG_SAE_PK=y

 # Remove debugging code that is printing out debug messages to stdout.
 # This can be used to reduce the size of the hostapd considerably if debugging
```

I configured manually the ip address of the AP and in NetworkManager of the client using CF-951AX. hostapd seems to be ok with the config file above, but the client's NM didn't connect after several attempts to authenticate. I saw in both logs that data was exchanged. I will perhaps try to use directly wpa_supplicant in the client..

morrownr, 2022-07-24:
> # IEEE 802.11be EHT support
 # CONFIG_IEEE80211AX is mandatory for setting CONFIG_IEEE80211BE.
@@ -166,10 +166,10 @@
 #CONFIG_IEEE80211BE=y

I think you need to uncomment the following when compiling as well:

CONFIG_IEEE80211BE=y

Give me a change to study the situation a little more,

Thanks for doing this.

yaslama, 2022-07-24:
> ```
> > # IEEE 802.11be EHT support
> > # CONFIG_IEEE80211AX is mandatory for setting CONFIG_IEEE80211BE.
> > @@ -166,10 +166,10 @@
> > #CONFIG_IEEE80211BE=y
> ```
> 
> I think you need to uncomment the following when compiling as well:
> 
> CONFIG_IEEE80211BE=y

CONFIG_IEEE80211BE is wifi7. It depends on CONFIG_IEEE80211AX but CONFIG_IEEE80211AX doesn't depend on CONFIG_IEEE80211BE.

bcdonadio, 2022-07-24:
@morrownr I _should_ be receiving (customs cleared, waiting for the postal service) the two CF-953AX adapters I bought sometime this week. Besides the tests above, anything else you want me to test?

Also, do you want me to setup one or two systems connected to them with SSH access for you to play with?

morrownr, 2022-07-24:
> > I think you need to uncomment the following when compiling as well:
> > CONFIG_IEEE80211BE=y
> 
> CONFIG_IEEE80211BE is wifi7. It depends on CONFIG_IEEE80211AX but CONFIG_IEEE80211AX doesn't depend on CONFIG_IEEE80211BE.

I got that one wrong. I need some time to soak in the new WiFi 6 stuff.

Here is one thought: I see you are activating WPA3. There is nothing wrong with that and I don't see any problems with your hostpad.conf in this regards but WPA3 requires a lot of things to be working right for everything to work right now. Maybe you should stick with WPA2 until you get 11AX working. Just a thought.

morrownr, 2022-07-24:
> @morrownr I _should_ be receiving (customs cleared, waiting for the postal service) the two CF-953AX adapters I bought sometime this week. Besides the tests above, anything else you want me to [...]

@bcdonadio 

That is a nice offer. We certainly need a product review or two to go with the [in-kernel links](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md) file but right now I think the highest priority is getting this 6 GHz capability sorted out.

Do you have something like a RasPi4B and an extra sd card that you could use to setup an AP with one of the adapters?

bcdonadio, 2022-07-24:
@morrownr I do, but I think it would be easier making it work on x86_64 first. I will plug them into two Intel NUCs with i7 and 32GB of RAM I have laying around first in the likely case a kernel build is needed. This way we also avoid the bugs of two chips connected using the same module connected to the same system.

I will also dig for a 6-7GHz transverter I _might_ have somewhere to plug into the spectrum scanner and see if indeed there are emissions in the 6E band.

morrownr, 2022-07-24:
@bcdonadio 

> I do, but I think it would be easier making it work on x86_64 first. 

It might be easier that way but it depends on your experience. Let me throw out the link to my Bridged Access Point guide as it might have some info to speed things along:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

I'm also working on a routed guide but it is not published yet.

Here is the current hostapd.conf from the main hostapd site:

https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf

I'm reading the above trying to figure out how to mode the hostapd.conf's in my guide so maybe I can be of help as you set things up.

Regards

bcdonadio, 2022-07-24:
The postal service will still take some days to deliver, but my week will be a quite busy with work. Likely I will only have time to dig deep into this next weekend.

I'm not really intimate with hostapd and wpa_supplicant besides what I needed to get 802.11ac/r/kv working on them some time ago on OpenWRT. Your guide will be really helpful.

I will plug the NUCs into the PDUs and switches and at least try to leave a bare Fedora 36 running on them, ready for when I receive the adapters. I will find a way to leave the adapters themselves not in the server room but close to me in case we need to keep unplugging and replugging them, and get them connected to the 10GbE switch to be able to get nice capture streams as well because I want to test the VLAN-aware Linux bridge code with them.

Regarding the spectrum analyzer, unfortunately I don't have a module for this band. I will definitely put that into the shopping list this year. :)

morrownr, 2022-07-24:
@bcdonadio 

> I'm not really intimate with hostapd and wpa_supplicant besides what I needed to get 802.11ac/r/kv working on them some time ago on OpenWRT. Your guide will be really helpful.

My guide is time tested. It will work on distros besides the RasPiOS but you may have to do some things like nuk networkmanager. For this situation it would probably be best to follow the parts of the guide for a single adapter and use the 5 GHz hostapd.conf.

@yaslama is making some good progress so I'll try to help him this week as he has time to report on his progress. The way I see it, if we have multiple confirmation of 6 GHz support then I think we can call it confirmed and can document it.

Darn sure wish I had two 953AX's to test with but I'm holding on until one of the places I do business with has them in stock.

Good times...

bcdonadio, 2022-07-24:
@morrownr I'm quite fond of NetworkManager, and use it even on servers when networking might be a little more dynamic than is convenient to handle with systemd-networkd. Are there needed features/parameters known to NetworkManager not have support to instruct hostapd and wpa_supplicant when bringing them up?

We can definitely start testing the adapters without NM, and with another distro (I have only one spare Raspi4B though, not two, if you want to go for ARM64), but if there isn't any hard roadblock I would like to help on documenting the setup with NM and/or contributing upstream to add the needed support if it's within my capabilities.

I had a number of issues before with that SMSC LAN9514 USB+Ethernet chip on previous Raspberries. I know that with the BCM2711 which is in the Raspi4B this isn't supposed to be an issue anymore, but that overall stack isn't known to be the most stable either. I want to avoid the known USB bugs, bus width and raw performance limitations since we're now trying to achieve past Gigabit speeds.

Your guide is supposed to setup a manual hostapd + systemd-networkd, right? I will go for a Fedora 36 Server basic setup just switching NM for networkd then, on two NUCs with Intel i7, plugged into the USB3.0 ports, with a backbone 10GBASE-SR link connected to the Thunderbolt 3 bus.

Let's see what @yaslama gets in the meantime. I'm eager to known whether 6E is indeed a thing on these adapters.

morrownr, 2022-07-25:
@bcdonadio 

> Let's see what @yaslama gets in the meantime. I'm eager to known whether 6E is indeed a thing on these adapters.

I am also eager to see if these adapters are 6E capable. Not having an adapter to test is a challenge to deal with.

> Your guide is supposed to setup a manual hostapd + systemd-networkd, right? 

That is correct. It is not the best solution for all situations. The goals I used to guide the development of my access point guide are:

- 24/7/365 (or close) uptime. This is what took NM out of the picture for me. It has been a while since I tried to use NM in an AP setup so things may be better now.
- Straightforward ability to use 2 or more usb wifi adapters so as to have a dual-band or tri-band access point. The guides gives good instructions for dual-band and the basic idea how to do tri-band. My RasPi4B currently runs tri-band with two 5 GHz usb radios in operation. 
- While the guide appears to be very long, it allows a user to copy and paste to the cli and set things up rapidly and dependably. I guess I could make a bash script and then it would be fast.

> We can definitely start testing the adapters without NM.

You should do what you are most comfortable with. If you have an extra sd card that is burned with a current RasPiOS then my guide would allow you to rapid get going but after the 6 GHz testing, you can format it and burning it again. I was just throwing something out that might speed things up.

> I have only one spare Raspi4B though, not two,

You don't need two. You need one to use as the AP. The other adapter can go in any client you want.

> I had a number of issues before with that SMSC LAN9514 USB+Ethernet chip on previous Raspberries. I know that with the BCM2711 which is in the Raspi4B this isn't supposed to be an issue anymore, but that overall stack isn't known to be the most stable either.

Been there, done that. The RasPi4B has what I will call errata in the USB3 hardware. It does not show in all situations but it does show in the rtl88x2bu driver that I maintain and I've seen reports that may be related with the mt7612u driver.

> I would like to help on documenting the setup with NM and/or contributing upstream to add the needed support if it's within my capabilities.

I'd like to work with you on that if it works out. 

Regards

morrownr, 2022-07-25:
I have been doing more digging for details about the new Mediatek USB chipsets. Information is hard to find at this point. Below is what I have come up with so far:

See below for the specifics for the two Mediatek USB Wifi chipsets. It appears the only difference between the two is that the MT7921AU can handle a channel width of 80 while the MT7922U can handle a channel width of 160. Both are capable of 6 GHz. Unless I am reading things wrong, the mt7921u driver is capable of handling both chipsets. I can't see where Mediatek is releasing a USB WiFi 6 chipset that is not capable of 6 GHz. It is not clear why Comfast is advertising the CF-953AX as a dual-band AX1800 adapter. It sure would be good to get confirmation that 6 GHz is functional.

```
MT7921AU (Filogic 330)
USB 3.0
abgn+ac+ax
2x2:2
Wi-Fi 6E
WPA3
OFDMA
Zero DFS
BT 5.2
MU-MIMO
1024QAM
HE80
LNA/PA
ESR
```

```
MT7922U (Filogic 330)
USB 3.0
abgn+ac+ax
2x2:2
Wi-Fi 6E
WPA3
OFDMA
Zero DFS
BT 5.2
MU-MIMO
1024QAM
HE160
LNA/PA
ESR
```

bcdonadio, 2022-07-26:
> It is not clear why Comfast is advertising the CF-953AX as a dual-band AX1800 adapter. It sure would be good to get confirmation that 6 GHz is functional.

Maybe they wanted to put an additional amplifier IC, and there wasn't one cheaply/readily available for the 6-7GHz band? Not sure whether the MT7921AU alone is already capable of maxing out the FCC-allowed EIRP ceilling.

bcdonadio, 2022-07-27:
@morrownr I'm afraid I don't have the best piece of news...

```
[root@eagle ~]# uname -r
5.19.0-0.rc7.20220722git68e77ffbfd06.56.fc36.x86_64
[root@eagle ~]# ls /lib/firmware/mediatek/ | grep 7961
BT_RAM_CODE_MT7961_1_2_hdr.bin
BT_RAM_CODE_MT7961_1_2_hdr.bin.xz
WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_MT7961_patch_mcu_1_2_hdr.bin.xz
WIFI_RAM_CODE_MT7961_1.bin
WIFI_RAM_CODE_MT7961_1.bin.xz
[root@eagle ~]# iw list
Wiphy phy1
	wiphy index: 1
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: Station
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce120dc0b306423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
[... 546 lines cut ...]
	Supported RX frame types:
		 * IBSS: 0x40 0xb0 0xc0 0xd0
		 * managed: 0x40 0xb0 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ BEACON_RATE_HT ]: HT beacon rate setting
		* [ BEACON_RATE_VHT ]: VHT beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

bcdonadio, 2022-07-27:
Hmmm, trying some stuff here to fix the regulatory domain. The database entry for Brazil (that is forbidding the 6E band apparently) is not up to date with the national legislation (which has already passed provisions allowing the use of this portion of the spectrum for WiFi).

Not sure if that is the reason for the channels to show up as disabled though.

morrownr, 2022-07-27:
> Not sure if that is the reason for the channels to show up as disabled though.

When I saw your first message today showing disabled, the first thing I thought of was regulatory domain. Wish I had an adapter so I could join the party.

bcdonadio, 2022-07-27:
Forced the US regdomain, rebooted, unplugged, replugged, and the 6E channels are all still disabled. Power limits on some channels changed, so I'm pretty sure it got applied.

yaslama, 2022-07-27:
@bcdonadio Can you try to force an EU country? I used FR in order to enable some 6Ghz channels. I have a working AP with a rpi400 and openwrt but I the connection is failing when using a CF-951AX with both wpa_supplicant and iwd.

bcdonadio, 2022-07-27:
Yep. Let me try France.

bcdonadio, 2022-07-27:
Actually, I was missed an additional zero. I was looking at the 60GHz band. Neither BR, US or FR is allowing something in the 6 to 7GHz band with my database. I'm gonna research how to update this.

```
[root@eagle ~]# iw phy2 reg get
global
country BR: DFS-FCC
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)

[root@eagle ~]# iw reg set US
[root@eagle ~]# iw phy2 reg get
global
country US: DFS-FCC
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

[root@eagle ~]# iw reg set FR
[root@eagle ~]# iw phy2 reg get
global
country FR: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

morrownr, 2022-07-27:
#88 is the post showing active 6 GHz channels on the 953AX. It appears the poster is in Germany.

A mystery to be solved.

bcdonadio, 2022-07-27:
I think I may have a clue. According [to this](https://wireless.wiki.kernel.org/en/developers/regulatory/processing_rules), the linux drivers restrict themselves voluntarily if any other device nearby is broadcasting that you can't use X or Y band, and that can even be just plain old information.

I can try to put it into a makeshift Faraday cage and see if the behavior changes.

yaslama, 2022-07-27:
This is my results (openwrt in the rpi400 and CF-953AX)
```
root@OpenWrtRpi400:~# iw reg get
global
country FR: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 71000 @ 2160), (N/A, 40), (N/A)

root@OpenWrtRpi400:~# iw reg set GR
root@OpenWrtRpi400:~# iw reg get
global
country GR: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5875 @ 80), (N/A, 13), (N/A)
        (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

bcdonadio, 2022-07-27:
I'm feeling a sudden urge to both jailbreak this thing AND put a 2kW hamradio linear amplifier just to make a statement...

morrownr, 2022-07-27:
>         (5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR

Hmmm...

@yaslama 

Which version of OpenWRT are you using?

Are you able to use the 953AX as an AP with 6 GHz?

yaslama, 2022-07-27:
> Which version of OpenWRT are you using?

```
root@OpenWrtRpi400:~# cat /etc/openwrt_version
r20156-0f301b0b1d
root@OpenWrtRpi400:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r20156-0f301b0b1d'
DISTRIB_TARGET='bcm27xx/bcm2711'
DISTRIB_ARCH='aarch64_cortex-a72'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r20156-0f301b0b1d'
DISTRIB_TAINTS=''
```

> 
> Are you able to use the 953AX as an AP with 6 GHz?

```
root@OpenWrtRpi400:~# iw dev wlan0 info
Interface wlan0
        ifindex 37
        wdev 0x12
        addr e0:e1:a9:34:a6:d1
        ssid testssid6
        type AP
        wiphy 0
        channel 53 (6215 MHz), width: 80 MHz, center1: 6225 MHz
        txpower 3.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       0       0       0       0       0       0               0
```
The AP is configured but I still wasn't able to connect to it (at 5Ghz it was ok). I see the the ssid but, the the connection fails.
I ordered some 7921k m.2 cards to test.

bcdonadio, 2022-07-27:
My WDR4300 with OpenWRT 22.03 and kernel 5.10 is much more liberal, even has the mt7921u module and is loading it to the correct USB device, but `iw` isn't recognizing it for some reason. Probably the USB2.0 port? I can try flashing the nightly snapshot of OpenWRT into the RPi4B.

```
root@heimdall:~# iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#2
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#1
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

morrownr, 2022-07-27:
> My WDR4300 with OpenWRT 22.03 and kernel 5.10 is much more liberal, even has the mt7921u module...

I'm running 22.03 on my Zyxel wifi router. I had not thought to check for the driver since it is running kernel 5.10. I smell a backport:

kmod-mt7921u

> Probably the USB2.0 port?

I doubt it. Is the firmware installed?

bcdonadio, 2022-07-27:
Done.

Before setting region on nightly snapshot of x86_64 OpenWRT:
```
root@OpenWrt:~# iw reg get
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
root@OpenWrt:~# iw list
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: managed
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
				OM Control
				Maximum A-MPDU Length Exponent: 3
				A-MSDU in A-MPDU
			HE PHY Capabilities: (0x2270ce120dc0b306423f00):
				HE40/2.4GHz
				242 tone RUs/2.4GHz
				Device Class: 1
				LDPC Coding in Payload
				HE SU PPDU with 1x HE-LTF and 0.8us GI
				NDP with 4x HE-LTF and 3.2us GI
				STBC Tx <= 80MHz
				STBC Rx <= 80MHz
				Full Bandwidth UL MU-MIMO
				Partial Bandwidth UL MU-MIMO
				DCM Max Constellation: 2
				DCM Max Constellation Rx: 2
[... 1056 lines cut ...]
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: 65535
	max scan plan iterations: 0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ BEACON_RATE_HT ]: HT beacon rate setting
		* [ BEACON_RATE_VHT ]: VHT beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
```

bcdonadio, 2022-07-27:
It's a backport, but it worked. I had to manually download the firmware though. I will prepare some performance tests later on.

morrownr, 2022-07-27:
> I had to manually download the firmware though. 

You might want to report the issue with lack of firmware package to:

https://github.com/openwrt/mt76

It is not like the OpenWRT folks don't know the firmware is needed as a couple of the OpenWRT devs work in linux-wireless. It could be that they are not aware that there has been a public release of mt7921au based adapters and we need that firmware package available in 22.03.

Here is the direct link to where I give instructions for the installation of the latest firmware files:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

Obviously you have already done it but others that read this later may want to know.

> I will prepare some performance tests later on.

Cool.

What does `iw dev` look like now that the firmware is installed?

bcdonadio, 2022-07-27:
> What does `iw dev` look like now that the firmware is installed?

@morrownr Nothing there just yet. I haven't connected in the other adapter yet, and if I connect on my own AP it is likely to restrict allowed bands again.

```
root@OpenWrt:~# iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr e0:e1:a9:34:a6:67
		type managed
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

[137 more replies]