morrownr/USB-WiFi #89: M.2 A+E key to PCIe WiFi cards with USB connectors on board.

croissantpetrichor, 2022-07-23:
Does anyone here have a [Gigabyte GC-WBAX200](https://wikidevi.wi-cat.ru/Gigabyte_GC-WBAX200) or a similar card?

Officially the only supported interface listed for this card is PCIe. Although I've seen reviews that state the adapter also works via a USB 2.0 interface. I think because of the bottleneck with newer cards that have speeds faster than the USB 2.0 interface permits they don't list this as a supported interface.

There are two USB connectors on the adapter labeled F_USB and R_USB which stands for Front panel USB and Rear USB respectively.

[image]

It only makes sense that the USB 2.0 interface is also supported, since the two major interfaces besides some debugging interfaces (UART, I2C...) that M.2 A+E supports are PCIe and USB 2.0. 

I don't have a card like that laying around here, so I'm unable to verify whether or not I'm right.

Could someone here with such a card try plugging it in solely via R_USB to USB 2.0?

morrownr, 2022-08-01:
I don't have a clue is my response to your questions but if you are looking for a wifi solution to a specific use case, there are several folks that stop by here that could throw out some suggestions.

Nick

joshschmelzle, 2022-08-01:
@croissantpetrichor Just because A+E supports both PCIe and USB signal transfers does not mean that Wi-Fi modules designed to use PCIe as the signal transfer will use USB instead.

Wi-Fi modules will typically use PCIe (with some exceptions like CNVio/2) as the signal transfer for the Wi-Fi radio. And Bluetooth will use USB. For example, see the System Interface Type row in the Intel AX210 specifications which explicitly calls this out. https://ark.intel.com/content/www/us/en/ark/products/204836/intel-wifi-6e-ax210-gig.html

What this means for adapters like the Gigabyte_GC-WBAX200 is the USB header needs connected in order for Bluetooth to work.

croissantpetrichor, 2022-08-01:
> @croissantpetrichor Just because A+E supports both PCIe and USB signal transfers does not mean that Wi-Fi modules designed to use PCIe as the signal transfer will use USB instead.

You're right, although I've seen reviews that state these adapters also work via USB, given that A+E keyed slots support USB and there is an extra USB connector on board there is an motive.

> Wi-Fi modules will typically use PCIe (with some exceptions like CNVio/2) as the signal transfer for the Wi-Fi radio. And Bluetooth will use USB. For example, see the System Interface Type row in [...]

The fact that A+E doesn't support USB 3.0 is beyond me... sucks.

croissantpetrichor, 2022-08-01:
> I don't have a clue is my response to your questions but if you are looking for a wifi solution to a specific use case, there are several folks that stop by here that could throw out some suggestions.
> 
> Nick

I want the versatility of using M.2 A+E modules via both PCIe, Thunderbolt (insecure but well,  if someone has physical access to your devices you've already lost) and USB.

I don't wanna use a wacky Chinese M.2 A+E to Mini PCIe --> Mini PCIe to usb adapter so I'm asking whether or not this board supports USB.

joshschmelzle, 2022-08-01:
I can't speak to the OP board, but I've had luck with this one on Debian and Windows with the Intel AX2xx. Just gotta connect the USB header for BT to function. https://www.amazon.com/gp/product/B01MFGYAX6