morrownr/USB-WiFi #90: ath11k.

croissantpetrichor, 2022-07-25:
https://wikidevi.wi-cat.ru/Ath11k
https://wireless.wiki.kernel.org/en/users/Drivers/ath11k

Seems to be mainlined, worth checking out.

https://wireless.wiki.kernel.org/en/users/drivers/ath11k
https://patchwork.kernel.org/project/linux-wireless/list/?state=*&q=ath11k

morrownr, 2022-07-25:
Hi @croissantpetrichor 

The ath11k wifi driver (module) has been mainlined for quite some time. ath10k and other older drivers are available as well. My main wifi router uses the ath10k driver.

The primary theme of this site is USB WiFi adapters and Qualcomm/Atheros has released no usb chipsets since the old ar9271 which is a 11n, WiFi 4 class chipset. Adapters that use the ar9271 are currently still available but probably won't be for much longer as the ar9271 went end of life last year if my information is correct.

Right now, the only two wifi chipset makers that are releasing usb chipsets are Mediatek and Realtek. Of those two companies, Mediatek is doing a really good job of supporting Linux the right way-with in-kernel drivers. Realtek seems stuck on providing non-standards compliant out-of-kernel drivers that are problematic. The driver for Realtek's Wifi 6 usb chipset, rtl8832au, is terrible. I try to warn Linux users to stay away from any usb adapter that uses the rtl8832au chipset.

Regards

croissantpetrichor, 2022-07-26:
>The ath11k wifi driver (module) has been mainlined for quite some time. ath10k and other older drivers are available as well.

I just found out recently, thanks for the info as always

>My main wifi router uses the ath10k driver.

 TP-Link Archer C7?
 
>The primary theme of this site is USB WiFi adapters and Qualcomm/Atheros has released no usb chipsets since the old ar9271 which is a 11n, WiFi 4 class chipset. Adapters that use the ar9271 are currently still available but probably won't be for much longer as the ar9271 went end of production last year if my information is correct.

Hopefully Qualcomm will grab their bearings and develop a AX USB WiFi chipset in the near future.

> Right now, the only two wifi chipset makers that are releasing usb chipsets are Mediatek and Realtek. Of those two companies, Mediatek is doing a really good job of supporting Linux the right [...]

It's harder to find MediaTek USB adapters in the wild (most people seem to use RealTek somehow) which is frustrating, but you're absolutely right.

closed 2022-07-26 by croissantpetrichor

morrownr, 2022-07-26:
> TP-Link Archer C7?

ZyXEL NBG6817

> It's harder to find MediaTek USB adapters in the wild (most people seem to use RealTek somehow) which is frustrating, but you're absolutely right.

See:

https://github.com/morrownr/USB-WiFi

Main Menu item 2 has a long list of adapters that use in-kernel driver and links where they can be purchased. With the new released of adapters using the mt7921au chipset, the list should grow rapidly over the next few months.

Regards