morrownr/USB-WiFi #92: Random network timeouts AWUS036ACM / AWUS036ACHM

CapitaineBidouille, 2022-07-27:
Hello !

I am using 2 ugreen hubs : https://www.amazon.fr/gp/product/B00OQ0CYFE/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
which are plugged to the PI 4B . Hubs are powered with power supply.
On each of them, there is hard drives and on the second 2 usb wifi : 
AWUS036ACM ==> mt7612u, running a 5GHz AP
AWUS036ACHM ==> mt7610u, running a 2,4GHz AP

USB scatter gather which caused complete crashes has been already fully disabled.

Sometimes, without generating any logs, my devices stay connected to the wifi but completely timeout everything during approx. 15 seconds on the 5GHz network. It's enough to stop playing a media on Kodi. I can't even ping the IP address of the PI and I lost my SSH connection if I was connected to the pi.
It seems to happen more frequently during stress tests (like file copying over the network), but it's mostly random.

I am totally sure it's coming from the Wifi network itself since routing my computer traffic over an USB Ethernet adapter is perfect, there is no disconnects at all.

Thank you very much for your help

Wifi 2.4 Ghz config file : 
`# /etc/hostapd/hostapd-2g.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2021-11-15

# SSID
ssid=WifiNetwork Legacy
# PASSPHRASE
wpa_passphrase=mypassword
# Band: a = 5g (a/n/ac), g = 2g (b/g/n)
hw_mode=g
# Channel
channel=11
# Country
country_code=FR
# WiFi interface
interface=wlx00c0caae6824

# nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers
driver=nl80211
#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0

beacon_int=100
dtim_period=2
max_num_sta=32
rts_threshold=2347
fragm_threshold=2346
#send_probe_response=1

# security
# auth_algs=3 is required for WPA-3 SAE and WPA-3 SAE Transitional
auth_algs=1
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2
wpa_pairwise=CCMP
# WPA-2 AES
wpa_key_mgmt=WPA-PSK
# WPA-3 SAE
#wpa_key_mgmt=SAE
#wpa_group_rekey=1800
rsn_pairwise=CCMP
# ieee80211w=2 is required for WPA-3 SAE
#ieee80211w=2
# If parameter is not set, 19 is the default value.
#sae_groups=19 20 21 25 26
#sae_require_mfp=1
# If parameter is not 9 set, 5 is the default value.
#sae_anti_clogging_threshold=10

# IEEE 802.11n
ieee80211n=1
wmm_enabled=1
#
# Note: Only one ht_capab= line should be active. The content of these lines is
# determined by the capabilities of your adapter.
#
# generic 20 NHz setting
#ht_capab=[SHORT-GI-20]
#
# RasPi4B internal wifi
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
#
# ar9271
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][RX-STBC1][DSSS_CCK-40]
#
# mt7612u - mt7610u
ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]
#
# rtl8812au - rtl8811au -  rtl8812bu - rtl8811cu
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# rtl8814au
#ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]

# generic setting
#vht_capab=[SHORT-GI-80]
#
# mt7610u
vht_capab=[SHORT-GI-80][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
# mt7612u
#vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
#
# rtl8812au - rtl8812bu
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8814au
#vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8811au
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8811cu
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# Note: [TX-STBC-2BY1] may cause problems with some Realtek drivers

# End of hostapd-2g.conf
`

Wifi 5 Ghz config file : 
`# /etc/hostapd/hostapd-5g.conf
# Documentation: https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
# 2022-02-11

# SSID
ssid=WifiNetwork
# PASSPHRASE
wpa_passphrase=mypassword
# Band: a = 5g (a/n/ac), g = 2g (b/g/n)
hw_mode=a
# Channel
channel=36
# Channel width
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
# Country code
country_code=FR
# WiFi interface
interface=wlx00c0cab02457

# nl80211 is used with all Linux mac80211 (in-kernel) and modern Realtek drivers
driver=nl80211
#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0

ieee80211d=1
# Enables support for 5GHz DFS channels
#ieee80211h=1

beacon_int=100
dtim_period=2
max_num_sta=32
macaddr_acl=0
rts_threshold=2347
fragm_threshold=2346
#send_probe_response=1

# security
# auth_algs=3 is required for WPA-3 SAE and WPA-3 SAE Transitional
auth_algs=1
ignore_broadcast_ssid=0
# wpa=2 is required for WPA2 and WPA3 (read the docs)
wpa=2
rsn_pairwise=CCMP
# only one wpa_key_mgmt= line should be active.
# wpa_key_mgmt=WPA-PSK is required for WPA2-AES
wpa_key_mgmt=WPA-PSK
# wpa_key_mgmt=SAE WPA-PSK is required for WPA3-AES Transitional
#wpa_key_mgmt=SAE WPA-PSK
# wpa_key_mgmt=SAE is required for WPA3-SAE
#wpa_key_mgmt=SAE
#wpa_group_rekey=1800
# ieee80211w=1 is required for WPA-3 SAE Transitional
# ieee80211w=2 is required for WPA-3 SAE
#ieee80211w=1
# if parameter is not set, 19 is the default value.
#sae_groups=19 20 21 25 26
# sae_require_mfp=1 is required for WPA-3 SAE Transitional
#sae_require_mfp=1
# if parameter is not 9 set, 5 is the default value.
#sae_anti_clogging_threshold=10

# Note: Capabilities can vary even between adapters with the same chipset.
#
# Note: Only one ht_capab= line and one vht_capab= should be active. The
# content of these lines is determined by the capabilities of your adapter.
#
# IEEE 802.11n
ieee80211n=1
wmm_enabled=1
#
# generic setting
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40]
#
# mt7612u - mt7610u
ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]
#
# rtl8812au - rtl8811au - rtl8811cu
#ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# rtl8812bu
#ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# rtl8814au
#ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]
#

# IEEE 802.11ac
ieee80211ac=1
#
# generic setting
vht_capab=[SHORT-GI-80]
#
# mt7610u
#vht_capab=[SHORT-GI-80][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
# mt7612u
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
#
# rtl8812au - rtl8812bu
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8814au
#vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8811au
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
# rtl8811cu
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT][MAX-A-MPDU-LEN-EXP7]
#
# Note: [TX-STBC-2BY1] may cause problems with some Realtek drivers

# end of hostapd-5g.conf
`

amisix, 2022-07-27:
What kind of power supply are you using to power the hub? Is it a phone charger? Don't use those, they often create under  voltage issues.

morrownr, 2022-07-27:
Hi @CapitaineBidouille 

Nice setup. My experience tells me we should first look at things related to USB3 first.

This is a setup with a lot of parts so let me just tell you what I would try to do and you can make the decisions about how best to do it.

I would try to rework things to where the usb adapters are plugged directly into ports on the RasPi4B. I realize that would only leave one USB3 port open for other things but maybe you can plug one powered hub into the Pi and serial the other port.

I've seen enough problems with RasPis and powered hubs to last a lifetime. I use a powered USB3 hub but I am careful how I use it. If I use two usb wifi adapters, I make sure they are both either plugged into the Pi or into the same powered hub while nothing else is on the powered hub.

What you could do to really narrow down the issue is burn a fresh sd with RasPiOS, 64 bit, and plug only the two adapters into the Pi and start building the setup so as to see if the wifi is stable. If it is stable, start adding your other components to the setup until you encounter the problem. I realize this is a lot of work but it is the best way to be for sure what is causing the problem.

Regards

CapitaineBidouille, 2022-07-28:
> What kind of power supply are you using to power the hub? Is it a phone charger? Don't use those, they often create under voltage issues.

I am using this one from UGreen : https://www.amazon.fr/gp/product/B07V9LZCXV/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1
With these USB3 cables : https://www.amazon.fr/gp/product/B01GN0M6NE/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
There is no cuts anymore at all on my 3 hard drives. Even the USB wifi does not really fully cut, since my devices stay connected on their network that recover itself after a few seconds. The 2.4 GHz seems also mostly unaffected from that issue, but I am not entirely sure since I am using less and only with certain old devices that does not support 5 GHz.

> Hi @CapitaineBidouille
>
> Nice setup. My experience tells me we should first look at things related to USB3 first.
>
> This is a setup with a lot of parts so let me just tell you what I would try to do [...]

Hi morrownr !
First I forgot the most important : thank you very much for your work there, it helped me a lot making my setup and make the right decisions to get the most I can get from my PI4.

My PI is really doing a lot of things now : 
- PLEX media server (no transcoding)
- NFS/SMB/FTP file sharing with OpenMediaVault (sharing 3x 5 TB)
- Can connect to VPN and share the VPN to all my devices through Ethernet and Wifi
- Torrent box (Deluge)
- DNSmasq DNS server

What is plug on the first USB hub : 
- 3 external hard drives

This is what I plug on the second USB hub : 
- The 2 USB Wifi Adapters
- TP Link UE300 Ethernet Wifi Adapter : https://www.amazon.fr/gp/product/B00V4BGDKU/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1 (which I recommend a lot, seems like it's one of the only one that can get full 1 GBps bandwidth routing without taking all the CPU)

What I am going to try to do first : 
- Move the USB Ethernet adapter on the other USB hub so the USB Wifi adapters will be alone on that Hub
- Remove power on that Hub so the PI will have the responsibility to give them power.
- If that's not enough, I will plug the AWUS036ACHM directly to an USB2 port of the PI, and AWUS036ACM will be plug directly to an USB3 port (directly on the PI too).

Kind regards

CapitaineBidouille, 2022-07-28:
Here is the result of my First try : 
USB3 Port1 : Powered Hub : 3x 5TB hard drives + USB Ethernet
USB3 Port2 : USB3 Extension cord connected to AWUS036ACM 
USB2 Port1 : AWUS036ACHM 
USB2 Port 2 : Nothing

It does not work. The system boot up, but if I connect to the 5 GHz network then start a speedtest, everything is crashing in 3 seconds. Not enough power I guess.
If I remove completely the AWUS036ACHM from the Pi then reboot it, I can connect to my 5GHz then complete a speedtest from start to finish : https://www.speedtest.net/result/13464842645

I am going to connect the second powered Hub to an USB2 port. On that hub I will connect only the AWUS036ACHM. Let's see if everything does not explode with that setup :D

CapitaineBidouille, 2022-07-28:
So, I am testing right now with that setup : 
USB3 Port1 : Powered Hub : 3x 5TB hard drives + USB Ethernet
USB3 Port2 : USB3 Extension cord connected to AWUS036ACM
USB2 Port1 : Powered Hub: AWUS036ACHM
USB2 Port 2 : Nothing

I connected to both Wifi network and started a speedtest from start to finish.
5GHz : https://www.speedtest.net/result/13464867452
2.4 GHz : https://www.speedtest.net/result/13464870687
I also made another one using Ethernet from my PC which is linked directly to the PI that has the USB Ethernet adapter plugged on the powered hub : https://www.speedtest.net/result/13464876298

There was no crash anymore, Wifis and hard drives are still running. Let's see what happen after some time, if I am still getting these random network timeouts.

CapitaineBidouille, 2022-07-28:
[image]
Let's transfer a huge virtual machine and see what happen. Performance is very good in 5Ghz, from around 300Mbps to 500Mbps. After some time copying it seems to stay around 275 Mbps.
In Ethernet it's still x2 to x3, this is why I connect my main PC through Ethernet : I want to take advantage of the maximum possible bandwidth on it. With Ethernet I can reach the whole gigabit when copying files, which saves also much time especially when I backup my whole internal hard drive regularly (using Macrium Reflect on Windows, which I recommend a lot : it saves everything in your PC very fast and can encrypt the whole backup. Very professionnal tool that seems not known enough imo.).

CapitaineBidouille, 2022-07-28:
[image]
It went from start to finish with perfect stability.
At the left, I was pinging Google to see if I was timeouting.

I could browse social networks and youtube on my phone on the same Wifi during the copy without issues as well.
I don't know if that test is enough and if it would had froze with the initial setup.
I could watch sometimes 2/3 hours of content with Plex without any single freezes, and got sometimes multiple freezes during the same time span.
Only time will give the definitive answer, but I don't see a reason that could justify another freezes. The 5GHz adapter is plugged directly on the PI now afterall, and the other one is alone on it's hub.

morrownr, 2022-07-28:
Hello again @CapitaineBidouille 

Here are some thoughts so far:

- To really pound specific connections, wired and wireless, I have found iperf3 to be valuable. You are familiar with my wireless AP guide, Bridged Wireless Access Point, which is currently located as [Main Menu item 9](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md). In this guide, there is a section called `Install and autostart iperf3.` This shows a good way to install and autostart iperf3 as a service on the RasPiOS. iperf3 is also available for Windows.

- Regarding the network setup. I have found bridging to be more stable and easier to maintain than a routed setup. If there is a reason for a routed setup, I understand. I am simply letting you know that unless there is a reason for a routed setup, a bridged setup is easier to maintain, slightly faster and more reliable based on my experience. FYI: I have been updating the Bridged Wireless Access Point document in a few places over the last week. If you see any problems, let me know.

- > USB scatter gather which caused complete crashes has been already fully disabled.

I have tested performance with and without USB Scatter-gather active in the driver and I simply have not been able to find any performance differences so I have to question whether defaulting to on is a good idea. I have not actually experienced problems that I can point to SG as the cause but I pay attention to the issue so that I can help others.

- In my opinion, the best OS to use on a RasPi4B is the 64 bit version of the RasPiOS. It took forever for the 64 bit version to be released but now that it is released and is stable with the 2022-04-04 release, it is better to move to it. The 32 bit version is simply going to continue to suffer from bit rot that can't be avoided due to so few 32 bit systems remaining in use. I'm not talking about RasPiOS specifically but rather the Linux kernel.

- Lastly, The RasPi4B with the RasPiOS can be very stable and do a lot of different things. My experience is that if I connect a LoT of things via USB, I may have to test to get a fully stable setup. The problem usually includes USB3 due to the chipset used in the USB3 controller but other problems related to power are at play also. The USB subsystem can only provide a maximum of 1200 mA of power. You need to know the power usage of the components used. There is also an issue of powered hubs backfeeding power into the Pi USB subsystem.  You also need to know the power usage of the components attached to the powered hub and what power the hub is capable of providing. With the required knowledge and testing, generally a very stable setup can be found.

Looking forward to your continued reports.

Regards

amisix, 2022-07-28:
The QC phone charger you have may not be outputting enough stable power for your purposes. Even though it says 18W, that's in QC mode which requires a negotiation to occur that's not happening when you're powering those hubs. At up to 900mA per device (USB3 spec) and 4 devices (3x HDDs & 1 Ethernet), you're already at the rated max of 18W... without Quick Charge mode being enabled. Do you know what the output of the power adapter is when not in QC mode? I couldn't find anything on Amazon. I ran into similar power gremlins when testing Samsung & Motorola QC capable phone chargers (with less overall load than the devices you're powering). Sometimes all it takes is one short voltage drop while an HDD spins up and that's that.

CapitaineBidouille, 2022-07-28:
> The QC phone charger you have may not be outputting enough stable power for your purposes. Even though it says 18W, that's in QC mode which requires a negotiation to occur that's not happening when you're powering those hubs. At up to 900mA per device (USB3 spec) and 4 devices (3x HDDs & 1 Ethernet), you're already at the rated max of 18W... without Quick Charge mode being enabled. Do you know what the output of the power adapter is when not in QC mode? I couldn't find anything on Amazon. I ran into similar (possible) power gremlins when testing Samsung & Motorola QC capable phone chargers (with less overall load than the devices you're powering). Sometimes all it takes is one short voltage drop while an HDD spins up and that's that.

It seems to be running just fine right now. The drives are not cutting themselves. So I guess it's ok as it is for the first USB powered hub.

> Hello again @CapitaineBidouille
>
> Here are some thoughts so far:
>
>     * To really pound specific connections, wired and wireless, I have found iperf3 to be valuable. You are familiar with my wireless [...]
- I am already running ARM64 RasPIOS for that reasons
- USB Scatter gather was causing total crashes of the Wifi networks randomly after a few minutes most of the time, so I disabled it.
- Right now, I am using routing so my local network will continue to work even if my home internet is completely down (which happen rarely but when it happen my router does not even give me a local IP address).
I also use routing rather than bridge to route VPN traffic easily.

To fix most issues with routing, you should add this IPTable rule : 
-A FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
It will fix any issue related to MTU, which prevent many internet site to work correctly when browsing the web.

Looks like I just had a complete timeout of 30 seconds :(. Shame. So the 5GHz network is still not perfectly stable, maybe the changes did not change anything.
Maybe these random timeout start to occur only after a few hours and not directly after a reboot.

morrownr, 2022-07-28:
@CapitaineBidouille 

You did say HDD and I should have asked you to clarify. Are those HDD's spinning rust drives? If so, oh my. Those things can pull some juice and it varies a lot depending on what you are doing. @amisix is speaking words of wisdom. Getting all the power stuff lined out is something that really needs to be done. I use a small in-line device that measures power usage. You might want to get one.

Regards

CapitaineBidouille, 2022-07-29:
> @CapitaineBidouille
>
> You did say HDD and I should have asked you to clarify. Are those HDD's spinning rust drives? If so, oh my. Those things can pull some juice and it varies a lot depending on [...]

It's 3x LaCie 5TB (mechanical hard drives).
Right now I do not have the tools to measure energy consumption.

morrownr, 2022-07-29:
> To fix most issues with routing, you should add this IPTable rule :
-A FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu
It will fix any issue related to MTU, which prevent many internet site to work correctly when browsing the web.

Hmmm... RasPiOS 2022-04-04 no longer uses ipTables. It uses NFtables. I think you just volunteered to help me work on a Routed Wireless Access Point document. <g>.

morrownr, 2022-07-29:
> It's 3x LaCie 5TB (mechanical hard drives).
Right now I do not have the tools to measure energy consumption.

You probably can get the power usage specifications for the drives and for the powered hubs so that we can figure out if there is a periodic brown-out possibility. That could account for what you are seeing.

CapitaineBidouille, 2022-07-29:
> 

Looks like both are installed on my PI. My NFtables does not have any rules.

Here is a result of iptables -S : 
-P INPUT ACCEPT
-P FORWARD ACCEPT
-P OUTPUT ACCEPT
-A FORWARD -i tun0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i wlan0 -o tun0 -j ACCEPT
-A FORWARD -p tcp -m tcp --tcp-flags SYN,RST SYN -j TCPMSS --clamp-mss-to-pmtu

> You probably can get the power usage specifications for the drives and for the powered hubs so that we can figure out if there is a periodic brown-out possibility. That could account for what you [...]

Tried another setup to balance better the power between the two hubs : 
USB3 Port1 : Powered Hub : 2x 5TB hard drives + USB Ethernet
USB3 Port2 : Powered Hub : 1x 5TB hard drive + AWUS036ACM
USB2 Port1 : AWUS036ACHM
USB2 Port 2 : Nothing

CapitaineBidouille, 2022-07-29:
I installed iperf3 server on the PI and the client on my Windows machine.
What commands should I use to start a good stress test that could give me logs ?

morrownr, 2022-07-29:
> Tried another setup to balance better the power between the two powered hubs :
USB3 Port1 : Powered Hub : 2x 5TB hard drives
USB3 Port2 : Powered Hub : 1x 5TB hard drive + AWUS036ACM + USB Ethernet
USB2 Port1 : AWUS036ACHM
USB2 Port 2 : Nothing

Something else to try that would get the wifi off of USB3:

```
USB3 Port1 : Powered Hub : 2x 5TB hard drives
USB3 Port2 : Powered Hub : 1x 5TB hard drive + USB Ethernet
USB2 Port1 : AWUS036ACHM
USB2 Port 2 : AWUS036ACM
```

That is just to test if the wifi is stable if the 5 GHz is not on USB3.

> What commands should I use to start a good stress test that could give me logs ?

Not sure on Windows but on Linux a good place to start would be:

$ iperf3 -c 192.168.1.1 -t 300

The ip address is the ip address of the iperf3 server. The 300 means for it to run for 300 seconds. iperf3 can really pound a setup.

CapitaineBidouille, 2022-07-29:
iperf3 is definitely an high value software.
Could timeout my wifi network in < than 5 minutes.
So that setup still does not work.

```Connecting to host 192.168.4.1, port 5201
[  4] local 192.168.4.176 port 24898 connected to 192.168.4.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  51.2 MBytes   429 Mbits/sec
[  4]   1.00-2.00   sec  51.6 MBytes   433 Mbits/sec
[  4]   2.00-3.00   sec  53.8 MBytes   452 Mbits/sec
[  4]   3.00-4.00   sec  52.5 MBytes   440 Mbits/sec
[  4]   4.00-5.00   sec  51.5 MBytes   433 Mbits/sec
[  4]   5.00-6.00   sec  50.4 MBytes   421 Mbits/sec
[  4]   6.00-7.00   sec  53.9 MBytes   454 Mbits/sec
[  4]   7.00-8.00   sec  49.9 MBytes   417 Mbits/sec
[  4]   8.00-9.00   sec  50.8 MBytes   427 Mbits/sec
[  4]   9.00-10.00  sec  50.2 MBytes   421 Mbits/sec
[  4]  10.00-11.00  sec  50.9 MBytes   426 Mbits/sec
[  4]  11.00-12.00  sec  51.0 MBytes   427 Mbits/sec
[  4]  12.00-13.00  sec  49.2 MBytes   415 Mbits/sec
[  4]  13.00-14.00  sec  48.8 MBytes   408 Mbits/sec
[  4]  14.00-15.00  sec  54.0 MBytes   452 Mbits/sec
[  4]  15.00-16.01  sec  49.4 MBytes   413 Mbits/sec
[  4]  16.01-17.00  sec  52.2 MBytes   439 Mbits/sec
[  4]  17.00-18.01  sec  50.6 MBytes   423 Mbits/sec
[  4]  18.01-19.00  sec  47.5 MBytes   400 Mbits/sec
[  4]  19.00-20.00  sec  50.6 MBytes   426 Mbits/sec
[  4]  20.00-21.01  sec  46.6 MBytes   390 Mbits/sec
[  4]  21.01-22.00  sec  49.2 MBytes   415 Mbits/sec
[  4]  22.00-23.00  sec  50.2 MBytes   422 Mbits/sec
[  4]  23.00-24.00  sec  52.8 MBytes   443 Mbits/sec
[  4]  24.00-25.00  sec  47.0 MBytes   394 Mbits/sec
[  4]  25.00-26.00  sec  49.4 MBytes   414 Mbits/sec
[  4]  26.00-27.00  sec  50.0 MBytes   420 Mbits/sec
[  4]  27.00-28.00  sec  51.1 MBytes   427 Mbits/sec
[  4]  28.00-29.00  sec  50.4 MBytes   424 Mbits/sec
[  4]  29.00-30.00  sec  50.9 MBytes   426 Mbits/sec
[  4]  30.00-31.00  sec  48.5 MBytes   407 Mbits/sec
[  4]  31.00-32.00  sec  50.9 MBytes   427 Mbits/sec
[  4]  32.00-33.00  sec  45.9 MBytes   384 Mbits/sec
[  4]  33.00-34.00  sec  49.8 MBytes   418 Mbits/sec
[  4]  34.00-35.00  sec  47.1 MBytes   395 Mbits/sec
[  4]  35.00-36.00  sec  49.5 MBytes   415 Mbits/sec
[  4]  36.00-37.01  sec  44.5 MBytes   371 Mbits/sec
[  4]  37.01-38.00  sec  39.5 MBytes   334 Mbits/sec
[  4]  38.00-39.00  sec  42.1 MBytes   353 Mbits/sec
[  4]  39.00-40.00  sec  27.5 MBytes   231 Mbits/sec
[  4]  40.00-41.00  sec  45.9 MBytes   385 Mbits/sec
[  4]  41.00-42.00  sec  44.4 MBytes   371 Mbits/sec
[  4]  42.00-43.00  sec  46.9 MBytes   394 Mbits/sec
[  4]  43.00-44.00  sec  49.4 MBytes   415 Mbits/sec
[  4]  44.00-45.00  sec  50.5 MBytes   423 Mbits/sec
[  4]  45.00-46.00  sec  51.0 MBytes   428 Mbits/sec
[  4]  46.00-47.00  sec  51.4 MBytes   431 Mbits/sec
[  4]  47.00-48.00  sec  48.2 MBytes   405 Mbits/sec
[  4]  48.00-49.00  sec  49.5 MBytes   415 Mbits/sec
[  4]  49.00-50.00  sec  49.5 MBytes   415 Mbits/sec
[  4]  50.00-51.00  sec  49.0 MBytes   411 Mbits/sec
[  4]  51.00-52.00  sec  47.1 MBytes   395 Mbits/sec
[  4]  52.00-53.00  sec  47.6 MBytes   400 Mbits/sec
[  4]  53.00-54.00  sec  48.2 MBytes   405 Mbits/sec
[  4]  54.00-55.00  sec  50.1 MBytes   420 Mbits/sec
[  4]  55.00-56.00  sec  51.2 MBytes   429 Mbits/sec
[  4]  56.00-57.00  sec  48.5 MBytes   407 Mbits/sec
[  4]  57.00-58.00  sec  53.8 MBytes   451 Mbits/sec
[  4]  58.00-59.00  sec  51.6 MBytes   434 Mbits/sec
[  4]  59.00-60.00  sec  51.2 MBytes   429 Mbits/sec
[  4]  60.00-61.00  sec  47.5 MBytes   398 Mbits/sec
[  4]  61.00-62.00  sec  54.8 MBytes   461 Mbits/sec
[  4]  62.00-63.00  sec  51.6 MBytes   433 Mbits/sec
[... 182 lines cut ...]
[  4] 245.00-246.00 sec  50.0 MBytes   419 Mbits/sec
[  4] 246.00-247.00 sec  49.6 MBytes   417 Mbits/sec
[  4] 247.00-248.01 sec  4.25 MBytes  35.5 Mbits/sec
[  4] 248.01-249.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 249.00-250.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 250.01-251.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 251.00-252.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 252.01-253.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 253.01-254.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 254.01-255.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 255.01-256.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 256.00-257.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 257.01-258.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 258.00-259.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 259.01-260.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 260.01-261.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 261.00-262.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 262.00-263.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 263.01-264.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 264.01-265.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 265.00-266.01 sec  0.00 Bytes  0.00 bits/sec
iperf3: error - unable to write to stream socket: Connection reset by peer

C:\WINDOWS\system32>
```

CapitaineBidouille, 2022-07-29:
Now I am going to make a 5 GHz wifi network from the USB2 port as you've suggested.

2.4 GHz could go from start to finish without issues : 
```
Connecting to host 192.168.4.1, port 5201
[  4] local 192.168.3.176 port 1054 connected to 192.168.4.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.01   sec  3.12 MBytes  26.0 Mbits/sec
[  4]   1.01-2.01   sec  1.38 MBytes  11.5 Mbits/sec
[  4]   2.01-3.00   sec  1.75 MBytes  14.8 Mbits/sec
[  4]   3.00-4.01   sec  1.62 MBytes  13.6 Mbits/sec
[  4]   4.01-5.01   sec  1.00 MBytes  8.36 Mbits/sec
[  4]   5.01-6.01   sec  1.50 MBytes  12.6 Mbits/sec
[  4]   6.01-7.00   sec  2.00 MBytes  16.8 Mbits/sec
[  4]   7.00-8.01   sec  1.38 MBytes  11.4 Mbits/sec
[  4]   8.01-9.00   sec  1.75 MBytes  14.8 Mbits/sec
[  4]   9.00-10.01  sec  3.62 MBytes  30.2 Mbits/sec
[  4]  10.01-11.00  sec  1.88 MBytes  15.8 Mbits/sec
[  4]  11.00-12.00  sec  2.38 MBytes  19.9 Mbits/sec
[  4]  12.00-13.01  sec  2.38 MBytes  19.9 Mbits/sec
[  4]  13.01-14.01  sec  2.12 MBytes  17.8 Mbits/sec
[  4]  14.01-15.00  sec  1.25 MBytes  10.6 Mbits/sec
[  4]  15.00-16.01  sec  1.25 MBytes  10.4 Mbits/sec
[  4]  16.01-17.01  sec  1.75 MBytes  14.7 Mbits/sec
[  4]  17.01-18.01  sec  1.50 MBytes  12.6 Mbits/sec
[  4]  18.01-19.00  sec  1.12 MBytes  9.48 Mbits/sec
[  4]  19.00-20.01  sec  1.88 MBytes  15.6 Mbits/sec
[  4]  20.01-21.00  sec  3.62 MBytes  30.6 Mbits/sec
[  4]  21.00-22.01  sec  1.50 MBytes  12.6 Mbits/sec
[  4]  22.01-23.01  sec  1.38 MBytes  11.4 Mbits/sec
[  4]  23.01-24.01  sec  1.88 MBytes  15.8 Mbits/sec
[  4]  24.01-25.01  sec  1.50 MBytes  12.6 Mbits/sec
[  4]  25.01-26.00  sec  3.75 MBytes  31.5 Mbits/sec
[  4]  26.00-27.01  sec  4.75 MBytes  39.4 Mbits/sec
[  4]  27.01-28.01  sec  4.25 MBytes  35.9 Mbits/sec
[  4]  28.01-29.01  sec  5.12 MBytes  42.8 Mbits/sec
[  4]  29.01-30.01  sec  5.75 MBytes  48.4 Mbits/sec
[  4]  30.01-31.01  sec  4.62 MBytes  38.9 Mbits/sec
[  4]  31.01-32.02  sec  2.12 MBytes  17.7 Mbits/sec
[  4]  32.02-33.01  sec  1.38 MBytes  11.6 Mbits/sec
[  4]  33.01-34.01  sec  1.38 MBytes  11.6 Mbits/sec
[  4]  34.01-35.00  sec  1.12 MBytes  9.49 Mbits/sec
[  4]  35.00-36.02  sec  1.62 MBytes  13.5 Mbits/sec
[  4]  36.02-37.00  sec  2.62 MBytes  22.3 Mbits/sec
[  4]  37.00-38.01  sec  2.38 MBytes  19.7 Mbits/sec
[  4]  38.01-39.00  sec  2.62 MBytes  22.2 Mbits/sec
[  4]  39.00-40.00  sec  5.38 MBytes  45.2 Mbits/sec
[  4]  40.00-41.01  sec  4.88 MBytes  40.5 Mbits/sec
[  4]  41.01-42.01  sec  5.75 MBytes  48.5 Mbits/sec
[  4]  42.01-43.00  sec  5.12 MBytes  43.2 Mbits/sec
[  4]  43.00-44.00  sec  5.88 MBytes  49.3 Mbits/sec
[  4]  44.00-45.01  sec  5.00 MBytes  41.7 Mbits/sec
[  4]  45.01-46.01  sec  2.50 MBytes  20.9 Mbits/sec
[  4]  46.01-47.01  sec  3.25 MBytes  27.3 Mbits/sec
[  4]  47.01-48.00  sec  4.50 MBytes  38.0 Mbits/sec
[  4]  48.00-49.01  sec  4.50 MBytes  37.3 Mbits/sec
[  4]  49.01-50.00  sec  3.88 MBytes  32.9 Mbits/sec
[  4]  50.00-51.00  sec  4.00 MBytes  33.6 Mbits/sec
[  4]  51.00-52.01  sec  2.75 MBytes  22.9 Mbits/sec
[  4]  52.01-53.00  sec  3.00 MBytes  25.2 Mbits/sec
[  4]  53.00-54.02  sec  2.75 MBytes  22.8 Mbits/sec
[  4]  54.02-55.01  sec  2.62 MBytes  22.1 Mbits/sec
[  4]  55.01-56.01  sec  4.75 MBytes  40.0 Mbits/sec
[  4]  56.01-57.01  sec  3.25 MBytes  27.3 Mbits/sec
[  4]  57.01-58.00  sec  2.25 MBytes  18.9 Mbits/sec
[  4]  58.00-59.01  sec  3.75 MBytes  31.4 Mbits/sec
[  4]  59.01-60.01  sec  2.62 MBytes  22.0 Mbits/sec
[  4]  60.01-61.01  sec  3.12 MBytes  26.2 Mbits/sec
[  4]  61.01-62.01  sec  2.62 MBytes  22.0 Mbits/sec
[  4]  62.01-63.00  sec  2.38 MBytes  20.1 Mbits/sec
[... 221 lines cut ...]
[  4] 284.00-285.01 sec  1.88 MBytes  15.6 Mbits/sec
[  4] 285.01-286.00 sec  3.75 MBytes  31.7 Mbits/sec
[  4] 286.00-287.00 sec  3.38 MBytes  28.4 Mbits/sec
[  4] 287.00-288.01 sec  6.12 MBytes  51.1 Mbits/sec
[  4] 288.01-289.01 sec  6.12 MBytes  51.3 Mbits/sec
[  4] 289.01-290.00 sec  5.38 MBytes  45.4 Mbits/sec
[  4] 290.00-291.01 sec  6.00 MBytes  50.0 Mbits/sec
[  4] 291.01-292.00 sec  2.12 MBytes  18.0 Mbits/sec
[  4] 292.00-293.01 sec  1.12 MBytes  9.34 Mbits/sec
[  4] 293.01-294.00 sec  1.00 MBytes  8.47 Mbits/sec
[  4] 294.00-295.01 sec   896 KBytes  7.29 Mbits/sec
[  4] 295.01-296.00 sec  1.75 MBytes  14.8 Mbits/sec
[  4] 296.00-297.01 sec  1.25 MBytes  10.4 Mbits/sec
[  4] 297.01-298.00 sec  1.25 MBytes  10.5 Mbits/sec
[  4] 298.00-299.01 sec  1.75 MBytes  14.6 Mbits/sec
[  4] 299.01-300.01 sec  1.38 MBytes  11.6 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-300.01 sec   832 MBytes  23.3 Mbits/sec                  sender
[  4]   0.00-300.01 sec   832 MBytes  23.3 Mbits/sec                  receiver

iperf Done.

C:\WINDOWS\system32>
```

CapitaineBidouille, 2022-07-29:
Unplugged the USB2 adapter. Kept only the 5GHz network on an USB3 port. Could do an iperf of 500 seconds from start to finish without errors.
Maybe these adapters can't work fine together ?

CapitaineBidouille, 2022-07-29:
I just finished another 500 seconds stress test, still in USB3. No timeout at all, it's perfect.
I started a 1000 second stress test.

morrownr, 2022-07-29:
> 2.4 GHz could go from start to finish without issues

There was more variability than I would have expected but that could be due to other traffic on the same channel. This technology is designed to share so it could be normal ups and downs owing to the traffic in your area.

I see that iperf3 showed 5 GHz shutting down at the 246 second mark. It will be interesting to see what happens when the adapter is in the USB2 port. Looking at dmesg and journalctl to see what is going on when it cuts out might help.

morrownr, 2022-07-29:
> Unplugged the USB2 adapter. Kept only the 5GHz network on an USB3 port. Could do an iperf of 500 seconds from start to finish without errors. Maybe these adapters can't work fine together ?

I have one of each of those adapters and have used them together without noticing problems. Another data point in the search for the truth.

CapitaineBidouille, 2022-07-29:
> I have one of each of those adapters and have used them together without noticing problems. Another data point in the search for the truth.

Do you use them together with USB scatter gather enabled or disabled ?

morrownr, 2022-07-29:
> Do you use them together with USB scatter gather enabled or disabled ?

Probably both enabled and disabled at times. I do a lot of testing for folks that are having problems and I can't remember all the details.

Since I have most of the same hardware, I'm wondering if it would help if I start trying to duplicate your setup to see what happens here. What do you think?

CapitaineBidouille, 2022-07-29:
I guess it's going to help.

Scatter gather was ON during this test. 2.4GHz adapter on USB2 port directly on the PI, 5GHz adapter on the powered hub.

```
[   17.546757] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: user_xattr,usrjquota=aquota.user,grpjquota=aquota.group,jqfmt=vfsv0,acl. Quota mode: journalled.
[   18.015514] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   18.015594] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   33.759327] cam-dummy-reg: disabling
[   80.942207] 8021q: 802.1Q VLAN Support v1.8
[   81.055140] Adding 102396k swap on /var/swap.  Priority:-2 extents:3 across:135168k SSFS
[   84.710733] NFSD: Using UMH upcall client tracking operations.
[   84.710770] NFSD: starting 90-second grace period (net f0000081)
[  550.983430] mt76x2u 2-1.4:1.0: error: mt76x02u_mcu_wait_resp failed with -110
^C
```

```
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 378.00-379.00 sec  20.7 MBytes   173 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 379.00-380.00 sec  19.7 MBytes   165 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 380.00-381.00 sec  20.4 MBytes   171 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 381.00-382.00 sec  20.5 MBytes   172 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 382.00-383.00 sec  20.5 MBytes   172 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 383.00-384.00 sec  19.0 MBytes   159 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 384.00-385.00 sec  20.4 MBytes   171 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 385.00-386.00 sec  18.2 MBytes   153 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 386.00-387.00 sec  20.6 MBytes   172 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 387.00-388.00 sec  19.9 MBytes   167 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 388.00-389.00 sec  20.2 MBytes   169 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 389.00-390.00 sec  18.9 MBytes   159 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 390.00-391.00 sec  20.1 MBytes   169 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 391.00-392.00 sec  19.9 MBytes   167 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 392.00-393.00 sec  20.0 MBytes   168 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 393.00-394.00 sec  20.5 MBytes   172 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 394.00-395.00 sec  19.5 MBytes   164 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 395.00-396.00 sec  20.2 MBytes   169 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 396.00-397.00 sec  19.3 MBytes   162 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 397.00-398.00 sec  19.5 MBytes   164 Mbits/sec
Jul 29 20:46:50 raspberrypi iperf3[1208]: [  5] 398.00-399.00 sec  19.7 MBytes   166 Mbits/sec
Jul 29 20:47:01 raspberrypi CRON[2456]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jul 29 20:47:01 raspberrypi CRON[2457]: (root) CMD (/usr/sbin/omv-ionice >/dev/null 2>&1)
Jul 29 20:47:01 raspberrypi CRON[2456]: pam_unix(cron:session): session closed for user root
Jul 29 20:47:22 raspberrypi kernel: mt76x2u 2-1.4:1.0: error: mt76x02u_mcu_wait_resp failed with -110
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 399.00-400.00 sec  20.1 MBytes   168 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 400.00-401.00 sec  20.4 MBytes   171 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 401.00-402.00 sec  20.5 MBytes   172 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 402.00-403.00 sec  19.3 MBytes   162 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 403.00-404.00 sec  20.3 MBytes   170 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 404.00-405.00 sec  19.7 MBytes   165 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 405.00-406.00 sec  20.1 MBytes   169 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 406.00-407.00 sec  19.4 MBytes   163 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 407.00-408.00 sec  19.5 MBytes   163 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 408.00-409.00 sec  19.4 MBytes   163 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 409.00-410.00 sec  20.2 MBytes   169 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 410.00-411.00 sec  20.3 MBytes   170 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 411.00-412.00 sec  19.7 MBytes   165 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 412.00-413.00 sec  19.9 MBytes   167 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 413.00-414.00 sec  20.5 MBytes   172 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 414.00-415.00 sec  20.5 MBytes   172 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 415.00-416.00 sec  18.9 MBytes   158 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 416.00-417.00 sec  20.0 MBytes   168 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 417.00-418.00 sec  19.1 MBytes   160 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 418.00-419.00 sec  20.1 MBytes   168 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 419.00-420.00 sec  19.2 MBytes   161 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 420.00-421.00 sec  19.9 MBytes   167 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 421.00-422.00 sec  19.2 MBytes   161 Mbits/sec
[... cut ...]
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 424.00-425.00 sec  20.4 MBytes   171 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 425.00-426.00 sec  19.2 MBytes   161 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 426.00-427.00 sec  20.2 MBytes   170 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 427.00-428.00 sec  20.2 MBytes   170 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 428.00-429.00 sec  19.9 MBytes   167 Mbits/sec
Jul 29 20:47:48 raspberrypi iperf3[1208]: [  5] 429.00-430.00 sec  18.6 MBytes   156 Mbits/sec
```

CapitaineBidouille, 2022-07-29:
[image]
Since the last crash occured I cannot connect at all to the network even in limited access. No logs when I try to connect.

CapitaineBidouille, 2022-07-29:
So, disabling scatter gather definitely changing the behaviour in my setup.
If it's enabled, when the network crashes it never reconnect and I get kicked from the network without being able to reconnect.
When disabled, it's just timeouting completely during a certain time.

CapitaineBidouille, 2022-07-29:
I finally started tests with this setup : 
USB3 Port1 : Powered Hub : 3x 5TB hard drives + USB Ethernet
USB3 Port2 : Powered Hub: AWUS036ACHM (2.4GHz network)
USB2 Port1 : AWUS036ACM (5GHz network)
USB2 Port 2 : Nothing

Will post iperf3 results when it will finish. It's around 200 MBPS, not so bad but 2x slower than USB3. But it's totally stable, it will be enough for me.

CapitaineBidouille, 2022-07-29:
It went from start to finish. Will start another one.
```
Connecting to host 192.168.4.1, port 5201
[  4] local 192.168.4.176 port 1032 connected to 192.168.4.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  23.2 MBytes   195 Mbits/sec
[  4]   1.00-2.01   sec  22.8 MBytes   190 Mbits/sec
[  4]   2.01-3.00   sec  23.4 MBytes   197 Mbits/sec
[  4]   3.00-4.00   sec  23.9 MBytes   200 Mbits/sec
[  4]   4.00-5.00   sec  24.0 MBytes   202 Mbits/sec
[  4]   5.00-6.00   sec  22.8 MBytes   191 Mbits/sec
[  4]   6.00-7.00   sec  24.0 MBytes   201 Mbits/sec
[  4]   7.00-8.00   sec  24.0 MBytes   201 Mbits/sec
[  4]   8.00-9.00   sec  22.6 MBytes   190 Mbits/sec
[  4]   9.00-10.00  sec  23.6 MBytes   198 Mbits/sec
[  4]  10.00-11.00  sec  22.8 MBytes   190 Mbits/sec
[  4]  11.00-12.00  sec  24.0 MBytes   202 Mbits/sec
[  4]  12.00-13.00  sec  23.9 MBytes   200 Mbits/sec
[  4]  13.00-14.01  sec  24.1 MBytes   202 Mbits/sec
[  4]  14.01-15.00  sec  23.8 MBytes   200 Mbits/sec
[  4]  15.00-16.01  sec  23.8 MBytes   199 Mbits/sec
[  4]  16.01-17.00  sec  22.6 MBytes   191 Mbits/sec
[  4]  17.00-18.00  sec  24.0 MBytes   201 Mbits/sec
[  4]  18.00-19.00  sec  21.9 MBytes   183 Mbits/sec
[  4]  19.00-20.00  sec  23.6 MBytes   198 Mbits/sec
[  4]  20.00-21.00  sec  22.6 MBytes   190 Mbits/sec
[  4]  21.00-22.01  sec  24.1 MBytes   201 Mbits/sec
[  4]  22.01-23.00  sec  23.6 MBytes   199 Mbits/sec
[  4]  23.00-24.00  sec  24.0 MBytes   201 Mbits/sec
[  4]  24.00-25.00  sec  22.9 MBytes   192 Mbits/sec
[  4]  25.00-26.00  sec  23.9 MBytes   200 Mbits/sec
[  4]  26.00-27.00  sec  24.1 MBytes   203 Mbits/sec
[  4]  27.00-28.00  sec  22.9 MBytes   191 Mbits/sec
[  4]  28.00-29.00  sec  22.6 MBytes   190 Mbits/sec
[  4]  29.00-30.00  sec  22.8 MBytes   190 Mbits/sec
[  4]  30.00-31.00  sec  23.9 MBytes   200 Mbits/sec
[  4]  31.00-32.00  sec  22.2 MBytes   187 Mbits/sec
[  4]  32.00-33.01  sec  23.9 MBytes   200 Mbits/sec
[  4]  33.01-34.00  sec  21.9 MBytes   184 Mbits/sec
[  4]  34.00-35.00  sec  23.9 MBytes   201 Mbits/sec
[  4]  35.00-36.00  sec  21.2 MBytes   178 Mbits/sec
[  4]  36.00-37.00  sec  24.1 MBytes   202 Mbits/sec
[  4]  37.00-38.00  sec  22.6 MBytes   190 Mbits/sec
[  4]  38.00-39.00  sec  24.0 MBytes   202 Mbits/sec
[  4]  39.00-40.00  sec  22.6 MBytes   190 Mbits/sec
[  4]  40.00-41.00  sec  23.8 MBytes   199 Mbits/sec
[  4]  41.00-42.01  sec  22.6 MBytes   189 Mbits/sec
[  4]  42.01-43.00  sec  23.1 MBytes   195 Mbits/sec
[  4]  43.00-44.00  sec  22.8 MBytes   190 Mbits/sec
[  4]  44.00-45.00  sec  24.0 MBytes   201 Mbits/sec
[  4]  45.00-46.00  sec  23.9 MBytes   200 Mbits/sec
[  4]  46.00-47.00  sec  22.6 MBytes   190 Mbits/sec
[  4]  47.00-48.00  sec  24.0 MBytes   202 Mbits/sec
[  4]  48.00-49.00  sec  22.6 MBytes   190 Mbits/sec
[  4]  49.00-50.00  sec  23.9 MBytes   200 Mbits/sec
[  4]  50.00-51.00  sec  23.5 MBytes   197 Mbits/sec
[  4]  51.00-52.00  sec  23.9 MBytes   200 Mbits/sec
[  4]  52.00-53.00  sec  23.2 MBytes   195 Mbits/sec
[  4]  53.00-54.00  sec  23.9 MBytes   200 Mbits/sec
[  4]  54.00-55.00  sec  23.4 MBytes   195 Mbits/sec
[  4]  55.00-56.00  sec  23.5 MBytes   198 Mbits/sec
[  4]  56.00-57.00  sec  22.4 MBytes   187 Mbits/sec
[  4]  57.00-58.00  sec  23.0 MBytes   193 Mbits/sec
[  4]  58.00-59.00  sec  22.1 MBytes   186 Mbits/sec
[  4]  59.00-60.00  sec  23.5 MBytes   197 Mbits/sec
[  4]  60.00-61.00  sec  23.4 MBytes   196 Mbits/sec
[  4]  61.00-62.00  sec  23.4 MBytes   196 Mbits/sec
[  4]  62.00-63.01  sec  21.1 MBytes   176 Mbits/sec
[  4]  63.01-64.01  sec  22.5 MBytes   189 Mbits/sec
[  4]  64.01-65.00  sec  22.0 MBytes   185 Mbits/sec
[... 917 lines cut ...]
[  4] 982.00-983.01 sec  20.2 MBytes   169 Mbits/sec
[  4] 983.01-984.00 sec  20.5 MBytes   172 Mbits/sec
[  4] 984.00-985.00 sec  20.0 MBytes   168 Mbits/sec
[  4] 985.00-986.00 sec  20.0 MBytes   168 Mbits/sec
[  4] 986.00-987.00 sec  19.4 MBytes   163 Mbits/sec
[  4] 987.00-988.00 sec  19.8 MBytes   165 Mbits/sec
[  4] 988.00-989.01 sec  19.5 MBytes   163 Mbits/sec
[  4] 989.01-990.00 sec  19.5 MBytes   165 Mbits/sec
[  4] 990.00-991.01 sec  20.4 MBytes   170 Mbits/sec
[  4] 991.01-992.01 sec  20.2 MBytes   170 Mbits/sec
[  4] 992.01-993.00 sec  17.0 MBytes   143 Mbits/sec
[  4] 993.00-994.00 sec  19.8 MBytes   166 Mbits/sec
[  4] 994.00-995.00 sec  18.9 MBytes   158 Mbits/sec
[  4] 995.00-996.01 sec  20.5 MBytes   171 Mbits/sec
[  4] 996.01-997.00 sec  19.5 MBytes   164 Mbits/sec
[  4] 997.00-998.00 sec  19.9 MBytes   167 Mbits/sec
[  4] 998.00-999.00 sec  19.6 MBytes   165 Mbits/sec
[  4] 999.00-1000.00 sec  20.0 MBytes   168 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1000.00 sec  20.9 GBytes   179 Mbits/sec                  sender
[  4]   0.00-1000.00 sec  20.9 GBytes   179 Mbits/sec                  receiver

iperf Done.
```

CapitaineBidouille, 2022-07-29:
2nd 1000 seconds test finished. Totally stable.

CapitaineBidouille, 2022-07-29:
That 2.4 GHz is performing perfectly, there is no regression.
Looks like I am getting the stable setup I wanted.

```
Connecting to host 192.168.4.1, port 5201
[  4] local 192.168.3.176 port 1033 connected to 192.168.4.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.01   sec  1.50 MBytes  12.5 Mbits/sec
[  4]   1.01-2.00   sec  1.75 MBytes  14.8 Mbits/sec
[  4]   2.00-3.01   sec   896 KBytes  7.25 Mbits/sec
[  4]   3.01-4.00   sec  1.12 MBytes  9.55 Mbits/sec
[  4]   4.00-5.01   sec  2.88 MBytes  23.8 Mbits/sec
[  4]   5.01-6.00   sec  1.88 MBytes  15.9 Mbits/sec
[  4]   6.00-7.01   sec  1.50 MBytes  12.5 Mbits/sec
[  4]   7.01-8.02   sec  1.12 MBytes  9.35 Mbits/sec
[  4]   8.02-9.00   sec   896 KBytes  7.44 Mbits/sec
[  4]   9.00-10.01  sec   512 KBytes  4.16 Mbits/sec
[  4]  10.01-11.01  sec   768 KBytes  6.31 Mbits/sec
[  4]  11.01-12.01  sec   768 KBytes  6.27 Mbits/sec
[  4]  12.01-13.00  sec   512 KBytes  4.22 Mbits/sec
[  4]  13.00-14.01  sec   512 KBytes  4.17 Mbits/sec
[  4]  14.01-15.00  sec   768 KBytes  6.32 Mbits/sec
[  4]  15.00-16.01  sec  2.88 MBytes  24.1 Mbits/sec
[  4]  16.01-17.02  sec  6.62 MBytes  55.0 Mbits/sec
[  4]  17.02-18.00  sec  4.62 MBytes  39.3 Mbits/sec
[  4]  18.00-19.01  sec  5.38 MBytes  44.8 Mbits/sec
[  4]  19.01-20.01  sec  4.50 MBytes  37.9 Mbits/sec
[  4]  20.01-21.00  sec  4.62 MBytes  39.0 Mbits/sec
[  4]  21.00-22.00  sec  5.38 MBytes  45.0 Mbits/sec
[  4]  22.00-23.00  sec  1.50 MBytes  12.6 Mbits/sec
[  4]  23.00-24.00  sec  1.62 MBytes  13.7 Mbits/sec
[  4]  24.00-25.00  sec  1.62 MBytes  13.6 Mbits/sec
[  4]  25.00-26.00  sec  1.00 MBytes  8.42 Mbits/sec
[  4]  26.00-27.00  sec   768 KBytes  6.28 Mbits/sec
[  4]  27.00-28.01  sec  1.38 MBytes  11.5 Mbits/sec
[  4]  28.01-29.00  sec  3.62 MBytes  30.6 Mbits/sec
[  4]  29.00-30.02  sec  5.12 MBytes  42.4 Mbits/sec
[  4]  30.02-31.00  sec  4.50 MBytes  38.2 Mbits/sec
[  4]  31.00-32.00  sec  5.88 MBytes  49.3 Mbits/sec
[  4]  32.00-33.00  sec  4.75 MBytes  39.9 Mbits/sec
[  4]  33.00-34.00  sec  4.12 MBytes  34.5 Mbits/sec
[  4]  34.00-35.00  sec  5.38 MBytes  45.1 Mbits/sec
[  4]  35.00-36.01  sec  3.00 MBytes  25.0 Mbits/sec
[  4]  36.01-37.00  sec  2.50 MBytes  21.1 Mbits/sec
[  4]  37.00-38.01  sec  1.25 MBytes  10.4 Mbits/sec
[  4]  38.01-39.01  sec  2.00 MBytes  16.9 Mbits/sec
[  4]  39.01-40.00  sec  4.00 MBytes  33.6 Mbits/sec
[  4]  40.00-41.00  sec  5.88 MBytes  49.4 Mbits/sec
[  4]  41.00-42.01  sec  3.88 MBytes  32.1 Mbits/sec
[  4]  42.01-43.01  sec  2.38 MBytes  20.1 Mbits/sec
[  4]  43.01-44.00  sec  1.00 MBytes  8.43 Mbits/sec
[  4]  44.00-45.00  sec  2.00 MBytes  16.7 Mbits/sec
[  4]  45.00-46.00  sec  1.50 MBytes  12.6 Mbits/sec
[  4]  46.00-47.01  sec   896 KBytes  7.28 Mbits/sec
[  4]  47.01-48.00  sec  1.25 MBytes  10.6 Mbits/sec
[  4]  48.00-49.00  sec  1.88 MBytes  15.7 Mbits/sec
[  4]  49.00-50.01  sec  2.38 MBytes  19.9 Mbits/sec
[  4]  50.01-51.00  sec  2.62 MBytes  22.2 Mbits/sec
[  4]  51.00-52.00  sec  3.50 MBytes  29.3 Mbits/sec
[  4]  52.00-53.01  sec  5.12 MBytes  42.6 Mbits/sec
[  4]  53.01-54.01  sec  2.88 MBytes  24.2 Mbits/sec
[  4]  54.01-55.00  sec  2.38 MBytes  20.0 Mbits/sec
[  4]  55.00-56.01  sec  1.88 MBytes  15.6 Mbits/sec
[  4]  56.01-57.00  sec  2.50 MBytes  21.2 Mbits/sec
[  4]  57.00-58.00  sec  2.62 MBytes  22.0 Mbits/sec
[  4]  58.00-59.01  sec  3.00 MBytes  25.0 Mbits/sec
[  4]  59.01-60.00  sec  3.75 MBytes  31.7 Mbits/sec
[  4]  60.00-61.01  sec  1.88 MBytes  15.6 Mbits/sec
[  4]  61.01-62.00  sec  2.00 MBytes  16.9 Mbits/sec
[  4]  62.00-63.00  sec   768 KBytes  6.30 Mbits/sec
[... 919 lines cut ...]
[  4] 982.01-983.01 sec  3.75 MBytes  31.5 Mbits/sec
[  4] 983.01-984.00 sec  5.75 MBytes  48.5 Mbits/sec
[  4] 984.00-985.01 sec  6.25 MBytes  51.9 Mbits/sec
[  4] 985.01-986.01 sec  7.38 MBytes  62.1 Mbits/sec
[  4] 986.01-987.00 sec  6.25 MBytes  52.7 Mbits/sec
[  4] 987.00-988.00 sec  4.75 MBytes  39.9 Mbits/sec
[  4] 988.00-989.00 sec  4.88 MBytes  40.8 Mbits/sec
[  4] 989.00-990.00 sec  6.25 MBytes  52.6 Mbits/sec
[  4] 990.00-991.00 sec  6.12 MBytes  51.4 Mbits/sec
[  4] 991.00-992.00 sec  3.88 MBytes  32.4 Mbits/sec
[  4] 992.00-993.01 sec  3.12 MBytes  26.0 Mbits/sec
[  4] 993.01-994.00 sec  2.38 MBytes  20.0 Mbits/sec
[  4] 994.00-995.01 sec  3.50 MBytes  29.2 Mbits/sec
[  4] 995.01-996.01 sec  2.88 MBytes  24.1 Mbits/sec
[  4] 996.01-997.01 sec  2.38 MBytes  20.0 Mbits/sec
[  4] 997.01-998.00 sec  3.12 MBytes  26.3 Mbits/sec
[  4] 998.00-999.01 sec  2.12 MBytes  17.7 Mbits/sec
[  4] 999.01-1000.00 sec  4.62 MBytes  39.1 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1000.00 sec  4.03 GBytes  34.6 Mbits/sec                  sender
[  4]   0.00-1000.00 sec  4.03 GBytes  34.6 Mbits/sec                  receiver

iperf Done.
```

CapitaineBidouille, 2022-07-29:
Started another 1000 seconds test on the Ethernet adapter.

morrownr, 2022-07-29:
So, when the USB3 adapter is in a USB2 port, which means it is running in USB2 mode, we are seeing fully stable operation. The USB2 adapter is still going to be in USB2 mode even if in a USB3 port because it is not capable of USB3.

The rat that I smell is USB3. 

Let me ask some questions to see if I understand your use case:

What kind of range do you need from 2.4 GHz?
Is 200 Mb/s adequate for 5 GHz? (for now)

I think I am going to try to setup a near duplicate of your setup to see what I can see. I like things that are 24/7/365 stable and the thing that causes concern with me is the two powered hubs working on USB3. That could be destabilizing Things with wifi.

CapitaineBidouille, 2022-07-29:
> 

I already tried previously to plug one USB3 powered hub on an USB2 port for the 2.4GHz adapter.
So only one powered hub was really using USB3, the other one was at USB2 speed. It did not work any better.

What kind of range do you need from 2.4 GHz?
10 meters minimum

Is 200 Mb/s adequate for 5 GHz? (for now)
It is enough. My main computer is plugged with Ethernet anyway.
It will be better to have more to get the most of my hardware but I will not feel much the difference in real use cases.

The iperf3 test in Ethernet went from start to finish without any issue. I was even watching plex using my phone connected to the 5GHz wifi network during the test to stress one hard drive a bit : 
```
Connecting to host 192.168.4.1, port 5201
[  4] local 192.168.5.115 port 8796 connected to 192.168.4.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   112 MBytes   939 Mbits/sec
[  4]   1.00-2.00   sec   112 MBytes   938 Mbits/sec
[  4]   2.00-3.00   sec   107 MBytes   901 Mbits/sec
[  4]   3.00-4.00   sec   112 MBytes   942 Mbits/sec
[  4]   4.00-5.00   sec   110 MBytes   926 Mbits/sec
[  4]   5.00-6.00   sec   111 MBytes   931 Mbits/sec
[  4]   6.00-7.00   sec   112 MBytes   941 Mbits/sec
[  4]   7.00-8.00   sec   112 MBytes   941 Mbits/sec
[  4]   8.00-9.00   sec   110 MBytes   924 Mbits/sec
[  4]   9.00-10.00  sec   111 MBytes   934 Mbits/sec
[  4]  10.00-11.00  sec   111 MBytes   933 Mbits/sec
[  4]  11.00-12.00  sec   112 MBytes   941 Mbits/sec
[  4]  12.00-13.00  sec   111 MBytes   928 Mbits/sec
[  4]  13.00-14.00  sec   112 MBytes   941 Mbits/sec
[  4]  14.00-15.00  sec   112 MBytes   943 Mbits/sec
[  4]  15.00-16.00  sec   112 MBytes   942 Mbits/sec
[  4]  16.00-17.00  sec   112 MBytes   942 Mbits/sec
[  4]  17.00-18.00  sec   112 MBytes   942 Mbits/sec
[  4]  18.00-19.00  sec   112 MBytes   939 Mbits/sec
[  4]  19.00-20.00  sec   112 MBytes   939 Mbits/sec
[  4]  20.00-21.00  sec   112 MBytes   943 Mbits/sec
[  4]  21.00-22.00  sec   112 MBytes   941 Mbits/sec
[  4]  22.00-23.00  sec   112 MBytes   942 Mbits/sec
[  4]  23.00-24.00  sec   112 MBytes   942 Mbits/sec
[  4]  24.00-25.00  sec   112 MBytes   943 Mbits/sec
[  4]  25.00-26.00  sec   112 MBytes   942 Mbits/sec
[  4]  26.00-27.00  sec   112 MBytes   942 Mbits/sec
[  4]  27.00-28.00  sec   112 MBytes   942 Mbits/sec
[  4]  28.00-29.00  sec   112 MBytes   943 Mbits/sec
[  4]  29.00-30.00  sec   112 MBytes   942 Mbits/sec
[  4]  30.00-31.00  sec   112 MBytes   939 Mbits/sec
[  4]  31.00-32.00  sec   112 MBytes   938 Mbits/sec
[  4]  32.00-33.00  sec   112 MBytes   942 Mbits/sec
[  4]  33.00-34.00  sec   112 MBytes   940 Mbits/sec
[  4]  34.00-35.00  sec   112 MBytes   940 Mbits/sec
[  4]  35.00-36.00  sec   111 MBytes   932 Mbits/sec
[  4]  36.00-37.00  sec   110 MBytes   923 Mbits/sec
[  4]  37.00-38.00  sec   111 MBytes   934 Mbits/sec
[  4]  38.00-39.00  sec   112 MBytes   936 Mbits/sec
[  4]  39.00-40.00  sec   111 MBytes   929 Mbits/sec
[  4]  40.00-41.00  sec   112 MBytes   937 Mbits/sec
[  4]  41.00-42.00  sec   112 MBytes   941 Mbits/sec
[  4]  42.00-43.00  sec   112 MBytes   937 Mbits/sec
[  4]  43.00-44.00  sec   112 MBytes   942 Mbits/sec
[  4]  44.00-45.00  sec   112 MBytes   937 Mbits/sec
[  4]  45.00-46.00  sec   110 MBytes   922 Mbits/sec
[  4]  46.00-47.00  sec   111 MBytes   930 Mbits/sec
[  4]  47.00-48.00  sec   111 MBytes   934 Mbits/sec
[  4]  48.00-49.00  sec   112 MBytes   942 Mbits/sec
[  4]  49.00-50.00  sec   112 MBytes   940 Mbits/sec
[  4]  50.00-51.00  sec   112 MBytes   938 Mbits/sec
[  4]  51.00-52.00  sec   112 MBytes   942 Mbits/sec
[  4]  52.00-53.00  sec   112 MBytes   942 Mbits/sec
[... 931 lines cut ...]
[  4] 984.00-985.00 sec   112 MBytes   942 Mbits/sec
[  4] 985.00-986.00 sec   112 MBytes   942 Mbits/sec
[  4] 986.00-987.00 sec   112 MBytes   942 Mbits/sec
[  4] 987.00-988.00 sec   112 MBytes   942 Mbits/sec
[  4] 988.00-989.00 sec   112 MBytes   942 Mbits/sec
[  4] 989.00-990.00 sec   112 MBytes   941 Mbits/sec
[  4] 990.00-991.00 sec   112 MBytes   941 Mbits/sec
[  4] 991.00-992.00 sec   112 MBytes   941 Mbits/sec
[  4] 992.00-993.00 sec   112 MBytes   941 Mbits/sec
[  4] 993.00-994.00 sec   112 MBytes   942 Mbits/sec
[  4] 994.00-995.00 sec   108 MBytes   903 Mbits/sec
[  4] 995.00-996.00 sec   112 MBytes   938 Mbits/sec
[  4] 996.00-997.00 sec   112 MBytes   941 Mbits/sec
[  4] 997.00-998.00 sec   112 MBytes   940 Mbits/sec
[  4] 998.00-999.00 sec   112 MBytes   937 Mbits/sec
[  4] 999.00-1000.00 sec   112 MBytes   941 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1000.00 sec   109 GBytes   937 Mbits/sec                  sender
[  4]   0.00-1000.00 sec   109 GBytes   937 Mbits/sec                  receiver

iperf Done.

C:\WINDOWS\system32>
```

morrownr, 2022-07-29:
> 937 Mbits/sec

Nothing to worry about with eth. iperf3 is fun isn't it. You can really pound the various connections in your local network with it. It is very good at finding problems.

> What kind of range do you need from 2.4 GHz? 10 meters minimum
>Is 200 Mb/s adequate for 5 GHz? (for now) It is enough. My main computer is plugged with Ethernet anyway.

Where I am going with this is would you be willing to test the following setup?

```
USB3 Port1 : Powered Hub : 2x 5TB hard drives 
USB3 Port2 : Powered Hub : 1x 5TB hard drives + USB Ethernet
USB2 Port1 : Powered Hub: AWUS036ACHM (5GHz network)
USB2 Port 2 : Nothing

2,4 GHz WiFi provided by RasPi internal wifi
```

I'm just looking for something totally stable that meets your current needs while we then take our time to figure out why the ACM is not fully stable when in USB3 mode. You would need to make some mods to your hostapd.conf files to account for changing the ACHM over to 5 GHz but that is doable and I can point you right to the lines to change. That stable system would give me time to setup and try to get to the bottom of this on my end. I can't promise a timeframe but will give it a try.

CapitaineBidouille, 2022-07-29:
I do not have three powered hubs unfortunately right now.

morrownr, 2022-07-29:
Typo. I meant:

```
USB3 Port1 : Powered Hub : 2x 5TB hard drives 
USB3 Port2 : Powered Hub : 1x 5TB hard drives + USB Ethernet
USB2 Port1 : AWUS036ACHM (5GHz network)
USB2 Port 2 : Nothing

2,4 GHz WiFi provided by RasPi internal wifi
```

CapitaineBidouille, 2022-07-29:
Looks like I'm still getting instabilities even with USB2.
Got timeout after around 1 hour of Twitch streaming + online gaming.
I am going to try your proposal later, let's hope it will be perfect.

CapitaineBidouille, 2022-07-29:
Trying another setup, pretty near of what you've suggested : 
USB3 Port1 : Powered Hub : 2x 5TB hard drives 
USB3 Port2 : Powered Hub : 1x 5TB hard drives + USB Ethernet + AWUS036ACM 
USB2 Port1 : Nothing
USB2 Port2 : Nothing

2,4 GHz WiFi provided by RasPi internal wifi

CapitaineBidouille, 2022-07-29:
Ok, it does not work : 
[  4] 364.00-365.00 sec  33.4 MBytes   280 Mbits/sec
[  4] 365.00-366.01 sec  27.6 MBytes   231 Mbits/sec
[  4] 366.01-367.00 sec  26.2 MBytes   221 Mbits/sec
[  4] 367.00-368.01 sec  8.12 MBytes  67.6 Mbits/sec
[  4] 368.01-369.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 369.01-370.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 370.01-371.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 371.01-372.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 372.00-373.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 373.01-374.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 374.01-375.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 375.01-376.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 376.01-377.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 377.01-378.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 378.00-379.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 379.01-380.02 sec  0.00 Bytes  0.00 bits/sec
[  4] 380.02-381.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 381.01-382.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 382.00-383.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 383.01-384.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 384.01-385.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 385.01-386.01 sec  0.00 Bytes  0.00 bits/sec
Will follow exactly your advice.

CapitaineBidouille, 2022-07-29:
Scatter gather : ON
Unbelievable: it's still not working with the setup you've suggested
I could reconnect to the AP a few seconds after and could get back Internet access.
```
Connecting to host 192.168.3.1, port 5201
[  4] local 192.168.3.176 port 13870 connected to 192.168.3.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  27.5 MBytes   230 Mbits/sec
[  4]   1.00-2.00   sec  25.4 MBytes   213 Mbits/sec
[  4]   2.00-3.00   sec  26.9 MBytes   225 Mbits/sec
[  4]   3.00-4.00   sec  27.1 MBytes   228 Mbits/sec
[  4]   4.00-5.00   sec  27.2 MBytes   228 Mbits/sec
[  4]   5.00-6.00   sec  27.6 MBytes   232 Mbits/sec
[  4]   6.00-7.00   sec  27.8 MBytes   233 Mbits/sec
[  4]   7.00-8.00   sec  27.5 MBytes   231 Mbits/sec
[  4]   8.00-9.00   sec  27.6 MBytes   232 Mbits/sec
[  4]   9.00-10.00  sec  27.6 MBytes   232 Mbits/sec
[  4]  10.00-11.00  sec  27.0 MBytes   227 Mbits/sec
[  4]  11.00-12.00  sec  27.9 MBytes   234 Mbits/sec
[  4]  12.00-13.00  sec  27.5 MBytes   230 Mbits/sec
[  4]  13.00-14.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  14.00-15.00  sec  27.2 MBytes   229 Mbits/sec
[  4]  15.00-16.00  sec  27.9 MBytes   233 Mbits/sec
[  4]  16.00-17.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  17.00-18.00  sec  28.0 MBytes   236 Mbits/sec
[  4]  18.00-19.00  sec  28.1 MBytes   236 Mbits/sec
[  4]  19.00-20.00  sec  28.0 MBytes   234 Mbits/sec
[  4]  20.00-21.00  sec  27.8 MBytes   233 Mbits/sec
[  4]  21.00-22.00  sec  27.6 MBytes   232 Mbits/sec
[  4]  22.00-23.00  sec  27.5 MBytes   230 Mbits/sec
[  4]  23.00-24.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  24.00-25.00  sec  28.2 MBytes   237 Mbits/sec
[  4]  25.00-26.00  sec  27.4 MBytes   230 Mbits/sec
[  4]  26.00-27.00  sec  26.9 MBytes   225 Mbits/sec
[  4]  27.00-28.00  sec  27.6 MBytes   232 Mbits/sec
[  4]  28.00-29.00  sec  27.8 MBytes   233 Mbits/sec
[  4]  29.00-30.00  sec  21.6 MBytes   181 Mbits/sec
[  4]  30.00-31.00  sec  23.5 MBytes   198 Mbits/sec
[  4]  31.00-32.00  sec  25.8 MBytes   216 Mbits/sec
[  4]  32.00-33.00  sec  28.1 MBytes   236 Mbits/sec
[  4]  33.00-34.00  sec  28.1 MBytes   236 Mbits/sec
[  4]  34.00-35.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  35.00-36.00  sec  27.8 MBytes   233 Mbits/sec
[  4]  36.00-37.00  sec  27.9 MBytes   233 Mbits/sec
[  4]  37.00-38.00  sec  28.4 MBytes   238 Mbits/sec
[  4]  38.00-39.01  sec  27.0 MBytes   225 Mbits/sec
[  4]  39.01-40.00  sec  27.5 MBytes   232 Mbits/sec
[  4]  40.00-41.00  sec  27.4 MBytes   229 Mbits/sec
[  4]  41.00-42.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  42.00-43.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  43.00-44.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  44.00-45.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  45.00-46.00  sec  28.1 MBytes   236 Mbits/sec
[  4]  46.00-47.00  sec  27.6 MBytes   231 Mbits/sec
[  4]  47.00-48.00  sec  28.2 MBytes   237 Mbits/sec
[  4]  48.00-49.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  49.00-50.00  sec  27.9 MBytes   234 Mbits/sec
[  4]  50.00-51.00  sec  27.9 MBytes   234 Mbits/sec
[  4]  51.00-52.00  sec  27.0 MBytes   226 Mbits/sec
[  4]  52.00-53.00  sec  27.6 MBytes   232 Mbits/sec
[  4]  53.00-54.00  sec  27.6 MBytes   231 Mbits/sec
[  4]  54.00-55.00  sec  27.6 MBytes   231 Mbits/sec
[  4]  55.00-56.01  sec  27.4 MBytes   229 Mbits/sec
[  4]  56.01-57.01  sec  27.8 MBytes   233 Mbits/sec
[  4]  57.01-58.00  sec  27.5 MBytes   232 Mbits/sec
[  4]  58.00-59.00  sec  27.6 MBytes   231 Mbits/sec
[  4]  59.00-60.00  sec  27.9 MBytes   235 Mbits/sec
[  4]  60.00-61.00  sec  28.1 MBytes   236 Mbits/sec
[  4]  61.00-62.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  62.00-63.00  sec  27.2 MBytes   228 Mbits/sec
[... 517 lines cut ...]
[  4] 580.01-581.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 581.00-582.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 582.01-583.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 583.01-584.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 584.01-585.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 585.01-586.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 586.00-587.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 587.00-588.00 sec  0.00 Bytes  0.00 bits/sec
[  4] 588.00-589.02 sec  0.00 Bytes  0.00 bits/sec
[  4] 589.02-590.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 590.01-591.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 591.01-592.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 592.01-593.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 593.01-594.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 594.01-595.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 595.01-596.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 596.01-597.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 597.01-598.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 598.01-599.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 599.01-600.01 sec  0.00 Bytes  0.00 bits/sec
[  4] 600.01-601.01 sec  0.00 Bytes  0.00 bits/sec
iperf3: error - unable to write to stream socket: Connection reset by peer

C:\WINDOWS\system32>
```

CapitaineBidouille, 2022-07-30:
Scather gather : OFF
Same setup.
This time the test was stable from start to finish : 
So looks like scatter gather is also related to USB2 devices.
```
Connecting to host 192.168.3.1, port 5201
[  4] local 192.168.3.176 port 24209 connected to 192.168.3.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.01   sec   640 KBytes  5.19 Mbits/sec
[  4]   1.01-2.01   sec   896 KBytes  7.37 Mbits/sec
[  4]   2.01-3.01   sec   512 KBytes  4.19 Mbits/sec
[  4]   3.01-4.00   sec   896 KBytes  7.35 Mbits/sec
[  4]   4.00-5.01   sec  1.88 MBytes  15.7 Mbits/sec
[  4]   5.01-6.01   sec  1.62 MBytes  13.6 Mbits/sec
[  4]   6.01-7.01   sec  2.12 MBytes  17.9 Mbits/sec
[  4]   7.01-8.01   sec  6.38 MBytes  53.4 Mbits/sec
[  4]   8.01-9.00   sec  7.75 MBytes  65.3 Mbits/sec
[  4]   9.00-10.01  sec  7.88 MBytes  65.3 Mbits/sec
[  4]  10.01-11.00  sec  7.25 MBytes  61.7 Mbits/sec
[  4]  11.00-12.01  sec  5.25 MBytes  43.6 Mbits/sec
[  4]  12.01-13.00  sec  6.25 MBytes  53.0 Mbits/sec
[  4]  13.00-14.01  sec  11.5 MBytes  95.4 Mbits/sec
[  4]  14.01-15.01  sec  7.62 MBytes  64.1 Mbits/sec
[  4]  15.01-16.01  sec  10.2 MBytes  86.1 Mbits/sec
[  4]  16.01-17.00  sec  23.8 MBytes   200 Mbits/sec
[  4]  17.00-18.00  sec  26.0 MBytes   219 Mbits/sec
[  4]  18.00-19.01  sec  26.2 MBytes   219 Mbits/sec
[  4]  19.01-20.00  sec  26.8 MBytes   226 Mbits/sec
[  4]  20.00-21.00  sec  28.8 MBytes   241 Mbits/sec
[  4]  21.00-22.00  sec  29.0 MBytes   244 Mbits/sec
[  4]  22.00-23.00  sec  28.8 MBytes   242 Mbits/sec
[  4]  23.00-24.00  sec  29.0 MBytes   243 Mbits/sec
[  4]  24.00-25.00  sec  29.5 MBytes   247 Mbits/sec
[  4]  25.00-26.00  sec  28.5 MBytes   239 Mbits/sec
[  4]  26.00-27.00  sec  29.2 MBytes   245 Mbits/sec
[  4]  27.00-28.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  28.00-29.00  sec  28.9 MBytes   242 Mbits/sec
[  4]  29.00-30.00  sec  29.0 MBytes   243 Mbits/sec
[  4]  30.00-31.00  sec  28.8 MBytes   241 Mbits/sec
[  4]  31.00-32.00  sec  29.2 MBytes   245 Mbits/sec
[  4]  32.00-33.00  sec  28.5 MBytes   239 Mbits/sec
[  4]  33.00-34.00  sec  25.5 MBytes   214 Mbits/sec
[  4]  34.00-35.00  sec  16.9 MBytes   141 Mbits/sec
[  4]  35.00-36.02  sec  23.2 MBytes   192 Mbits/sec
[  4]  36.02-37.00  sec  22.6 MBytes   193 Mbits/sec
[  4]  37.00-38.00  sec  25.5 MBytes   214 Mbits/sec
[  4]  38.00-39.00  sec  27.9 MBytes   233 Mbits/sec
[  4]  39.00-40.00  sec  29.1 MBytes   245 Mbits/sec
[  4]  40.00-41.00  sec  28.8 MBytes   241 Mbits/sec
[  4]  41.00-42.00  sec  26.2 MBytes   220 Mbits/sec
[  4]  42.00-43.00  sec  28.0 MBytes   234 Mbits/sec
[  4]  43.00-44.00  sec  27.9 MBytes   235 Mbits/sec
[  4]  44.00-45.00  sec  29.1 MBytes   244 Mbits/sec
[  4]  45.00-46.00  sec  28.9 MBytes   243 Mbits/sec
[  4]  46.00-47.00  sec  28.8 MBytes   241 Mbits/sec
[  4]  47.00-48.00  sec  29.1 MBytes   244 Mbits/sec
[  4]  48.00-49.00  sec  28.6 MBytes   240 Mbits/sec
[  4]  49.00-50.00  sec  28.8 MBytes   242 Mbits/sec
[  4]  50.00-51.00  sec  28.6 MBytes   240 Mbits/sec
[  4]  51.00-52.00  sec  28.8 MBytes   241 Mbits/sec
[  4]  52.00-53.00  sec  28.5 MBytes   239 Mbits/sec
[  4]  53.00-54.00  sec  29.2 MBytes   246 Mbits/sec
[  4]  54.00-55.00  sec  29.0 MBytes   243 Mbits/sec
[  4]  55.00-56.00  sec  27.5 MBytes   231 Mbits/sec
[  4]  56.00-57.00  sec  29.1 MBytes   244 Mbits/sec
[  4]  57.00-58.00  sec  29.2 MBytes   245 Mbits/sec
[  4]  58.00-59.00  sec  28.6 MBytes   241 Mbits/sec
[  4]  59.00-60.00  sec  29.1 MBytes   244 Mbits/sec
[  4]  60.00-61.00  sec  28.1 MBytes   235 Mbits/sec
[  4]  61.00-62.00  sec  28.5 MBytes   239 Mbits/sec
[... 822 lines cut ...]
[  4] 884.00-885.00 sec  28.4 MBytes   238 Mbits/sec
[  4] 885.00-886.00 sec  27.8 MBytes   233 Mbits/sec
[  4] 886.00-887.00 sec  28.4 MBytes   238 Mbits/sec
[  4] 887.00-888.00 sec  27.6 MBytes   232 Mbits/sec
[  4] 888.00-889.00 sec  27.9 MBytes   235 Mbits/sec
[  4] 889.00-890.00 sec  27.2 MBytes   228 Mbits/sec
[  4] 890.00-891.00 sec  27.5 MBytes   231 Mbits/sec
[  4] 891.00-892.01 sec  27.5 MBytes   229 Mbits/sec
[  4] 892.01-893.00 sec  27.0 MBytes   227 Mbits/sec
[  4] 893.00-894.00 sec  27.2 MBytes   229 Mbits/sec
[  4] 894.00-895.00 sec  27.0 MBytes   226 Mbits/sec
[  4] 895.00-896.01 sec  27.4 MBytes   228 Mbits/sec
[  4] 896.01-897.00 sec  27.4 MBytes   231 Mbits/sec
[  4] 897.00-898.00 sec  27.4 MBytes   230 Mbits/sec
[  4] 898.00-899.00 sec  27.4 MBytes   229 Mbits/sec
[  4] 899.00-900.00 sec  27.2 MBytes   229 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-900.00 sec  24.0 GBytes   229 Mbits/sec                  sender
[  4]   0.00-900.00 sec  24.0 GBytes   229 Mbits/sec                  receiver

iperf Done.

C:\WINDOWS\system32>
```

CapitaineBidouille, 2022-07-30:
Started an ultimate 1800 seconds test, let's go

CapitaineBidouille, 2022-07-30:
Worked !
Now let's start a 5H stress test while I will sleep :)

C:\WINDOWS\system32>iperf3 -c 192.168.3.1 -t 1800
Connecting to host 192.168.3.1, port 5201
[  4] local 192.168.3.176 port 10149 connected to 192.168.3.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  26.0 MBytes   217 Mbits/sec
[  4]   1.00-2.00   sec  26.0 MBytes   219 Mbits/sec
[  4]   2.00-3.00   sec  26.6 MBytes   224 Mbits/sec
[  4]   3.00-4.00   sec  27.4 MBytes   229 Mbits/sec
[  4]   4.00-5.00   sec  26.9 MBytes   225 Mbits/sec
[  4]   5.00-6.00   sec  27.4 MBytes   230 Mbits/sec
[  4]   6.00-7.00   sec  27.1 MBytes   227 Mbits/sec
[  4]   7.00-8.00   sec  27.2 MBytes   228 Mbits/sec
[  4]   8.00-9.00   sec  27.2 MBytes   229 Mbits/sec
[  4]   9.00-10.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  10.00-11.00  sec  27.2 MBytes   229 Mbits/sec
[  4]  11.00-12.01  sec  27.8 MBytes   232 Mbits/sec
[  4]  12.01-13.00  sec  27.0 MBytes   228 Mbits/sec
[  4]  13.00-14.00  sec  28.0 MBytes   234 Mbits/sec
[  4]  14.00-15.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  15.00-16.00  sec  27.0 MBytes   227 Mbits/sec
[  4]  16.00-17.00  sec  27.2 MBytes   229 Mbits/sec
[  4]  17.00-18.00  sec  27.4 MBytes   229 Mbits/sec
[  4]  18.00-19.00  sec  27.6 MBytes   231 Mbits/sec
[  4]  19.00-20.00  sec  27.9 MBytes   234 Mbits/sec
[  4]  20.00-21.00  sec  27.0 MBytes   227 Mbits/sec
[  4]  21.00-22.00  sec  28.5 MBytes   240 Mbits/sec
[  4]  22.00-23.00  sec  27.5 MBytes   230 Mbits/sec
[  4]  23.00-24.00  sec  26.9 MBytes   226 Mbits/sec
[  4]  24.00-25.00  sec  26.8 MBytes   224 Mbits/sec
[  4]  25.00-26.00  sec  27.0 MBytes   227 Mbits/sec
[  4]  26.00-27.00  sec  27.1 MBytes   227 Mbits/sec
[  4]  27.00-28.00  sec  27.5 MBytes   231 Mbits/sec
[  4]  28.00-29.00  sec  27.1 MBytes   227 Mbits/sec
[  4]  29.00-30.00  sec  27.2 MBytes   228 Mbits/sec
[  4]  30.00-31.00  sec  26.6 MBytes   224 Mbits/sec
[  4]  31.00-32.00  sec  27.4 MBytes   230 Mbits/sec
[  4]  32.00-33.00  sec  27.6 MBytes   232 Mbits/sec
[  4]  33.00-34.00  sec  26.8 MBytes   224 Mbits/sec
[  4]  34.00-35.00  sec  28.4 MBytes   238 Mbits/sec
[  4]  35.00-36.00  sec  27.0 MBytes   227 Mbits/sec
[  4]  36.00-37.00  sec  26.6 MBytes   223 Mbits/sec
[  4]  37.00-38.00  sec  26.4 MBytes   221 Mbits/sec
[  4]  38.00-39.00  sec  26.1 MBytes   219 Mbits/sec
[  4]  39.00-40.00  sec  26.6 MBytes   224 Mbits/sec
[  4]  40.00-41.00  sec  28.0 MBytes   235 Mbits/sec
[  4]  41.00-42.00  sec  26.6 MBytes   224 Mbits/sec
[  4]  42.00-43.00  sec  27.0 MBytes   227 Mbits/sec
[  4]  43.00-44.00  sec  27.4 MBytes   229 Mbits/sec
[  4]  44.00-45.01  sec  27.2 MBytes   228 Mbits/sec
[  4]  45.01-46.00  sec  27.0 MBytes   227 Mbits/sec
[  4]  46.00-47.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  47.00-48.00  sec  26.6 MBytes   223 Mbits/sec
[  4]  48.00-49.01  sec  26.8 MBytes   223 Mbits/sec
[  4]  49.01-50.00  sec  26.2 MBytes   222 Mbits/sec
[  4]  50.00-51.00  sec  26.9 MBytes   226 Mbits/sec
[  4]  51.00-52.01  sec  27.2 MBytes   227 Mbits/sec
[  4]  52.01-53.00  sec  26.8 MBytes   226 Mbits/sec
[  4]  53.00-54.00  sec  27.4 MBytes   229 Mbits/sec
[  4]  54.00-55.00  sec  25.2 MBytes   212 Mbits/sec
[  4]  55.00-56.00  sec  26.8 MBytes   224 Mbits/sec
[  4]  56.00-57.00  sec  26.5 MBytes   222 Mbits/sec
[  4]  57.00-58.00  sec  26.1 MBytes   219 Mbits/sec
[  4]  58.00-59.00  sec  26.2 MBytes   221 Mbits/sec
[  4]  59.00-60.00  sec  27.1 MBytes   228 Mbits/sec
[  4]  60.00-61.00  sec  26.6 MBytes   223 Mbits/sec
[  4]  61.00-62.00  sec  26.4 MBytes   221 Mbits/sec
[  4]  62.00-63.00  sec  26.2 MBytes   221 Mbits/sec
[... 1720 lines cut ...]
[  4] 1783.00-1784.00 sec  28.2 MBytes   237 Mbits/sec
[  4] 1784.00-1785.00 sec  28.1 MBytes   236 Mbits/sec
[  4] 1785.00-1786.00 sec  27.9 MBytes   233 Mbits/sec
[  4] 1786.00-1787.00 sec  27.6 MBytes   232 Mbits/sec
[  4] 1787.00-1788.00 sec  28.1 MBytes   236 Mbits/sec
[  4] 1788.00-1789.00 sec  28.5 MBytes   239 Mbits/sec
[  4] 1789.00-1790.00 sec  28.2 MBytes   237 Mbits/sec
[  4] 1790.00-1791.00 sec  28.4 MBytes   238 Mbits/sec
[  4] 1791.00-1792.00 sec  28.5 MBytes   238 Mbits/sec
[  4] 1792.00-1793.00 sec  28.0 MBytes   236 Mbits/sec
[  4] 1793.00-1794.00 sec  28.2 MBytes   236 Mbits/sec
[  4] 1794.00-1795.01 sec  27.6 MBytes   231 Mbits/sec
[  4] 1795.01-1796.00 sec  28.4 MBytes   238 Mbits/sec
[  4] 1796.00-1797.00 sec  28.1 MBytes   236 Mbits/sec
[  4] 1797.00-1798.00 sec  28.5 MBytes   239 Mbits/sec
[  4] 1798.00-1799.00 sec  27.9 MBytes   235 Mbits/sec
[  4] 1799.00-1800.00 sec  28.4 MBytes   238 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1800.00 sec  47.7 GBytes   228 Mbits/sec                  sender
[  4]   0.00-1800.00 sec  47.7 GBytes   228 Mbits/sec                  receiver

iperf Done.

C:\WINDOWS\system32>

CapitaineBidouille, 2022-07-30:
It crashed at the beginning of the test :(
C:\WINDOWS\system32>iperf3 -c 192.168.3.1 -t 18000
Connecting to host 192.168.3.1, port 5201
[  4] local 192.168.3.176 port 8284 connected to 192.168.3.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.01   sec  1.25 MBytes  10.3 Mbits/sec
[  4]   1.01-2.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   2.00-3.01   sec  0.00 Bytes  0.00 bits/sec
[  4]   3.01-4.01   sec  0.00 Bytes  0.00 bits/sec
[  4]   4.01-5.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   5.00-6.01   sec  0.00 Bytes  0.00 bits/sec
[  4]   6.01-7.00   sec  0.00 Bytes  0.00 bits/sec
[  4]   7.00-8.01   sec  0.00 Bytes  0.00 bits/sec
[  4]   8.01-9.01   sec  0.00 Bytes  0.00 bits/sec
[  4]   9.01-10.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  10.01-11.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  11.01-12.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  12.01-13.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  13.01-14.00  sec  0.00 Bytes  0.00 bits/sec
[  4]  14.00-15.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  15.01-16.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  16.01-17.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  17.01-18.01  sec  0.00 Bytes  0.00 bits/sec
[  4]  18.01-19.01  sec  0.00 Bytes  0.00 bits/sec
iperf3: error - unable to write to stream socket: Connection reset by peer

CapitaineBidouille, 2022-07-30:
I guess I will obtain the best stability and speed using a real Wifi router plugged in Ethernet to the PI...

CapitaineBidouille, 2022-07-30:
I am going to try another setup : 
USB3 Port1 : Powered Hub : 2x 5TB hard drives
USB3 Port2 : Powered Hub : 1x 5TB hard drives + USB Ethernet + USB Ethernet (the one I own that is throttled by the CPU of the PI at 600 mbps, but I do not care since the TP Link will not be able to give a Wifi network faster than 300 Mbps)
USB2 Port1 : Nothing
USB2 Port2 : Nothing

Second USB ethernet will be plugged to a mini TP Link Router which will take it's power directly from the power strip, Model : TL-WR802N which is marketed to give me a 300 MBPS wifi network. I will configure that router in 5GHz mode to get the most of it. 

- 2.4 GHz Wifi Network hosted by the PI. Just made an iperf, it's great at making an 2.4 GHz AP. It's strangely much more stable than using the AWUS to do the same task even if the range is probably much lower, my PC at 1 meter get a really nice bandwidth from it that seems to never vary. It's full 54 Mbps : 
```
C:\WINDOWS\system32>iperf3 -c 192.168.6.1 -t 900
Connecting to host 192.168.6.1, port 5201
[  4] local 192.168.6.176 port 15984 connected to 192.168.6.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  6.50 MBytes  54.4 Mbits/sec
[  4]   1.00-2.01   sec  6.38 MBytes  53.3 Mbits/sec
[  4]   2.01-3.00   sec  6.50 MBytes  54.7 Mbits/sec
[  4]   3.00-4.00   sec  6.50 MBytes  54.5 Mbits/sec
[  4]   4.00-5.00   sec  6.38 MBytes  53.7 Mbits/sec
[  4]   5.00-6.01   sec  6.50 MBytes  54.0 Mbits/sec
[  4]   6.01-7.00   sec  6.25 MBytes  52.8 Mbits/sec
[  4]   7.00-8.01   sec  6.62 MBytes  55.4 Mbits/sec
[  4]   8.01-9.00   sec  5.25 MBytes  44.3 Mbits/sec
[  4]   9.00-10.00  sec  6.38 MBytes  53.3 Mbits/sec
[  4]  10.00-11.01  sec  6.50 MBytes  54.2 Mbits/sec
[  4]  11.01-12.01  sec  6.38 MBytes  53.3 Mbits/sec
[  4]  12.01-13.00  sec  6.50 MBytes  55.2 Mbits/sec
[  4]  13.00-14.01  sec  6.12 MBytes  51.1 Mbits/sec
[  4]  14.01-15.01  sec  6.25 MBytes  52.4 Mbits/sec
[  4]  15.01-16.00  sec  6.38 MBytes  53.7 Mbits/sec
[  4]  16.00-17.01  sec  6.50 MBytes  53.9 Mbits/sec
[  4]  17.01-18.00  sec  4.50 MBytes  38.1 Mbits/sec
[  4]  18.00-19.01  sec  6.75 MBytes  56.6 Mbits/sec
[  4]  19.01-20.01  sec  6.38 MBytes  53.3 Mbits/sec
[  4]  20.01-21.00  sec  6.25 MBytes  52.8 Mbits/sec
[  4]  21.00-22.00  sec  6.38 MBytes  53.4 Mbits/sec
[  4]  22.00-23.01  sec  6.38 MBytes  53.3 Mbits/sec
[  4]  23.01-24.00  sec  6.38 MBytes  53.7 Mbits/sec
[  4]  24.00-25.00  sec  4.88 MBytes  41.0 Mbits/sec
[  4]  25.00-26.01  sec  6.50 MBytes  54.2 Mbits/sec
[  4]  26.01-27.00  sec  6.50 MBytes  54.6 Mbits/sec
[  4]  27.00-28.01  sec  6.38 MBytes  53.5 Mbits/sec
[  4]  28.01-29.01  sec  5.25 MBytes  43.8 Mbits/sec
[  4]  29.01-30.00  sec  5.88 MBytes  49.7 Mbits/sec
[  4]  30.00-31.01  sec  6.38 MBytes  53.3 Mbits/sec
[  4]  31.01-32.00  sec  6.50 MBytes  54.7 Mbits/sec
[  4]  32.00-33.00  sec  6.50 MBytes  54.6 Mbits/sec
[  4]  33.00-34.00  sec  6.38 MBytes  53.4 Mbits/sec
[  4]  34.00-35.01  sec  6.62 MBytes  54.9 Mbits/sec
[  4]  35.01-36.01  sec  4.75 MBytes  40.0 Mbits/sec
[  4]  36.01-37.01  sec  6.38 MBytes  53.6 Mbits/sec
[  4]  37.01-38.00  sec  6.38 MBytes  53.8 Mbits/sec
[  4]  38.00-39.01  sec  6.62 MBytes  55.3 Mbits/sec
[  4]  39.01-40.00  sec  6.38 MBytes  53.6 Mbits/sec
[  4]  40.00-41.00  sec  6.38 MBytes  53.5 Mbits/sec
[  4]  41.00-42.00  sec  6.50 MBytes  54.7 Mbits/sec
[  4]  42.00-43.00  sec  5.25 MBytes  44.0 Mbits/sec
[  4]  43.00-44.01  sec  6.50 MBytes  54.1 Mbits/sec
[  4]  44.01-45.00  sec  6.38 MBytes  53.9 Mbits/sec
[  4]  45.00-46.01  sec  6.25 MBytes  52.1 Mbits/sec
[  4]  46.01-47.01  sec  5.62 MBytes  47.3 Mbits/sec
[  4]  47.01-48.00  sec  6.00 MBytes  50.6 Mbits/sec
[  4]  48.00-49.01  sec  6.62 MBytes  54.9 Mbits/sec
[  4]  49.01-50.01  sec  6.38 MBytes  53.6 Mbits/sec
[  4]  50.01-51.01  sec  6.50 MBytes  54.7 Mbits/sec
[  4]  51.01-52.00  sec  6.25 MBytes  52.7 Mbits/sec
[  4]  52.00-53.00  sec  6.50 MBytes  54.5 Mbits/sec
[  4]  53.00-54.00  sec  6.25 MBytes  52.4 Mbits/sec
[  4]  54.00-55.00  sec  6.25 MBytes  52.6 Mbits/sec
[  4]  55.00-56.01  sec  6.62 MBytes  54.9 Mbits/sec
[... 828 lines cut ...]
[  4] 884.00-885.01 sec  6.50 MBytes  54.0 Mbits/sec
[  4] 885.01-886.01 sec  6.38 MBytes  53.7 Mbits/sec
[  4] 886.01-887.00 sec  6.25 MBytes  52.7 Mbits/sec
[  4] 887.00-888.00 sec  6.50 MBytes  54.6 Mbits/sec
[  4] 888.00-889.00 sec  6.38 MBytes  53.6 Mbits/sec
[  4] 889.00-890.00 sec  6.50 MBytes  54.4 Mbits/sec
[  4] 890.00-891.01 sec  6.38 MBytes  53.0 Mbits/sec
[  4] 891.01-892.00 sec  6.50 MBytes  54.9 Mbits/sec
[  4] 892.00-893.01 sec  6.50 MBytes  54.2 Mbits/sec
[  4] 893.01-894.01 sec  6.38 MBytes  53.2 Mbits/sec
[  4] 894.01-895.01 sec  6.50 MBytes  54.8 Mbits/sec
[  4] 895.01-896.00 sec  6.25 MBytes  53.0 Mbits/sec
[  4] 896.00-897.00 sec  6.50 MBytes  54.3 Mbits/sec
[  4] 897.00-898.00 sec  6.38 MBytes  53.5 Mbits/sec
[  4] 898.00-899.00 sec  6.62 MBytes  55.6 Mbits/sec
[  4] 899.00-900.00 sec  6.38 MBytes  53.5 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-900.00 sec  5.59 GBytes  53.4 Mbits/sec                  sender
[  4]   0.00-900.00 sec  5.59 GBytes  53.4 Mbits/sec                  receiver

iperf Done.

C:\WINDOWS\system32>
```

CapitaineBidouille, 2022-07-30:
I am getting extreme stability with it just like I remembered when I used it in the past.
Bandwidth is disappointing though, since the ethernet port of that router can't go faster than 100 MBPS.
It's also not a 5GHz router, only 2.4. I ordered this one : https://www.amazon.fr/gp/product/B01MY5JIJ0/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 that should arrive next Wednesday. I guess it will be far better.

```Connecting to host 192.168.7.1, port 5201
[  4] local 192.168.8.100 port 1036 connected to 192.168.7.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  9.00 MBytes  75.5 Mbits/sec
[  4]   1.00-2.01   sec  11.1 MBytes  92.8 Mbits/sec
[  4]   2.01-3.00   sec  11.2 MBytes  94.8 Mbits/sec
[  4]   3.00-4.01   sec  11.4 MBytes  94.9 Mbits/sec
[  4]   4.01-5.00   sec  11.2 MBytes  94.8 Mbits/sec
[  4]   5.00-6.00   sec  11.2 MBytes  94.5 Mbits/sec
[  4]   6.00-7.00   sec  11.4 MBytes  95.3 Mbits/sec
[  4]   7.00-8.01   sec  11.4 MBytes  94.8 Mbits/sec
[  4]   8.01-9.00   sec  11.1 MBytes  94.0 Mbits/sec
[  4]   9.00-10.00  sec  11.4 MBytes  95.2 Mbits/sec
[  4]  10.00-11.01  sec  11.4 MBytes  94.9 Mbits/sec
[  4]  11.01-12.00  sec  11.1 MBytes  94.1 Mbits/sec
[  4]  12.00-13.01  sec  11.4 MBytes  94.9 Mbits/sec
[  4]  13.01-14.00  sec  11.4 MBytes  95.6 Mbits/sec
[  4]  14.00-15.01  sec  11.4 MBytes  94.8 Mbits/sec
[  4]  15.01-16.00  sec  11.1 MBytes  94.4 Mbits/sec
[  4]  16.00-17.00  sec  11.4 MBytes  95.4 Mbits/sec
[  4]  17.00-18.00  sec  11.1 MBytes  93.2 Mbits/sec
[  4]  18.00-19.00  sec  11.2 MBytes  94.2 Mbits/sec
[  4]  19.00-20.00  sec  11.4 MBytes  95.5 Mbits/sec
[  4]  20.00-21.01  sec  11.4 MBytes  95.2 Mbits/sec
[  4]  21.01-22.00  sec  11.2 MBytes  94.8 Mbits/sec
[  4]  22.00-23.00  sec  11.2 MBytes  94.2 Mbits/sec
[  4]  23.00-24.00  sec  11.4 MBytes  95.6 Mbits/sec
[  4]  24.00-25.01  sec  11.2 MBytes  94.0 Mbits/sec
[  4]  25.01-26.00  sec  11.2 MBytes  94.8 Mbits/sec
[  4]  26.00-27.00  sec  11.2 MBytes  94.4 Mbits/sec
[  4]  27.00-28.01  sec  11.4 MBytes  94.7 Mbits/sec
[  4]  28.01-29.00  sec  11.2 MBytes  94.9 Mbits/sec
[  4]  29.00-30.00  sec  11.4 MBytes  95.4 Mbits/sec
[  4]  30.00-31.00  sec  11.0 MBytes  92.1 Mbits/sec
[  4]  31.00-32.00  sec  11.4 MBytes  95.5 Mbits/sec
[  4]  32.00-33.01  sec  11.2 MBytes  94.1 Mbits/sec
[  4]  33.01-34.00  sec  11.2 MBytes  94.9 Mbits/sec
[  4]  34.00-35.01  sec  11.5 MBytes  95.6 Mbits/sec
[  4]  35.01-36.00  sec  11.1 MBytes  93.9 Mbits/sec
[  4]  36.00-37.01  sec  11.5 MBytes  95.8 Mbits/sec
[  4]  37.01-38.00  sec  11.1 MBytes  94.1 Mbits/sec
[  4]  38.00-39.00  sec  11.4 MBytes  95.4 Mbits/sec
[  4]  39.00-40.01  sec  11.4 MBytes  94.9 Mbits/sec
[  4]  40.01-41.00  sec  11.2 MBytes  94.9 Mbits/sec
[  4]  41.00-42.01  sec  11.2 MBytes  94.0 Mbits/sec
[  4]  42.01-43.00  sec  11.2 MBytes  94.6 Mbits/sec
[  4]  43.00-44.00  sec  11.4 MBytes  95.4 Mbits/sec
[  4]  44.00-45.00  sec  11.2 MBytes  94.5 Mbits/sec
[  4]  45.00-46.00  sec  7.25 MBytes  60.8 Mbits/sec
[  4]  46.00-47.01  sec  11.2 MBytes  94.0 Mbits/sec
[  4]  47.01-48.00  sec  11.2 MBytes  94.8 Mbits/sec
[  4]  48.00-49.00  sec  11.4 MBytes  95.3 Mbits/sec
[  4]  49.00-50.01  sec  11.2 MBytes  94.1 Mbits/sec
[  4]  50.01-51.01  sec  11.4 MBytes  95.1 Mbits/sec
[  4]  51.01-52.01  sec  11.2 MBytes  94.7 Mbits/sec
[  4]  52.01-53.00  sec  11.2 MBytes  94.8 Mbits/sec
[  4]  53.00-54.00  sec  11.4 MBytes  95.1 Mbits/sec
[  4]  54.00-55.00  sec  11.2 MBytes  94.6 Mbits/sec
[  4]  55.00-56.00  sec  11.1 MBytes  93.1 Mbits/sec
[  4]  56.00-57.00  sec  11.4 MBytes  95.5 Mbits/sec
[  4]  57.00-58.01  sec  11.1 MBytes  92.9 Mbits/sec
[  4]  58.01-59.01  sec  11.2 MBytes  94.1 Mbits/sec
[  4]  59.01-60.00  sec  11.2 MBytes  95.3 Mbits/sec
[  4]  60.00-61.00  sec  11.2 MBytes  94.0 Mbits/sec
[  4]  61.00-62.00  sec  11.4 MBytes  95.6 Mbits/sec
[  4]  62.00-63.00  sec  11.2 MBytes  94.6 Mbits/sec
[... 1121 lines cut ...]
[  4] 1184.00-1185.00 sec  11.4 MBytes  95.7 Mbits/sec
[  4] 1185.00-1186.01 sec  11.4 MBytes  94.8 Mbits/sec
[  4] 1186.01-1187.01 sec  11.2 MBytes  94.4 Mbits/sec
[  4] 1187.01-1188.00 sec  11.2 MBytes  94.7 Mbits/sec
[  4] 1188.00-1189.00 sec  11.2 MBytes  94.4 Mbits/sec
[  4] 1189.00-1190.00 sec  11.2 MBytes  94.5 Mbits/sec
[  4] 1190.00-1191.01 sec  11.4 MBytes  94.6 Mbits/sec
[  4] 1191.01-1192.00 sec  11.2 MBytes  95.1 Mbits/sec
[  4] 1192.00-1193.01 sec  11.4 MBytes  94.8 Mbits/sec
[  4] 1193.01-1194.00 sec  11.1 MBytes  94.1 Mbits/sec
[  4] 1194.00-1195.00 sec  11.2 MBytes  94.3 Mbits/sec
[  4] 1195.00-1196.01 sec  11.4 MBytes  94.9 Mbits/sec
[  4] 1196.01-1197.00 sec  11.2 MBytes  94.7 Mbits/sec
[  4] 1197.00-1198.01 sec  11.4 MBytes  94.9 Mbits/sec
[  4] 1198.01-1199.00 sec  11.2 MBytes  94.9 Mbits/sec
[  4] 1199.00-1200.01 sec  11.4 MBytes  95.0 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1200.01 sec  13.0 GBytes  93.2 Mbits/sec                  sender
[  4]   0.00-1200.01 sec  13.0 GBytes  93.2 Mbits/sec                  receiver

iperf Done.

C:\WINDOWS\system32>
```

CapitaineBidouille, 2022-07-30:
I ordered that router, that will have the responsibility to share the VPN connection of the PI in Wifi6 : https://www.amazon.fr/gp/product/B08F74YX99/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
I should reach the gigabit with this one and UE300 TPLINK USB adapter.
The money spend on it worth the time I am going to save in order to make work a complex setup that even at it's best, will not be able to deliver more than 400 Mbit/s.

morrownr, 2022-07-30:
Well, the adventures of the RasPi.

There is a reason I own a usb inline power meter. I learned that trying to connect a lot of things into the USB subsystem of a RasPi can be problematic. I decided to try to duplicate, as best I can, your setup this morning. It took me about 10 tries to get things working together and I have done this a lot and I have a good idea of what the best hardware is. RasPis are cool and we can do a lot of with them but it takes some experience to get a stable system. If you are interested, as I have time, I will post the RasPi setup and results.

My main concern with your setup was the 2 powered hubs and the 3 HDDs. While they may seem to be the stable part of your setup, it is likely that they had a hand in causing the problem that showed up in wifi.

Regards

morrownr, 2022-07-31:
@CapitaineBidouille 

Here is my RasPi4B in a stable access point configuration:

```
$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=mt76x2u, 5000M
    |__ Port 2: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=Vendor Specific Class, Driver=mt76x0u, 480M
        |__ Port 4: Dev 4, If 2, Class=Video, Driver=uvcvideo, 480M
        |__ Port 4: Dev 4, If 0, Class=Audio, Driver=snd-usb-audio, 480M
        |__ Port 4: Dev 4, If 3, Class=Video, Driver=uvcvideo, 480M
        |__ Port 4: Dev 4, If 1, Class=Audio, Driver=snd-usb-audio, 480M
```

Note that the ALFA ACM is on port 1 (USB3) and the ALFA ACHM is on port 3 (USB2). Why does this matter? I have long suspected that the USB ports that are near level with the board suffer from interference more than the ports away from the board. USB WiFi adapters may not be able to handle what is happening as well as some other devices. This is also true of other USB WiFi adapters. When last I brought a new version of the rtl88x2bu driver online, the team and I could not find a stable solution for 5 GHz AP mode when USB3 was being used. 

https://github.com/morrownr/88x2bu-20210702

From the FAQ for the above driver:

```
Question: Are there any known problems with AP mode?

Answer: Overall this driver does a good job with AP mode. During testing and work prior to making this driver available, the team working on this driver noticed some problems in AP mode if used with a Raspberry Pi 4B. We were unable to discover or fix the exact cause of the problem but the workaround is to keep the driver in USB2 mode. This workaround only applies to AP mode with Raspberry Pi 4B. No problems were noted with systems that use x86 or amd64 processors.
```

I have seen simiar problems with other Realtek based adapters. In fact, the Mediatek based adapters seem to handle it better than the Realtek based adapters.

The stable configuration has something plugged into each of the 4 ports. There are no powered hubs attached. There is a webcam to stream video on my local network, a USB3 nano flash drive, and the 2 ALFA adapters. I run headless. This setup stays below the 1200 mA limit of the RasPi4 USB subsystem and there is no possibility of backfeeding due to no use of a powered hub.

Switching to a different subject: As I noted in the previous message, I decided to try to duplicate your setup as best I can. I plugged in the following:

USB3 powered hub with 4 ports
USB3 to sata SDD (500 gb)
ALFA ACM
ALFA ACHM
2 x USB flash drives
USB2 webcam

In this configuration I was easily able to duplicate the drop outs that your saw with the ALFA ACM. This problem does not show with the RasPi3B and + as they don't have USB3 capability. If you want to try to get a stable access point setup while using the powered hubs and HDDs, you might want to just use the ALFA ACHM (it can get you 200 Mb/s in 5 GHz) and sit the ACM aside for another project.

My 2 cents. Your mileage may vary as you have already seen.

Nick

CapitaineBidouille, 2022-07-31:
Il already tried to make an access point with the ACHM connected directly to an usb2 port. It didn’t work well, I was experiencing the same instabilities. It’s not working well.
Tomorrow I am going to receive an asus wifi 6 router. I will connect it’s Wan port to the Pi, I then should be able to share one full gigabit over wifi. I guess I will not even have to connect my PC on Ethernet anymore.

morrownr, 2022-08-01:
> It’s not working well.

I understand. You may not have success building a wifi router with a RasPi4B if you have a powered USB hub in the mix. I've been using Pi's for a few years and if there is something that I can pass on to your that could help, it is that before spending any money on something to go with the Pi, do a LOT of research. The Pi's were designed to be low cost educational tools, not fire breathing servers. My experience is that the USB subsystem is a weakness in the design. Another serious source of problems is having good quality power supplies.

Here is the iperf3 output from the stable setup I posted earlier. This is using the ALFA ACM as the AP and while maximum throughput could be around 400 Mb/s, it is not for a few reasons in this test. The main reason is that the channel (36) that I am using for this test is very heavily congested and since this is sharing technology. the rest of the bandwidth is being mostly eaten up by others on the same channel. There is also an issue that there are 2 walls and some distance from my RasPi to the desktop system I am using. The system is using WPA3. It works well with the adapters that use in-kernel drivers. Note that there are no drop offs and no retrys. Very stable.

```
$ iperf3 -c 192.168.1.197 -t 300
Connecting to host 192.168.1.197, port 5201
[  5] local 192.168.1.235 port 37652 connected to 192.168.1.197 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  19.1 MBytes   160 Mbits/sec    0    568 KBytes       
[  5]   1.00-2.00   sec  16.6 MBytes   139 Mbits/sec    0    631 KBytes       
[  5]   2.00-3.00   sec  15.0 MBytes   126 Mbits/sec    0    676 KBytes       
[  5]   3.00-4.00   sec  16.2 MBytes   136 Mbits/sec    0    725 KBytes       
[  5]   4.00-5.00   sec  16.2 MBytes   136 Mbits/sec    0    762 KBytes       
[  5]   5.00-6.00   sec  17.5 MBytes   147 Mbits/sec    0    762 KBytes       
[  5]   6.00-7.00   sec  17.5 MBytes   147 Mbits/sec    0    762 KBytes       
[  5]   7.00-8.00   sec  15.0 MBytes   126 Mbits/sec    0    762 KBytes       
[  5]   8.00-9.00   sec  16.2 MBytes   136 Mbits/sec    0    837 KBytes       
[  5]   9.00-10.00  sec  17.5 MBytes   147 Mbits/sec    0    837 KBytes       
[  5]  10.00-11.00  sec  15.0 MBytes   126 Mbits/sec    0    887 KBytes       
[  5]  11.00-12.00  sec  16.2 MBytes   136 Mbits/sec    0    887 KBytes       
[  5]  12.00-13.00  sec  16.2 MBytes   136 Mbits/sec    0   1.19 MBytes       
[  5]  13.00-14.00  sec  16.2 MBytes   136 Mbits/sec    0   1.19 MBytes       
[  5]  14.00-15.00  sec  16.2 MBytes   136 Mbits/sec    0   1.19 MBytes       
[  5]  15.00-16.00  sec  16.2 MBytes   136 Mbits/sec    0   1.19 MBytes       
[  5]  16.00-17.00  sec  16.2 MBytes   136 Mbits/sec    0   1.19 MBytes       
[  5]  17.00-18.00  sec  16.2 MBytes   136 Mbits/sec    0   1.19 MBytes       
[  5]  18.00-19.00  sec  13.8 MBytes   115 Mbits/sec    0   1.19 MBytes       
[  5]  19.00-20.00  sec  16.2 MBytes   136 Mbits/sec    0   1.19 MBytes       
[  5]  20.00-21.00  sec  18.8 MBytes   157 Mbits/sec    0   1.19 MBytes       
[  5]  21.00-22.00  sec  17.5 MBytes   147 Mbits/sec    0   1.78 MBytes       
[  5]  22.00-23.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  23.00-24.00  sec  17.5 MBytes   147 Mbits/sec    0   1.78 MBytes       
[  5]  24.00-25.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  25.00-26.00  sec  18.8 MBytes   157 Mbits/sec    0   1.78 MBytes       
[  5]  26.00-27.00  sec  17.5 MBytes   147 Mbits/sec    0   1.78 MBytes       
[  5]  27.00-28.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  28.00-29.00  sec  15.0 MBytes   126 Mbits/sec    0   1.78 MBytes       
[  5]  29.00-30.00  sec  15.0 MBytes   126 Mbits/sec    0   1.78 MBytes       
[  5]  30.00-31.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  31.00-32.00  sec  17.5 MBytes   147 Mbits/sec    0   1.78 MBytes       
[  5]  32.00-33.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  33.00-34.00  sec  18.8 MBytes   157 Mbits/sec    0   1.78 MBytes       
[  5]  34.00-35.00  sec  17.5 MBytes   147 Mbits/sec    0   1.78 MBytes       
[  5]  35.00-36.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  36.00-37.00  sec  15.0 MBytes   126 Mbits/sec    0   1.78 MBytes       
[  5]  37.00-38.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  38.00-39.00  sec  13.8 MBytes   115 Mbits/sec    0   1.78 MBytes       
[  5]  39.00-40.00  sec  15.0 MBytes   126 Mbits/sec    0   1.78 MBytes       
[  5]  40.00-41.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  41.00-42.00  sec  15.0 MBytes   126 Mbits/sec    0   1.78 MBytes       
[  5]  42.00-43.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  43.00-44.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  44.00-45.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  45.00-46.00  sec  17.5 MBytes   147 Mbits/sec    0   1.78 MBytes       
[  5]  46.00-47.00  sec  17.5 MBytes   147 Mbits/sec    0   1.78 MBytes       
[  5]  47.00-48.00  sec  16.2 MBytes   136 Mbits/sec    0   1.78 MBytes       
[  5]  48.00-49.00  sec  15.0 MBytes   126 Mbits/sec    0   1.78 MBytes       
[  5]  49.00-50.00  sec  15.0 MBytes   126 Mbits/sec    0   1.78 MBytes       
[  5]  50.00-51.00  sec  17.5 MBytes   147 Mbits/sec    0   1.78 MBytes       
[... cut ...]
[  5] 297.00-298.00 sec  18.8 MBytes   157 Mbits/sec    0   2.65 MBytes       
[  5] 298.00-299.00 sec  21.2 MBytes   178 Mbits/sec    0   2.65 MBytes       
[  5] 299.00-300.00 sec  20.0 MBytes   168 Mbits/sec    0   2.65 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-300.00 sec  4.89 GBytes   140 Mbits/sec    0             sender
[  5]   0.00-300.02 sec  4.88 GBytes   140 Mbits/sec                  receiver

iperf Done.
```

CapitaineBidouille, 2022-08-01:
Some news from the Wifi6 network project, I just received the ASUS router.
It's WAN port is connected to the U300 Tplink USB adapter.
https://www.speedtest.net/result/13482953872 This is the speed I get when I connect to it using Wifi6. It's pretty crazy speed for a Wifi network.

CapitaineBidouille, 2022-08-01:
Some more benchs using the ASUS RT-AX55 Wifi6 router. It's wan port is connect to the PI through the TP Link USB adapter : 
Ookla speedtest : https://www.speedtest.net/result/13483143529

5 GHz : 
Connecting to host 192.168.3.1, port 5201
[  4] local 192.168.2.29 port 28282 connected to 192.168.3.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  51.1 MBytes   428 Mbits/sec
[  4]   1.00-2.00   sec  48.6 MBytes   407 Mbits/sec
[  4]   2.00-3.00   sec  49.4 MBytes   415 Mbits/sec
[  4]   3.00-4.00   sec  38.0 MBytes   319 Mbits/sec
[  4]   4.00-5.00   sec  43.8 MBytes   367 Mbits/sec
[  4]   5.00-6.00   sec  50.0 MBytes   420 Mbits/sec
[  4]   6.00-7.00   sec  41.2 MBytes   346 Mbits/sec
[  4]   7.00-8.00   sec  48.2 MBytes   405 Mbits/sec
[  4]   8.00-9.00   sec  51.5 MBytes   432 Mbits/sec
[  4]   9.00-10.00  sec  50.8 MBytes   427 Mbits/sec
[  4]  10.00-11.00  sec  51.4 MBytes   430 Mbits/sec
[  4]  11.00-12.00  sec  50.9 MBytes   427 Mbits/sec
[  4]  12.00-13.00  sec  50.8 MBytes   425 Mbits/sec
[  4]  13.00-14.00  sec  50.2 MBytes   422 Mbits/sec
[  4]  14.00-15.00  sec  48.9 MBytes   410 Mbits/sec
[  4]  15.00-16.00  sec  49.9 MBytes   418 Mbits/sec
[  4]  16.00-17.00  sec  50.2 MBytes   421 Mbits/sec
[  4]  17.00-18.00  sec  50.4 MBytes   423 Mbits/sec
[  4]  18.00-19.00  sec  49.9 MBytes   419 Mbits/sec
[  4]  19.00-20.00  sec  50.8 MBytes   425 Mbits/sec
[  4]  20.00-21.00  sec  49.5 MBytes   416 Mbits/sec
[  4]  21.00-22.00  sec  51.5 MBytes   432 Mbits/sec
[  4]  22.00-23.00  sec  49.9 MBytes   419 Mbits/sec
[  4]  23.00-24.00  sec  48.9 MBytes   410 Mbits/sec
[  4]  24.00-25.00  sec  49.9 MBytes   418 Mbits/sec
[  4]  25.00-26.00  sec  50.1 MBytes   421 Mbits/sec
[  4]  26.00-27.00  sec  50.4 MBytes   423 Mbits/sec
[  4]  27.00-28.00  sec  50.2 MBytes   421 Mbits/sec
[  4]  28.00-29.00  sec  50.5 MBytes   423 Mbits/sec
[  4]  29.00-30.00  sec  49.9 MBytes   418 Mbits/sec
[  4]  30.00-31.00  sec  50.2 MBytes   422 Mbits/sec
[  4]  31.00-32.00  sec  50.1 MBytes   420 Mbits/sec
[  4]  32.00-33.00  sec  51.6 MBytes   433 Mbits/sec
[  4]  33.00-34.00  sec  51.4 MBytes   430 Mbits/sec
[  4]  34.00-35.00  sec  49.5 MBytes   416 Mbits/sec
[  4]  35.00-36.00  sec  50.4 MBytes   422 Mbits/sec
[  4]  36.00-37.00  sec  49.8 MBytes   418 Mbits/sec
[  4]  37.00-38.00  sec  49.8 MBytes   417 Mbits/sec
[  4]  38.00-39.00  sec  50.6 MBytes   424 Mbits/sec
[  4]  39.00-40.00  sec  50.2 MBytes   422 Mbits/sec
[  4]  40.00-41.00  sec  50.4 MBytes   421 Mbits/sec
[  4]  41.00-42.00  sec  50.8 MBytes   427 Mbits/sec
[  4]  42.00-43.00  sec  50.6 MBytes   424 Mbits/sec
[  4]  43.00-44.00  sec  50.0 MBytes   419 Mbits/sec
[  4]  44.00-45.00  sec  50.1 MBytes   420 Mbits/sec
[  4]  45.00-46.00  sec  49.9 MBytes   419 Mbits/sec
[  4]  46.00-47.00  sec  50.6 MBytes   424 Mbits/sec
[  4]  47.00-48.00  sec  49.9 MBytes   419 Mbits/sec
[  4]  48.00-49.00  sec  49.8 MBytes   418 Mbits/sec
[  4]  49.00-50.00  sec  49.9 MBytes   419 Mbits/sec
[  4]  50.00-51.00  sec  50.2 MBytes   421 Mbits/sec
[  4]  51.00-52.00  sec  49.1 MBytes   412 Mbits/sec
[  4]  52.00-53.00  sec  50.5 MBytes   424 Mbits/sec
[  4]  53.00-54.00  sec  50.0 MBytes   419 Mbits/sec
[  4]  54.00-55.00  sec  49.2 MBytes   414 Mbits/sec
[  4]  55.00-56.00  sec  48.1 MBytes   403 Mbits/sec
[  4]  56.00-57.00  sec  49.6 MBytes   416 Mbits/sec
[  4]  57.00-58.00  sec  48.8 MBytes   410 Mbits/sec
[  4]  58.00-59.00  sec  50.2 MBytes   421 Mbits/sec
[  4]  59.00-60.00  sec  50.5 MBytes   423 Mbits/sec
[  4]  60.00-61.00  sec  49.8 MBytes   418 Mbits/sec
[  4]  61.00-62.00  sec  50.2 MBytes   421 Mbits/sec
[  4]  62.00-63.00  sec  50.5 MBytes   422 Mbits/sec
[... 538 lines cut ...]
[  4] 289.01-290.00 sec  17.9 MBytes   151 Mbits/sec
[  4] 290.00-291.00 sec  18.6 MBytes   156 Mbits/sec
[  4] 291.00-292.00 sec  18.2 MBytes   153 Mbits/sec
[  4] 292.00-293.00 sec  16.2 MBytes   136 Mbits/sec
[  4] 293.00-294.00 sec  18.8 MBytes   157 Mbits/sec
[  4] 294.00-295.01 sec  19.5 MBytes   163 Mbits/sec
[  4] 295.01-296.01 sec  19.0 MBytes   160 Mbits/sec
[  4] 296.01-297.00 sec  18.9 MBytes   158 Mbits/sec
[  4] 297.00-298.01 sec  19.1 MBytes   160 Mbits/sec
[  4] 298.01-299.00 sec  17.4 MBytes   147 Mbits/sec
[  4] 299.00-300.00 sec  18.2 MBytes   152 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-300.00 sec  5.38 GBytes   154 Mbits/sec                  sender
[  4]   0.00-300.00 sec  5.38 GBytes   154 Mbits/sec                  receiver

iperf Done.

C:\WINDOWS\system32>

The most elegant way to get a killer network was to not give this responsibility to the PI.
The PI is really good at Ethernet routing though, there is a very low overhead even on gigabit.
It's also very good as serving Plex and content that is located on external hard drives. It's stable for days without any issue.
Making a Wifi network can be great with the PI as long as it's a 2.4GHz for legacy devices or if the PI is dedicated for that using OpenWRT I guess.
Right now I decided to not connect anymore with an Ethernet cable. It's not necessary anymore. The Wifi6 bandwidth seems enough.

CapitaineBidouille, 2022-08-01:
When the PC is connected via Ethernet to the Asus Router.

```
Connecting to host 192.168.3.1, port 5201
[  4] local 192.168.2.52 port 28064 connected to 192.168.3.1 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   112 MBytes   934 Mbits/sec
[  4]   1.00-2.00   sec   112 MBytes   939 Mbits/sec
[  4]   2.00-3.00   sec   111 MBytes   928 Mbits/sec
[  4]   3.00-4.00   sec   112 MBytes   939 Mbits/sec
[  4]   4.00-5.00   sec   112 MBytes   938 Mbits/sec
[  4]   5.00-6.00   sec   112 MBytes   938 Mbits/sec
[  4]   6.00-7.00   sec   112 MBytes   936 Mbits/sec
[  4]   7.00-8.00   sec   112 MBytes   936 Mbits/sec
[  4]   8.00-9.00   sec   112 MBytes   938 Mbits/sec
[  4]   9.00-10.00  sec   112 MBytes   937 Mbits/sec
[  4]  10.00-11.00  sec   111 MBytes   934 Mbits/sec
[  4]  11.00-12.00  sec   112 MBytes   938 Mbits/sec
[  4]  12.00-13.00  sec   110 MBytes   920 Mbits/sec
[  4]  13.00-14.00  sec   112 MBytes   936 Mbits/sec
[  4]  14.00-15.00  sec   112 MBytes   938 Mbits/sec
[  4]  15.00-16.00  sec   112 MBytes   938 Mbits/sec
[  4]  16.00-17.00  sec   112 MBytes   941 Mbits/sec
[  4]  17.00-18.00  sec   112 MBytes   939 Mbits/sec
[  4]  18.00-19.00  sec   112 MBytes   939 Mbits/sec
[  4]  19.00-20.00  sec   112 MBytes   939 Mbits/sec
[  4]  20.00-21.00  sec   112 MBytes   936 Mbits/sec
[  4]  21.00-22.00  sec   112 MBytes   937 Mbits/sec
[  4]  22.00-23.00  sec   111 MBytes   935 Mbits/sec
[  4]  23.00-24.00  sec   111 MBytes   935 Mbits/sec
[  4]  24.00-25.00  sec   112 MBytes   939 Mbits/sec
[  4]  25.00-26.00  sec   112 MBytes   939 Mbits/sec
[  4]  26.00-27.00  sec   112 MBytes   939 Mbits/sec
[  4]  27.00-28.00  sec   110 MBytes   927 Mbits/sec
[  4]  28.00-29.00  sec   112 MBytes   941 Mbits/sec
[  4]  29.00-30.00  sec   112 MBytes   938 Mbits/sec
[  4]  30.00-31.00  sec   111 MBytes   934 Mbits/sec
[  4]  31.00-32.00  sec   112 MBytes   937 Mbits/sec
[  4]  32.00-33.00  sec   112 MBytes   939 Mbits/sec
[  4]  33.00-34.00  sec   112 MBytes   939 Mbits/sec
[  4]  34.00-35.00  sec   112 MBytes   940 Mbits/sec
[  4]  35.00-36.00  sec   111 MBytes   928 Mbits/sec
[  4]  36.00-37.00  sec   112 MBytes   940 Mbits/sec
[  4]  37.00-38.00  sec   112 MBytes   937 Mbits/sec
[  4]  38.00-39.00  sec   112 MBytes   940 Mbits/sec
[  4]  39.00-40.00  sec   112 MBytes   940 Mbits/sec
[  4]  40.00-41.00  sec   112 MBytes   937 Mbits/sec
[  4]  41.00-42.00  sec   112 MBytes   940 Mbits/sec
[  4]  42.00-43.00  sec   111 MBytes   932 Mbits/sec
[  4]  43.00-44.00  sec   112 MBytes   937 Mbits/sec
[  4]  44.00-45.00  sec   112 MBytes   937 Mbits/sec
[  4]  45.00-46.00  sec   112 MBytes   937 Mbits/sec
[  4]  46.00-47.00  sec   111 MBytes   932 Mbits/sec
[  4]  47.00-48.00  sec   112 MBytes   936 Mbits/sec
[  4]  48.00-49.00  sec   112 MBytes   935 Mbits/sec
[  4]  49.00-50.00  sec   112 MBytes   938 Mbits/sec
[  4]  50.00-51.00  sec   112 MBytes   936 Mbits/sec
[  4]  51.00-52.00  sec   112 MBytes   937 Mbits/sec
[  4]  52.00-53.00  sec   112 MBytes   937 Mbits/sec
[  4]  53.00-54.00  sec   112 MBytes   937 Mbits/sec
[  4]  54.00-55.00  sec   112 MBytes   936 Mbits/sec
[  4]  55.00-56.00  sec   112 MBytes   935 Mbits/sec
[  4]  56.00-57.00  sec   112 MBytes   940 Mbits/sec
[  4]  57.00-58.00  sec   111 MBytes   933 Mbits/sec
[  4]  58.00-59.00  sec   112 MBytes   939 Mbits/sec
[  4]  59.00-60.00  sec   112 MBytes   938 Mbits/sec
[  4]  60.00-61.00  sec   111 MBytes   933 Mbits/sec
[  4]  61.00-62.00  sec   112 MBytes   936 Mbits/sec
[  4]  62.00-63.00  sec   112 MBytes   939 Mbits/sec
[  4]  63.00-64.00  sec   112 MBytes   941 Mbits/sec
[... 220 lines cut ...]
[  4] 284.00-285.00 sec   111 MBytes   930 Mbits/sec
[  4] 285.00-286.00 sec   112 MBytes   937 Mbits/sec
[  4] 286.00-287.00 sec   111 MBytes   933 Mbits/sec
[  4] 287.00-288.00 sec   112 MBytes   938 Mbits/sec
[  4] 288.00-289.00 sec   112 MBytes   937 Mbits/sec
[  4] 289.00-290.00 sec   111 MBytes   934 Mbits/sec
[  4] 290.00-291.00 sec   111 MBytes   928 Mbits/sec
[  4] 291.00-292.00 sec   111 MBytes   934 Mbits/sec
[  4] 292.00-293.00 sec   112 MBytes   937 Mbits/sec
[  4] 293.00-294.00 sec   112 MBytes   935 Mbits/sec
[  4] 294.00-295.00 sec   111 MBytes   933 Mbits/sec
[  4] 295.00-296.00 sec   110 MBytes   926 Mbits/sec
[  4] 296.00-297.00 sec   111 MBytes   935 Mbits/sec
[  4] 297.00-298.00 sec   112 MBytes   936 Mbits/sec
[  4] 298.00-299.00 sec   111 MBytes   931 Mbits/sec
[  4] 299.00-300.00 sec   112 MBytes   936 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-300.00 sec  32.7 GBytes   935 Mbits/sec                  sender
[  4]   0.00-300.00 sec  32.7 GBytes   935 Mbits/sec                  receiver

iperf Done.

C:\WINDOWS\system32>
```

morrownr, 2022-08-01:
It looks like you are getting closer to where you want to be.

> Making a Wifi network can be great with the PI as long as it's a 2.4GHz for legacy devices or if the PI is dedicated for that using OpenWRT I guess.

This is a good assessment but please allow me to expand on it a little.

Making a WiFi access point with a RasPi can be great but there are things to be aware of. Using a powered hub does not always work well. Many, probably most, powered hubs just simply do not work well with RasPi's and while a specific powered hub maybe seem to work well with a specific task, it may cause other problems, such as with wifi. Another critical issue is to make sure you are using good quality power supplies and cables that are providing the power that is needed.

OpenWRT is a good solution and it does not provide the flexibility of setup based on the RasPiOS but OpenWRT is not totally a dedicated use situation as OpenWRT provides numerous packages for functions in addition to being wifi router software. Example: It has a package that supports connecting a webcam so that it can serve video over your lan or save snapshot files. OpenWRT has packages that contain numerous drivers for usb adapters including ones for the mt7612u, mt7610u and even one for the new mt7921au (wifi 6) in the 22.03 version.

I think I need to add something along the lines of the above to my access point guide so that folks are aware of some things.

Regards

Nick

CapitaineBidouille, 2022-08-06:
5 days after, time to give some news.
I'm still with that network setup. This is what I was looking for. That pretty small router is doing the job perfectly to give a stable network for every devices based on an Ethernet link to the PI, which has now even more free ports available.
I guess I can probably jump to 20 or 25TB of shared space with 2 more hard drives and it will probably still work fine since there will just be hard drives on these hubs. But I am not gonna try it before a long of time since it's enough to me.
As it is, that setup seems to be stable 24/7.

The wifi network is high quality with the ASUS router. But on my PC, I stayed with Ethernet plugged on the ASUS router to get a stable gigabit bandwidth when I make backups.

Updating your guide seems like a great idea. I had the idea of making my PI as an USB router by looking at your guide and what you advise us to buy to get the best of it.
People who decide to go on that setup should know that it will really work well only if the PI is dedicated for that or if they want to stabilize that using a lot of their free time. The cost/benefit ratio in my case was so much better on a dedicated router rather than on the 2 USB adapters. The router cost me 95€, so less than 100€, and was plug & play : very easy to configure and start doing it's job.

Anyway, I do not regrest at all trying the USB wifi setup. It was definitely interesting to explore and debug. It was also a good reason to make me practice networking on Debian.