morrownr/USB-WiFi #93: Android x86 Support (Kernel 4.19.110)

stakeout55, 2022-08-01:
Hey there, and thank you SO much for this list - this is amazing. I had a couple questions for you regarding Android x86 / kernel 4.19.110. Their [release notes located here](https://www.android-x86.org/releases/releasenote-9-0-r2.html) indicated that Android x86 9.0-r2 has LTS kernel 4.19.110 - and I was wondering based off your pages here if that means [AC1200 / AC1300](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md#ac1200--ac1300---usb-3---24-ghz-and-5-ghz-wifi-5), and [AC580 / AC600](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md#ac580--ac600---usb-2---24-ghz-and-5-ghz-wifi-5) will work out of the box? 

Also, does that mean any AC1200, AC1300, AC580 and AC600 will innately support linux kernel 4.19 out of the box???

I am a little new to kernel support and am just trying to learn as I go here for a "project" build I am working on. Thank you so much for your time, help, and patience.

morrownr, 2022-08-01:
Hi @stakeout55 

> I was wondering based off your pages here if that means AC1200 / AC1300, and AC580 / AC600 will work out of the box?

That is a good question and I am sure others might want to know as well. Hopefully I can give you some homework and you can answer that question.

Adapters that work with in-kernel drivers require a driver (module) and firmware to be installed. I can't say whether the Android x86 project is compiling the drivers into their kernel but you should be able to check by looking for the files. In the distro I use here on my dev system, the drivers are located as such:

mt7912u:
/usr/lib/modules/5.15.0-41-generic/kernel/drivers/net/wireless/mediatek/mt76/mt76x2/mt76x2u.ko
/usr/lib/modules/5.15.0-41-generic/kernel/drivers/net/wireless/mediatek/mt76/mt76x2/mt76x2-common.ko

mt7610u:
/usr/lib/modules/5.15.0-41-generic/kernel/drivers/net/wireless/mediatek/mt76/mt76x0/mt76x0u.ko
/usr/lib/modules/5.15.0-41-generic/kernel/drivers/net/wireless/mediatek/mt76/mt76x0/mt76x0-common.ko

Note the `5.15.0-41-generic` as that will be `4.19-bla bla bla` or so since your kernel is different than mine. If the .ko files are there then the driver is there.

For the firmware, I have good instructions for installing or checking the firmware:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

When that homework is complete, come back and let us know what you found.

Regards,

Nick

stakeout55, 2022-08-01:
Hi there and thank you for this. According to more reading, the Androidx86 wifi drivers are located under /lib/firmware and under respective brand folders (brcm, mwlwifi, and more) - there is no folder alluding to "mt" anything really, but I am still investigating. Also the suffixes of .ko are non-existant within android x86 it seems, rather .bin -- so it seems I will need to do more investigating to see which wifi chipsets are supported in general!

morrownr, 2022-08-01:
> /lib/firmware

It looks like Google has changed a few things. That's fine. The drivers are in the kernel ready to be compiled if the details are worked out. I will assume the firmware is going in the above directory and the drivers are going in the folders below this directory.

I'd say having a chat about the subject with one of the devs of the distro would be helpful. If they are willing to work with you on the subject, you might even talk me into helping.

Nick

closed 2022-11-22 by morrownr