morrownr/USB-WiFi #99: WiFi HaLow

croissantpetrichor, 2022-08-11:
I found the [HaLow-U by Alfa](https://www.alfa.com.tw/products/halow-u) it utilizes [NRC7292 chipset](https://newracom.com/products/nrc7292) which has official open source drivers for [OpenWRT](https://github.com/newracom/nrc7292_openwrt), [Linux](https://github.com/newracom/nrc7292_sw_pkg) and a [SDK](https://github.com/newracom/nrc7292_sdk).

They actually appear to be the first manufacturer that have developed complete WiFi HaLow solutions. 

Although the drivers haven't been mainlined yet, it's still a new chipset and since the official drivers are already open source there is potential to integrate them into the Linux kernel.

This chipset only has a 15MB/s throughput however the IEEE 802.11ah standard specifies that if the chipset uses a 16MHz channel bandwidth you can achieve rates as high as 347 MB/s.

Since WiFi HaLow uses the 900 MHz bands this could prove to be a real game changer for long range bridges. The standard advertises that it can cover a 1KM radius due to the low frequency and this coverage with a simply an omnidirectional antenna, we can only imagine what range we'd get with a good LOS, Fresnel zone and a directional antenna.

I can't be the only one excited about this right? What does everyone here think, maybe someone has tried HaLow already?

morrownr, 2022-08-12:
This is interesting. Thanks for the links.

Sami32, 2022-11-01:
https://www.alfa.com.tw/products/amlga7921?variant=39651154624584

bjlockie, 2022-11-01:
Shouldn't the description say triband?

Sami32, 2022-11-03:
Quadri-band?

bjlockie, 2022-11-03:
Triband
2.4GHz, 5GHz, 6GHz.

Sami32, 2022-11-03:
+ WiFi HaLow (802.11ah) --> sub 1G

Sami32, 2022-11-09:
Product informations corrected and updated.

closed 2022-11-22 by morrownr

codebling, 2022-12-01:
Does anyone have a purchase link for this adapter?