morrownr/USB-WiFi #100: CF-953AX dropping connection with Fritzbox 6660 cable

UnknownProgrammer, 2022-08-13:
To **test the CF-953AX with Xubuntu 22.04** I installed kernel versions 5.18, 5.18.15 and 5.19 from https://kernel.ubuntu.com/~kernel-ppa/mainline/.

At first I used the firmware already provided by 22.04.

The router is ax capable -> Fritzbox 6660 cable set to channel 36 at 5Ghz. Channel width can't be changed and can be up to 160Mhz. Firmware version is the latest 7.29. But the firmware is controlled by the cable provider, which takes quite some time to review and deliver the manufacturer firmware.
 
The client is set to use only the 5Ghz band and connects with 2x2, 80Mhz channel width.

The distance from client to router is around 7m with two thin ceilings between.

#### Results:

Kernel 5.18 didn't have stable throughput rates. They where fluctuating between 50 and 300 Mbit using a two minute connection test to the router with iperf.

So I switched to 5.18.15. Throughput became stable at ~300 Mbit. But after about two hours the client got deauthenticated from the router (Reason: 16=GROUP_KEY_HANDSHAKE_TIMEOUT) and a reboot was necessary to reestablish the connection.

After that I tied 5.19 (Daily/current -> 2022-08-11). Throughput stayed at ~300 Mbit. Client got deauthenticated after about two hours again.

As next step I replaced the provided firmware (Files: BT_RAM_CODE_MT7961_1_2_hdr.bin, WIFI_MT7961_patch_mcu_1_2_hdr.bin, WIFI_RAM_CODE_MT7961_1.bin) with the ones here https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek. Checking the size and date on the files showed that they differ from the included ones. Even with the new firmware the client got deauthenticated again, sadly.

Tested also the combination kernel version 5.18.15 with new firmware. -> connection drop.

Switching to channel 100 in the 5Ghz band and setting the client fix to 2.4Ghz showed the same connection drop.

I tested other usb-wifi devices on the same port in the same location and they stay connected or reconnect if issues occur.

Speeds are quite good, but it would be nice if the driver had some recovery mechanism to reestablish a connection after deauthentication.

#### Expected behavior:

Stay connected and/or reestablish connection after deauthentication.

#### Actual behavior:

Drops connection after 2-3 hours and needs reboot to reconnect.

#### Additional Info:

- Group key request occurs every 10 minutes.
- Looks like the time-out timer for rekeying is set to 3 seconds, because the deauthentication happens 3 seconds after the rekeying should have happened.
- Can't see any connection between load on the wifi device and deauthentication. 
- I rebooted every time I changed something.
- Syslog msg: `Aug 11 16:59:47 ... kernel: [ 8765.480856] wlan0: deauthenticated from ... (Reason: 16=GROUP_KEY_HANDSHAKE_TIMEOUT)`
- The outcome of deauthentication is the inability to establish any network connection, the indicator still shows signal-strength as if wifi is connected.

morrownr, 2022-08-13:
Hi @UnknownProgrammer 

The first thing that popped into my mind was wpa_supplicant but there are a lot of things that depend on each other so this is a puzzle to solve.

I don't have a cf-953ax yet so I can't test locally but if this issue was happening to me, the first step I would take is to connect the adapter to a different access point to see if the same thing happens. I happen to have a RasPi4B that is set up as a dual band access point on the primary network so it would be easy for me to do such as that gives me two separate access points that are based on different setups. My primary wifi router runs OpenWRT and the RasPi runs hostapd. Do you have the capability to try to connect the adapter to another access point?

I'll keep an eye out for others that may be seeing this. There are others that stop by here that may be able to add something to the conversation.

Of note: there are reports from others that have the cf-953ax that they are able to use it on 6 GHz channels. If you can check that, it would help confirm the capability.

Do you have the ability to change the version of WPA in the Fritzbox? If so, is it using WPA2, WPA3 or mixed?

Regards,

Nick

UnknownProgrammer, 2022-08-16:
> The first thing that popped into my mind was wpa_supplicant [...]

Good guess, restarting wpa_supplicant.service let the device reconnect. Since Canonical has plans to replace wpa_supplicant with iwd in the next release and I'm planning to upgrade to it as soon as 22.10 beta is available anyway. **I switched from wpa_supplicant to iwd and the connection became stable.**

Before that I tried everything you suggested with wpa_supplicant still installed:

> [...] try to connect the adapter to another access point?

I tried to recreate the connection drop with an AC router (Netgear R6220) running OpenWRT 21.02.3. But setting the time interval for rekeying GTK to 600 seconds only creates the rekeying event in the client log and caused no connection drop as long as I was observing (> 3h). I also set it to 300 seconds in the hopes that the drop happens faster. Might be something else causing it.

Good news for other users without Fritzbox I guess.

> Do you have the ability to change the version of WPA in the Fritzbox? If so, is it using WPA2, WPA3 or mixed?

I have two options to chose from:

`WPA2` and `ẀPA2+WPA3`

Switching to `ẀPA2+WPA3` and selecting WPA3 Personal on the client makes no difference. Drop happens regardless.

> Of note: there are reports from others that have the cf-953ax that they are able to use it on 6 GHz channels. If you can check that, it would help confirm the capability.

I looked into it. Installed the latest snapshot for rpi-4 the day before yesterday. And with it luci, usbutils, kmod-mt7921u and the latest firmware files. If I set the domain to DE `iw reg set 'DE'` I get the same results as on my ubuntu machine, which are:

	$iw reg get

	global
	country DE: DFS-ETSI
		(2400 - 2483 @ 40), (N/A, 20), (N/A)
		(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
		(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
		(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
		(5725 - 5875 @ 80), (N/A, 13), (N/A)
		(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
		(57000 - 66000 @ 2160), (N/A, 40), (N/A)

	$iw phy0 channels

	Band 1:
		* 2412 MHz [1] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40+
		* 2417 MHz [2] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40+
		* 2422 MHz [3] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40+
		* 2427 MHz [4] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40+
		* 2432 MHz [5] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40- HT40+
		* 2437 MHz [6] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40- HT40+
		* 2442 MHz [7] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40- HT40+
		* 2447 MHz [8] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40- HT40+
		* 2452 MHz [9] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40- HT40+
		* 2457 MHz [10] 
		  Maximum TX power: 20.0 dBm
		  Channel widths: 20MHz HT40-
		* 2462 MHz [11] 
[... 364 lines cut ...]
		wiphy 2
		channel 81 (6355 MHz), width: 40 MHz, center1: 6365 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	99	0	0	0	0	12307		109

 
On client:

	$ iw wlp33s0 info
	Interface wlp33s0
		ifindex 3
		wdev 0x1
		addr xx:xx:xx:xx:xx:xx
		ssid OpenWrt
		type managed
		wiphy 0
		channel 81 (6355 MHz), width: 40 MHz, center1: 6365 MHz
		txpower 22.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0

@morrownr so what should we do with this issue? Replacing wpa_supplicant is only a workaround.

morrownr, 2022-08-16:
@UnknownProgrammer 

> so what should we do with this issue?

Give me chance to digest the information. 

> Replacing wpa_supplicant is only a workaround.

Well, yes, but it may be a clue as to where the real problem is.

Sometimes I wonder how we get connections at all. Within the last few years we have added 80211ax, 6 GHz channels and WPA3...and WiFi 7 stuff is already going in.

morrownr, 2022-08-17:
@UnknownProgrammer 

The two key issues in my opinion are:

> I switched from wpa_supplicant to iwd and the connection became stable.

> I tested other usb-wifi devices on the same port in the same location and they stay connected or reconnect if issues occur.

This makes it look like there is an issue in the driver and it appears there may be an issue with wpa_supplicant. I think you should post a bug report.

https://wireless.wiki.kernel.org/en/users/documentation/reporting_bugs

> https://wireless.wiki.kernel.org/en/users/drivers/mediatek

Remember that the linux-wireless mailing list is text only, no html and the subject should be such that the correct Mediatek dev will know if it is something that he/she works on:

Suggested Subject: [BUG} mt7921u: COMFAST CF-953AX, mt7921au chipset, dropping connection with Fritzbox 6660

closed 2022-11-22 by morrownr